b-rad-NDi/Ubuntu-media-tree-kernel-builder #68: WinTV-NOVA-TD-500 second tuner missing

Saentist, 2019-03-20:
On Ubuntu 16.04 with standard repo kernel both tuners working,
but after add 2x QuadHD + kernel+mediatree+hauppauge
second tuner is missing

https://linuxtv.org/wiki/index.php/Hauppauge_WinTV-NOVA-T-500#Losing_one_tuner.3F

manual is not working.

any advices?

b-rad-NDi, 2019-03-22:
Well, is everything working together when not using my ppa? You say it worked with the stock kernel, but then you add multiple variables and say it no longer works. Do you have dmesg logs for me?

Saentist, 2019-03-23:
dmesg
https://paste.ubuntu.com/p/KZk2ZRY6rh/

with stock kernel 

I play little bit and with manual adding to
/etc/modprobe.d/options.conf  
```
options usbcore autosuspend=-1
```
after few restarts second tuner have position 3 ???
other tuners are Quad, very strange arange 

```
Info:  DVB:  ==================================
Info:  DVB:          DVB CARDS LISTING
Info:  DVB:  ==================================
Info:  DVB:  =========== Card 0 - Tuner 0 ===========
Info:  DVB:   Frontend : DiBcom 7000PC
Info:  DVB:   Terrestrial (DVB-T) card
Info:  DVB:   Frequency: 45000 kHz to 860000 kHz
Deb0:  DVB:   == Card capabilities ==
Deb0:  DVB:  FE_CAN_INVERSION_AUTO
Deb0:  DVB:  FE_CAN_FEC_1_2
Deb0:  DVB:  FE_CAN_FEC_2_3
Deb0:  DVB:  FE_CAN_FEC_3_4
Deb0:  DVB:  FE_CAN_FEC_5_6
Deb0:  DVB:  FE_CAN_FEC_7_8
Deb0:  DVB:  FE_CAN_FEC_AUTO
Deb0:  DVB:  FE_CAN_QPSK
Deb0:  DVB:  FE_CAN_QAM_16
Deb0:  DVB:  FE_CAN_QAM_64
Deb0:  DVB:  FE_CAN_QAM_AUTO
Deb0:  DVB:  FE_CAN_TRANSMISSION_MODE_AUTO
Deb0:  DVB:  FE_CAN_GUARD_INTERVAL_AUTO
Deb0:  DVB:  FE_CAN_HIERARCHY_AUTO
Deb0:  DVB:  FE_CAN_RECOVER /* frontend can recover from a cable unplug automatically */
Info:  DVB:
Info:  DVB:  =========== Card 1 - Tuner 0 ===========
Info:  DVB:   Frontend : Silicon Labs Si2168
Info:  DVB:   Terrestrial (DVB-T) card
Info:  DVB:   Frequency: 48000 kHz to 870000 kHz
Deb0:  DVB:   == Card capabilities ==
Deb0:  DVB:  FE_CAN_INVERSION_AUTO
Deb0:  DVB:  FE_CAN_FEC_1_2
Deb0:  DVB:  FE_CAN_FEC_2_3
Deb0:  DVB:  FE_CAN_FEC_3_4
Deb0:  DVB:  FE_CAN_FEC_5_6
Deb0:  DVB:  FE_CAN_FEC_7_8
Deb0:  DVB:  FE_CAN_FEC_AUTO
Deb0:  DVB:  FE_CAN_QPSK
Deb0:  DVB:  FE_CAN_QAM_16
Deb0:  DVB:  FE_CAN_QAM_32
Deb0:  DVB:  FE_CAN_QAM_64
Deb0:  DVB:  FE_CAN_QAM_128
Deb0:  DVB:  FE_CAN_QAM_256
Deb0:  DVB:  FE_CAN_QAM_AUTO
Deb0:  DVB:  FE_CAN_TRANSMISSION_MODE_AUTO
Deb0:  DVB:  FE_CAN_GUARD_INTERVAL_AUTO
Deb0:  DVB:  FE_CAN_HIERARCHY_AUTO
Deb0:  DVB:  FE_CAN_2G_MODULATION /* frontend supports '2nd generation modulation' (DVB-S2) */
Deb0:  DVB:  FE_CAN_MUTE_TS /* frontend can stop spurious TS data output */
Info:  DVB:
Info:  DVB:  =========== Card 2 - Tuner 0 ===========
Info:  DVB:   Frontend : Silicon Labs Si2168
Info:  DVB:   Terrestrial (DVB-T) card
Info:  DVB:   Frequency: 48000 kHz to 870000 kHz
Deb0:  DVB:   == Card capabilities ==
Deb0:  DVB:  FE_CAN_INVERSION_AUTO
Deb0:  DVB:  FE_CAN_FEC_1_2
[... 165 lines cut ...]
Info:  DVB:   Frontend : Silicon Labs Si2168
Info:  DVB:   Terrestrial (DVB-T) card
Info:  DVB:   Frequency: 48000 kHz to 870000 kHz
Deb0:  DVB:   == Card capabilities ==
Deb0:  DVB:  FE_CAN_INVERSION_AUTO
Deb0:  DVB:  FE_CAN_FEC_1_2
Deb0:  DVB:  FE_CAN_FEC_2_3
Deb0:  DVB:  FE_CAN_FEC_3_4
Deb0:  DVB:  FE_CAN_FEC_5_6
Deb0:  DVB:  FE_CAN_FEC_7_8
Deb0:  DVB:  FE_CAN_FEC_AUTO
Deb0:  DVB:  FE_CAN_QPSK
Deb0:  DVB:  FE_CAN_QAM_16
Deb0:  DVB:  FE_CAN_QAM_32
Deb0:  DVB:  FE_CAN_QAM_64
Deb0:  DVB:  FE_CAN_QAM_128
Deb0:  DVB:  FE_CAN_QAM_256
Deb0:  DVB:  FE_CAN_QAM_AUTO
Deb0:  DVB:  FE_CAN_TRANSMISSION_MODE_AUTO
Deb0:  DVB:  FE_CAN_GUARD_INTERVAL_AUTO
Deb0:  DVB:  FE_CAN_HIERARCHY_AUTO
Deb0:  DVB:  FE_CAN_2G_MODULATION /* frontend supports '2nd generation modulation' (DVB-S2) */
Deb0:  DVB:  FE_CAN_MUTE_TS /* frontend can stop spurious TS data output */

```

b-rad-NDi, 2019-03-25:
Device order can potentially change every single boot, the process is asynchronous to a certain degree. I'll give a look at what's changed in the dibcom driver since 4.4. Shouldn't really be any reason for breakage, but totally possible a regression happened.

Saentist, 2019-03-25:
Device order can be controlled by PARM settings
if there is some short manual in wiki part of this git or official page, will be helpfully 

```
modinfo cx23885
parm:           adapter_nr:DVB adapter numbers (array of short)

 modinfo dvb_usb_dib0700
parm:           adapter_nr:DVB adapter numbers (array of short)
```

b-rad-NDi, 2019-03-26:
https://linuxtv.org is where all documentation/wiki exist.

I just reviewed kernel history for the affected drivers back to 4.4 release date and see nothing suspect. Do both tuners and the quadhd's work for you if you use:
```
options usbcore autosuspend=-1
```
?
Does this fix the issue completely?

Saentist, 2019-03-26:
Yes it fix it, but its not for beginners.
Also why with stock kernel work, but with "extra" modules need to make extra modifications how modules to be loaded.

b-rad-NDi, 2019-03-26:
Well, without doing a bisection through 5 years of kernel code I'm guessing something was done in the media tree relating to suspend that is affecting you. I imagine you'd encounter the exact same issue if you tried running mainline kernel.

Saentist, 2019-03-26:
PC is fresh installed with Ubuntu 16.04 x86-64 and updated with all updates.
Twin dvb tuner working perfect, week later added 2xQuadHD with not working, until added PPA fallowing [THIS](http://www.hauppauge.com/pages/support/support_linux.html) manual
then second tuner of TD-500 disappear, but both quad tuners start to work.
```
/etc/modprobe.d/options.conf
options usbcore autosuspend=-1
```

Restart make second tuner of TD-500 reappear.

https://unix.stackexchange.com/questions/91027/how-to-disable-usb-autosuspend-on-kernel-3-7-10-or-above
Problem maby start little earlier, but as fact not appear with 4.4.
Possible some kernel configuration

b-rad-NDi, 2019-03-27:
That's what I'm saying. The stock kernel is nothing but 4.4 stable, which was released 4 years ago. This project integrates the entire media tree. That means almost everything in drivers/media/ is 4 years newer. Some regression in usb suspend must have developed. That's unfortunate, but hard for me to diagnose, especially without affected equipment.

closed 2022-01-27 by b-rad-NDi