b-rad-NDi/Ubuntu-media-tree-kernel-builder #69: WinTV-quadHD mpeg risc op code error / dvb-t2 scan not working

plntyk, 2019-03-25:
Reading #17 and #51 suggest that non-AMD platforms and mainline kernel should not exhibit "mpeg risc op code error" anymore

`dvbv5-scan -C de simple.conf`
with content 
````
[CHANNEL]
	DELIVERY_SYSTEM = DVBT2
	FREQUENCY = 506000000
	BANDWIDTH_HZ = 8000000
	CODE_RATE_HP = 3/5
	CODE_RATE_LP = NONE
	MODULATION = QAM/64
	TRANSMISSION_MODE = 32K
	GUARD_INTERVAL = 19/256
	HIERARCHY = NONE
	INVERSION = AUTO
````
fails.

The error goes away with
`options cx23885 debug=7`
in /etc/modprobe.d/cx23885.conf - as is suggested in the other issue and the scan is successful.

VLC works without any error in syslog via cmdline / tuning seems fine
`vlc dvb-t2://frequency=506000000:bandwidth=8`
A short test with tvheadend shows similar errors but recording seems to work.

System:
Intel i7-3770, Intel Z77 Mainboard
OS: Archlinux 5.0.4-arch1-1-ARCH
Location: Berlin, Germany DVB-T2

error:

````
[  921.973337] dvb_frontend: dvb_frontend_get_frequency_limits: frequency interval: tuner: 42000000...870000000, frontend: 0...0
[  922.393092] cx23885: cx23885[0]: mpeg risc op code error
[  922.393103] cx23885: cx23885[0]: TS1 B - dma channel status dump
[  922.393109] cx23885: cx23885[0]:   cmds: init risc lo   : 0xffeff000
[  922.393116] cx23885: cx23885[0]:   cmds: init risc hi   : 0x00000000
[  922.393122] cx23885: cx23885[0]:   cmds: cdt base       : 0x00010870
[  922.393129] cx23885: cx23885[0]:   cmds: cdt size       : 0x0000000a
[  922.393136] cx23885: cx23885[0]:   cmds: iq base        : 0x00010630
[  922.393143] cx23885: cx23885[0]:   cmds: iq size        : 0x00000010
[  922.393149] cx23885: cx23885[0]:   cmds: risc pc lo     : 0xffeff018
[  922.393156] cx23885: cx23885[0]:   cmds: risc pc hi     : 0x00000000
[  922.393163] cx23885: cx23885[0]:   cmds: iq wr ptr      : 0x00004192
[  922.393169] cx23885: cx23885[0]:   cmds: iq rd ptr      : 0x0000418c
[  922.393176] cx23885: cx23885[0]:   cmds: cdt current    : 0x00010888
[  922.393183] cx23885: cx23885[0]:   cmds: pci target lo  : 0xffffa340
[  922.393189] cx23885: cx23885[0]:   cmds: pci target hi  : 0x00000000
[  922.393196] cx23885: cx23885[0]:   cmds: line / byte    : 0x000d0000
[  922.393203] cx23885: cx23885[0]:   risc0: 
[  922.393204] 0x1c0002f0 [ write sol eol count=752 ]
[  922.393212] cx23885: cx23885[0]:   risc1: 
[  922.393213] 0xffffa050 [ INVALID sol eol irq2 irq1 23 22 21 20 19 18 cnt1 cnt0 resync 13 count=80 ]
[  922.393225] cx23885: cx23885[0]:   risc2: 
[  922.393226] 0x00000000 [ INVALID count=0 ]
[  922.393232] cx23885: cx23885[0]:   risc3: 
[  922.393233] 0x1c0002f0 [ write sol eol count=752 ]
[  922.393239] cx23885: cx23885[0]:   (0x00010630) iq 0: 
[  922.393240] 0x1c0002f0 [ write sol eol count=752 ]
[  922.393249] cx23885: cx23885[0]:   iq 1: 0xffffa630 [ arg #1 ]
[  922.393256] cx23885: cx23885[0]:   iq 2: 0x00000000 [ arg #2 ]
[  922.393262] cx23885: cx23885[0]:   (0x0001063c) iq 3: 
[  922.393263] 0x1c0002f0 [ write sol eol count=752 ]
[  922.393269] cx23885: cx23885[0]:   iq 4: 0xffffa920 [ arg #1 ]
[  922.393276] cx23885: cx23885[0]:   iq 5: 0x00000000 [ arg #2 ]
[  922.393282] cx23885: cx23885[0]:   (0x00010648) iq 6: 
[  922.393283] 0x00000000 [ INVALID count=0 ]
[  922.393289] cx23885: cx23885[0]:   (0x0001064c) iq 7: 
[  922.393290] 0x1c0002f0 [ write sol eol count=752 ]
[  922.393296] cx23885: cx23885[0]:   iq 8: 0xffffa340 [ arg #1 ]
[  922.393303] cx23885: cx23885[0]:   iq 9: 0x00000000 [ arg #2 ]
[  922.393309] cx23885: cx23885[0]:   (0x00010658) iq a: 
[  922.393310] 0x00000000 [ INVALID count=0 ]
[  922.393316] cx23885: cx23885[0]:   (0x0001065c) iq b: 
[  922.393316] 0x1c0002f0 [ write sol eol count=752 ]
[  922.393323] cx23885: cx23885[0]:   iq c: 0xffff9780 [ arg #1 ]
[  922.393329] cx23885: cx23885[0]:   iq d: 0x00000000 [ arg #2 ]
[  922.393334] cx23885: cx23885[0]:   (0x00010668) iq e: 
[  922.393334] 0x1c0002f0 [ write sol eol count=752 ]
[  922.393341] cx23885: cx23885[0]:   iq f: 0xffff9a70 [ arg #1 ]
[  922.393347] cx23885: cx23885[0]:   iq 10: 0x48652910 [ arg #2 ]
[  922.393348] cx23885: cx23885[0]: fifo: 0x00005000 -> 0x6000
[  922.393350] cx23885: cx23885[0]: ctrl: 0x00010630 -> 0x10690
[  922.393355] cx23885: cx23885[0]:   ptr1_reg: 0x00005690
[  922.393360] cx23885: cx23885[0]:   ptr2_reg: 0x00010898
[  922.393367] cx23885: cx23885[0]:   cnt1_reg: 0x0000000b
[  922.393372] cx23885: cx23885[0]:   cnt2_reg: 0x00000005
[  925.124976] dvb_frontend: dvb_frontend_get_frequency_limits: frequency interval: tuner: 42000000...870000000, frontend: 0...0
````

b-rad-NDi, 2019-03-28:
Did you try setting the kernel parameter to force the amd fix?
```
options cx23885 dma_reset_workaround=2
```

plntyk, 2019-04-13:
Forcing the AMD fix seems to fix the issue. Recorded some stuff via timers and I didnt see any of these strange log messages.

regarding those - Is this because I am using as Kernel parameter
````
intel_iommu=on,igfx_off
````
aka IOMMU/Virtualization VT-D with device passthrough / PCI or PCIe - passthrough ?

PS: my use case 

I sometimes spin up a Windows 7/8 VM with DVB Viewer (with another DVB device) - now splitting 4 Tuners the of winTV QuadHD between Windows VM and Linux Host OS seems to work now too.

So this "bug" is either IOMMU - iirc that was mentioned on the linux-media mailing list or an issue with Mainboard/Chipset/IOMMU environment (Intel Z77: DZ77RE-75K)

b-rad-NDi, 2019-05-08:
Glad that option fixed you up. If the root cause is indeed the IOMMU like is suspected, then unfortunately it's outside of my realm of expertise.
Can you post the results of:
```lspci -v -nn -k```
So I can keep a tally of affected hardware?

I'm going to close this issue for now.

closed 2019-05-08 by b-rad-NDi