b-rad-NDi/Ubuntu-media-tree-kernel-builder #70: Backport MDADM patch into 4.18 kernel

albertcard, 2019-04-20:
This is a wishlist item but is it possible to get this patch applied for a fix on systems using MDADM with RAID-10 and XFS backported to the 4.18 kernel that was patched in 4.19? 

Patch: https://lkml.org/lkml/2019/2/11/1281
Bug: https://bugs.launchpad.net/ubuntu/+bug/1767992

b-rad-NDi, 2019-05-08:
I will give this a look in case the patch isn't backported to previous stable trees.

b-rad-NDi, 2019-05-29:
Ubuntu still hasn't fixed this. I have added it. 5cb621260be52de32ac130b0386a07f171e1ca2e

I only apply it to 4.15 and 4.18 builds, 4.4 is different. Builds are available in my testing ppa.

https://launchpad.net/~b-rad/+archive/ubuntu/test-test-hw-staging

closed 2019-05-29 by b-rad-NDi