b-rad-NDi/Ubuntu-media-tree-kernel-builder #71: Ubuntu 18.04-HVR4000-Firmware not loading

mralapete, 2019-04-24:
Just installed a fresh copy of Ubuntu 18.04 and followed your exact instructions for installing the PPA. This was the output of dmesg before I ran Kaffeine to test the HVR4000

15.789695] media: Linux media interface: v0.10
[   16.195949] videodev: Linux video capture interface: v2.00
[   16.346087] PCI Interrupt Link [APC2] enabled at IRQ 17
[   16.346138] cx88xx: subsystem: 0070:6902, board: Hauppauge WinTV-HVR4000 DVB-S/S2/T/Hybrid [card=68,autodetected], frontend(s): 2
[   16.474655] cx88xx: i2c init: enabling analog demod on HVR1300/3000/4000 tuner
[   16.505667] PCI Interrupt Link [AAZA] enabled at IRQ 23
[   16.505690] snd_hda_intel 0000:00:09.0: Disabling MSI
[   16.522643] PCI Interrupt Link [APC6] enabled at IRQ 16
[   16.522803] checking generic (d1000000 130000) vs hw (c0000000 10000000)
[   16.522805] checking generic (d1000000 130000) vs hw (d0000000 2000000)
[   16.522806] fb: switching to nouveaufb from VESA VGA
[   16.522860] Console: switching to colour dummy device 80x25
[   16.524422] nouveau 0000:02:00.0: NVIDIA G94 (094280a1)
[   16.647859] nouveau 0000:02:00.0: bios: version 62.94.6e.00.00
[   16.670716] nouveau 0000:02:00.0: bios: M0203T not found
[   16.670719] nouveau 0000:02:00.0: bios: M0203E not matched!
[   16.670722] nouveau 0000:02:00.0: fb: 1024 MiB DDR2
[   16.729847] intel_powerclamp: No package C-state available
[   16.747710] [TTM] Zone  kernel: Available graphics memory: 2019232 kiB
[   16.747712] [TTM] Initializing pool allocator
[   16.747718] [TTM] Initializing DMA pool allocator
[   16.747742] nouveau 0000:02:00.0: DRM: VRAM: 1024 MiB
[   16.747745] nouveau 0000:02:00.0: DRM: GART: 1048576 MiB
[   16.747750] nouveau 0000:02:00.0: DRM: TMDS table version 2.0
[   16.747752] nouveau 0000:02:00.0: DRM: DCB version 4.0
[   16.747755] nouveau 0000:02:00.0: DRM: DCB outp 00: 02000300 00020028
[   16.747758] nouveau 0000:02:00.0: DRM: DCB outp 01: 01000302 00020030
[   16.747760] nouveau 0000:02:00.0: DRM: DCB outp 02: 04011310 00020028
[   16.747763] nouveau 0000:02:00.0: DRM: DCB outp 03: 02011312 00020030
[   16.747765] nouveau 0000:02:00.0: DRM: DCB outp 04: 04022342 00c20020
[   16.747768] nouveau 0000:02:00.0: DRM: DCB conn 00: 00001030
[   16.747770] nouveau 0000:02:00.0: DRM: DCB conn 01: 00002130
[   16.747772] nouveau 0000:02:00.0: DRM: DCB conn 02: 00010261
[   16.749085] [drm] Supports vblank timestamp caching Rev 2 (21.10.2013).
[   16.749086] [drm] Driver supports precise vblank timestamp query.
[   16.771467] nouveau 0000:02:00.0: DRM: MM: using CRYPT for buffer copies
[   16.872154] nouveau 0000:02:00.0: DRM: allocated 1920x1080 fb: 0x70000, bo 00000000472196b7
[   16.872325] fbcon: nouveaufb (fb0) is primary device
[   16.872563] Console: switching to colour frame buffer device 240x67
[   16.872625] nouveau 0000:02:00.0: fb0: nouveaufb frame buffer device
[   16.888042] [drm] Initialized nouveau 1.3.1 20120801 for 0000:02:00.0 on minor 0
[   17.340909] tda9887 2-0043: creating new instance
[   17.340911] tda9887 2-0043: tda988[5/6/7] found
[   17.341714] tuner: 2-0043: Tuner 74 found with type(s) Radio TV.
[   17.344342] tuner: 2-0061: Tuner -1 found with type(s) Radio TV.
[   17.382694] tveeprom: Hauppauge model 69009, rev B2D3, serial# 4031907689
[   17.382697] tveeprom: MAC address is 00:0d:fe:52:07:69
[   17.382698] tveeprom: tuner model is Philips FMD1216MEX (idx 133, type 78)
[   17.382700] tveeprom: TV standards PAL(B/G) PAL(I) SECAM(L/L') PAL(D/D1/K) ATSC/DVB Digital (eeprom 0xf4)
[   17.382701] tveeprom: audio processor is CX882 (idx 33)
[   17.382702] tveeprom: decoder processor is CX882 (idx 25)
[   17.382703] tveeprom: has radio, has IR receiver, has no IR transmitter
[   17.382705] cx88xx: hauppauge eeprom: model=69009
[   17.499790] tuner-simple 2-0061: creating new instance
[   17.499794] tuner-simple 2-0061: type set to 78 (Philips FMD1216MEX MK3 Hybrid Tuner)
[   17.604017] Registered IR keymap rc-hauppauge
[   17.662658] IR RC5(x/sz) protocol handler initialized
[   17.688083] rc rc0: cx88 IR (Hauppauge WinTV-HVR400 as /devices/pci0000:00/0000:00:0a.0/0000:01:06.0/rc/rc0
[   17.688145] input: cx88 IR (Hauppauge WinTV-HVR400 as /devices/pci0000:00/0000:00:0a.0/0000:01:06.0/rc/rc0/input9
[   17.688276] rc rc0: lirc_dev: driver cx88xx registered at minor = 0, raw IR receiver, no transmitter
[   17.688346] cx8800: found at 0000:01:06.0, rev: 5, irq: 17, latency: 0, mmio: 0xd6000000
[   17.688403] cx8802: cx2388x 8802 Driver Manager
[   17.688521] cx88_alsa: cx88[0]: alsa: CX88x/0: ALSA support for cx2388x boards
[   17.688568] cx8802: found at 0000:01:06.2, rev: 5, irq: 17, latency: 64, mmio: 0xd8000000
[   17.903939] cx88_dvb: cx2388x dvb driver version 1.0.0 loaded
[   17.903942] cx8802: registering cx8802 driver, type: dvb access: shared
[   17.903945] cx8802: subsystem: 0070:6902, board: Hauppauge WinTV-HVR4000 DVB-S/S2/T/Hybrid [card=68]
[   17.903946] cx88_dvb: cx2388x based DVB/ATSC card
[... 30 lines cut ...]
[  353.232084] cx8800 0000:01:06.0: Direct firmware load for dvb-fe-cx24116.fw failed with error -2
[  353.232086] cx24116_firmware_ondemand: Waiting for firmware upload(2)...
[  353.232087] cx24116_firmware_ondemand: No firmware uploaded (timeout or file not found?)
[  353.232091] cx24116_cmd_execute(): Unable initialise the firmware
[  466.011301] perf: interrupt took too long (2548 > 2500), lowering kernel.perf_event_max_sample_rate to 78250
[  584.321559] perf: interrupt took too long (3187 > 3185), lowering kernel.perf_event_max_sample_rate to 62750
[  642.784755] cx24116_firmware_ondemand: Waiting for firmware upload (dvb-fe-cx24116.fw)...
[  642.784773] cx8800 0000:01:06.0: Direct firmware load for dvb-fe-cx24116.fw failed with error -2
[  642.784776] cx24116_firmware_ondemand: Waiting for firmware upload(2)...
[  642.784777] cx24116_firmware_ondemand: No firmware uploaded (timeout or file not found?)
[  642.784781] cx24116_cmd_execute(): Unable initialise the firmware
[  642.785280] cx88-mpeg driver manager 0000:01:06.2: DVB: adapter 0 frontend 0 frequency 10714250 out of range (950000..2150000)
[  642.787293] cx88-mpeg driver manager 0000:01:06.2: DVB: adapter 0 frontend 0 frequency 10729000 out of range (950000..2150000)
[  642.788632] cx88-mpeg driver manager 0000:01:06.2: DVB: adapter 0 frontend 0 frequency 10744000 out of range (950000..2150000)
[  642.789490] cx88-mpeg driver manager 0000:01:06.2: DVB: adapter 0 frontend 0 frequency 10758000 out of range (950000..2150000)
[  642.790377] cx88-mpeg driver manager 0000:01:06.2: DVB: adapter 0 frontend 0 frequency 10773000 out of range (950000..2150000)
[  642.791250] cx88-mpeg driver manager 0000:01:06.2: DVB: adapter 0 frontend 0 frequency 10788000 out of range (950000..2150000)

The HVR4000 was working fine in Ubuntu 14.04 LTS and is still working fine in Windows.

I ran uname -a

4.18.0-18-generic #19~18.04.1-Ubuntu SMP Fri Apr 5 10:22:13 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux

I don't see any mention of mediatree here

mralapete, 2019-04-27:
I came across this link. Would this install the necessary drivers to get the card working https://ubuntu.pkgs.org/18.04/ubuntu-updates-main-amd64/linux-modules-extra-4.18.0-1009-gcp_4.18.0-1009.10~18.04.1_amd64.deb.html

b-rad-NDi, 2019-04-28:
You are missing a firmware:
```/lib/firmware/dvb-fe-cx24116.fw```
Either you originally found it elsewhere or it was removed from linux-firmware-nonfree. I will locate it and add it to my linux-firmware-hauppauge package.

mralapete, 2019-04-29:
Yes I thought that was the problem. I was searching for a copy of that firmware so I could copy it into /lib/firmware folder manually to see if that would resolve the problem but any of the sources I found were either broken links or just non existent.

It was so much easier installing Linux firmware nonfree back in Ubuntu 14.04 LTS and I only updated as this version was EOL. Linux-firmware-nonfree is no longer available from the repo. I did notice that Launchpad has it https://launchpad.net/ubuntu/+source/linux-firmware-nonfree and assumed by installing it from this source it woukd solve the problem. I had to revert back to Ubuntu 14.04 LTS while this problem was being investigated.

Are you confident that this is the only issue at play here. Should installing Linux firmware-nonfree from Launchpad resolve the problem. What exactly do I need you PPA for. What exactly will I need to do after I install Ubuntu 18.04 LTS which installs kernel 4.15 and after sudo upgrade installs kernel 4.18 to get the HVR 4000 recognised by the system

mralapete, 2019-04-29:
Will this get you the driver you need to locate http://www.hauppauge.de/files/drivers/hcw88/

mralapete, 2019-05-01:
I meant to ask earlier. Could I just add the dvb-fe-cx24116.fw driver to my /lib/firmware folder on 18.04 LTS to solve this problem. 

The HVR 4000 card was a very popular sale item back in the day and I'm sure I'm not the only user who will want to upgrade to 18.06 LTS or greater at some stage. I'm puzzled why the cx24116 wasn't included in your package when you created it. As far a I remember the Nova DVB S2 card used this firmware too.

Do you have a HVR 4000 to hand that you could test to see if it works. 

I do understand that you are probably quite busy at the moment but I really would like to get 18.04 up and running as 14.04 EOL is right here now.

b-rad-NDi, 2019-05-01:
Like I said earlier. All you are missing is the firmware file. The firmware is unrelated to the driver. If the driver worked previously, it will/should still work now if you put the firmware it wants in the correct location.
Until someone tells me my firmware package is missing firmwares I do not add them. Now that I know I will add this one.

mralapete, 2019-05-01:
Ok. I will attempt an 18.04 installation once again and report back for other users benefit too.

Thanks!!!!

mralapete, 2019-05-02:
The good news is the cx24116 driver did the trick. There's a number of different versions of this driver floating about but I can confirm that following these instructions exactly as they are posted got the HVR 4000 up and running under 18.04 LTS (as of the 02/05/2019). I suggest if anyone needs to use them read the instructions carefully so you download, extract and install the driver correctly. Version Firmware v1.26.90.0 is what you are after. 

https://www.linuxtv.org/wiki/index.php/Hauppauge_WinTV-HVR-4000

Once again thanks for pointing me in the right direction.

closed 2019-05-08 by mralapete