b-rad-NDi/Ubuntu-media-tree-kernel-builder #73: WinTV-quadHD issue with kernel 4.18 in Ubuntu 18.04

marcusboon, 2019-05-08:
I just noticed that there were HWE enabled kernel versions and so I installed them to try out, channel tuning seems to work fine but when trying to view live TV in plex, it comes up with:

> Could not tune channel. Please check your tuner or antenna.

The output of /var/log/syslog when clicking into a live TV channel is as follows:

> May  8 15:05:22 malachor kernel: [ 3264.523589] dvb_frontend: dvb_frontend_get_frequency_limits: frequency interval: tuner: 42000000...870000000, frontend: 48000000...870000000
> May  8 15:05:22 malachor kernel: [ 3584.984092] dvb_frontend: dvb_frontend_get_frequency_limits: frequency interval: tuner: 42000000...870000000, frontend: 48000000...870000000
>...
> May  8 15:05:23 malachor kernel: [ 3585.218113] si2157 5-0060: found a 'Silicon Labs Si2157-A30'
> May  8 15:05:23 malachor kernel: [ 3585.240903] si2157 5-0060: firmware version: 3.0.5
> ...
> May  8 15:05:29 malachor kernel: [ 3587.988610] dvb_frontend: dvb_frontend_get_frequency_limits: frequency interval: tuner: 42000000...870000000, frontend: 48000000...870000000

And sudo dmesg gives:
> [ 1751.922072] dvb_frontend: dvb_frontend_get_frequency_limits: frequency interval: tuner: 42000000...870000000, frontend: 48000000...870000000
> [ 1751.930156] si2168 16-0066: downloading firmware from file 'dvb-demod-si2168-b40-01.fw'
> [ 1751.930351] si2168 16-0064: downloading firmware from file 'dvb-demod-si2168-b40-01.fw'
> [ 1753.176947] si2168 16-0066: firmware version: B 4.0.11
> [ 1753.179231] si2168 16-0064: firmware version: B 4.0.11
> [ 1753.187691] si2157 17-0062: found a 'Silicon Labs Si2157-A30'
> [ 1753.190375] si2157 17-0060: found a 'Silicon Labs Si2157-A30'
> [ 1753.237852] si2157 17-0062: firmware version: 3.0.5
> [ 1753.240144] si2157 17-0060: firmware version: 3.0.5
> [ 1753.244990] dvb_frontend: dvb_frontend_get_frequency_limits: frequency interval: tuner: 42000000...870000000, frontend: 48000000...870000000
> ...
> [ 1830.094543] cx23885: cx23885[1]: mpeg risc op code error
> [ 1830.094551] cx23885: cx23885[1]: TS1 B - dma channel status dump
> [ 1830.094555] cx23885: cx23885[1]:   cmds: init risc lo   : 0xea3aa000
> [ 1830.094558] cx23885: cx23885[1]:   cmds: init risc hi   : 0x00000000
> [ 1830.094562] cx23885: cx23885[1]:   cmds: cdt base       : 0x00010580
> [ 1830.094565] cx23885: cx23885[1]:   cmds: cdt size       : 0x0000000a
> [ 1830.094568] cx23885: cx23885[1]:   cmds: iq base        : 0x00010400
> [ 1830.094572] cx23885: cx23885[1]:   cmds: iq size        : 0x00000010
> [ 1830.094575] cx23885: cx23885[1]:   cmds: risc pc lo     : 0xea3aa048
> [ 1830.094578] cx23885: cx23885[1]:   cmds: risc pc hi     : 0x00000000
> [ 1830.094582] cx23885: cx23885[1]:   cmds: iq wr ptr      : 0x00004100
> [ 1830.094585] cx23885: cx23885[1]:   cmds: iq rd ptr      : 0x00004104
> [ 1830.094588] cx23885: cx23885[1]:   cmds: cdt current    : 0x00010588
> [ 1830.094592] cx23885: cx23885[1]:   cmds: pci target lo  : 0xeab28000
> [ 1830.094595] cx23885: cx23885[1]:   cmds: pci target hi  : 0x00000000
> [ 1830.094598] cx23885: cx23885[1]:   cmds: line / byte    : 0x00000000
> [ 1830.094601] cx23885: cx23885[1]:   risc0:
> [ 1830.094602] 0x1c0002f0 [ write sol eol count=752 ]
> [ 1830.094607] cx23885: cx23885[1]:   risc1:
> [ 1830.094608] 0xeab28000 [ INVALID sol irq2 23 21 20 cnt1 resync count=0 ]
> [ 1830.094614] cx23885: cx23885[1]:   risc2:
> [ 1830.094615] 0x00000000 [ INVALID count=0 ]
> [ 1830.094618] cx23885: cx23885[1]:   risc3:
> [ 1830.094619] 0x1c0002f0 [ write sol eol count=752 ]
> [ 1830.094624] cx23885: cx23885[1]:   (0x00010400) iq 0:
> [ 1830.094624] 0x70000000 [ jump count=0 ]
> [ 1830.094628] cx23885: cx23885[1]:   iq 1: 0x1c0002f0 [ arg #1 ]
> [ 1830.094632] cx23885: cx23885[1]:   iq 2: 0xeab28000 [ arg #2 ]
> [ 1830.094635] cx23885: cx23885[1]:   (0x0001040c) iq 3:
> [ 1830.094636] 0x00000000 [ INVALID count=0 ]
> [ 1830.094640] cx23885: cx23885[1]:   (0x00010410) iq 4:
> [ 1830.094640] 0x1c0002f0 [ write sol eol count=752 ]
> [ 1830.094645] cx23885: cx23885[1]:   iq 5: 0xeab282f0 [ arg #1 ]
> [ 1830.094648] cx23885: cx23885[1]:   iq 6: 0x00000000 [ arg #2 ]
> [ 1830.094652] cx23885: cx23885[1]:   (0x0001041c) iq 7:
> [ 1830.094652] 0x1c0002f0 [ write sol eol count=752 ]
> [ 1830.094657] cx23885: cx23885[1]:   iq 8: 0xeab285e0 [ arg #1 ]
> [ 1830.094660] cx23885: cx23885[1]:   iq 9: 0x00000000 [ arg #2 ]
> [ 1830.094663] cx23885: cx23885[1]:   (0x00010428) iq a:
> [ 1830.094664] 0x1c0002f0 [ write sol eol count=752 ]
> [ 1830.094669] cx23885: cx23885[1]:   iq b: 0xeab288d0 [ arg #1 ]
> [ 1830.094672] cx23885: cx23885[1]:   iq c: 0x00000000 [ arg #2 ]
> [ 1830.094675] cx23885: cx23885[1]:   (0x00010434) iq d:
> [ 1830.094676] 0x1c0002f0 [ write sol eol count=752 ]
> [ 1830.094680] cx23885: cx23885[1]:   iq e: 0xeab28bc0 [ arg #1 ]
> [ 1830.094684] cx23885: cx23885[1]:   iq f: 0x00000000 [ arg #2 ]
> [ 1830.094685] cx23885: cx23885[1]: fifo: 0x00005000 -> 0x6000
> [ 1830.094685] cx23885: cx23885[1]: ctrl: 0x00010400 -> 0x10460
> [ 1830.094689] cx23885: cx23885[1]:   ptr1_reg: 0x00005000
> [ 1830.094692] cx23885: cx23885[1]:   ptr2_reg: 0x00010588
> [ 1830.094695] cx23885: cx23885[1]:   cnt1_reg: 0x00000000
> [ 1830.094698] cx23885: cx23885[1]:   cnt2_reg: 0x00000009

b-rad-NDi, 2019-05-08:
Did you already review the other posted issues about this? What cpu do you use?

Are you saying that the "original 4.15" kernel did not do this, but the HWE kernel did?

marcusboon, 2019-05-09:
Hi @b-rad-NDi,

Yes I have already reviewed the other related issues on the matter (#17, #51, #69) and while it looks like it should be resolved, it isn't. I am on a Ryzen system (1600X) with an Asrock B350 motherboard with the latest BIOS (5.80).

Here are the results of `lspci -v -nn -k`:

> 00:00.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) Root Complex [1022:1450]
>         Subsystem: ASRock Incorporation Family 17h (Models 00h-0fh) Root Complex [1849:1450]
>         Flags: fast devsel
> 
> 00:00.2 IOMMU [0806]: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) I/O Memory Management Unit [1022:1451]
>         Subsystem: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) I/O Memory Management Unit [1022:1451]
>         Flags: fast devsel, IRQ 25
>         Capabilities: [40] Secure device <?>
>         Capabilities: [64] MSI: Enable+ Count=1/4 Maskable- 64bit+
>         Capabilities: [74] HyperTransport: MSI Mapping Enable+ Fixed+
> 
> 00:01.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) PCIe Dummy Host Bridge [1022:1452]
>         Flags: fast devsel
> 
> 00:01.3 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) PCIe GPP Bridge [1022:1453] (prog-if 00 [Normal decode])
>         Flags: bus master, fast devsel, latency 0, IRQ 26
>         Bus: primary=00, secondary=03, subordinate=1c, sec-latency=0
>         I/O behind bridge: 0000f000-0000ffff
>         Memory behind bridge: fc400000-fc9fffff
>         Prefetchable memory behind bridge: 00000000f0300000-00000000f03fffff
>         Capabilities: [50] Power Management version 3
>         Capabilities: [58] Express Root Port (Slot+), MSI 00
>         Capabilities: [a0] MSI: Enable+ Count=1/1 Maskable- 64bit+
>         Capabilities: [c0] Subsystem: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) PCIe GPP Bridge [1022:1453]
>         Capabilities: [c8] HyperTransport: MSI Mapping Enable+ Fixed+
>         Capabilities: [100] Vendor Specific Information: ID=0001 Rev=1 Len=010 <?>
>         Capabilities: [150] Advanced Error Reporting
>         Capabilities: [270] #19
>         Capabilities: [2a0] Access Control Services
>         Capabilities: [370] L1 PM Substates
>         Capabilities: [3c4] #23
>         Kernel driver in use: pcieport
> 
> 00:02.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) PCIe Dummy Host Bridge [1022:1452]
>         Flags: fast devsel
> 
> 00:03.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) PCIe Dummy Host Bridge [1022:1452]
>         Flags: fast devsel
> 
> 00:03.1 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) PCIe GPP Bridge [1022:1453] (prog-if 00 [Normal decode])
>         Flags: bus master, fast devsel, latency 0, IRQ 27
>         Bus: primary=00, secondary=1d, subordinate=1d, sec-latency=0
>         I/O behind bridge: 0000e000-0000efff
>         Memory behind bridge: fce00000-fcefffff
>         Prefetchable memory behind bridge: 00000000e0000000-00000000f01fffff
>         Capabilities: [50] Power Management version 3
>         Capabilities: [58] Express Root Port (Slot+), MSI 00
>         Capabilities: [a0] MSI: Enable+ Count=1/1 Maskable- 64bit+
>         Capabilities: [c0] Subsystem: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) PCIe GPP Bridge [1022:1453]
>         Capabilities: [c8] HyperTransport: MSI Mapping Enable+ Fixed+
>         Capabilities: [100] Vendor Specific Information: ID=0001 Rev=1 Len=010 <?>
>         Capabilities: [150] Advanced Error Reporting
>         Capabilities: [270] #19
>         Capabilities: [2a0] Access Control Services
>         Capabilities: [370] L1 PM Substates
>         Capabilities: [3c4] #23
>         Kernel driver in use: pcieport
> 
> 00:04.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) PCIe Dummy Host Bridge [1022:1452]
>         Flags: fast devsel
> 
> 00:07.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) PCIe Dummy Host Bridge [1022:1452]
>         Flags: fast devsel
> 
[... 330 lines cut ...]
>         Capabilities: [50] Power Management version 3
>         Capabilities: [64] Express Endpoint, MSI 00
>         Capabilities: [a0] MSI: Enable+ Count=1/16 Maskable- 64bit+
>         Capabilities: [d0] SATA HBA v1.0
>         Capabilities: [100] Vendor Specific Information: ID=0001 Rev=1 Len=010 <?>
>         Capabilities: [150] Advanced Error Reporting
>         Capabilities: [2a0] Access Control Services
>         Kernel driver in use: ahci
>         Kernel modules: ahci
> 
> 1f:00.3 Audio device [0403]: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) HD Audio Controller [1022:1457]
>         Subsystem: ASRock Incorporation Family 17h (Models 00h-0fh) HD Audio Controller [1849:6893]
>         Flags: bus master, fast devsel, latency 0, IRQ 52
>         Memory at fcd00000 (32-bit, non-prefetchable) [size=32K]
>         Capabilities: [48] Vendor Specific Information: Len=08 <?>
>         Capabilities: [50] Power Management version 3
>         Capabilities: [64] Express Endpoint, MSI 00
>         Capabilities: [a0] MSI: Enable+ Count=1/1 Maskable- 64bit+
>         Capabilities: [100] Vendor Specific Information: ID=0001 Rev=1 Len=010 <?>
>         Capabilities: [150] Advanced Error Reporting
>         Capabilities: [2a0] Access Control Services
>         Kernel driver in use: snd_hda_intel
>         Kernel modules: snd_hda_intel

And yes kernel 4.15 used to work, I'll need to downgrade the kernel to double confirm but the latest linux-hwe-* stuff with kernel 4.18 doesn't work.

marcusboon, 2019-05-09:
Hmm actually belay that, there was a new version of the plexmediaserver (1.15.4.994) and that seems to have resolved the issue. `/var/log/syslog` still gets spammed with:

> May  9 10:42:19 malachor kernel: [ 2988.419982] dvb_frontend: dvb_frontend_get_frequency_limits: frequency interval: tuner: 42000000...870000000, frontend: 48000000...870000000

But at least now live TV shows up though I have yet to test out timed recordings.

marcusboon, 2019-05-09:
Ok so it looks like all it needed was a new version of plexmediaserver and now the mpeg risc op error code is gone and live TV and recording is working again.

closed 2019-05-09 by marcusboon

b-rad-NDi, 2019-05-10:
It's incredibly disturbing to know that this could be user land related. I'll have to instrument and review for theoretical race conditions.

mtraxiot, 2019-05-18:
I'm getting this symptoms using my quadhd hauppauge tuner ie dmesgj mpeg risc op code error
note I'm using mythtv.
what output do you need to debug it?

mtraxiot, 2019-05-18:
kernel log

> May 19 09:15:51 mythbox kernel: [63631.130604] cx23885: cx23885[0]: mpeg risc op code error
May 19 09:15:51 mythbox kernel: [63631.130614] cx23885: cx23885[0]: TS1 B - dma channel status dump
May 19 09:15:51 mythbox kernel: [63631.130619] cx23885: cx23885[0]:   cmds: init risc lo   : 0xfaebe000
May 19 09:15:51 mythbox kernel: [63631.130623] cx23885: cx23885[0]:   cmds: init risc hi   : 0x00000000
May 19 09:15:51 mythbox kernel: [63631.130627] cx23885: cx23885[0]:   cmds: cdt base       : 0x00010870
May 19 09:15:51 mythbox kernel: [63631.130631] cx23885: cx23885[0]:   cmds: cdt size       : 0x0000000a
May 19 09:15:51 mythbox kernel: [63631.130635] cx23885: cx23885[0]:   cmds: iq base        : 0x00010630
May 19 09:15:51 mythbox kernel: [63631.130639] cx23885: cx23885[0]:   cmds: iq size        : 0x00000010
May 19 09:15:51 mythbox kernel: [63631.130643] cx23885: cx23885[0]:   cmds: risc pc lo     : 0xfaebe018
May 19 09:15:51 mythbox kernel: [63631.130647] cx23885: cx23885[0]:   cmds: risc pc hi     : 0x00000000
May 19 09:15:51 mythbox kernel: [63631.130651] cx23885: cx23885[0]:   cmds: iq wr ptr      : 0x00004192
May 19 09:15:51 mythbox kernel: [63631.130655] cx23885: cx23885[0]:   cmds: iq rd ptr      : 0x0000418c
May 19 09:15:51 mythbox kernel: [63631.130658] cx23885: cx23885[0]:   cmds: cdt current    : 0x00010898
May 19 09:15:51 mythbox kernel: [63631.130662] cx23885: cx23885[0]:   cmds: pci target lo  : 0xffc72920
May 19 09:15:51 mythbox kernel: [63631.130666] cx23885: cx23885[0]:   cmds: pci target hi  : 0x00000000
May 19 09:15:51 mythbox kernel: [63631.130670] cx23885: cx23885[0]:   cmds: line / byte    : 0x000e0000
May 19 09:15:51 mythbox kernel: [63631.130674] cx23885: cx23885[0]:   risc0:
May 19 09:15:51 mythbox kernel: [63631.130676] 0x1c0002f0 [ write sol eol count=752 ]
May 19 09:15:51 mythbox kernel: [63631.130682] cx23885: cx23885[0]:   risc1:
May 19 09:15:51 mythbox kernel: [63631.130683] 0xffc72630 [ INVALID sol eol irq2 irq1 23 22 18 cnt1 cnt0 13 count=1584 ]
May 19 09:15:51 mythbox kernel: [63631.130692] cx23885: cx23885[0]:   risc2:
May 19 09:15:51 mythbox kernel: [63631.130693] 0x00000000 [ INVALID count=0 ]
May 19 09:15:51 mythbox kernel: [63631.130697] cx23885: cx23885[0]:   risc3:
May 19 09:15:51 mythbox kernel: [63631.130698] 0x1c0002f0 [ write sol eol count=752 ]
May 19 09:15:51 mythbox kernel: [63631.130704] cx23885: cx23885[0]:   (0x00010630) iq 0:
May 19 09:15:51 mythbox kernel: [63631.130705] 0x1c0002f0 [ write sol eol count=752 ]
May 19 09:15:51 mythbox kernel: [63631.130711] cx23885: cx23885[0]:   iq 1: 0xffc72c10 [ arg #1 ]
May 19 09:15:51 mythbox kernel: [63631.130715] cx23885: cx23885[0]:   iq 2: 0x00000000 [ arg #2 ]
May 19 09:15:51 mythbox kernel: [63631.130719] cx23885: cx23885[0]:   (0x0001063c) iq 3:
May 19 09:15:51 mythbox kernel: [63631.130720] 0x1c0002f0 [ write sol eol count=752 ]
May 19 09:15:51 mythbox kernel: [63631.130725] cx23885: cx23885[0]:   iq 4: 0xffc72f00 [ arg #1 ]
May 19 09:15:51 mythbox kernel: [63631.130729] cx23885: cx23885[0]:   iq 5: 0x00000000 [ arg #2 ]
May 19 09:15:51 mythbox kernel: [63631.130733] cx23885: cx23885[0]:   (0x00010648) iq 6:
May 19 09:15:51 mythbox kernel: [63631.130734] 0x00000000 [ INVALID count=0 ]

mtraxiot, 2019-05-18:
should I open another issue?

mtraxiot, 2019-05-19:
also seeing these messages:

>y 19 19:55:33 mythbox kernel: [ 1954.010362] dvb_frontend: dvb_frontend_get_frequency_limits: frequency interval: tuner: 42000000...870000000, frontend: 48000000...870000000
May 19 19:55:35 mythbox kernel: [ 1956.037140] dvb_frontend: dvb_frontend_get_frequency_limits: frequency interval: tuner: 42000000...870000000, frontend: 48000000...870000000
May 19 19:55:37 mythbox kernel: [ 1958.081029] dvb_frontend: dvb_frontend_get_frequency_limits: frequency interval: tuner: 42000000...870000000, frontend: 48000000...870000000
May 19 19:55:39 mythbox kernel: [ 1960.132670] dvb_frontend: dvb_frontend_get_frequency_limits: frequency interval: tuner: 42000000...870000000, frontend: 48000000...870000000