b-rad-NDi/Ubuntu-media-tree-kernel-builder #74: PCTV 292e/Si2168 unable to scan for channels on ESXi 6.7.0 Ubuntu VM

xamguy, 2019-05-11:
Hi

I'm running Ubuntu 18.4.2 LTS with the latest 4.15.0 kernel patched from here in a VM on ESXi 6.7.0U1 (13004448), using a PCTV 292e USB tuner. Unfortunately when scanning for channels in Tvheadend I get 'no data' or 'poll timeout' errors, and using w_scan I get 'no data from PAT after 2 seconds' for each mux it finds.

```
root@kraz:~# w_scan -ft -c GB -X
w_scan -ft -c GB -X
w_scan version 20170107 (compiled for DVB API 5.10)
using settings for UNITED KINGDOM
DVB aerial
DVB-T GB
scan type TERRESTRIAL, channellist 6
output format czap/tzap/szap/xine
WARNING: could not guess your codepage. Falling back to 'UTF-8'
output charset 'UTF-8', use -C <charset> to override
Info: using DVB adapter auto detection.
        /dev/dvb/adapter0/frontend0 -> TERRESTRIAL "Silicon Labs Si2168": very good :-))

Using TERRESTRIAL frontend (adapter /dev/dvb/adapter0/frontend0)
-_-_-_-_ Getting frontend capabilities-_-_-_-_
Using DVB API 5.11
frontend 'Silicon Labs Si2168' supports
DVB-T2
INVERSION_AUTO
QAM_AUTO
TRANSMISSION_MODE_AUTO
GUARD_INTERVAL_AUTO
HIERARCHY_AUTO
FEC_AUTO
BANDWIDTH_AUTO not supported, trying 6/7/8 MHz.
FREQ (48.00MHz ... 870.00MHz)
-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_
Scanning DVB-T...
Scanning 8MHz frequencies...
```
```
578000: (time: 01:37.625)         signal ok:    QAM_AUTO f = 578000 kHz I999B8C999D999T999G999Y999 (0:0:0)
        Info: no data from PAT after 2 seconds
        deleting (QAM_AUTO f = 578000 kHz I999B8C999D999T999G999Y999 (0:0:0))
```

No channels are detected after a full scan. My dmesg looks like this:

```
root@kraz:~# dmesg | egrep "dvb|frontend|firmware|si2"
[    0.005071] Spectre V2 : Enabling Restricted Speculation for firmware calls
[    8.200062] em28xx 1-2:1.0: dvb set to isoc mode.
[    8.239267] si2168 1-0064: Silicon Labs Si2168-B40 successfully identified
[    8.239268] si2168 1-0064: firmware version: B 4.0.2
[    8.247686] si2157 2-0060: Silicon Labs Si2147/2148/2157/2158 successfully attached
[    8.247701] dvbdev: DVB: registering new adapter (1-2:1.0)
[    8.247703] em28xx 1-2:1.0: DVB: registering adapter 0 frontend 0 (Silicon Labs Si2168)...
[    8.248068] em28xx: Registered (Em28xx dvb Extension) extension
[   11.029806] si2168 1-0064: downloading firmware from file 'dvb-demod-si2168-b40-01.fw'
[   12.343549] si2168 1-0064: firmware version: B 4.0.25
[   12.351695] si2157 2-0060: found a 'Silicon Labs Si2157-A30'
[   12.382345] si2157 2-0060: firmware version: 3.0.5
[   12.393732] dvb_frontend: dvb_frontend_get_frequency_limits: frequency interval: tuner: 42000000...870000000, frontend: 48000000...870000000
[   12.393736] em28xx 1-2:1.0: DVB: adapter 0 frontend 0 frequency 0 out of range (48000000..870000000)
[   12.393738] dvb_frontend: dvb_frontend_get_frequency_limits: frequency interval: tuner: 42000000...870000000, frontend: 48000000...870000000
[   12.418389] si2168 1-0064: downloading firmware from file 'dvb-demod-si2168-b40-01.fw'
[   14.253929] si2168 1-0064: firmware version: B 4.0.25
[   14.270339] si2157 2-0060: found a 'Silicon Labs Si2157-A30'
[   14.297170] si2157 2-0060: firmware version: 3.0.5
```

I've been searching around and landed here after finding [this issue](https://github.com/b-rad-NDi/Ubuntu-media-tree-kernel-builder/issues/31) which also seemed to relate to VMware, having already found [this forum post](https://communities.vmware.com/thread/608870) that speaks about exactly the same issue and a possible avenue for exploration with the isoc/bulk mode.

I did have this setup working perfectly using ESXi 6.0.0 and Ubuntu 14 LTS, but only realised today that my upgrade to 6.7.0U1 a couple of weeks ago had broken the playback via Tvheadend. Rebuilding the VM with Ubuntu 18.4.2 LTS and a fresh install of Tvheadend made no difference. Changing the firmware used from the default to 4.0.25 did clear some dmesg errors but didn't solve the problem. The PPA from here did not seem to change anything either. No combination of removing the USB device or USB controller from the VM, changing from a USB3 to USB2 controller, or different physical ports on the host have made any difference. I would try a VT-d passthrough but unfortunately my host CPU doesn't support it. Pretty sure VMware have changed something between 6 and 6.7.

Any ideas?! Happy to help debug the problem! Any input very welcome at this stage :)

Thanks

b-rad-NDi, 2019-05-16:
Well, you've discovered that no matter what VMWare says about passthru being passthru, it's not. VMWare cannot handle loads of usb communication, similar to what is generated by media device i2c. Setting a device to bulk transfer mode *should* work, but one of the users from that thread says they now see artifacts. I have a work in progress Linux utility to convert your isoc device to bulk mode. If you prefer a more tested conversion utility that uses the official empia windows driver just reach out to hauppauge support for it.

I've told VMWare engineers about this issue at conferences and they had the gall to say it's not their problem and if we wanted to hire their big guns then they'd look into it. The kernel drivers are fine and these issues *only* pop up in VMWware, and not KVM/qemu, so as far as I'm concerned this problem lies solely with VMWare. Unfortunately they don't seem to care.

xamguy, 2019-05-16:
Hey Brad, thanks for your insight. I guess for Hauppauge this is really an edge case situation, and likewise for VMware! Just a pity VMware can't look into it, it's clearly something they've introduced since 6.0.0.

I'd be happy to test out the WIP Linux util to convert to bulk mode. Maybe I'll get lucky and not have artifact issues. Presumably it can be changed back to isoc easily enough?

b-rad-NDi, 2019-05-16:
The utility is eeprom-tinker. I think I've pushed up all my commits so far, but I'll double check. It currently does not do the switch for you, instead it outputs i2c-set commands for you to execute. Please be careful and ask any questions. It is the same process to convert between bulk & isoc, so no problem if you for some reason want to go back to isoc.

b-rad-NDi, 2019-05-16:
Looks like eeprom-tinker is up to date.
```
# device discovery
./eeprom-tinker -d -v

# device modification instructions
./eeprom-tinker -v -b <bus_num> -m both
```

closed 2019-05-16 by b-rad-NDi

xamguy, 2019-05-16:
Ok got eeprom-tinker going, but, and I suspect I know the answer to this, it's not going to work from within a VMware VM is it?

```
# ./eeprom-tinker -d -v
EEPROM Tinker
        Welcome to the danger zone
        Be careful, this utility can harm your system if misused!

=========================================
0 em28xx devices found
=========================================
```

b-rad-NDi, 2019-05-16:
Hmmm...that stage should have worked, even in vmware. It's really just scanning sysfs looking at em28xx devices. Can you file a ticket about this? I'll want to see how your sysfs is formatted or add more debugging info.

I do not recommend trying to actually do the conversion in a vm though.