b-rad-NDi/Ubuntu-media-tree-kernel-builder #75: [Kernel 5.1] workqueue hangs on em28xx disconneting.

comio, 2019-05-13:
Hi,

I already know that your patches are for ubuntu's kernel. Today I faced the following kernel kernel trace on a custom 5.1.1 kernel. (But I faced the same also with the previous 5.1).

Thanks in advance for your effort and your help to solve this.

Ciao

luigi

```
[44471.312219] usb 1-1.2: USB disconnect, device number 5
[44471.312368] em28xx 1-1.2:1.0: Disconnecting em28xx #1
[44471.312428] WARNING: CPU: 1 PID: 29308 at kernel/workqueue.c:3030 __flush_work+0x1d2/0x1f0
[44471.312501] Modules linked in: veth nf_conntrack_netlink xfrm_user xfrm_algo br_netfilter cfg80211 bridge stp llc xt_conntrack nft_counter nft_chain_nat xt_nat xt_tcpudp ipt_MASQUERADE nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 xt_addrtype em28xx_rc nft_compat rc_core si2157 si2168 nft_chain_route_ipv4 i2c_mux em28xx_dvb dvb_core nf_tables nfnetlink intel_rapl intel_powerclamp coretemp kvm_intel snd_hda_codec_hdmi kvm snd_hda_codec_realtek snd_hda_codec_generic snd_hda_intel irqbypass snd_hda_codec punit_atom_debug input_leds iTCO_wdt led_class crct10dif_pclmul mei_hdcp crc32_pclmul iTCO_vendor_support ghash_clmulni_intel r8169 snd_hwdep snd_hda_core realtek snd_pcm snd_timer libphy em28xx aesni_intel snd tveeprom aes_x86_64 v4l2_common crypto_simd uas videodev cryptd usb_storage glue_helper hid_generic i2c_i801 media pcspkr lpc_ich mei_txe mei soundcore acpi_pad intel_int0002_vgpio rtc_cmos pcc_cpufreq vfat fat zstd sch_fq_codel binfmt_misc zram ip_tables x_tables xhci_pci xhci_hcd
[44471.312565]  ahci libahci i2c_hid pwm_lpss_platform pwm_lpss pinctrl_cherryview nct6775 hwmon_vid
[44471.313381] CPU: 1 PID: 29308 Comm: kworker/1:1 Tainted: G                T 5.1.1-gentoo #1
[44471.313454] Hardware name: To Be Filled By O.E.M. To Be Filled By O.E.M./N3700-ITX, BIOS P2.00 04/16/2018
[44471.313541] Workqueue: usb_hub_wq hub_event
[44471.313582] RIP: 0010:__flush_work+0x1d2/0x1f0
[44471.313624] Code: 00 00 45 31 ed e9 74 ff ff ff 0f 0b e9 6d ff ff ff 4c 89 f7 c6 07 00 0f 1f 40 00 fb 66 0f 1f 44 00 00 45 31 ed e9 54 ff ff ff <0f> 0b 45 31 ed e9 4a ff ff ff e8 2f 0d fe ff 66 66 2e 0f 1f 84 00
[44471.313782] RSP: 0000:ffffb16b0305fba8 EFLAGS: 00010246
[44471.313830] RAX: 0000000000000000 RBX: ffff9a2015c93060 RCX: 0000000000000000
[44471.313893] RDX: 0000000000000007 RSI: 0000000000000000 RDI: ffff9a2015c93060
[44471.313956] RBP: ffff9a201ad10000 R08: 00000000000006cd R09: ffffb16b3037821c
[44471.314019] R10: 0000000000aaaaaa R11: 0000000000000000 R12: ffff9a2026afd000
[44471.314081] R13: 0000000000000001 R14: ffff9a20241ce400 R15: ffff9a20241ce430
[44471.314145] FS:  0000000000000000(0000) GS:ffff9a2030280000(0000) knlGS:0000000000000000
[44471.314215] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[44471.314266] CR2: 00007f3d3c749000 CR3: 000000026a34c000 CR4: 00000000001006e0
[44471.314330] Call Trace:
[44471.314362]  ? _dev_info+0x6c/0x83
[44471.314404]  em28xx_usb_disconnect.cold+0x4b/0xc2 [em28xx]
[44471.314458]  usb_unbind_interface+0x86/0x250
[44471.314501]  device_release_driver_internal+0xe4/0x1d0
[44471.314549]  bus_remove_device+0xe5/0x150
[44471.314590]  device_del+0x159/0x370
[44471.314625]  usb_disable_device+0x93/0x240
[44471.314665]  usb_disconnect+0xba/0x250
[44471.314702]  ? set_port_feature+0x33/0x40
[44471.314741]  hub_event+0xd1c/0x14d0
[44471.314777]  ? native_usergs_sysret64+0x1/0x10
[44471.314821]  process_one_work+0x1da/0x3c0
[44471.314860]  worker_thread+0x4d/0x3f0
[44471.314896]  kthread+0xfb/0x130
[44471.314928]  ? process_one_work+0x3c0/0x3c0
[44471.314967]  ? kthread_park+0x90/0x90
[44471.315002]  ret_from_fork+0x35/0x40
[44471.315038] ---[ end trace 0986dc28b53c73bd ]---
[44471.315085] em28xx 1-1.2:1.0: Disconnecting em28xx
[44471.315132] em28xx 1-1.2:1.0: Closing DVB extension
[44471.317250] em28xx 1-1.2:1.0: Closing DVB extension
[44471.319399] em28xx 1-1.2:1.0: Closing input extension
[44471.319460] em28xx 1-1.2:1.0: Closing input extension
[44471.392196] em28xx 1-1.2:1.0: Freeing device
[44471.392253] em28xx 1-1.2:1.0: Freeing device

```

parkerlreed, 2019-05-13:
Do subsequent plugs work? I think I saw this on 5.0.x but the device still operated fine.

comio, 2019-05-13:
Maybe a hw fault... Argh.

Ciao

Luigi

closed 2019-05-13 by comio

parkerlreed, 2019-05-13:
Umm, if this is an issue, you should leave it open. Does this happen randomly or only on unplug?

comio, 2019-05-13:
My tuner is not shown by lsusb and the power doesn't light. I think that the random errors were a preview of the imminent dead.

parkerlreed, 2019-05-17:
Maybe try a reboot. My old tuner the LED died but it still worked fine. 

If the module crashed it'll take a reboot for it to come back.

comio, 2019-05-21:
To confirm that I was wrong ad I had an hw fault, I replaced my broken DualHD and it works of course.

ciao

luigi