b-rad-NDi/Ubuntu-media-tree-kernel-builder #76: Ubuntu 18.04 QuadHD  mpeg risc op code error

mtraxiot, 2019-05-20:
I'm using MythTV and swapped my legacy PCI cards with Hauppauge-QuadHD PCIe but can't seem to get this working on my server.
I've installed media tree kernel : 4.15.0-47201904171620-generic #0+mediatree+hauppauge-Ubuntu
but when MythTV tries to used this adapter I get  :  mpeg risc op code error

> y 20 17:05:27 mythbox mythtv.desktop[23511]: 2019-05-20 17:05:27.570457 I  VDPAU: Added 2 output surfaces (total 4, max 4)
May 20 17:05:28 mythbox kernel: [78149.419875] dvb_frontend: dvb_frontend_get_frequency_limits: frequency interval: tuner: 42000000...870000000, frontend: 48000000...870000000
May 20 17:05:28 mythbox kernel: [78150.367823] cx23885: cx23885[0]: mpeg risc op code error
May 20 17:05:28 mythbox kernel: [78150.367832] cx23885: cx23885[0]: TS1 B - dma channel status dump
May 20 17:05:28 mythbox kernel: [78150.367837] cx23885: cx23885[0]:   cmds: init risc lo   : 0xfb69e000
May 20 17:05:28 mythbox kernel: [78150.367841] cx23885: cx23885[0]:   cmds: init risc hi   : 0x00000000
May 20 17:05:28 mythbox kernel: [78150.367845] cx23885: cx23885[0]:   cmds: cdt base       : 0x00010870
May 20 17:05:28 mythbox kernel: [78150.367849] cx23885: cx23885[0]:   cmds: cdt size       : 0x0000000a
May 20 17:05:28 mythbox kernel: [78150.367853] cx23885: cx23885[0]:   cmds: iq base        : 0x00010630
May 20 17:05:28 mythbox kernel: [78150.367857] cx23885: cx23885[0]:   cmds: iq size        : 0x00000010
May 20 17:05:28 mythbox kernel: [78150.367861] cx23885: cx23885[0]:   cmds: risc pc lo     : 0xfb69e048
May 20 17:05:28 mythbox kernel: [78150.367865] cx23885: cx23885[0]:   cmds: risc pc hi     : 0x00000000
May 20 17:05:28 mythbox kernel: [78150.367869] cx23885: cx23885[0]:   cmds: iq wr ptr      : 0x00004194

mtraxiot, 2019-05-20:
full syslog output:

> ay 20 17:05:27 mythbox kernel: [78141.977427] dvb_frontend: dvb_frontend_get_frequency_limits: frequency interval: tuner: 45000000...864000000, frontend: 177000000...858000000
May 20 17:05:27 mythbox kernel: [78149.389685] si2157 5-0060: found a 'Silicon Labs Si2157-A30'
May 20 17:05:27 mythbox kernel: [78149.412217] si2157 5-0060: firmware version: 3.0.5
May 20 17:05:27 mythbox kernel: [78149.415366] dvb_frontend: dvb_frontend_get_frequency_limits: frequency interval: tuner: 42000000...870000000, frontend: 48000000...870000000

snip
>May 20 17:05:27 mythbox kernel: [78149.415366] dvb_frontend: dvb_frontend_get_frequency_limits: frequency interval: tuner: 42000000...870000000, frontend: 48000000...870000000
May 20 17:05:27 mythbox mythtv: mythtv recording:card=1,CH=3021, Fake Or Fortune?-Gauguin
May 20 17:05:27 mythbox mythtv.desktop[23511]: 2019-05-20 17:05:27.426674 E  ALSA: snd_pcm_info_get_card: Operation not permitted
May 20 17:05:27 mythbox mythtv.desktop[23511]: 2019-05-20 17:05:27.441948 N  AudioPlayer: Enabling Audio
May 20 17:05:27 mythbox mythtv.desktop[23511]: 2019-05-20 17:05:27.443718 I  Clearing OpenGL painter cache.
May 20 17:05:27 mythbox mythtv.desktop[23511]: 2019-05-20 17:05:27.485184 I  VDPAU: Created 2 output surfaces.
May 20 17:05:27 mythbox mythtv.desktop[23511]: 2019-05-20 17:05:27.485195 I  VDPAU: Created VDPAU render device 1920x1080
May 20 17:05:27 mythbox mythtv.desktop[23511]: 2019-05-20 17:05:27.511609 I  Player(1): Video timing method: USleep with busy wait
May 20 17:05:27 mythbox mythtv.desktop[23511]: 2019-05-20 17:05:27.514867 E  LiveTVChain(live-mythbox.salsola-2019-05-20T07:05:19Z): SwitchTo() not switching to current
May 20 17:05:27 mythbox mythtv.desktop[23511]: 2019-05-20 17:05:27.570457 I  VDPAU: Added 2 output surfaces (total 4, max 4)
May 20 17:05:28 mythbox kernel: [78149.419875] dvb_frontend: dvb_frontend_get_frequency_limits: frequency interval: tuner: 42000000...870000000, frontend: 48000000...870000000
May 20 17:05:28 mythbox kernel: [78150.367823] cx23885: cx23885[0]: mpeg risc op code error
May 20 17:05:28 mythbox kernel: [78150.367832] cx23885: cx23885[0]: TS1 B - dma channel status dump
May 20 17:05:28 mythbox kernel: [78150.367837] cx23885: cx23885[0]:   cmds: init risc lo   : 0xfb69e000
May 20 17:05:28 mythbox kernel: [78150.367841] cx23885: cx23885[0]:   cmds: init risc hi   : 0x00000000
May 20 17:05:28 mythbox kernel: [78150.367845] cx23885: cx23885[0]:   cmds: cdt base       : 0x00010870
May 20 17:05:28 mythbox kernel: [78150.367849] cx23885: cx23885[0]:   cmds: cdt size       : 0x0000000a
May 20 17:05:28 mythbox kernel: [78150.367853] cx23885: cx23885[0]:   cmds: iq base        : 0x00010630
May 20 17:05:28 mythbox kernel: [78150.367857] cx23885: cx23885[0]:   cmds: iq size        : 0x00000010
May 20 17:05:28 mythbox kernel: [78150.367861] cx23885: cx23885[0]:   cmds: risc pc lo     : 0xfb69e048
May 20 17:05:28 mythbox kernel: [78150.367865] cx23885: cx23885[0]:   cmds: risc pc hi     : 0x00000000
May 20 17:05:28 mythbox kernel: [78150.367869] cx23885: cx23885[0]:   cmds: iq wr ptr      : 0x00004194
May 20 17:05:28 mythbox kernel: [78150.367873] cx23885: cx23885[0]:   cmds: iq rd ptr      : 0x00004198
May 20 17:05:28 mythbox kernel: [78150.367877] cx23885: cx23885[0]:   cmds: cdt current    : 0x000108a8
May 20 17:05:28 mythbox kernel: [78150.367881] cx23885: cx23885[0]:   cmds: pci target lo  : 0xe41f8000
May 20 17:05:28 mythbox kernel: [78150.367885] cx23885: cx23885[0]:   cmds: pci target hi  : 0x00000000
May 20 17:05:28 mythbox kernel: [78150.367889] cx23885: cx23885[0]:   cmds: line / byte    : 0x00000000
May 20 17:05:28 mythbox kernel: [78150.367893] cx23885: cx23885[0]:   risc0:
May 20 17:05:28 mythbox kernel: [78150.367894] 0x1c0002f0 [ write sol eol count=752 ]
May 20 17:05:28 mythbox kernel: [78150.367900] cx23885: cx23885[0]:   risc1:
May 20 17:05:28 mythbox kernel: [78150.367901] 0xe41f8000 [ INVALID eol 20 19 18 cnt1 cnt0 resync count=0 ]
May 20 17:05:28 mythbox kernel: [78150.367909] cx23885: cx23885[0]:   risc2:
May 20 17:05:28 mythbox kernel: [78150.367910] 0x00000000 [ INVALID count=0 ]
May 20 17:05:28 mythbox kernel: [78150.367914] cx23885: cx23885[0]:   risc3:
May 20 17:05:28 mythbox kernel: [78150.367915] 0x1c0002f0 [ write sol eol count=752 ]
May 20 17:05:28 mythbox kernel: [78150.367921] cx23885: cx23885[0]:   (0x00010630) iq 0:
May 20 17:05:28 mythbox kernel: [78150.367921] 0xe41f8000 [ INVALID eol 20 19 18 cnt1 cnt0 resync count=0 ]
May 20 17:05:28 mythbox kernel: [78150.367929] cx23885: cx23885[0]:   (0x00010634) iq 1:
May 20 17:05:28 mythbox kernel: [78150.367930] 0x00000000 [ INVALID count=0 ]
May 20 17:05:28 mythbox kernel: [78150.367934] cx23885: cx23885[0]:   (0x00010638) iq 2:
May 20 17:05:28 mythbox kernel: [78150.367935] 0x1c0002f0 [ write sol eol count=752 ]
May 20 17:05:28 mythbox kernel: [78150.367940] cx23885: cx23885[0]:   iq 3: 0xe41f82f0 [ arg #1 ]
May 20 17:05:28 mythbox kernel: [78150.367945] cx23885: cx23885[0]:   iq 4: 0x00000000 [ arg #2 ]
May 20 17:05:28 mythbox kernel: [78150.367948] cx23885: cx23885[0]:   (0x00010644) iq 5:
May 20 17:05:28 mythbox kernel: [78150.367949] 0x1c0002f0 [ write sol eol count=752 ]
May 20 17:05:28 mythbox kernel: [78150.367954] cx23885: cx23885[0]:   iq 6: 0xe41f8bc0 [ arg #1 ]
May 20 17:05:28 mythbox kernel: [78150.367958] cx23885: cx23885[0]:   iq 7: 0x00000000 [ arg #2 ]
[... cut ...]
May 20 17:05:38 mythbox mythtv.desktop[23511]: 2019-05-20 17:05:38.470064 I  VDPAU Painter: Clearing VDPAU painter cache.
May 20 17:05:38 mythbox mythtv.desktop[23511]: 2019-05-20 17:05:38.493259 I  VDPAU Painter: Clearing VDPAU painter cache.
May 20 17:05:38 mythbox kernel: [78160.697943] si2157 5-0062: found a 'Silicon Labs Si2157-A30'
May 20 17:05:38 mythbox mythtv.desktop[23511]: 2019-05-20 17:05:38.576851 I  MythCoreContext::ConnectCommandSocket(): Connecting to backend server: 192.168.0.2:6543 (try 1 of 1)
May 20 17:05:38 mythbox kernel: [78160.720251] si2157 5-0062: firmware version: 3.0.5

mtraxiot, 2019-05-23:
Further note I'm running a AMD Ryzen 3 2200G if that is relevant.

b-rad-NDi, 2019-05-23:
How about the output from this:
```
 lspci -v -nn -k
```
and in case your pcie is not the variant listed in the driver, did you try the modprobe module option forced fix for this?
```
options cx23885 dma_reset_workaround=2
```

mtraxiot, 2019-05-23:
Thanks Brad,
I've created a mydvb.conf in my modprobe.conf with your settings and it
looks a lot better now

Question do I need the kernel packages for ubuntu 18.04 ?

thanks
Paul

00:00.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Device [1022:15d0]
	Subsystem: Advanced Micro Devices, Inc. [AMD] Device [1022:15d0]
	Flags: fast devsel

00:00.2 IOMMU [0806]: Advanced Micro Devices, Inc. [AMD] Device [1022:15d1]
	Subsystem: Advanced Micro Devices, Inc. [AMD] Device [1022:15d1]
	Flags: bus master, fast devsel, latency 0, IRQ 27
	Capabilities: [40] Secure device <?>
	Capabilities: [64] MSI: Enable+ Count=1/4 Maskable- 64bit+
	Capabilities: [74] HyperTransport: MSI Mapping Enable+ Fixed+

00:01.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) PCIe Dummy Host Bridge [1022:1452]
	Flags: fast devsel

00:01.1 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Device [1022:15d3] (prog-if 00 [Normal decode])
	Flags: bus master, fast devsel, latency 0, IRQ 28
	Bus: primary=00, secondary=01, subordinate=01, sec-latency=0
	I/O behind bridge: 0000f000-0000ffff
	Memory behind bridge: f5000000-f60fffff
	Prefetchable memory behind bridge: 00000000e8000000-00000000f1ffffff
	Capabilities: [50] Power Management version 3
	Capabilities: [58] Express Root Port (Slot+), MSI 00
	Capabilities: [a0] MSI: Enable+ Count=1/1 Maskable- 64bit+
	Capabilities: [c0] Subsystem: Advanced Micro Devices, Inc. [AMD] Device [1022:1453]
	Capabilities: [c8] HyperTransport: MSI Mapping Enable+ Fixed+
	Capabilities: [100] Vendor Specific Information: ID=0001 Rev=1 Len=010 <?>
	Capabilities: [150] Advanced Error Reporting
	Capabilities: [270] #19
	Capabilities: [2a0] Access Control Services
	Capabilities: [370] L1 PM Substates
	Capabilities: [3c4] #23
	Kernel driver in use: pcieport
	Kernel modules: shpchp

00:01.2 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Device [1022:15d3] (prog-if 00 [Normal decode])
	Flags: bus master, fast devsel, latency 0, IRQ 29
	Bus: primary=00, secondary=02, subordinate=0c, sec-latency=0
	I/O behind bridge: 0000e000-0000efff
	Memory behind bridge: f6400000-f72fffff
	Capabilities: [50] Power Management version 3
	Capabilities: [58] Express Root Port (Slot+), MSI 00
	Capabilities: [a0] MSI: Enable+ Count=1/1 Maskable- 64bit+
	Capabilities: [c0] Subsystem: Advanced Micro Devices, Inc. [AMD] Device [1022:1453]
	Capabilities: [c8] HyperTransport: MSI Mapping Enable+ Fixed+
	Capabilities: [100] Vendor Specific Information: ID=0001 Rev=1 Len=010 <?>
	Capabilities: [150] Advanced Error Reporting
	Capabilities: [270] #19
	Capabilities: [2a0] Access Control Services
	Capabilities: [370] L1 PM Substates
	Capabilities: [3c4] #23
	Kernel driver in use: pcieport
	Kernel modules: shpchp

00:08.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) PCIe Dummy Host Bridge [1022:1452]
	Flags: fast devsel

00:08.1 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Device [1022:15db] (prog-if 00 [Normal decode])
	Flags: bus master, fast devsel, latency 0, IRQ 30
	Bus: primary=00, secondary=0d, subordinate=0d, sec-latency=0
	Memory behind bridge: f7400000-f77fffff
	Capabilities: [50] Power Management version 3
[... 366 lines cut ...]
0d:00.6 Audio device [0403]: Advanced Micro Devices, Inc. [AMD] Device [1022:15e3]
	Subsystem: Gigabyte Technology Co., Ltd Device [1458:a182]
	Flags: bus master, fast devsel, latency 0, IRQ 66
	Memory at f7700000 (32-bit, non-prefetchable) [size=32K]
	Capabilities: [48] Vendor Specific Information: Len=08 <?>
	Capabilities: [50] Power Management version 3
	Capabilities: [64] Express Endpoint, MSI 00
	Capabilities: [a0] MSI: Enable+ Count=1/1 Maskable- 64bit+
	Capabilities: [100] Vendor Specific Information: ID=0001 Rev=1 Len=010 <?>
	Kernel driver in use: snd_hda_intel
	Kernel modules: snd_hda_intel

0e:00.0 SATA controller [0106]: Advanced Micro Devices, Inc. [AMD] FCH SATA Controller [AHCI mode] [1022:7901] (rev 61) (prog-if 01 [AHCI 1.0])
	Subsystem: Gigabyte Technology Co., Ltd FCH SATA Controller [AHCI mode] [1458:b002]
	Flags: bus master, fast devsel, latency 0, IRQ 63
	Memory at f7800000 (32-bit, non-prefetchable) [size=2K]
	Capabilities: [48] Vendor Specific Information: Len=08 <?>
	Capabilities: [50] Power Management version 3
	Capabilities: [64] Express Endpoint, MSI 00
	Capabilities: [a0] MSI: Enable+ Count=2/2 Maskable- 64bit+
	Capabilities: [d0] SATA HBA v1.0
	Capabilities: [100] Vendor Specific Information: ID=0001 Rev=1 Len=010 <?>
	Capabilities: [270] #19
	Kernel driver in use: ahci
	Kernel modules: ahci

b-rad-NDi, 2019-05-23:
Gimme output from
```
uname -a
```
on 18.04.
If you don't use analog then you might be fine if the kernel is new enough.

mtraxiot, 2019-05-24:
sorry for the late reply was at work:
output:
Linux mythbox.salsola 4.15.0-47201904171620-generic
#0+mediatree+hauppauge-Ubuntu SMP Wed Apr 17 22:20:12 UTC 2019 x86_64
x86_64 x86_64 GNU/Linux

b-rad-NDi, 2019-05-24:
It's possible 4.15 is not new enough, so you'd still require this ppa.

mtraxiot, 2019-05-24:
Thanks Brad,
What's your opinion on installing newer kernel, as I have a ryzen CPU as
well?
is it advisable to use a 4.18 or v5 kernel?
Are they stable enough?
Thanks Paul

mtraxiot, 2019-05-26:
rebooted again and getting a slightly different error..  "Audio risc op code error"

>May 26 13:05:33 mythbox kernel: [   25.813793] dvb_frontend: dvb_frontend_get_frequency_limits: frequency interval: tuner: 42000000...870000000, frontend: 48000000...870000000
May 26 13:05:33 mythbox kernel: [   26.649156] cx23885: cx23885[0]/1: Audio risc op code error
May 26 13:05:33 mythbox kernel: [   26.649161] cx23885: cx23885[0]: TV Audio - dma channel status dump
May 26 13:05:33 mythbox kernel: [   26.649165] cx23885: cx23885[0]:   cmds: init risc lo   : 0xfecc2000
May 26 13:05:33 mythbox kernel: [   26.649168] cx23885: cx23885[0]:   cmds: init risc hi   : 0x00000000
May 26 13:05:33 mythbox kernel: [   26.649172] cx23885: cx23885[0]:   cmds: cdt base       : 0x00010930
May 26 13:05:33 mythbox kernel: [   26.649175] cx23885: cx23885[0]:   cmds: cdt size       : 0x00000008
May 26 13:05:33 mythbox kernel: [   26.649178] cx23885: cx23885[0]:   cmds: iq base        : 0x000106b0
May 26 13:05:33 mythbox kernel: [   26.649181] cx23885: cx23885[0]:   cmds: iq size        : 0x00000010
May 26 13:05:33 mythbox kernel: [   26.649185] cx23885: cx23885[0]:   cmds: risc pc lo     : 0xfecc2004
May 26 13:05:33 mythbox kernel: [   26.649188] cx23885: cx23885[0]:   cmds: risc pc hi     : 0x00000000
May 26 13:05:33 mythbox kernel: [   26.649191] cx23885: cx23885[0]:   cmds: iq wr ptr      : 0x000041ad
May 26 13:05:33 mythbox kernel: [   26.649194] cx23885: cx23885[0]:   cmds: iq rd ptr      : 0x000041ac
May 26 13:05:33 mythbox kernel: [   26.649198] cx23885: cx23885[0]:   cmds: cdt current    : 0x00000000
May 26 13:05:33 mythbox kernel: [   26.649201] cx23885: cx23885[0]:   cmds: pci target lo  : 0x00000000
May 26 13:05:33 mythbox kernel: [   26.649204] cx23885: cx23885[0]:   cmds: pci target hi  : 0x00000000
May 26 13:05:33 mythbox kernel: [   26.649207] cx23885: cx23885[0]:   cmds: line / byte    : 0x00000000
May 26 13:05:33 mythbox kernel: [   26.649210] cx23885: cx23885[0]:   risc0:
May 26 13:05:33 mythbox kernel: [   26.649211] 0x00000000 [ INVALID count=0 ]
May 26 13:05:33 mythbox kernel: [   26.649215] cx23885: cx23885[0]:   risc1:

b-rad-NDi, 2019-05-29:
Audio opcode errors are related to the analog capability. They should not affect digital television.

mtraxiot, 2019-05-30:
Thanks do you recommend updating kernel version to v5?

b-rad-NDi, 2019-05-30:
No reason to upgrade unless you have one.
I'd still like output from this:
```
 lspci -v -nn -k
```

closed 2019-05-30 by b-rad-NDi

mtraxiot, 2019-05-30:
attached output
[lspci.txt](https://github.com/b-rad-NDi/Ubuntu-media-tree-kernel-builder/files/3238717/lspci.txt)

b-rad-NDi, 2019-05-30:
Thank you, I'll submit a patch to get your system included upstream.

mtraxiot, 2019-05-30:
much appreciated.