b-rad-NDi/Ubuntu-media-tree-kernel-builder #77: Add support for SAA 7162 DVB-S chipsets

tapwag, 2019-05-24:
As discussed on IRC I am filing a request that I would entail incorporating support for DVB-S chipsets. 

According to "/sbin/lspci" I am using the following device:

`01:00.0 Multimedia controller: Philips Semiconductors SAA7162 (rev 01)`

Linuxtv.org also has a [listing for my card](https://www.linuxtv.org/wiki/index.php/Pinnacle_PCTV_7010iX).

/sbin/lspci -vnn on my current openSUSE system currently gives me:

```
01:00.0 Multimedia controller [0480]: Philips Semiconductors SAA7162
 [1131:7162] (rev 01) Subsystem: Pinnacle Systems Inc. Pinnacle PCTV 
7010iX TV Card [11bd:0101] Flags: bus master, fast devsel, latency
0, IRQ 5 Memory at ffa00000 (64-bit, non-prefetchable) [size=1M] 
Capabilities: <access denied>
```
Please let me know should you require further information.

b-rad-NDi, 2019-05-24:
The only driver "missing" is the DVB-S tuner, ZL10037. The zl10039 driver appears to be closely compatible. All of the registers are the same, so I'm hoping the driver is ok as is.
I'll make a build and upload it somewhere soon.

b-rad-NDi, 2019-05-24:
I'll need output from:
```
grep 11bd /proc/bus/pci/devices | awk '{ printf "%s  %s\n", $1, $2; }'
```
There should be one more subvendor:subdevice pair.

tapwag, 2019-05-24:
I copied and pasted the string to my terminal but unfortunately it doesn't return anything.

b-rad-NDi, 2019-05-24:
Hmmm...the first pairing is 11bd:0101 so I don't know why it wouldn't show up.
How about manually looking in /proc/bus/pci/devices. The output is very wide, so you'll have to stretch your window. The subvendor:deviceid is the second column. You should see 11bd0101 in there. You should also see another 11bd entry. That second one is what I need, if this bridge works like the pci bridge we currently use.

tapwag, 2019-05-24:
I did a 

cat /proc/bus/pci/devices > output.txt

as a regular user and I am uploading the file created. 

[output.txt](https://github.com/b-rad-NDi/Ubuntu-media-tree-kernel-builder/files/3218449/output.txt)

Is this of help?

b-rad-NDi, 2019-05-24:
Hmmm...I only see the main pcie device 1131:7162 in there. Will deal with that detail later. I'll see about enabling only the first pair for now.

b-rad-NDi, 2019-05-24:
This should do it:
```
sudo apt-get install discover
discover --vendor-id --model-id pci | uniq
```

tapwag, 2019-05-24:
I am currently on openSUSE "Tumbleweed" and discover is not available for this distribution. There is a tool available for hardware detection though called "hwinfo". 
[hwinfo.txt](https://github.com/b-rad-NDi/Ubuntu-media-tree-kernel-builder/files/3218536/hwinfo.txt)

Does this file contain the information you require?

b-rad-NDi, 2019-05-24:
That worked, thanks. I only see a single device id. I'll review some drivers.

tapwag, 2019-05-25:
Thanks, in the meantime I signed up for a Wiki account at linuxtv.org - Please let me know if there is any information which needs to be documented.