b-rad-NDi/Ubuntu-media-tree-kernel-builder #78: Signal measurement documentation

Saentist, 2019-06-02:
Is there any table to convert measure signal?
example:
/QuadHD on DVB-C Ubuntu/  
```
# uname -a
Linux tvserv3 4.4.0-148201905291620-generic #0+mediatree+hauppauge-Ubuntu SMP Wed May 29 21:34:40 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux
# femon -a 5
FE: Silicon Labs Si2168 (DVBC)
status SCVYL | signal 7d70 | snr 0172 | ber 00000000 | unc 00000000 | FE_HAS_LOCK
status SCVYL | signal 7d70 | snr 0174 | ber 00000000 | unc 00000000 | FE_HAS_LOCK
status SCVYL | signal 7d70 | snr 0174 | ber 00000000 | unc 00000000 | FE_HAS_LOCK
status SCVYL | signal 7d70 | snr 0174 | ber 00000000 | unc 00000000 | FE_HAS_LOCK
status SCVYL | signal 7d70 | snr 0174 | ber 00000000 | unc 00000000 | FE_HAS_LOCK

# femon -a 5 -H
FE: Silicon Labs Si2168 (DVBC)
status SCVYL | signal  48% | snr   0% | ber 0 | unc 0 | FE_HAS_LOCK
status SCVYL | signal  48% | snr   0% | ber 0 | unc 0 | FE_HAS_LOCK
status SCVYL | signal  48% | snr   0% | ber 0 | unc 0 | FE_HAS_LOCK
status SCVYL | signal  48% | snr   0% | ber 0 | unc 0 | FE_HAS_LOCK
```
In same time I see in TVHeadend  SNR=36.8dB Signal=-53dBm

What are range of values and how to guess when are in % or dB

crazycat69, 2019-06-02:
Use dvb-fe-tool -m -vThis show extended signal statistic like tvheadend.

Saentist, 2019-06-03:
@crazycat69 
0070:6b28, board: Hauppauge WinTV-QuadHD-DVB /dvb-c/
```
dvb-fe-tool -m -v -a 2
Device Silicon Labs Si2168 (/dev/dvb/adapter2/frontend0) capabilities:
     CAN_2G_MODULATION
     CAN_FEC_1_2
     CAN_FEC_2_3
     CAN_FEC_3_4
     CAN_FEC_5_6
     CAN_FEC_7_8
     CAN_FEC_AUTO
     CAN_GUARD_INTERVAL_AUTO
     CAN_HIERARCHY_AUTO
     CAN_INVERSION_AUTO
     CAN_MULTISTREAM
     CAN_MUTE_TS
     CAN_QAM_16
     CAN_QAM_32
     CAN_QAM_64
     CAN_QAM_128
     CAN_QAM_256
     CAN_QAM_AUTO
     CAN_QPSK
     CAN_TRANSMISSION_MODE_AUTO
DVB API Version 5.11, Current v5 delivery system: DVBC/ANNEX_A
Supported delivery systems:
     DVBT
     DVBT2
    [DVBC/ANNEX_A]
Got parameters for DVBC/ANNEX_A:
FREQUENCY = 474000000
MODULATION = QAM/AUTO
INVERSION = OFF
SYMBOL_RATE = 6900000
INNER_FEC = AUTO
DELIVERY_SYSTEM = DVBC/ANNEX_A
Lock   (0x1f) Signal= -48.00dBm C/N= 37.50dB UCB= 0 postBER= 0
Lock   (0x1f) Signal= -48.00dBm C/N= 37.50dB UCB= 0 postBER= 0
Lock   (0x1f) Signal= -48.00dBm C/N= 37.25dB UCB= 0 postBER= 0
Lock   (0x1f) Signal= -48.00dBm C/N= 37.50dB UCB= 0 postBER= 0

femon -a 2
FE: Silicon Labs Si2168 (DVBC)
status SCVYL | signal 851e | snr 0177 | ber 00000000 | unc 00000000 | FE_HAS_LOCK
status SCVYL | signal 851e | snr 0174 | ber 00000000 | unc 00000000 | FE_HAS_LOCK
status SCVYL | signal 851e | snr 0177 | ber 00000000 | unc 00000000 | FE_HAS_LOCK
status SCVYL | signal 851e | snr 0177 | ber 00000000 | unc 00000000 | FE_HAS_LOCK

femon -a 2 -H
FE: Silicon Labs Si2168 (DVBC)
status SCVYL | signal  51% | snr   0% | ber 0 | unc 0 | FE_HAS_LOCK
status SCVYL | signal  51% | snr   0% | ber 0 | unc 0 | FE_HAS_LOCK
status SCVYL | signal  51% | snr   0% | ber 0 | unc 0 | FE_HAS_LOCK
status SCVYL | signal  51% | snr   0% | ber 0 | unc 0 | FE_HAS_LOCK
```

This is not so practical in case I use Minisatip https://github.com/catalinii/minisatip to send signal to TVHeadend. 
Range of values with Femon read is % but in case this vales are in what range 
If for signal
0000 (hex) = 0%
FFFF (hex) 65535 (dec) = 100%
851E (hex) = 51%
step of 655 =1%

then SNR are in with range ?
if 0000 = 0%
then X = 100% 

And other device 
0070:6906, board: Hauppauge WinTV-HVR4000(Lite) DVB-S/S2
```
dvb-fe-tool -m -v
Device Conexant CX24116/CX24118 (/dev/dvb/adapter0/frontend0) capabilities:
     CAN_2G_MODULATION
     CAN_FEC_1_2
     CAN_FEC_2_3
     CAN_FEC_3_4
     CAN_FEC_4_5
     CAN_FEC_5_6
     CAN_FEC_6_7
     CAN_FEC_7_8
     CAN_FEC_AUTO
     CAN_INVERSION_AUTO
     CAN_QPSK
     CAN_RECOVER
DVB API Version 5.10, Current v5 delivery system: DVBS2
Supported delivery systems:
     DVBS
    [DVBS2]
Got parameters for DVBS2:
FREQUENCY = 4293344296
INVERSION = AUTO
SYMBOL_RATE = 13381000
INNER_FEC = 3/4
MODULATION = QPSK
PILOT = AUTO
ROLLOFF = AUTO
POLARIZATION = OFF
STREAM_ID = 4294967295
DELIVERY_SYSTEM = DVBS2
Lock   (0x1f) Signal= 85,26% C/N= 97,50% UCB= 63898 postBER= 0
Lock   (0x1f) Signal= 85,26% C/N= 97,50% UCB= 63898 postBER= 0
Lock   (0x1f) Signal= 85,26% C/N= 96,88% UCB= 63488 postBER= 0
Lock   (0x1f) Signal= 85,26% C/N= 96,88% UCB= 63488 postBER= 0
Lock   (0x1f) Signal= 85,26% C/N= 96,88% UCB= 63488 postBER= 0
Lock   (0x1f) Signal= 85,26% C/N= 96,88% UCB= 63488 postBER= 0
Lock   (0x1f) Signal= 85,26% C/N= 97,50% UCB= 63898 postBER= 0

 femon
FE: Conexant CX24116/CX24118 (DVBS)
status SCVYL | signal da40 | snr f99a | ber 00000000 | unc 00000000 | FE_HAS_LOCK
status SCVYL | signal da40 | snr f800 | ber 00000000 | unc 00000000 | FE_HAS_LOCK
status SCVYL | signal da40 | snr f800 | ber 00000000 | unc 00000000 | FE_HAS_LOCK

 femon -H
FE: Conexant CX24116/CX24118 (DVBS)
status SCVYL | signal  84% | snr  96% | ber 0 | unc 0 | FE_HAS_LOCK
status SCVYL | signal  84% | snr  96% | ber 0 | unc 0 | FE_HAS_LOCK
status SCVYL | signal  83% | snr  96% | ber 0 | unc 0 | FE_HAS_LOCK
```

easy to guess that SNR value FFFF is equal to 100%

crazycat69, 2019-06-04:
femon show only legacy signal strength and snr as 16-bit value (usual scaled 0-100%). but many frontend driver like si2168 report only extended signal statistic. You can try [media_build](https://bitbucket.org/CrazyCat/media_build) where fixed legacy signal statistic  for si2168 ( Strength(%) = 100 + RFLevel (dBm), SNR(%) = SNR(dB) * 2).

P.S. Conexant demod drivers usual report some "crazy" leghacy signal statistic close to 100/100% :)

Saentist, 2019-06-04:
@crazycat69 
Do you agree that all this is mess, and need standardization?

Some software developers make optional value multiplier to "fix" that
```
* -M multiplies the strength and snr of the DVB adapter with the specified values
        * If the snr or the strength multipliers are set to 0, minisatip will override the value received from the adapter and will report always full signal 100%
        * eg: -M 4-6:1.2-1.3 - multiplies the strength with 1.2 and the snr with 1.3 for adapter 4, 5 and 6
        * eg: -M *:1.5-1.6 - multiplies the strength with 1.5 and the snr with 1.6 for all adapters
```

b-rad-NDi, 2019-06-04:
Every single driver is different and every single author has done stats their own way. Sure, it would be amazing the unify stats across all drivers so that they produced similar numbers for similar signals, but without detailed datasheets for every chip this is not done. A lot of this code is just translated from a usb capture of a windows driver.

b-rad-NDi, 2019-06-04:
@crazycat69 : I believe one of my patches splits up all the stats functions and provides v3 "old" format. So if using this project that is already done, based on silicon labs reference code.

b-rad-NDi, 2019-06-04:
https://github.com/b-rad-NDi/Ubuntu-media-tree-kernel-builder/blob/master/patches/mainline-extra/tip/20.missing.stat.bits/0003-si2168-Add-some-various-v3-stat-ioctls.patch

Saentist, 2019-06-04:
@b-rad-NDi 
If some one not start standardization, no one will.
Hauppauge make windows drivers also, 
maby they can provide more information of measurement.

b-rad-NDi, 2019-06-04:
That's the thing. Even though Hauppauge provides windows drivers a lot of times those windows drivers are in fact just black box reference drivers. Some companies just won't provide datasheets to us. You use their reference driver and just provide additional code to hook into it when producing new devices.
I'm absolutely not against statistic standardization, it's just something difficult to unify.

Saentist, 2019-06-04:
@b-rad-NDi 
Can YOU personally, in this repo to make measurement unification, 
with hope other to do same, with their drivers.
I can't believe this will make negative result to any one.

b-rad-NDi, 2019-06-04:
I'll at least review the few demods that i'm familiar with and look into unifying them. I do not have a DVB generator, so it's not very easy for me to handle that half.

Saentist, 2019-06-04:
Is I see some devices measure from 1-65535 (0000-FFFF) other from 1-4065 (0000-0FFF) im see some other brand to measure from 0-256 (0000-00FF) in % range 
dBm range is not available to most of devices.

Koumes, 2019-06-14:
> https://github.com/b-rad-NDi/Ubuntu-media-tree-kernel-builder/blob/master/patches/mainline-extra/tip/20.missing.stat.bits/0003-si2168-Add-some-various-v3-stat-ioctls.patch

Hi @b-rad-NDi, there is an issue in the SNR calculation si2168_read_snr(). The resulting percentage is always less than 1 %.

Fix:

```diff
--- a/drivers/media/dvb-frontends/si2168.c
+++ b/drivers/media/dvb-frontends/si2168.c
@@ -239,7 +239,7 @@ static int si2168_read_snr(struct dvb_frontend *fe, u16 *snr)
 		goto err;
 
 	if (status & FE_HAS_LOCK) {
-		stat_snr = c->cnr.stat[0].svalue;
+		stat_snr = c->cnr.stat[0].svalue * 131;
 		do_div(stat_snr, 100);
 		*snr = (u16)stat_snr;
 	}
```