b-rad-NDi/Ubuntu-media-tree-kernel-builder #79: Incompatibility with Geniatech MyGica T230C v2

Koumes, 2019-06-08:
Hello,

patch https://github.com/b-rad-NDi/Ubuntu-media-tree-kernel-builder/blob/master/patches/mainline-extra/tip/07.si2157.analog.tuning/0002-si2157-Check-error-status-bit-on-cmd-execute.patch is not compatible with Si2141 tuner based devices (e.g. Geniatech MyGica T230C v2, Evolveo Sigma T2, Tesla Proxy T2). Details: https://github.com/CoreELEC/CoreELEC/pull/209.

**Fix:**

```
Fix compatibility with Si2141 tuner (e.g. MyGica T230C v2)
which needs a next command for proper initialization.

--- a/drivers/media/tuners/si2157.c
+++ b/drivers/media/tuners/si2157.c
@@ -120,7 +120,7 @@ static int si2157_init(struct dvb_fronte
 	}
	cmd.rlen = 1;
 	ret = si2157_cmd_execute(client, &cmd);
-	if (ret)
+	if (ret && (dev->chiptype != SI2157_CHIPTYPE_SI2141 || ret != -EAGAIN))
 		goto err;
 
 	/* Si2141 needs a second command before it answers the revision query */

```

b-rad-NDi, 2019-06-08:
Please make separate tickets for separate issues.
I will fix the error status bit patch soon, if you'd like me to add tested-by tag to the patch you can supply it here.

Koumes, 2019-06-08:
Okay. I split them up. Thanks.

b-rad-NDi, 2019-08-09:
Patch has been live in my testing ppa for a long time, now it's part of the repo.
Fixed by: c65020f48532f0dd783f8c881f9a90047366fe20

closed 2019-08-09 by b-rad-NDi

b-rad-NDi, 2019-08-15:
@Koumes btw this patch is bad. You essentially broke everything *NOT* si2141, *ALL* the time. And then I didn't test it out before I pushed, so bad on me.

Koumes, 2019-08-15:
@b-rad-NDi My patch is correct.

I think your commit https://github.com/b-rad-NDi/Ubuntu-media-tree-kernel-builder/commit/c65020f48532f0dd783f8c881f9a90047366fe20 changes different piece of code. This should be changed in si2157_init() and not in si2157_probe().

b-rad-NDi, 2019-08-15:
my bad then.

b-rad-NDi, 2019-08-15:
I don't understand why you're ignoring -EAGAIN though. If this only affects 2141 is the check for that single model not good enough? This does not affect 2157, nor 2177.

b-rad-NDi, 2019-08-15:
The only place -EAGAIN should be ignored is the single spot in probe. It should not be ignored after that. The patch I'm including now only ignores si2141 errors at your patches location. If you think this is too broad, then please display the exact error[s] you're receiving.

74ca7cdc6408192a5515b9cd8c9d64556be25e22

Hopefully that is sufficient for si2141.

Koumes, 2019-08-15:
I think it is better to ignore only a specific error -EAGAIN on Si2141.

For evidence:
Unable to tune and -11 == -EAGAIN errors occurred with debug enabled.

```
Jun 03 20:00:11 CoreELEC kernel: si2168 1-0064: Silicon Labs Si2168-D60 successfully identified
Jun 03 20:00:11 CoreELEC kernel: si2168 1-0064: firmware version: D 6.0.1
Jun 03 20:00:11 CoreELEC kernel: si2157 2-0060: Silicon Labs Si2141 successfully attached
Jun 03 20:00:11 CoreELEC kernel: usb 1-2: DVB: registering adapter 0 frontend 0 (Silicon Labs Si2168)...
Jun 03 20:00:11 CoreELEC kernel: dvbdev: dvb_create_media_entity: media entity 'Silicon Labs Si2168' registered.
Jun 03 20:00:11 CoreELEC kernel: usb 1-2: dvb_usb_v2: 'MyGica Mini DVB-T2 USB Stick T230C v2' successfully initiali
zed and connected
Jun 03 20:00:11 CoreELEC kernel: usbcore: registered new interface driver dvb_usb_dvbsky
Jun 03 20:00:11 CoreELEC kernel: si2168 1-0064: downloading firmware from file 'dvb-demod-si2168-d60-01.fw'
Jun 03 20:00:12 CoreELEC kernel: si2168 1-0064: firmware version: D 6.0.13
Jun 03 20:00:12 CoreELEC kernel: si2157 2-0060: found a 'Silicon Labs Si2141-A10'
Jun 03 20:00:12 CoreELEC kernel: si2157 2-0060: downloading firmware from file 'dvb-tuner-si2141-a10-01.fw'
Jun 03 20:00:13 CoreELEC kernel: si2157 2-0060: firmware version: 1.1.11

...

Jun 03 20:02:59 CoreELEC kernel: si2168 1-0064: si2168_ts_bus_ctrl acquire: 1
Jun 03 20:02:59 CoreELEC kernel: si2168 1-0064: cmd execution took 0 ms
Jun 03 20:02:59 CoreELEC kernel: si2157 2-0060: 
Jun 03 20:02:59 CoreELEC kernel: si2157 2-0060: cmd execution took 0 ms, status=c0
Jun 03 20:02:59 CoreELEC kernel: si2157 2-0060: failed=-11
Jun 03 20:02:59 CoreELEC kernel: si2157 2-0060: cmd execution took 10 ms, status=c0
Jun 03 20:02:59 CoreELEC kernel: si2157 2-0060: failed=-11
Jun 03 20:02:59 CoreELEC kernel: si2157 2-0060: failed=-11
Jun 03 20:02:59 CoreELEC kernel: dvb_frontend: dvb_frontend_get_frequency_limits: frequency interval: tuner: 420000
00...870000000, frontend: 0...0
Jun 03 20:02:59 CoreELEC kernel: si2157 2-0060: 
Jun 03 20:02:59 CoreELEC kernel: si2157 2-0060: cmd execution took 0 ms, status=80
Jun 03 20:02:59 CoreELEC kernel: si2168 1-0064: 
Jun 03 20:02:59 CoreELEC kernel: si2168 1-0064: si2168_ts_bus_ctrl acquire: 0
Jun 03 20:02:59 CoreELEC kernel: si2168 1-0064: cmd execution took 0 ms
Jun 03 20:02:59 CoreELEC kernel: si2168 1-0064: 
Jun 03 20:02:59 CoreELEC kernel: si2168 1-0064: cmd execution took 0 ms
Jun 03 20:02:59 CoreELEC kernel: si2168 1-0064: downloading firmware from file 'dvb-demod-si2168-d60-01.fw'
Jun 03 20:03:00 CoreELEC kernel: si2168 1-0064: cmd execution took 50 ms

...

Jun 03 20:03:01 CoreELEC kernel: si2168 1-0064: si2168_ts_bus_ctrl acquire: 1
Jun 03 20:03:01 CoreELEC kernel: si2168 1-0064: cmd execution took 0 ms
Jun 03 20:03:01 CoreELEC kernel: si2157 2-0060: 
Jun 03 20:03:01 CoreELEC kernel: si2157 2-0060: cmd execution took 0 ms, status=c0
Jun 03 20:03:01 CoreELEC kernel: si2157 2-0060: failed=-11
Jun 03 20:03:01 CoreELEC kernel: si2157 2-0060: cmd execution took 0 ms, status=c0
Jun 03 20:03:01 CoreELEC kernel: si2157 2-0060: failed=-11
Jun 03 20:03:01 CoreELEC kernel: si2157 2-0060: failed=-11
Jun 03 20:03:01 CoreELEC kernel: dvb_frontend: dvb_frontend_get_frequency_limits: frequency interval: tuner: 420000
00...870000000, frontend: 0...0
Jun 03 20:03:01 CoreELEC kernel: dvb_frontend: dvb_frontend_get_frequency_limits: frequency interval: tuner: 420000
00...870000000, frontend: 0...0
Jun 03 20:03:01 CoreELEC kernel: si2168 1-0064: delivery_system=3 modulation=6 frequency=474000000 bandwidth_hz=800
0000 symbol_rate=0 inversion=0 stream_id=4294967295
Jun 03 20:03:01 CoreELEC kernel: si2157 2-0060: delivery_system=3 frequency=474000000 bandwidth_hz=8000000
Jun 03 20:03:01 CoreELEC kernel: si2157 2-0060: failed=-11
Jun 03 20:03:01 CoreELEC kernel: si2168 1-0064: failed=-11
Jun 03 20:03:02 CoreELEC kernel: si2168 1-0064: cmd execution took 0 ms
Jun 03 20:03:02 CoreELEC kernel: si2168 1-0064: cmd execution took 0 ms
Jun 03 20:03:02 CoreELEC kernel: si2168 1-0064: failed=-121
Jun 03 20:03:02 CoreELEC kernel: si2168 1-0064: cmd execution took 0 ms
Jun 03 20:03:02 CoreELEC kernel: si2168 1-0064: failed=-121
Jun 03 20:03:02 CoreELEC kernel: si2168 1-0064: failed=-121
Jun 03 20:03:02 CoreELEC kernel: si2168 1-0064: failed=-121
Jun 03 20:03:02 CoreELEC kernel: si2168 1-0064: cmd execution took 0 ms
Jun 03 20:03:02 CoreELEC kernel: si2168 1-0064: cmd execution took 0 ms
Jun 03 20:03:02 CoreELEC kernel: si2168 1-0064: failed=-121
Jun 03 20:03:02 CoreELEC kernel: si2168 1-0064: cmd execution took 0 ms
Jun 03 20:03:02 CoreELEC kernel: si2168 1-0064: failed=-121
Jun 03 20:03:02 CoreELEC kernel: si2168 1-0064: failed=-121
Jun 03 20:03:02 CoreELEC kernel: si2168 1-0064: failed=-121
```

b-rad-NDi, 2019-08-15:
Are you testing my latest patch or your own build? Can you set debugfs to print line number. If you're testing my patch can then then give all the associated dmesg. If you're running your own custom patch set can you put that si2157.c somewhere. I would like to know exactly what commands are failing. My goal is to upstream here, so we need to narrow down the minimum to fix 2141.

b-rad-NDi, 2019-08-15:
hmmm...actually line number might not matter. You might have to enable all debugfs with line numbers included to narrow down what commands fail.

Koumes, 2019-08-17:
This log was obtained from the driver without any patch.

Your fix may also work, but with Si2141 it will ignore all errors and not only -EAGAIN. But I can be wrong and it doesn't have to be important.