b-rad-NDi/Ubuntu-media-tree-kernel-builder #80: Auto detect DVB-T/T2 incompatibility with MyGica T230C v2

Koumes, 2019-06-08:
PLP functionality (DVB-T/T2 autodetect) from patch https://github.com/b-rad-NDi/Ubuntu-media-tree-kernel-builder/blob/master/patches/mainline-extra/tip/10.random.patches/0001-si2168-Change-from-DVB-T-to-DVB-T-T2-autodetect.patch does not work properly with Si2141 tuner based devices (e.g. Geniatech MyGica T230C v2, Evolveo Sigma T2, Tesla Proxy T2). Sometimes there is "No signal" error when tunning DVB-T.

PLP patch description informs about default disabled but it is enabled.

```
The module parameter is default disabled, to be backwards compatible
with current behaviour.

static int dvbt_auto_plp = 1;
```

Debug log (with https://github.com/b-rad-NDi/Ubuntu-media-tree-kernel-builder/issues/79 patch applied and PLP enabled):

```
Jun 08 18:45:59.431468 CoreELEC tvheadend[3176]: linuxdvb: Silicon Labs Si2168 #0 : DVB-T #0 - starting 674MHz in DVB-T
Jun 08 18:45:59.431519 CoreELEC tvheadend[3176]: subscription: 0051: "127.0.0.1 [  | Kodi Media Center ]" subscribing on channel "Retro Music TV", weight: 150, adapter: "Silicon Labs Si2168 #0 : DVB-T #0", network: "DVB-T", mux: "674MHz", provider: "PDTV", service: "Retro Music TV", profile="htsp", hostname="127.0.0.1", username="127.0.0.1", client="Kodi Media Center"
Jun 08 18:45:59.444419 CoreELEC kernel: dvb_frontend: dvb_frontend_get_frequency_limits: frequency interval: tuner: 42000000...870000000, frontend: 0...0
Jun 08 18:45:59.445191 CoreELEC kernel: si2168 1-0064: delivery_system=3 modulation=3 frequency=674000000 bandwidth_hz=8000000 symbol_rate=0 inversion=1 stream_id=4294967295
Jun 08 18:45:59.446120 CoreELEC kernel: si2157 2-0060: delivery_system=3 frequency=674000000 bandwidth_hz=8000000
Jun 08 18:45:59.449182 CoreELEC kernel: si2157 2-0060: cmd execution took 0 ms, status=85
Jun 08 18:45:59.450527 CoreELEC kernel: si2157 2-0060: cmd execution took 0 ms, status=85
Jun 08 18:45:59.451357 CoreELEC kernel: si2157 2-0060: cmd execution took 10 ms, status=80
Jun 08 18:45:59.484453 CoreELEC kernel: si2157 2-0060: tuning took 30 ms, status=0x85
Jun 08 18:45:59.485411 CoreELEC kernel: si2168 1-0064: cmd execution took 0 ms
Jun 08 18:45:59.486070 CoreELEC kernel: si2168 1-0064: cmd execution took 0 ms
Jun 08 18:45:59.486696 CoreELEC kernel: si2168 1-0064: cmd execution took 0 ms
Jun 08 18:45:59.487382 CoreELEC kernel: si2168 1-0064: cmd execution took 0 ms
Jun 08 18:45:59.488092 CoreELEC kernel: si2168 1-0064: cmd execution took 0 ms
Jun 08 18:45:59.488708 CoreELEC kernel: si2168 1-0064: cmd execution took 10 ms
Jun 08 18:45:59.489289 CoreELEC kernel: si2168 1-0064: cmd execution took 0 ms
Jun 08 18:45:59.489878 CoreELEC kernel: si2168 1-0064: cmd execution took 0 ms
Jun 08 18:45:59.490479 CoreELEC kernel: si2168 1-0064: cmd execution took 0 ms
Jun 08 18:45:59.494412 CoreELEC kernel: si2168 1-0064: cmd execution took 0 ms
Jun 08 18:45:59.495220 CoreELEC kernel: si2168 1-0064: cmd execution took 0 ms
Jun 08 18:45:59.495851 CoreELEC kernel: si2168 1-0064: cmd execution took 10 ms
Jun 08 18:45:59.496451 CoreELEC kernel: si2168 1-0064: cmd execution took 0 ms
Jun 08 18:45:59.504506 CoreELEC kernel: si2168 1-0064: cmd execution took 0 ms
Jun 08 18:45:59.505479 CoreELEC kernel: si2168 1-0064: cmd execution took 0 ms
Jun 08 18:45:59.514404 CoreELEC kernel: si2168 1-0064: cmd execution took 10 ms
Jun 08 18:45:59.515206 CoreELEC kernel: si2168 1-0064: si2168_ts_bus_ctrl acquire: 1
Jun 08 18:45:59.515817 CoreELEC kernel: si2168 1-0064: cmd execution took 0 ms
Jun 08 18:45:59.516372 CoreELEC kernel: si2168 1-0064: cmd execution took 0 ms
Jun 08 18:45:59.516920 CoreELEC kernel: si2168 1-0064: cmd execution took 0 ms
Jun 08 18:45:59.517489 CoreELEC kernel: si2168 1-0064: status=00 args=80 00 08 00 00 00 00 00 03 1b 00 00 01 00
Jun 08 18:45:59.544404 CoreELEC kernel: si2168 1-0064: cmd execution took 0 ms
Jun 08 18:45:59.545323 CoreELEC kernel: si2168 1-0064: cmd execution took 10 ms
Jun 08 18:45:59.545941 CoreELEC kernel: si2168 1-0064: status=00 args=80 00 08 00 00 00 00 00 03 1b 00 00 01 00
Jun 08 18:45:59.584428 CoreELEC kernel: si2168 1-0064: cmd execution took 0 ms
Jun 08 18:45:59.585450 CoreELEC kernel: si2168 1-0064: cmd execution took 10 ms
Jun 08 18:45:59.586248 CoreELEC kernel: si2168 1-0064: failed=-121
Jun 08 18:45:59.587039 CoreELEC kernel: si2168 1-0064: cmd execution took 0 ms
Jun 08 18:45:59.587751 CoreELEC kernel: si2168 1-0064: status=00 args=80 00 08 00 18 00 86 ff 03 2d 00 00 01 00
Jun 08 18:45:59.644417 CoreELEC kernel: si2168 1-0064: cmd execution took 0 ms
Jun 08 18:45:59.645265 CoreELEC kernel: si2168 1-0064: cmd execution took 10 ms
Jun 08 18:45:59.646020 CoreELEC kernel: si2168 1-0064: failed=-121
Jun 08 18:45:59.646659 CoreELEC kernel: si2168 1-0064: cmd execution took 0 ms
Jun 08 18:45:59.647230 CoreELEC kernel: si2168 1-0064: status=00 args=80 00 08 00 18 00 ad ff 03 2d 00 00 01 00
Jun 08 18:45:59.692864 CoreELEC tvheadend[3176]: linuxdvb: Silicon Labs Si2168 #0 : DVB-T #0 - FE_READ_STATUS error Remote I/O error
Jun 08 18:45:59.694430 CoreELEC kernel: si2168 1-0064: cmd execution took 0 ms
Jun 08 18:45:59.695286 CoreELEC kernel: si2168 1-0064: cmd execution took 0 ms
Jun 08 18:45:59.695887 CoreELEC kernel: si2168 1-0064: failed=-121
Jun 08 18:45:59.696439 CoreELEC kernel: si2168 1-0064: cmd execution took 0 ms
Jun 08 18:45:59.697207 CoreELEC kernel: si2168 1-0064: failed=-121
Jun 08 18:45:59.697991 CoreELEC kernel: si2168 1-0064: failed=-121
Jun 08 18:45:59.698681 CoreELEC kernel: si2168 1-0064: failed=-121
Jun 08 18:46:00.424398 CoreELEC kernel: si2168 1-0064: cmd execution took 0 ms
Jun 08 18:46:00.425298 CoreELEC kernel: si2168 1-0064: cmd execution took 0 ms
Jun 08 18:46:00.425954 CoreELEC kernel: si2168 1-0064: status=1f args=80 00 06 71 18 00 dd ff 09 12 2d 01 17
Jun 08 18:46:00.426510 CoreELEC kernel: si2168 1-0064: cmd execution took 0 ms
Jun 08 18:46:00.427071 CoreELEC kernel: si2168 1-0064: cmd execution took 0 ms
[... 73 lines cut ...]
Jun 08 18:46:05.865165 CoreELEC kernel: si2168 1-0064: status=1f args=80 00 06 71 18 00 dd ff 09 12 2d 01 17
Jun 08 18:46:05.865720 CoreELEC kernel: si2168 1-0064: cmd execution took 0 ms
Jun 08 18:46:05.866321 CoreELEC kernel: si2168 1-0064: post_bit_error=0 post_bit_count=100000000 ber=0*10^-8
Jun 08 18:46:05.866924 CoreELEC kernel: si2168 1-0064: cmd execution took 0 ms
Jun 08 18:46:05.867484 CoreELEC kernel: si2168 1-0064: cmd execution took 10 ms
Jun 08 18:46:05.868020 CoreELEC kernel: si2168 1-0064: status=1f args=80 00 06 71 18 00 dd ff 09 12 2d 01 17
Jun 08 18:46:05.868681 CoreELEC kernel: si2168 1-0064: cmd execution took 0 ms
Jun 08 18:46:05.869248 CoreELEC kernel: si2168 1-0064: block_error=0
Jun 08 18:46:06.336847 CoreELEC tvheadend[3176]: subscription: 0051: service instance is bad, reason: No input detected
Jun 08 18:46:06.337059 CoreELEC tvheadend[3176]: linuxdvb: Silicon Labs Si2168 #0 : DVB-T #0 - stopping 674MHz in DVB-T
Jun 08 18:46:07.014397 CoreELEC kernel: si2157 2-0060: 
Jun 08 18:46:07.015317 CoreELEC kernel: si2157 2-0060: cmd execution took 0 ms, status=85
Jun 08 18:46:07.337175 CoreELEC tvheadend[3176]: mpegts: 778MHz in DVB-T - tuning on Silicon Labs Si2168 #0 : DVB-T #0
Jun 08 18:46:07.337360 CoreELEC tvheadend[3176]: linuxdvb: Silicon Labs Si2168 #0 : DVB-T #0 - starting 778MHz in DVB-T
Jun 08 18:46:07.337398 CoreELEC tvheadend[3176]: subscription: 0053: "epggrab" subscribing to mux "778MHz", weight: 4, adapter: "Silicon Labs Si2168 #0 : DVB-T #0", network: "DVB-T", service: "Raw PID Subscription"
Jun 08 18:46:07.344402 CoreELEC kernel: dvb_frontend: dvb_frontend_get_frequency_limits: frequency interval: tuner: 42000000...870000000, frontend: 0...0
Jun 08 18:46:07.344750 CoreELEC kernel: si2168 1-0064: delivery_system=3 modulation=3 frequency=778000000 bandwidth_hz=8000000 symbol_rate=0 inversion=1 stream_id=4294967295
Jun 08 18:46:07.345521 CoreELEC kernel: si2157 2-0060: delivery_system=3 frequency=778000000 bandwidth_hz=8000000
Jun 08 18:46:07.346181 CoreELEC kernel: si2157 2-0060: cmd execution took 0 ms, status=85
Jun 08 18:46:07.346751 CoreELEC kernel: si2157 2-0060: cmd execution took 10 ms, status=85
Jun 08 18:46:07.354388 CoreELEC kernel: si2157 2-0060: cmd execution took 0 ms, status=80
Jun 08 18:46:07.394394 CoreELEC kernel: si2157 2-0060: tuning took 40 ms, status=0x85
Jun 08 18:46:07.395322 CoreELEC kernel: si2168 1-0064: cmd execution took 0 ms

```

b-rad-NDi, 2019-06-13:
The upstream submission had the module parameter disabled like the text states. I will disable it here as well. It turns out this patch can adversely affect DVB-T reception if your signal is too hot.

b-rad-NDi, 2019-08-09:
We were able to repro this in the lab.
Fixed by: 86bda82badd0d3a7e39d4f8c08bbc58341e41507

closed 2019-08-09 by b-rad-NDi