b-rad-NDi/Ubuntu-media-tree-kernel-builder #81: Ubuntu 18.04 LTS (Bionic) released a new HWE kernel: 5.0.0

sindastra, 2019-08-01:
Ubuntu 18.04 LTS (Bionic) released a new HWE kernel: 5.0.0

Please update accordingly, thanks.

b-rad-NDi, 2019-08-01:
Hi, issues are being sorted out. If you'd like to test please see my staging repo:
https://launchpad.net/~b-rad/+archive/ubuntu/test-test-hw-staging/

b-rad-NDi, 2019-08-08:
This is installable and boots fine, but is still untested, on the above linked ppa.

sindastra, 2019-08-08:
What kind of testing is required?

b-rad-NDi, 2019-08-08:
Just stating that Hauppauge testers have not run through and validated hardware. There's nothing that should have broken, so everything should still be ok.

sindastra, 2019-08-08:
I will try to test it tomorrow.

sindastra, 2019-08-11:
I just wanted to go ahead and test it on my end and added the testing repo. I hoped it would simply prompt to upgrade the kernel but it did not. How would I install the testing kernel?

b-rad-NDi, 2019-08-11:
I'll add virtual package for you, that should trigger the upgrade.

sindastra, 2019-08-11:
I feel stupid now to ask but:

```
# apt list --upgradable
Listing... Done
linux-hwe-headers-mediatree/bionic,bionic 0.18.9+bionic all [upgradable from: 0.18.8+bionic]
linux-hwe-image-mediatree/bionic,bionic 0.18.9+bionic all [upgradable from: 0.18.8+bionic]
```
but then:
```
# apt upgrade
Reading package lists... Done
Building dependency tree
Reading state information... Done
Calculating upgrade... Done
0 to upgrade, 0 to newly install, 0 to remove and 0 not to upgrade.
```
And yes I did ```apt update``` beforehand.

b-rad-NDi, 2019-08-11:
Unsure, I use apt-get still.

sindastra, 2019-08-12:
I merely meant to say that somehow the update isn't happening. Maybe something wrong with the virtual package?

b-rad-NDi, 2019-08-12:
fudge, you're correct. There's missing tags in the package names. Unsure how that happened.

b-rad-NDi, 2019-08-13:
Should be good to go now.

closed 2019-08-13 by b-rad-NDi

sindastra, 2019-08-15:
Works on my end, thanks! 👍