b-rad-NDi/Ubuntu-media-tree-kernel-builder #83: config-4.15.0-55201908121620-generic WinTV-dualHD not found Plex

cdevelc, 2019-08-15:
Updated to config-4.15.0-55201908121620-generic and Plex no longer sees my WinTV-dualHD device.

Backed down to config-4.15.0-50201905291620-generic and Plex is happy, finding WinTV-dualHD again without difficulty.  

Plex Media Server Version 1.16.3.1402, (on both versions of mediatree+hauppauge-Ubuntu)
Intel i3-4130

b-rad-NDi, 2019-08-15:
You should have saved full dmesg before down grading.

b-rad-NDi, 2019-08-15:
Actually I suspected a contributed patch, and after reviewing it now, it is bad and it pretty much broke everything.

radu022003, 2019-08-15:
same here
[    7.088840] tveeprom: Hauppauge model 204109, rev B3I6, serial# 13942608
[    7.088841] tveeprom: tuner model is SiLabs Si2157 (idx 186, type 4)
[    7.088842] tveeprom: TV standards PAL(B/G) NTSC(M) PAL(I) SECAM(L/L') PAL(D/D1/K) ATSC/DVB Digital (eeprom 0xfc)
[    7.088842] tveeprom: audio processor is None (idx 0)
[    7.088843] tveeprom: has no radio, has IR receiver, has no IR transmitter
[    7.088845] em28xx 3-1:1.0: dvb ts2 set to isoc mode.
[    7.287447] usbcore: registered new interface driver em28xx
[    7.378193] WARNING: You are using an experimental version of the media stack.
                As the driver is backported to an older kernel, it doesn't offer
                enough quality for its usage in production.
                Use it with care.
               Latest git patches (needed if you report a bug to linux-media@vger.kernel.org):
                1e0d0a5fd38192f23304ea2fc2b531fea7c74247 media: s5p-mfc: fix incorrect bus assignment in virtual child device
                07758747a2d7963593421b4e782b0f20a80f58d3 media: platform: Fix missing spin_lock_init()
                3eefa620d08dc6aa9323e409f1a307783c528004 media: Documentation: staging/ipu3-imgu: Add license information
[    7.384549] em28xx 3-1:1.0: Binding DVB extension
[    7.392811] i2c i2c-9: Added multiplexed i2c bus 12
[    7.392814] si2168 9-0064: Silicon Labs Si2168-B40 successfully identified
[    7.392815] si2168 9-0064: firmware version: B 4.0.2
[    7.395804] si2157: probe of 12-0060 failed with error -11
[    7.395900] em28xx 3-1:1.0: Binding DVB extension
[    7.402592] i2c i2c-11: Added multiplexed i2c bus 12
[    7.402594] si2168 11-0067: Silicon Labs Si2168-B40 successfully identified
[    7.402596] si2168 11-0067: firmware version: B 4.0.2
[    7.403876] si2157: probe of 12-0063 failed with error -11
[    7.403961] em28xx: Registered (Em28xx dvb Extension) extension

b-rad-NDi, 2019-08-15:
I'm pushing up a patch now. New builds should be available in two hours or so. Apologies and thank you for reporting.

radu022003, 2019-08-15:
thank you for such a fast patch.

b-rad-NDi, 2019-08-15:
All fresh packages are in the ppa, this should be fixed. Please confirm.

radu022003, 2019-08-16:
I can confirm that at least my problem was fixed. Everything seems to be fine at the moment. Thank you very much.

cdevelc, 2019-08-16:
I updated to 4.15.0-55201908151420-generic and WinTV-dualHD is working.  Fix confirmed.  And thank you for the fast response.

b-rad-NDi, 2019-08-16:
Excellent.

closed 2019-08-16 by b-rad-NDi