b-rad-NDi/Ubuntu-media-tree-kernel-builder #84: WinTV dualHD on Arch Linux

chrisnharvey, 2019-08-21:
I know this is a tool for building a mediatree kernel on Ubuntu, but I'm wondering if someone could help me.

I have been using this PPA on Ubuntu and it seems to work great with the Hauppague WinTV dualHD, exactly as expected, but I have another machine that runs Arch Linux that I'd like to get this tuner working on.

I have compiled the LinuxTV mediatree on Arch Linux, which allows me to scan for SD channels (DVB-T) in Plex, but no HD channels (DVB-T2) seem to show up.

So I applied [this patch](https://github.com/b-rad-NDi/Ubuntu-media-tree-kernel-builder/blob/master/patches/mainline-extra/tip/10.random.patches/0001-si2168-Change-from-DVB-T-to-DVB-T-T2-autodetect.patch) as I figured it would help with my problem. It kind-of worked, but now I'm only seeing HD (DVB-T2) channels and no SD (DVB-T) channels show up.

Does anyone know what patches I may need to apply to get both DVB-T and DVB-T2 channels showing up when scanning for channels in Plex?

b-rad-NDi, 2019-10-17:
Try including this patch as well:

0003-si2168-extend-i2c-timeout-insert-cmd-status-poll-del.patch

closed 2020-05-14 by b-rad-NDi

parkerlreed, 2022-05-04:
Last I checked, DualHD works out of the box for Arch. Both tuners at least on the ATSC model run great. 

What errors are you seeing when trying to use it? dmesg/journalctl log may help.

Davetheraveuk, 2022-05-05:
@chrisnharvey @b-rad-NDi
Hi, 
From what I am seeing in the dmesg output from the terminal, there are no errors. 
Command I used: **sudo dmesg | egrep -i '(ivtv|tveeprom|tuner|dualhd|em28xx)'**
Here's the weird thing: In Arch Kaffeine, the signal is very strong, but no channels are detected. I also have a virtual machine running Ubuntu with Kaffeine which also has a strong signal and **is** able to pickup channels. 

If everything looks sweet in the Arch terminal output then how come I cannot receive any channels despite the signal strength being at maximum? Its strange.
I am using the most recent ZEN kernel.
Hope you can help or advise.
Thanks for getting back to me.
David.

Davetheraveuk, 2022-05-05:
@chrisnharvey @b-rad-NDi
UPDATE: I converted the following packages from *.deb to *.pkg.tar using debtap:
linux-mediatree-0.2.4.impish-1-any
linux-firmware-hauppauge_0.1.8+impish_all

In arch + kaffeine the signal is strong and channels **are** being detected, but no sound or (at best) a garbled picture is displayed.
David.

b-rad-NDi, 2022-05-11:
No sound and garbled picture sound like issues you should not encounter in digital. Perhaps attach your full dmesg here?

parkerlreed, 2022-05-11:
Garbled happens frequently if the signal isn't strong enough. On ATSC I can frequently get enough signal for a lock but not enough to actually play it. Does your antenna setup and this card work elsewhere?

[image]

shspvr, 2022-05-11:
> 

Can tell us more about antenna setup please post a photo of this and your exact location where you live at based on this photo it looks like you're get ghosting signal often caused when a TV signal travels by two different paths to a receiving antenna, with a slight difference in timing.

Davetheraveuk, 2022-05-15:
@b-rad-NDi 
Hi, below is my dmesg output:

```
[    1.847492] usb 1-3: Product: dualHD
[   19.301827] em28xx 1-3:1.0: New device HCW dualHD @ 480 Mbps (2040:8265, interface 0, class 0)
[   19.301836] em28xx 1-3:1.0: DVB interface 0 found: bulk
[   19.354676] em28xx 1-3:1.0: chip ID is em28174
[   20.493974] em28xx 1-3:1.0: EEPROM ID = 26 00 01 00, EEPROM hash = 0x80ecca4c
[   20.493982] em28xx 1-3:1.0: EEPROM info:
[   20.493984] em28xx 1-3:1.0: 	microcode start address = 0x0004, boot configuration = 0x01
[   20.501218] em28xx 1-3:1.0: 	AC97 audio (5 sample rates)
[   20.501225] em28xx 1-3:1.0: 	500mA max power
[   20.501228] em28xx 1-3:1.0: 	Table at offset 0x27, strings=0x0e6a, 0x1888, 0x087e
[   20.553612] em28xx 1-3:1.0: Identified as Hauppauge WinTV-dualHD DVB (card=99)
[   20.556057] tveeprom: Hauppauge model 204209, rev C3I6, serial# 14318696
[   20.556066] tveeprom: tuner model is SiLabs Si2157 (idx 186, type 4)
[   20.556070] tveeprom: TV standards PAL(B/G) NTSC(M) PAL(I) SECAM(L/L') PAL(D/D1/K) ATSC/DVB Digital (eeprom 0xfc)
[   20.556077] tveeprom: audio processor is None (idx 0)
[   20.556080] tveeprom: has no radio, has IR receiver, has no IR transmitter
[   20.556088] em28xx 1-3:1.0: We currently don't support analog TV or stream capture on dual tuners.
[   20.556093] em28xx 1-3:1.0: dvb set to bulk mode.
[   20.607699] em28xx 1-3:1.0: chip ID is em28174
[   21.749758] em28xx 1-3:1.0: EEPROM ID = 26 00 01 00, EEPROM hash = 0x80ecca4c
[   21.749769] em28xx 1-3:1.0: EEPROM info:
[   21.749772] em28xx 1-3:1.0: 	microcode start address = 0x0004, boot configuration = 0x01
[   21.757010] em28xx 1-3:1.0: 	AC97 audio (5 sample rates)
[   21.757017] em28xx 1-3:1.0: 	500mA max power
[   21.757020] em28xx 1-3:1.0: 	Table at offset 0x27, strings=0x0e6a, 0x1888, 0x087e
[   21.808657] em28xx 1-3:1.0: Identified as Hauppauge WinTV-dualHD DVB (card=99)
[   21.811960] tveeprom: Hauppauge model 204209, rev C3I6, serial# 14318696
[   21.811967] tveeprom: tuner model is SiLabs Si2157 (idx 186, type 4)
[   21.811971] tveeprom: TV standards PAL(B/G) NTSC(M) PAL(I) SECAM(L/L') PAL(D/D1/K) ATSC/DVB Digital (eeprom 0xfc)
[   21.811976] tveeprom: audio processor is None (idx 0)
[   21.811978] tveeprom: has no radio, has IR receiver, has no IR transmitter
[   21.811984] em28xx 1-3:1.0: dvb ts2 set to bulk mode.
[   22.012417] usbcore: registered new interface driver em28xx
[   22.112592] em28xx 1-3:1.0: Binding DVB extension
[   22.787412] em28xx 1-3:1.0: DVB: registering adapter 0 frontend 0 (Silicon Labs Si2168)...
[   22.790086] em28xx 1-3:1.0: DVB extension successfully initialized
[   22.790095] em28xx 1-3:1.0: Binding DVB extension
[   23.943541] em28xx 1-3:1.0: DVB: registering adapter 1 frontend 0 (Silicon Labs Si2168)...
[   23.945509] em28xx 1-3:1.0: DVB extension successfully initialized
[   23.945514] em28xx: Registered (Em28xx dvb Extension) extension
[   24.175362] em28xx 1-3:1.0: Registering input extension
[   24.401121] rc rc0: Hauppauge WinTV-dualHD DVB as /devices/pci0000:00/0000:00:12.2/usb1/1-3/1-3:1.0/rc/rc0
[   24.401215] rc rc0: lirc_dev: driver em28xx registered at minor = 0, scancode receiver, no transmitter
[   24.401363] input: Hauppauge WinTV-dualHD DVB as /devices/pci0000:00/0000:00:12.2/usb1/1-3/1-3:1.0/rc/rc0/input17
[   24.401479] em28xx 1-3:1.0: Input extension successfully initialized
[   24.401495] em28xx 1-3:1.0: Remote control support is not available for this card.
[   24.401498] em28xx: Registered (Em28xx Input Extension) extension
[ 7050.316845] em28xx 1-3:1.0: Disconnecting em28xx #1
[ 7050.316850] em28xx 1-3:1.0: Disconnecting em28xx
[ 7050.316854] em28xx 1-3:1.0: Closing DVB extension
[ 7050.320535] em28xx 1-3:1.0: Closing DVB extension
[ 7050.323959] em28xx 1-3:1.0: Closing input extension
[ 7050.323966] em28xx 1-3:1.0: Closing input extension
[ 7050.353507] em28xx 1-3:1.0: Freeing device
[ 7050.353515] em28xx 1-3:1.0: Freeing device
[ 7050.706060] usb 1-3: Product: dualHD
[ 7050.706451] em28xx 1-3:1.0: New device HCW dualHD @ 480 Mbps (2040:8265, interface 0, class 0)
[ 7050.706460] em28xx 1-3:1.0: DVB interface 0 found: bulk
[ 7050.766143] em28xx 1-3:1.0: unknown em28xx chip ID (0)
[ 7050.850374] em28xx 1-3:1.0: reading from i2c device at 0xa0 failed (error=-5)
[ 7050.850384] em28xx 1-3:1.0: board has no eeprom
[ 7050.918107] em28xx 1-3:1.0: Identified as Hauppauge WinTV-dualHD DVB (card=99)
[ 7050.918117] em28xx 1-3:1.0: We currently don't support analog TV or stream capture on dual tuners.
[ 7050.918120] em28xx 1-3:1.0: dvb set to bulk mode.
[ 7050.994107] em28xx 1-3:1.0: unknown em28xx chip ID (0)
[ 7051.078460] em28xx 1-3:1.0: reading from i2c device at 0xa0 failed (error=-5)
[... 46 lines cut ...]
[ 7054.445666] em28xx 1-3:1.0: DVB: registering adapter 0 frontend 0 (Silicon Labs Si2168)...
[ 7054.446880] em28xx 1-3:1.0: DVB extension successfully initialized
[ 7054.446882] em28xx 1-3:1.0: Binding DVB extension
[ 7054.458405] em28xx 1-3:1.0: DVB: registering adapter 1 frontend 0 (Silicon Labs Si2168)...
[ 7054.459903] em28xx 1-3:1.0: DVB extension successfully initialized
[ 7054.459912] em28xx 1-3:1.0: Registering input extension
[ 7054.460652] rc rc0: Hauppauge WinTV-dualHD DVB as /devices/pci0000:00/0000:00:12.2/usb1/1-3/1-3:1.0/rc/rc0
[ 7054.460677] rc rc0: lirc_dev: driver em28xx registered at minor = 0, scancode receiver, no transmitter
[ 7054.460710] input: Hauppauge WinTV-dualHD DVB as /devices/pci0000:00/0000:00:12.2/usb1/1-3/1-3:1.0/rc/rc0/input19
[... cut ...]
> Can tell us more about antenna setup please post a photo of this antenna and your exact location where you live at based on this photo it looks like you're get ghosting signal often caused when a TV signal travels by two different paths to a receiving antenna, with a slight difference in timing.

My antenna is attached to the external chimney on top of the property (this is the highest point in the property). I also live (geographically) in a high area. My exact location is Loughton, Essex, UK. Postal code: IG10 2BY.

shspvr, 2022-05-15:
> 
> My antenna is attached to the external chimney on top of the property (this is the highest point in the property). I also live (geographically) in a high area. My exact location is Loughton, Essex, UK. Postal code: IG10 2BY.
I see you in UK
I could be wrong and I'm sure brad will know but your dmesg output seem to be missing the firmware has loaded which look something like this with in dmesg
downloading firmware from file '????????????????.fw'
firmware version: ???????

parkerlreed, 2022-05-16:
@shspvr This was just a quick test setup demonstrating that *some* signal works and can present weirdness. Wasn't anywhere near a permanent setup (inside a building with metal walls)

What's much more common is getting full screen randomness while it's trying to lock on.

My area in general is terrible for reception so I've just gotten used to having to fiddle with it constantly.

b-rad-NDi, 2022-05-17:
@Davetheraveuk Can you copy the dmesg output from when you're scanning or viewing a channel?

Davetheraveuk, 2022-10-11:
Hi,
From what I am seeing in the dmesg output from the terminal, there are no
errors.
Command I used: *sudo dmesg | egrep -i
'(ivtv|tveeprom|tuner|dualhd|em28xx)'*
Here's the weird thing: In Arch Kaffeine, the signal is very strong, but no
channels are detected. I also have a virtual machine running Ubuntu with
Kaffeine which also has a strong signal and *is* able to pickup channels.

If everything looks sweet in the terminal output then how come I cannot
receive any channels despite the signal strength being at maximum?
I am using the most recent ZEN kernel.
Hope you can help or advise.
Thanks for getting back to me.
David.