b-rad-NDi/Ubuntu-media-tree-kernel-builder #86: WinTV quadHD USB

shspvr, 2019-11-08:
Hi brad are you work on new quadHD USB model ?, If so do got any tips for make it work under PoP! OS 19.10

b-rad-NDi, 2019-11-08:
One is potentially in the mail to me now. Unfortunately this device uses a new usb bridge without any Linux support. I am working to get documentation and vendor assistance.

shspvr, 2019-11-08:
> One is potentially in the mail to me now. Unfortunately this device uses a new usb bridge without any Linux support. I am working to get documentation and vendor assistance.

Ok drop a message when you have something.
P.S. Are you aware of NextPVR now support Linux ?.

b-rad-NDi, 2019-11-08:
Nope, I'll let the bosses know in case they want to reach out to them though. I use Plex for DVR nowadays, and pretty much don't pay attention to anything else.

shspvr, 2019-11-08:
Are you refer Ken P at Hauppauge he know

b-rad-NDi, 2019-11-08:
yup Ken, if he knows then all is good.

b-rad-NDi, 2019-12-30:
Ok, I finally have a usb quadhd and I will be working on the driver support.

shspvr, 2019-12-30:
Cool

b-rad-NDi, 2020-01-23:
Progress report : I have the bridge code in place and can talk with all four tuners, including tune/lock/read. I just need to develop the demod/tuner driver from the materials I have.

shspvr, 2020-01-23:
that sound awesome bard

ElliotFriend, 2020-02-10:
Great to hear you're working on this. I wish I wouldn't have assumed the USB model would work, if the PCI models were listed as working.

Anything we can do to help?

siuside, 2020-02-14:
@ElliotFriend 
Ditto! Just got one from Amazon and trying to get this going in Raspberry Pi 4. I hope @b-rad-NDi is closer to a release.

b-rad-NDi, 2020-02-21:
If you can compile your own kernel then head over to my media tree repo and peruse the branch usb.quadhd.mxl692.a 
:D
I ran out of time this week to add it to the ppa.

shspvr, 2020-02-22:
Thanks I guest I have wait
for PPA as I have no experience in kennel compiling

siuside, 2020-02-22:
For the brave it is located here https://github.com/b-rad-NDi/media_tree/tree/usb.quadhd.mxl692.a
The RPi4 comes with 4.19.x and I will wait for the PPA release too.

b-rad-NDi, 2020-02-26:
As long as you don't run xenial-hwe or bionic 4.15.0 then the ppa should get your fully up and running now. The 4.15.0 releases will not build due to the way the build system depends on certain packages, and launchpad removed those packages...

Note: you will need linux-firmware-hauppauge as well.

If it works please note here and I'll close this issue.

siuside, 2020-02-26:
I get the following 
`pi@raspberrypi:~ $ apt list | grep -i hauppauge

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

linux-firmware-hauppauge/bionic,now 0.1.5+bionic all [installed]
linux-headers-4.15.0-74202001131313/bionic 4.15.0-74202001131313.0+mediatree+hauppauge all
linux-headers-5.3.0-41202002241620/bionic 5.3.0-41202002241620.0+mediatree+hauppauge~hwe all
pi@raspberrypi:~ $ 
`
and dmesg

`[  633.562448] usb 1-1.2.1: new high-speed USB device number 6 using xhci_hcd
[  633.692821] usb 1-1.2.1: New USB device found, idVendor=04e2, idProduct=0404, bcdDevice= 1.11
[  633.692837] usb 1-1.2.1: New USB device strings: Mfr=0, Product=1, SerialNumber=0
[  633.692847] usb 1-1.2.1: Product: USB 2.0 Hub
[  633.694191] hub 1-1.2.1:1.0: USB hub found
[  633.694268] hub 1-1.2.1:1.0: 4 ports detected
[  634.012484] usb 1-1.2.1.1: new high-speed USB device number 7 using xhci_hcd
[  634.143296] usb 1-1.2.1.1: New USB device found, idVendor=2040, idProduct=846d, bcdDevice= 1.00
[  634.143315] usb 1-1.2.1.1: New USB device strings: Mfr=3, Product=1, SerialNumber=2
[  634.143328] usb 1-1.2.1.1: Product: quadHD-A
[  634.143340] usb 1-1.2.1.1: Manufacturer: HCW
[  634.143352] usb 1-1.2.1.1: SerialNumber: 0014073174
[  634.242423] usb 1-1.2.1.3: new high-speed USB device number 8 using xhci_hcd
[  634.383284] usb 1-1.2.1.3: New USB device found, idVendor=2040, idProduct=846d, bcdDevice= 1.00
[  634.383303] usb 1-1.2.1.3: New USB device strings: Mfr=3, Product=1, SerialNumber=2
[  634.383316] usb 1-1.2.1.3: Product: quadHD-A
[  634.383330] usb 1-1.2.1.3: Manufacturer: HCW
[  634.383342] usb 1-1.2.1.3: SerialNumber: 0014073174

`
I don't see the mxl692

shspvr, 2020-02-26:
> As long as you don't run xenial-hwe or bionic 4.15.0 then the ppa should get your fully up and running now. The 4.15.0 releases will not build due to the way the build system depends on certain [...]

So far looking good under PoP OS! 19.10 with NextPVR and Kaffeine and is seeing all 4 tuner and seem to working good here Bard

b-rad-NDi, 2020-02-26:
@siuside : no where does it state arm compatibility for my ppa. There aren't even installable kernel packages for raspbian...If you want rpi support you must either use a distro that integrates my patches or compile your own kernel.

Thanks for the update @shspvr closing this issue.

closed 2020-02-26 by b-rad-NDi