b-rad-NDi/Ubuntu-media-tree-kernel-builder #88: LTS / HWE support.

LordTricky, 2019-11-25:
Hi, is there any news on when this will be supported?

I had my HDR4000 working perfect under ubuntu 18.04 with kernel 5.0.0.36 / 5.0.0.35... but all of a sudden it stopped working, i then removed linux-image-mediatree linux-headers-mediatree with the intention of reinstalling. I now get errors "linux-image-mediatree : Depends: linux-mediatree but it is not going to be installed
E: Unable to correct problems, you have held broken packages." even though linux-mediatree is installed. And modprobe: ERROR: could not insert 'cx88_dvb': No such device

I am at the point of accepting defeat and losing satellite completely but any help will be appreciated.

b-rad-NDi, 2019-11-25:
There is LTS support for Xenial and Bionic.

In addition I did some virtual package renaming. I tested it out on a variety of platforms and it worked for me, but it seems like there's edge cases. Essentially linux-image-mediatree and linux-headers-mediatree are deprecated. Now there is one single package linux-mediatree. There is also linux-hwe-mediatree. If you wipe out the old packages and ensure only the new one and it's dependency (the 11191317 kernel) then things should work.

I do not touch the cx88 driver, so if it stopped working then that would be an unfortunate side effect of updating the media tree drivers at the beginning of the month. Something in cx88 might have been broken at that instant in time. I'll review the logs.

LordTricky, 2019-11-25:
Thank you very much, so if i apt remove everything i have installed from the ppa (currently all hwe variants) and linux-mediatree, reboot and find / install kernel version 5.0.0.11191317 then reinstall linux-hwe-mediatree it should come back?

At the moment it doesn't even show up in dmesg, i know i've broke it (assumed originally from a microcode update) then worse by trying to reinstall it, i just cant reconfigure the whole server as i would lose too much data and time getting it back to where i had it.

b-rad-NDi, 2019-11-25:
There is definitely not any reason to re-install. All you should ever have to install is the virtual package, once, thereafter as it updates it installs new kernel and deprecates the old one for auto-removal. By installing (or just leaving it around) an Ubuntu kernel, and removing all mediatree packages you would be at vanilla.

LordTricky, 2019-11-25:
So all i need installed is linux-mediatree? does this provide an alternate driver from cx88? Sorry i'm not great with linux but usually get by.

b-rad-NDi, 2019-11-25:
linux-mediatree is a virtual package. It's only dependencies are the actual kernel packages, including all drivers, updated to the latest mainline tip Linux kernel version. This includes everything in drivers/media, so cx88 is the most up to date driver available at the sync time.
It's done this way to ensure updates happen. When linux-mediatree updates, you get a new version, which depends on new kernel packages. Everything is kept up to date automagically without you having to be specific.

LordTricky, 2019-11-25:
ok cool, should i also make clean in linuxtv media_build?

Do you have any idea why it just stopped working about 3 days ago? after reboots i had to modprobe cx88-dvb to get it working but now the card isnt even recognised but i have been messing trying to fix it.

b-rad-NDi, 2019-11-25:
Wait...are you building this yourself and installing/using those packages? Are you downloading the linuxtv.org media_build system, compiling it, then installing those drivers?

LordTricky, 2019-11-25:
yes im building my own kernel with linuxtv, i was under the impression that i needed linuxtv media_build for my astrometa DVB-C usb tuner (which works perfectly) and linuxtv says they support the HDR4000 as well, so i dont think i am installing any driver as its already in media_build at least thats how i have took it. I have the firmware files in lib/firmware/ dvb-fe-cx24116.fw dvb-fe-cx24116-1.26.90.0.fw but i did also had linux-image-mediatree and linux-headers-mediatree installed from the ppa.

I now cant reinstall linux-image-mediatree linux-headers-mediatree due to apparently held packages or the deprecation, aptitude says 
The following packages have unmet dependencies:
 linux-mediatree : Breaks: linux-image-mediatree (> 0) but 0.19.7+bionic is to be installed
The following actions will resolve these dependencies:

     Keep the following packages at their current version:
1)     linux-image-mediatree [Not Installed]

b-rad-NDi, 2019-11-25:
That's the thing. You are essentially overriding the entire purpose of this project. This project uses the linuxtv.org media_build system to produce a set of tip drivers that are then backported and integrated into whatever kernel I offer. When you compile media_build and then install those drivers you are nuking everything included by my packages. Unless you require a commit more recent than last month, you have no reason to use the media_build system. That is why you are using my ppa.

Now onto the dependency issue:
linux-mediatree == linux-image-mediatree + linux-headers-mediatree
The packages are intentionally broken. You should only install linux-mediatree.

LordTricky, 2019-11-25:
ok so for the HDR4000 card my problem is a conflict between linuxtv and your ppa, can you suggest my best option for getting back to your kernel instead of linuxtv please. I assume make clean wont remove linuxtv kernel?

LordTricky, 2019-11-26:
ok so i have tried to build your kernel, at least i assume thats what im doing. I have tried
 git clone https://github.com/b-rad-NDi/Ubuntu-media-tree-kernel-builder.git brad
cd brad
 ./mediatree-kbuild.sh -i -p -x
cd media_build
./build
make install
which seems to have reset my drivers in /lib/modules
i now get this in dmesg
 dmesg | grep cx
[    7.707975] cx88xx: func 1: Can't get MMIO memory @ 0x0, subsystem: 0070:6902
[    7.736900] Modules linked in: dvb_core(OE) videobuf2_dma_sg(OE) videobuf2_vmalloc(OE) cx88_alsa(OE+) cx88xx(OE) videobuf2_memops(OE) videobuf2_v4l2(OE) tveeprom(OE) sparse_keymap(+) videobuf2_common(OE) joydev input_leds wmi_bmof ch341 videodev(OE) mc(OE) snd_hdmi_lpe_audio(+) usbserial lpc_ich snd_pcm snd_timer intel_xhci_usb_role_switch roles snd soundcore mei_txe hci_uart mei btqca parport_pc btrtl parport btbcm btintel bluetooth ecdh_generic rfkill_gpio mac_hid intel_int0002_vgpio sch_fq_codel ib_iser rdma_cm iw_cm ib_cm ib_core iscsi_tcp libiscsi_tcp libiscsi scsi_transport_iscsi ip_tables x_tables autofs4 btrfs zstd_compress raid10 raid456 async_raid6_recov async_memcpy async_pq async_xor async_tx xor raid6_pq libcrc32c raid1 raid0 multipath linear hid_generic usbhid i915 crct10dif_pclmul crc32_pclmul kvmgt vfio_mdev ghash_clmulni_intel mdev vfio_iommu_type1 vfio kvm irqbypass i2c_algo_bit drm_kms_helper aesni_intel syscopyarea sysfillrect sysimgblt fb_sys_fops aes_x86_64
[    7.737020]  snd_cx88_dev_free+0x20/0x40 [cx88_alsa]
[    7.737056]  cx88_audio_initdev+0x1ba/0x460 [cx88_alsa]
[    7.737103]  cx88_audio_pci_driver_init+0x23/0x1000 [cx88_alsa]
[    7.756136] RIP: 0010:cx88_core_put+0xa/0x100 [cx88xx]
[    7.798737]  snd_cx88_dev_free+0x2c/0x40 [cx88_alsa]
[    7.823478]  cx88_audio_initdev+0x1ba/0x460 [cx88_alsa]
[    7.878218]  cx88_audio_pci_driver_init+0x23/0x1000 [cx88_alsa]
[    7.941922] Modules linked in: dvb_core(OE) videobuf2_dma_sg(OE) videobuf2_vmalloc(OE) cx88_alsa(OE+) cx88xx(OE) videobuf2_memops(OE) videobuf2_v4l2(OE) tveeprom(OE) sparse_keymap(+) videobuf2_common(OE) joydev input_leds wmi_bmof ch341 videodev(OE) mc(OE) snd_hdmi_lpe_audio(+) usbserial lpc_ich snd_pcm snd_timer intel_xhci_usb_role_switch roles snd soundcore mei_txe hci_uart mei btqca parport_pc btrtl parport btbcm btintel bluetooth ecdh_generic rfkill_gpio mac_hid intel_int0002_vgpio sch_fq_codel ib_iser rdma_cm iw_cm ib_cm ib_core iscsi_tcp libiscsi_tcp libiscsi scsi_transport_iscsi ip_tables x_tables autofs4 btrfs zstd_compress raid10 raid456 async_raid6_recov async_memcpy async_pq async_xor async_tx xor raid6_pq libcrc32c raid1 raid0 multipath linear hid_generic usbhid i915 crct10dif_pclmul crc32_pclmul kvmgt vfio_mdev ghash_clmulni_intel mdev vfio_iommu_type1 vfio kvm irqbypass i2c_algo_bit drm_kms_helper aesni_intel syscopyarea sysfillrect sysimgblt fb_sys_fops aes_x86_64
[    7.972389] RIP: 0010:cx88_core_put+0xa/0x100 [cx88xx]
[    7.972666] cx88xx: func 2: Can't get MMIO memory @ 0x0, subsystem: 0070:6902
[    7.989136] cx88-mpeg driver manager: probe of 0000:04:00.2 failed with error -22
[    8.055101] cx88xx: func 0: Can't get MMIO memory @ 0x0, subsystem: 0070:6902
[    8.070237] cx8800: probe of 0000:04:00.0 failed with error -22
[    8.271149] i2c i2c-9: cxd2841er_attach(): I2C adapter 00000000be2ffac5 SLVX addr 6e SLVT addr 6c
[    8.284448] i2c i2c-9: cxd2841er_attach(): attaching CXD2837ER DVB-C/T/T2 frontend
[    8.289095] i2c i2c-9: cxd2841er_attach(): chip ID 0xb1 OK.
[  109.410058] cx88_dvb: cx2388x dvb driver version 1.0.0 loaded
[  109.410144] cx8802: registering cx8802 driver, type: dvb access: shared

What am i misunderstanding or doing wrong please?

b-rad-NDi, 2019-11-26:
You are not ever supposed to touch anything in that directory. None of my (minimal) instructions indicate as such.
```
# help
./mediatree-kbuild.sh -h
# initialize
./mediatree-kbuild.sh -i
# patch
./mediatree-kbuild.sh -px
# clean
./mediatree-kbuild.sh -c
# build
./mediatree-kbuild.sh -b
# to start fresh
./mediatree-kbuild.sh -r
```

LordTricky, 2019-11-26:
yes i was guessing how to use it, and trying to work around an error, ./mediatree-kbuild.sh -c results in 
./mediatree-kbuild.sh -c
for i in debian.master/control.stub.in; do                                      \
  new=`echo $i | sed 's/\.in$//'`;                                      \
  cat $i | sed -e 's/PKGVER/4.15.0/g'                              \
        -e 's/ABINUM/70201911261414/g'                                       \
        -e 's/SRCPKGNAME/linux/g'                             \
        -e 's/=HUMAN=/64 bit x86/g'                                    \
  > $new;                                                               \
done
flavours="debian.master/control.d/vars.generic debian.master/control.d/vars.generic-lpae debian.master/control.d/vars.lowlatency";\
for i in $flavours; do                                                  \
  /bin/bash -e debian/scripts/control-create $i "" |    \
        sed -e 's/PKGVER/4.15.0/g'                                  \
        -e 's/ABINUM/70201911261414/g'                                       \
        -e 's/SRCPKGNAME/linux/g'                             \
        -e 's/=HUMAN=/64 bit x86/g'                                    \
        >> debian.master/control.stub;                                          \
done
echo "# placebo control.stub for kernel-wedge flow change" >debian/control.stub
cp debian.master/control.stub debian/control
export KW_DEFCONFIG_DIR=debian.master/d-i && \
export KW_CONFIG_DIR=debian.master/d-i && \
LANG=C kernel-wedge gen-control 4.15.0-70201911261414 | \
        perl -f debian/scripts/misc/kernel-wedge-arch.pl amd64 \
        >>/home/tricky/brad/ubuntu-bionic/debian/control
/bin/bash: line 2: kernel-wedge: command not found
dh_testdir
/bin/bash: dh_testdir: command not found
debian/rules:116: recipe for target 'clean' failed
make: *** [clean] Error 127

am i missing a dependency or something?

LordTricky, 2019-11-26:
was missing kernel-wedge. Now building / compiling, will report back in an hour or so once its finished.

Thank you for your help :)

LordTricky, 2019-11-26:
wow, almost 4 hours of building then failed with some package not found errors

II: dkms-build downloading spl (spl-dkms_0.7.5-1ubuntu2_all.deb)
--2019-11-26 18:31:03--  http://ppa.launchpad.net/b-rad/kernel+mediatree+hauppauge/ubuntu/pool/universe/s/spl-linux/spl-dkms_0.7.5-1ubuntu2_all.deb
Resolving ppa.launchpad.net (ppa.launchpad.net)... 91.189.95.83, 2001:67c:1560:8008::15
Connecting to ppa.launchpad.net (ppa.launchpad.net)|91.189.95.83|:80... connected.
HTTP request sent, awaiting response... 404 Not Found
2019-11-26 18:31:03 ERROR 404: Not Found.

--2019-11-26 18:31:03--  http://ppa.launchpad.net/b-rad/kernel+mediatree+hauppauge/ubuntu/pool/main/s/spl-linux/spl-dkms_0.7.5-1ubuntu2_all.deb
Resolving ppa.launchpad.net (ppa.launchpad.net)... 91.189.95.83, 2001:67c:1560:8008::15
Connecting to ppa.launchpad.net (ppa.launchpad.net)|91.189.95.83|:80... connected.
HTTP request sent, awaiting response... 404 Not Found
2019-11-26 18:31:03 ERROR 404: Not Found.

--2019-11-26 18:31:03--  http://security.ubuntu.com/ubuntu/pool/universe/s/spl-linux/spl-dkms_0.7.5-1ubuntu2_all.deb
Resolving security.ubuntu.com (security.ubuntu.com)... 91.189.91.26, 91.189.91.23, 91.189.88.162, ...
Connecting to security.ubuntu.com (security.ubuntu.com)|91.189.91.26|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 392728 (384K) [application/x-debian-package]
Saving to: 'spl-dkms_0.7.5-1ubuntu2_all.deb'

spl-dkms_0.7.5-1ubuntu2_all.deb                      100%[=====================================================================================================================>] 383.52K   836KB/s    in 0.5s

2019-11-26 18:31:04 (836 KB/s) - 'spl-dkms_0.7.5-1ubuntu2_all.deb' saved [392728/392728]

II: dkms-build building spl
debian/scripts/dkms-build: line 120: /usr/sbin/dkms: No such file or directory
debian/scripts/dkms-build: line 132: /home/tricky/brad/ubuntu-bionic/debian/build/build-generic/_____________________________________dkms/build/spl/0.7.5/4.15.0-70201911261500-generic/*/log/make.log: No such file or directory
debian/rules.d/2-binary-arch.mk:112: recipe for target 'install-generic' failed
make: *** [install-generic] Error 1

real    210m0.149s
user    629m30.948s
sys     106m10.105s

Any idea's?

b-rad-NDi, 2019-11-26:
This is something dumb that Ubuntu did recently. They hardcoded package versions into their build system. Packages which update more quickly than the kernel does and which are removed from the archives the moment they're deprecated. You basically need to bump the version number yourself in debian/dkms-versions. There's three packages that couple have changed, so you might as well look each up the first time. They're all in the dkms-versions file. You basically look through universe/multiverse repos for the correct version. Yes, it's a hassle.

LordTricky, 2019-11-27:
ok  i did 
apt-get install dkms spl-dkms 
sudo ./mediatree-kbuild.sh -c
sudo ./mediatree-kbuild.sh -b
And after almost 5 hours i think it has finally built, with the last part of the output saying 
Debug: install-perarch
# Add the tools.
Debug: binary-perarch

real    257m42.386s
user    676m45.149s
sys     130m23.804s

Now i really don't want to screw this up and wait another 5 hours. 
How do i now install the kernel? or is it already installed by your script?
It sounds like ./mediatree-kbuild.sh -r will wipe out the last 5 hours of compiling?

LordTricky, 2019-11-27:
should i be doing
sudo dpkg -i linux-image-4.15.0-70201911261949-generic_4.15.0-70201911261949.1+mediatree+hauppauge_amd64.deb 
linux-modules-4.15.0-70201911261949-generic_4.15.0-70201911261949.1+mediatree+hauppauge_amd64.deb
linux-modules-extra-4.15.0-70201911261949-generic_4.15.0-70201911261949.1+mediatree+hauppauge_amd64.deb
linux-buildinfo-mediatree-4.15.0-70201911261949-generic_4.15.0-70201911261949.1+mediatree+hauppauge_amd64.deb
linux-headers-4.15.0-70201911261949-generic_4.15.0-70201911261949.1+mediatree+hauppauge_amd64.deb

does the order matter?
i also have 2 other header files but they were created hours ago so i assume they are from the failed builds, should i ignore those?

b-rad-NDi, 2019-11-27:
As you've found, the debs are produced in the top level directory. Now that you've built it (4+ hours ouch, mine takes under 60min) you will only have to reset the directories to do a fresh build. This will not affect the built debs.
All packages are time stamped, and one headers package is created very early so will exist even in the case of failed build. You will need two headers packages, image, modules, and modules-extra. If you tell dpkg to install all of them at once it will figure things out.

LordTricky, 2019-11-27:
yeah i went with sudo dpkg -i linux*4.15.0-70201911261949*
it installed, rebooted but still had no DVR4000 :(

LordTricky, 2019-11-27:
can you see anything that is glaringly obvious to you but i'm blind to?

i do see something about trying to free an already free IRQ but have no idea how to handle it.

dmesg
[    0.000000] microcode: microcode updated early to revision 0x368, date = 2019-04-23
[    0.000000] Linux version 4.15.0-70201911261949-generic (root@ubuntu) (gcc version 7.4.0 (Ubuntu 7.4.0-1ubuntu1~18.04.1)) #1+mediatree+hauppauge SMP Tue Nov 26 19:52:26 GMT 2019 (Ubuntu 4.15.0-70201911261949.1+mediatree+hauppauge-generic 4.15.18)
[    0.000000] Command line: BOOT_IMAGE=/vmlinuz-4.15.0-70201911261949-generic root=/dev/mapper/ubuntu--vg-root ro
[    0.000000] KERNEL supported cpus:
[    0.000000]   Intel GenuineIntel
[    0.000000]   AMD AuthenticAMD
[    0.000000]   Centaur CentaurHauls
[    0.000000] x86/fpu: x87 FPU will use FXSAVE
[    0.000000] e820: BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000003efff] usable
[    0.000000] BIOS-e820: [mem 0x000000000003f000-0x000000000003ffff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x0000000000040000-0x000000000009ffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000003fffffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000040000000-0x00000000401fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000040200000-0x00000000725e2fff] usable
[    0.000000] BIOS-e820: [mem 0x00000000725e3000-0x0000000072660fff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000072661000-0x0000000072672fff] ACPI data
[    0.000000] BIOS-e820: [mem 0x0000000072673000-0x000000007355efff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x000000007355f000-0x000000007389afff] reserved
[    0.000000] BIOS-e820: [mem 0x000000007389b000-0x00000000738f2fff] type 20
[    0.000000] BIOS-e820: [mem 0x00000000738f3000-0x0000000073adffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000073ae0000-0x0000000073fc2fff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000073fc3000-0x0000000073ffffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000e0000000-0x00000000e3ffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fea00000-0x00000000feafffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec00000-0x00000000fec00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed01000-0x00000000fed01fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed03000-0x00000000fed03fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed06000-0x00000000fed06fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed08000-0x00000000fed09fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed1c000-0x00000000fed1cfff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed80000-0x00000000fedbffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fee00000-0x00000000fee00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ff800000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000027fffffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] efi: EFI v2.40 by American Megatrends
[    0.000000] efi:  ESRT=0x73897218  ACPI=0x72664000  ACPI 2.0=0x72664000  SMBIOS=0xf05b0  MPS=0xfd610 
[    0.000000] secureboot: Secure boot could not be determined (mode 0)
[    0.000000] SMBIOS 2.8 present.
[    0.000000] DMI: ASUS All Series/N3150M-E, BIOS 0304 06/09/2015
[    0.000000] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.000000] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.000000] e820: last_pfn = 0x280000 max_arch_pfn = 0x400000000
[    0.000000] MTRR default type: uncachable
[    0.000000] MTRR fixed ranges enabled:
[    0.000000]   00000-9FFFF write-back
[    0.000000]   A0000-BFFFF uncachable
[    0.000000]   C0000-FFFFF write-protect
[    0.000000] MTRR variable ranges enabled:
[    0.000000]   0 base 000000000 mask F80000000 write-back
[    0.000000]   1 base 076B00000 mask FFFF00000 uncachable
[    0.000000]   2 base 076C00000 mask FFFC00000 uncachable
[    0.000000]   3 base 077000000 mask FFF000000 uncachable
[    0.000000]   4 base 078000000 mask FF8000000 uncachable
[    0.000000]   5 base 100000000 mask F00000000 write-back
[    0.000000]   6 base 200000000 mask F80000000 write-back
[    0.000000]   7 disabled
[    0.000000] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WP  UC- WT  
[    0.000000] total RAM covered: 8043M
[    0.000000] Found optimal setting for mtrr clean up
[    0.000000]  gran_size: 64K 	chunk_size: 256M 	num_reg: 7  	lose cover RAM: 0G
[    0.000000] e820: update [mem 0x76b00000-0xffffffff] usable ==> reserved
[    0.000000] e820: last_pfn = 0x74000 max_arch_pfn = 0x400000000
[    0.000000] found SMP MP-table at [mem 0x000fd7e0-0x000fd7ef]
[... 841 lines cut ...]
[    8.702680] usbcore: registered new interface driver dvb_usb_rtl28xxu
[    8.706617] EXT4-fs (sda2): mounting ext2 file system using the ext4 subsystem
[    8.713332] EXT4-fs (sda2): mounted filesystem without journal. Opts: (null)
[    9.742883] audit: type=1400 audit(1574823104.048:2): apparmor="STATUS" operation="profile_load" profile="unconfined" name="/usr/bin/lxc-start" pid=860 comm="apparmor_parser"
[    9.850459] audit: type=1400 audit(1574823104.156:3): apparmor="STATUS" operation="profile_load" profile="unconfined" name="/usr/bin/ubuntu-core-launcher" pid=864 comm="apparmor_parser"
[    9.864702] random: crng init done
[    9.868186] random: 7 urandom warning(s) missed due to ratelimiting
[... cut ...]
[   15.564755] IPv6: ADDRCONF(NETDEV_CHANGE): enp2s0: link becomes ready
[  131.479883] kauditd_printk_skb: 8 callbacks suppressed
[  131.479888] audit: type=1400 audit(1574823225.820:20): apparmor="DENIED" operation="capable" profile="/sbin/dhclient" pid=1254 comm="dhclient" capability=16  capname="sys_module"
[  258.002119] usb 1-5.2: DVB: adapter 0 frontend 0 frequency 0 out of range (174000000..862000000)
[  258.801767] usb 1-5.2: DVB: adapter 0 frontend 1 frequency 0 out of range (42000000..1002000000)

LordTricky, 2019-11-27:
lspci -v
04:00.0 Multimedia video controller: Conexant Systems, Inc. CX23880/1/2/3 PCI Video and Audio Decoder (rev 05)
	Subsystem: Hauppauge computer works Inc. WinTV HVR-4000-HD
	Flags: medium devsel, IRQ 19
	Capabilities: [44] Vital Product Data
	Capabilities: [4c] Power Management version 2
	Kernel modules: cx8800

04:00.1 Multimedia controller: Conexant Systems, Inc. CX23880/1/2/3 PCI Video and Audio Decoder [Audio Port] (rev 05)
	Subsystem: Hauppauge computer works Inc. WinTV HVR-4000-HD
	Flags: bus master, medium devsel, latency 32, IRQ 19
	Capabilities: [4c] Power Management version 2
	Kernel driver in use: cx88_audio
	Kernel modules: cx88_alsa

04:00.2 Multimedia controller: Conexant Systems, Inc. CX23880/1/2/3 PCI Video and Audio Decoder [MPEG Port] (rev 05)
	Subsystem: Hauppauge computer works Inc. WinTV HVR-4000-HD
	Flags: medium devsel, IRQ 10
	Capabilities: [4c] Power Management version 2
	Kernel modules: cx8802

04:00.4 Multimedia controller: Conexant Systems, Inc. CX23880/1/2/3 PCI Video and Audio Decoder [IR Port] (rev 05)
	Subsystem: Hauppauge computer works Inc. WinTV HVR-4000-HD
	Flags: medium devsel, IRQ 10
	Capabilities: [4c] Power Management version 2

surely i'm missing something simple?

LordTricky, 2019-11-29:
i'm really having no joy with this, is there any way i can still get the pre deprecated files that did work for me? please?

LordTricky, 2019-12-02:
Sorry to pester but i'm still having no joy even after a fresh install of ubuntu 18.04 on a spare ssd. I'm starting to think the problem could be around the pci to pcie adapter i have the hdr4000 in. I'm also thinking i need to do something with "setpci" but i can't get my head around it. Can you make any suggestions please?

lspci -vv 
02:00.0 PCI bridge: ASMedia Technology Inc. ASM1083/1085 PCIe to PCI Bridge (rev 04) (prog-if 00 [Normal decode])
        Control: I/O+ Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
        Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
        Latency: 0, Cache Line Size: 64 bytes
        Interrupt: pin A routed to IRQ 17
        Bus: primary=02, secondary=03, subordinate=03, sec-latency=32
        Secondary status: 66MHz+ FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort+ <SERR- <PERR+
        BridgeCtl: Parity- SERR- NoISA- VGA- MAbort- >Reset- FastB2B-
                PriDiscTmr- SecDiscTmr- DiscTmrStat- DiscTmrSERREn-
        Capabilities: <access denied>

03:00.0 Multimedia video controller: Conexant Systems, Inc. CX23880/1/2/3 PCI Video and Audio Decoder (rev 05)
        Subsystem: Hauppauge computer works Inc. WinTV HVR-4000-HD
        Control: I/O- Mem- BusMaster- SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
        Status: Cap+ 66MHz- UDF- FastB2B+ ParErr- DEVSEL=medium >TAbort- <TAbort- <MAbort- >SERR- <PERR+ INTx-
        Interrupt: pin A routed to IRQ 17
        Capabilities: <access denied>
        Kernel modules: cx8800

03:00.1 Multimedia controller: Conexant Systems, Inc. CX23880/1/2/3 PCI Video and Audio Decoder [Audio Port] (rev 05)
        Subsystem: Hauppauge computer works Inc. WinTV HVR-4000-HD
        Control: I/O- Mem- BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
        Status: Cap+ 66MHz- UDF- FastB2B+ ParErr- DEVSEL=medium >TAbort- <TAbort- <MAbort- >SERR- <PERR+ INTx-
        Latency: 32 (1000ns min, 63750ns max), Cache Line Size: 64 bytes
        Interrupt: pin A routed to IRQ 17
        Capabilities: <access denied>
        Kernel driver in use: cx88_audio
        Kernel modules: cx88_alsa

03:00.2 Multimedia controller: Conexant Systems, Inc. CX23880/1/2/3 PCI Video and Audio Decoder [MPEG Port] (rev 05)
        Subsystem: Hauppauge computer works Inc. WinTV HVR-4000-HD
        Control: I/O- Mem- BusMaster- SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
        Status: Cap+ 66MHz- UDF- FastB2B+ ParErr- DEVSEL=medium >TAbort- <TAbort- <MAbort- >SERR- <PERR+ INTx-
        Interrupt: pin A routed to IRQ 3
        Capabilities: <access denied>
        Kernel modules: cx8802

03:00.4 Multimedia controller: Conexant Systems, Inc. CX23880/1/2/3 PCI Video and Audio Decoder [IR Port] (rev 05)
        Subsystem: Hauppauge computer works Inc. WinTV HVR-4000-HD
        Control: I/O- Mem- BusMaster- SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
        Status: Cap+ 66MHz- UDF- FastB2B+ ParErr- DEVSEL=medium >TAbort- <TAbort- <MAbort- >SERR- <PERR+ INTx-
        Interrupt: pin A routed to IRQ 3
        Capabilities: <access denied>

b-rad-NDi, 2019-12-05:
For me to further address this issue you will have to start running a mainline kernel. If you refile this issue on my media_tree repository I will assist you there with the cx88 issues.

b-rad-NDi, 2019-12-05:
Also, I have no idea about pci to pcie converters or issues associated with them. I'd have to look that up

LordTricky, 2019-12-05:
no worries, i have since done a fresh install on spare ssd of ubuntu 18.04, ubuntu 19.10 and ubuntu 16.04, all of which gave me no positive results. I currently have a mainline kernel running on my ubuntu 18.04 (original setup) 5.4.1-050401-generic.

I have also tested the HVR4000 in a windows machine and that is still working, i also tried it with the pci to pcie adaptor and had no success, i'm not sure if this is windows driver related or not, but i have now ordered a new pci to pcie adaptor to rule it out, new one should be here in about 3 weeks (damn china post). 

I will update once the new pci to pcie adaptor arrives, its strange because both windows and linux see the adaptor, but only linux see's the HVR4000 in the adaptor. if you are interested the adaptor is ASM1083 https://linux-hardware.org/index.php?id=pci:1b21-1080-1043-8489&page=6

Thank you once more for your help.

b-rad-NDi, 2019-12-05:
The fact it worked once upon a time seems to indicate a regression introduced somewhere. If that regression is not easy to determine then the only solution is to do a kernel bisect via git. Essentially you go back in the past on mainline, compile and test that (hoping it works as a baseline), then do a binary search from there to tip to determine where the regression was introduced. It's tedious, but not terribly so.

LordTricky, 2019-12-16:
I owe you an apology, i'm very sorry. My new pci to pci-e adaptor has now arrived and my HVR4000 is working as expected once more. It seems it was just coincidental that the old one broke around the time of the last update. 
Thank you for your help in the matter.
This issue can now be closed.

b-rad-NDi, 2019-12-16:
d'oh! It happens, glad to hear you're back in order.

closed 2019-12-16 by b-rad-NDi

LordTricky, 2020-01-07:
Hi Brad, me again. I have bought a motorised dish and i'm struggling to get some satellites 16E being the main 1 i want, but i'm running into apparent driver problems.

this is the output from w_scan for 27.5W
trying 'S  f = 3714 kHz L SR =  2915  7/8 0,35  QPSK  (0:0:0)'
(time: 00:13.223) 	 skipped: (freq 6.04 unsupported by driver: min=0.95, max=2.15)

initial_tune:2813: Setting frontend failed S  f = 3714 kHz L SR =  2915  7/8 0,35  QPSK  (0:0:0)
trying 'S  f = 3715 kHz R SR = 12000  1/2 0,35  QPSK  (0:0:0)'
(time: 00:14.120) 	 skipped: (freq 6.04 unsupported by driver: min=0.95, max=2.15)

initial_tune:2813: Setting frontend failed S  f = 3715 kHz R SR = 12000  1/2 0,35  QPSK  (0:0:0)
trying 'S  f = 3764 kHz R SR = 24450  3/4 0,35  QPSK  (0:0:0)'
(time: 00:15.028) 	 skipped: (freq 5.99 unsupported by driver: min=0.95, max=2.15)

initial_tune:2813: Setting frontend failed S  f = 3764 kHz R SR = 24450  3/4 0,35  QPSK  (0:0:0)
trying 'S  f = 3829 kHz R SR =  5787  3/4 0,35  QPSK  (0:0:0)'
(time: 00:15.924) 	 skipped: (freq 5.92 unsupported by driver: min=0.95, max=2.15)

initial_tune:2813: Setting frontend failed S  f = 3829 kHz R SR =  5787  3/4 0,35  QPSK  (0:0:0)
trying 'S  f = 3838 kHz R SR =  7223  3/4 0,35  QPSK  (0:0:0)'
(time: 00:16.832) 	 skipped: (freq 5.91 unsupported by driver: min=0.95, max=2.15)

initial_tune:2813: Setting frontend failed S  f = 3838 kHz R SR =  7223  3/4 0,35  QPSK  (0:0:0)
trying 'S  f = 3898 kHz R SR =  2893  3/4 0,35  QPSK  (0:0:0)'
(time: 00:17.752) 	 skipped: (freq 5.85 unsupported by driver: min=0.95, max=2.15)

initial_tune:2813: Setting frontend failed S  f = 3898 kHz R SR =  2893  3/4 0,35  QPSK  (0:0:0)
trying 'S  f = 3936 kHz R SR =  4550  1/2 0,35  QPSK  (0:0:0)'
(time: 00:18.660) 	 skipped: (freq 5.81 unsupported by driver: min=0.95, max=2.15)

initial_tune:2813: Setting frontend failed S  f = 3936 kHz R SR =  4550  1/2 0,35  QPSK  (0:0:0)
trying 'S  f = 4048 kHz R SR =  7000  3/4 0,35  QPSK  (0:0:0)'
(time: 00:19.576) 	 skipped: (freq 5.70 unsupported by driver: min=0.95, max=2.15)

initial_tune:2813: Setting frontend failed S  f = 4048 kHz R SR =  7000  3/4 0,35  QPSK  (0:0:0)
trying 'S2 f = 11495 kHz V SR = 44100 9/10 0,AUTO  QPSK  (0:0:0)'
(time: 00:20.525) 
trying 'S  f = 11666 kHz V SR =  9765  5/6 0,35  QPSK  (0:0:0)'
(time: 00:23.433) 

And 16E
trying 'S  f = 11070 kHz H SR =   333  1/2 0,35  QPSK  (0:0:0)'
(time: 00:58.434) 	 skipped: (srate 333000 unsupported by driver)

initial_tune:2813: Setting frontend failed S  f = 11070 kHz H SR =   333  1/2 0,35  QPSK  (0:0:0)

Is this a hardware limitation? or is it just that the driver is not yet built to support these other satellites?

I really don't want to be replacing my HVR4000 if i can help it.

b-rad-NDi, 2020-06-07:
@LordTricky : this issue slipped by past me due to this issue being closed. If you are still having issue please file a fresh ticket and I'll see if I can get to the bottom of your problem.