b-rad-NDi/Ubuntu-media-tree-kernel-builder #89: quadhd only two tuners available

skalnet, 2019-12-11:
recently got a quadhd pcie but somehow i can't get all of the frontends to show up.

```
uname -a
Linux omv1 5.3.0-25201912061620-generic #0+mediatree+hauppauge~hwe-Ubuntu SMP Sat Dec 7 04:27:41 UTC 201 x86_64 x86_64 x86_64 GNU/Linux

```

since i saw in another thread someone should create a modprobe conf file with
```
option cx23885 debug=1
option cx25840 debug=1
options tuner debug=1
options dvb-core frontend_debug=1
options dvb-core debug=1
```

```
dmesg | egrep 'si2|lgdt|cx2|dvb|tune|frontend|adapter'
[    4.887643] cx23885: cx23885 driver version 0.0.4 loaded
[    4.887799] cx23885: CORE cx23885[0]: subsystem: 0070:6a28, board: Hauppauge WinTV-QuadHD-DVB(885) [card=60,autodetected]
[    5.237714] tveeprom: tuner model is SiLabs Si2157 (idx 186, type 4)
[    5.237720] cx23885: cx23885[0]: hauppauge eeprom: model=166200
[    5.237722] cx23885: cx23885_dvb_register() allocating 1 frontend(s)
[    5.237724] cx23885: cx23885[0]: cx23885 based dvb card
[    5.237726] cx23885: dvb_register(): board=60 port=1
[    5.252263] si2168 8-0064: Silicon Labs Si2168-B40 successfully identified
[    5.252264] si2168 8-0064: firmware version: B 4.0.2
[    5.255876] si2157 9-0060: Silicon Labs Si2147/2148/2157/2158 successfully attached
[    5.255897] dvbdev: DVB: registering new adapter (cx23885[0])
[    5.255900] cx23885 0000:04:00.0: DVB: registering adapter 0 frontend 0 (Silicon Labs Si2168)...
[    5.256454] cx23885: cx23885_dvb_register() allocating 1 frontend(s)
[    5.256455] cx23885: cx23885[0]: cx23885 based dvb card
[    5.256458] cx23885: dvb_register(): board=60 port=2
[    5.265913] si2168 8-0066: Silicon Labs Si2168-B40 successfully identified
[    5.265914] si2168 8-0066: firmware version: B 4.0.2
[    5.269207] si2157 9-0062: Silicon Labs Si2147/2148/2157/2158 successfully attached
[    5.269251] dvbdev: DVB: registering new adapter (cx23885[0])
[    5.269255] cx23885 0000:04:00.0: DVB: registering adapter 1 frontend 0 (Silicon Labs Si2168)...
[    5.269655] cx23885: cx23885_dev_checkrevision() Hardware revision = 0xa5
[    5.269661] cx23885: cx23885[0]/0: found at 0000:04:00.0, rev: 4, irq: 17, latency: 0, mmio: 0xfba00000
[    5.269823] cx23885: CORE cx23885[1]: subsystem: 0070:6b28, board: Hauppauge WinTV-QuadHD-DVB(885) [card=60,autodetected]
[    5.623985] tveeprom: tuner model is SiLabs Si2157 (idx 186, type 4)
[    5.623988] cx23885: cx23885[1]: hauppauge eeprom: model=166201
[    5.623990] cx23885: cx23885_dvb_register() allocating 1 frontend(s)
[    5.623991] cx23885: cx23885[1]: cx23885 based dvb card
[    5.623993] cx23885: dvb_register(): board=60 port=1
[    5.932345] si2168 13-0064: probe failed = -110
[    5.932354] si2168: probe of 13-0064 failed with error -110
[    5.932363] cx23885: cx23885_dvb_register() dvb_register failed err = -22
[    5.932366] cx23885: cx23885_dev_setup() Failed to register dvb adapters on VID_B
[    5.932368] cx23885: cx23885_dvb_register() allocating 1 frontend(s)
[    5.932369] cx23885: cx23885[1]: cx23885 based dvb card
[    5.932372] cx23885: dvb_register(): board=60 port=2
[    6.238427] si2168 13-0066: probe failed = -110
[    6.238437] si2168: probe of 13-0066 failed with error -110
[    6.238446] cx23885: cx23885_dvb_register() dvb_register failed err = -22
[    6.238449] cx23885: cx23885_dev_setup() Failed to register dvb on VID_C
[    6.238453] cx23885: cx23885_dev_checkrevision() Hardware revision = 0xa5
[    6.238459] cx23885: cx23885[1]/0: found at 0000:05:00.0, rev: 4, irq: 18, latency: 0, mmio: 0xfb800000
```

b-rad-NDi, 2019-12-11:
The second pcie pair is failing to initialize. Haven't had something like reported to me. I don't have a DVB quadhd, but my atsc version is fine. Can you try booting the original ubuntu kernel and seeing if there is different results?

b-rad-NDi, 2019-12-11:
The related patches haven't been touched in a long time, and were the same in the last two releases, so this is a strange error.

skalnet, 2019-12-12:
Hi,
same Problem with original Kernel. Maybe i'm able to test the Card in a windows PC later, to rule out a hardware defect.

b-rad-NDi, 2019-12-12:
I will request the main office do a sanity test with a dvb quadhd to ensure nothing broke in the code somewhere.

Saentist, 2020-03-17:
I have same problem,
there are two quadhd working long time but after restart 

```
dmesg | grep 885_dvb
[    5.307809] cx23885: cx23885_dvb_register() allocating 1 frontend(s)
[    5.372194] cx23885: cx23885_dvb_register() allocating 1 frontend(s)
[    5.723689] cx23885: cx23885_dvb_register() allocating 1 frontend(s)
[    5.730851] cx23885: cx23885_dvb_register() allocating 1 frontend(s)
[    6.080572] cx23885: cx23885_dvb_register() allocating 1 frontend(s)
[    6.087861] cx23885: cx23885_dvb_register() allocating 1 frontend(s)
[    6.436621] cx23885: cx23885_dvb_register() allocating 1 frontend(s)
[    6.443662] cx23885: cx23885_dvb_register() dvb_register failed err = -22
[    6.445577] cx23885: cx23885_dvb_register() allocating 1 frontend(s)
[    6.452669] cx23885: cx23885_dvb_register() dvb_register failed err = -22
```

is there a list for errors in this case -22

_if there is need for some other logs just ask._

b-rad-NDi, 2020-06-06:
@Saentist : is this still an active issue? Are you running my kernel? 

When something like this happens on a stock kernel it's usually because of a hardcoded limit of 8 dvb devices allowed system wide, which I override in my builds to 24 or 32.

Saentist, 2020-06-07:
Can be closed

closed 2020-06-07 by b-rad-NDi