b-rad-NDi/Ubuntu-media-tree-kernel-builder #90: HVR 2250 (model 88061 C4F2)  different signal strengths

emveepee, 2019-12-26:
Running Debian 10 Buster and my HVR 2250 is detected fine and ATSC on adapter 0 is fine, and analog on both tuners are fine.

On the adapter 1 tuner  the signal strength is greatly reduced and doesn't go above 55% in dvbv5 scan so most channels can't be watched.  Forcing a zap with the adapter 0 conf file doesn't work.  Oddly the stations that are picked up are not the strongest signals I get.

In Windows both tuners work so It doesn't look like hardware related.  Is there something I need to add in the conf file card=7 is the only option I can find

b-rad-NDi, 2019-12-30:
I know nothing about this model, but I'll look into it as I have time.

techiegirl2day, 2020-02-06:
Hi, I just saw your post emveepee. I have 2255 cards running on MxLinux 19 with default 4.19.06 kernel ( it is a buster based distro )  I have found my signal strengths are finicky in linux.  What has helped me is playing around with antenna position & using an amp to boost the signal.  I have found the the reception can vary between the two tuners on the same card.  The only thing that has helped me is playing around with the antenna.  In your case it might mean relocating the antenna or maybe getting a different model  if adding an amp or checking all the lines & connections from the antenna to the tuner don't yield any improvements. Maybe running a different distro will be enough of a change to settings to make a positive improvement?  Best of Luck!

techiegirl2day, 2020-02-13:
I am experiencing similar issues with Hauppauge tuners. I have another brand of usb tuner hooked up to same antenna & am getting 80 - 90 % signal strength. With the Hauppauge linux drivers I am only getting between 50 - 60 % signal. Due to this issue my application drops tuners as lost or undetected. I have a range of devices.
Quad LG chip
2255  LG chip
2255 Samsung chip

I have included my dmesg with hardware specs if that helps to troubleshoot this
[Uploading dmesg Feb13,2020ed.txt…]()

b-rad-NDi, 2020-06-06:
@techiegirl2day : you didn't actually upload any logs, that's a link to this issue. If you have a problem still please file a new issue.

emveepee, 2020-06-06:
@b-rad-NDi  what log do you need for my original post?

b-rad-NDi, 2020-06-07:
i'm unsure of the drivers involved, since I haven't looked them up, but if you could enable kernel dynamicdebug for bridge/tuner/demod using option *+pfl* that would get me at least started. All the logs from boot through several zaps that work on don't work.

No reason why one tuner should perform much less than the other.

emveepee, 2020-06-11:
I tried with modules saa7164, tda18271 and tveeprom but I can't figure out out to to that sorry.  I do have a debugfs but I get permission denied when I try things like 

``sudo echo -n 'module saa7164 +pfl' > /sys/kernel/debug/dynamic_debug/control
``
so I tried with modprobe and only tveeprom gave any extra logging

``
[ 2851.145575] [2211] 503: tveeprom: Tag [04] + 8 bytes: 04 20 77 00 40 39 b4 6f f0
[ 2851.145580] [2211] 503: tveeprom: Tag [05] + 2 bytes: 05 2b 00
[ 2851.145582] [2211] 503: tveeprom: Tag [06] + 7 bytes: 06 fd 57 01 00 92 49 8d
[ 2851.145585] [2211] 503: tveeprom: Tag [07] + 1 bytes: 07 70
[ 2851.145587] [2211] 503: tveeprom: Tag [09] + 2 bytes: 09 28 5f
[ 2851.145589] [2211] 503: tveeprom: Tag [0a] + 2 bytes: 0a 88 98
[ 2851.145591] [2211] 503: tveeprom: Tag [0b] + 1 bytes: 0b 2a
[ 2851.145593] [2211] 503: tveeprom: Tag [0e] + 1 bytes: 0e 01
[ 2851.145595] [2211] 503: tveeprom: Tag [0f] + 1 bytes: 0f 1d
[ 2851.145597] [2211] 503: tveeprom: Tag [10] + 1 bytes: 10 01
[ 2851.145599] [2211] 644: tveeprom: Not sure what to do with tag [10]
[ 2851.145601] [2211] 503: tveeprom: Tag [11] + 1 bytes: 11 ff
[ 2851.145603] [2211] 644: tveeprom: Not sure what to do with tag [11]
[ 2851.145605] [2211] 503: tveeprom: Tag [13] + 2 bytes: 13 74 6d
[ 2851.145606] [2211] 644: tveeprom: Not sure what to do with tag [13]
``

I wanted to confirm my original problem since it has been months and here are both tuners with the strongest two channels on the same PC via RDP (it is headless) and scanning doesn't find either channel on tuner 2 on Debian 10 (same scan file)

[image]

b-rad-NDi, 2020-06-11:
you can't use echo via sudo for whatever reason, you have to sudo su first.

emveepee, 2020-06-11:
I did also try su but I didn't see any change to the existing control file and I could find any information on a dvbv5-scan in any logs

emveepee, 2020-06-14:
I believe I may have solved the issue if I force the analog device to the tuner

 v4l2-ctl -i 0 -d /dev/v4l/by-path/pci-0000\:01\:00.0-video-index1

I get the same signals on both tuners.