b-rad-NDi/Ubuntu-media-tree-kernel-builder #91: Signal breakup when using multiple Hauppauge 461e Tuners

grumpydolphin, 2020-01-08:
I have two Hauppauge 461e Tuners.  When I use any single tuner I have no issues, if I use both tuners at once I see errors on both tuners.

The hardware revision on my tuners is B6H9

b-rad-NDi, 2020-01-08:
Do you have logs for me?

What type of system?

How are the tuners connected? Are they on a single usb bus?

I need information if I'm expected to even consider an issue. I don't have access to signal, nor do I have a B6H9, so the odus is on you to supply as much information as you can generate.

With that said, I've had people tell me they use four 461e on one pc.

Saentist, 2020-01-08:
Raspberry PI 4 is device OS Ubuntu, (as it report in some external forum)
RPi4 have single USB bus.

b-rad-NDi, 2020-01-08:
If both tuners work together on a pc, but not on an rpi, then the issue has to do with rpi limitations. If this is caused by an rpi driver issue, then it's out of my hands.

If it can be shown that it is *not* an rpi limitation, then I will try to sort it out.

Nothing can be done, whatsoever though, until I get full logs.

grumpydolphin, 2020-01-09:
Here are some logs, let me know if you need anything else.

The messages in the syslog were generated when the second tuner was activated.

[kern.log](https://github.com/b-rad-NDi/Ubuntu-media-tree-kernel-builder/files/4042728/kern.log)
[syslog.log](https://github.com/b-rad-NDi/Ubuntu-media-tree-kernel-builder/files/4042729/syslog.log)
[dmesg.log](https://github.com/b-rad-NDi/Ubuntu-media-tree-kernel-builder/files/4042730/dmesg.log)

b-rad-NDi, 2020-06-05:
@grumpydolphin : would you be interested in converting your device to bulk mode to see if this fixes your issue?

b-rad-NDi, 2020-06-06:
Converting to bulk mode has fixed a number of problematic satellites for one user.

This is a tentative confirmed fix.

jksj461, 2020-07-22:
High i have a Hauppauge 461e rev B8H9 and a Dual HD both of which work well and are reliable on there own. However when used simultaneously I get bad recordings. They seem to use the same driver so I thought it was a multithreading issue. However reading this it may be a USB issue. How do I change the mode to bulk mode?

b-rad-NDi, 2020-07-22:
Follow the instructions for my tool eeprom-tinker. You will find the source code if you just browse my repositories

jksj461, 2020-07-23:
Thanks found that, looks fun. 
Trying to get the 461e rev B8H9 working on P1 4 which is now running kernel 5.4.51.
Fetched the latest media tree from //linuxtv.org/media_build.git
Should this repository include your committ  [https://github.com/b-rad-NDi/media_tree/commit/abf86d2e894c0adaa4c934edbda585c0ee1fdf7e#diff-447a37e70749e46662f9ab41ab9bed1e](url)
dated b-rad-NDi committed on 13 Jun 2019 
It looks to me that some but not all of your changes are there but it may have been updated since.
The build failed anyway but am I wasting my time, will this build method not work on a Pi.

  Building modules, stage 2.
  MODPOST 692 modules
ERROR: "__devm_regmap_init_sccb" [/home/pi/media_build/v4l/ov9650.ko] undefined!

b-rad-NDi, 2020-07-23:
media_build definitely works on cross compilation. I use it extensively on more than a dozen different architectures. I do not use the build script, I do it manually.
```
make release DIR=/path/to/kernel/and/kernel/config
make -C v4l/
```

b-rad-NDi, 2020-07-23:
If there's modules that still don't compile for some odd reason, then look for the config option in v4l/ and disable it manually. I doubt you need that image sensors driver.

jksj461, 2020-07-24:
Thanks, dumping the ov9650 driver allowed it to build but there are lots of errors at run time, presumably some specific  configuration is needed for rasbian. It looks like the Rasbian video driver is not happy. 
I am guessing that I would just have to install the dvb drivers and not the other stuff or would I be better off trying ubuntu on the Pi?
How do I install just the DVB drivers?

The /devdvb adaptor is created for the 461E but there are errors loading the module and the firmware does not load
Errors

[syslog.txt](https://github.com/b-rad-NDi/Ubuntu-media-tree-kernel-builder/files/4973490/syslog.txt)

b-rad-NDi, 2020-07-24:
Building modules requires your exact kernel config and exact headers in order to build compatible symbols. If you built your own kernel, then installed it and verified it worked, *then* pointed media_build at that directory and built then everything would probably be happy symbol wise. Your exact kernel config can usually be found in /proc/config.gz if you want to try and just build against the headers again.

I can't say I've ever seen a message 'firmware cannot run'.

In addition, if you go the full works and just compile your own kernel you could probably just place the files em28xx and m88ds3103 drivers into the 5.4 tree at the correct spot and compile it all together. There's probably only very minor, if any, fix up required.

jksj461, 2020-07-25:
Thanks copying the firmware from your ububnu 20.04 ppa made it load.
m88ds3103 12-006a: firmware version: 3.B
It tries to tune but fails to lock will test more over the weekend.
Many thanks for your help.

jksj461, 2020-07-27:
The card works fine - load tested it with a Dual HD running a total of three recordings and watching one (remotely) on a Pi4B 4G memory.
Applying the media tree to 5.4 on the Pi takes out Kodi for some reason but I am ok with running the pi headless so that's fine.
No doubt that will get fixed in a future kernel update.

psyborg55, 2022-10-29:
It's a manufacturer driver problem. I see this in windows, when one device has strong signal (90%) and another weak (48%) both signals are averaged and both devices show 69% when simultaneously in use. It is wrong.
On linux, there is no issue while using modified drivers.