b-rad-NDi/Ubuntu-media-tree-kernel-builder #92: Build Error - can't apply patch

thomisus, 2020-01-10:
Hi, 

this is the error i get trying to apply the patches with ./mediatree-kbuild.sh -p

```
Applied patch include/uapi/linux/media.h cleanly.
Applied patch include/uapi/linux/usb/video.h cleanly.
Applied patch include/uapi/linux/uvcvideo.h cleanly.
Applied patch include/uapi/linux/v4l2-common.h cleanly.
Applied patch include/uapi/linux/v4l2-controls.h cleanly.
Applied patch include/uapi/linux/v4l2-mediabus.h cleanly.
Applied patch include/uapi/linux/v4l2-subdev.h cleanly.
Applied patch include/uapi/linux/videodev2.h cleanly.
Applied patch sound/pci/bt87x.c cleanly.
Patch failed at 0001 Linuxtv.org media tree sync - 2019-10-11-12-13
Use 'git am --show-current-patch' to see the failed patch
When you have resolved this problem, run "git am --continue".
If you prefer to skip this patch, run "git am --skip" instead.
To restore the original branch and stop patching, run "git am --abort".
Patch failure: /opt/Ubuntu-media-tree-kernel-builder/patches/ubuntu-bionic-4.15.0/0001-Linuxtv.org-media-tree-sync-2019-10-11-12-13.patch
    Aborting git am
patch failure, exiting 

```

Could you help me?

b-rad-NDi, 2020-01-10:
You're trying to build against an Ubuntu kernel revision the failing patch was not generated against. Never fear though, I'm updating all builds now. So the patch will apply clean in an hour or so.

Solifuge80, 2020-01-11:
Any news @b-rad-NDi ?

I'm in the same situation as thomisus and really excited of getting things functional :)

b-rad-NDi, 2020-01-13:
Sorry, time ran out on friday. Everything is up to date now.

closed 2020-01-13 by b-rad-NDi