b-rad-NDi/Ubuntu-media-tree-kernel-builder #94: m88ds3103 fails to compile at recent media_tree

CvH, 2020-02-08:
Building recent media_tree (with media_build) with your patches from `patches/mainline-extra/tip/99.montage.3103b` applied resulting in an build error.
The "old" one (not maintained anymore??) from https://git.linuxtv.org/brad/media_tree.git/log/?h=Montage-3103b.v2 is still working.

```
/builddir/v4l/m88ds3103.c: In function 'm88ds3103_probe':
/builddir/v4l/m88ds3103.c:1906:20: error: implicit declaration of function 'i2c_new_dummy' [-Werror=implicit-function-declaration]
 1906 |   dev->dt_client = i2c_new_dummy(client->adapter, dev->dt_addr);
      |                    ^~~~~~~~~~~~~
/builddir/v4l/m88ds3103.c:1906:18: warning: assignment to 'struct i2c_client *' from 'int' makes pointer from integer without a cast [-Wint-conversion]
 1906 |   dev->dt_client = i2c_new_dummy(client->adapter, dev->dt_addr);
      |                  ^
/builddir/v4l/m88ds3103.c: In function 'm88ds3103_set_frontend':
/builddir/v4l/m88ds3103.c:908:6: warning: this statement may fall through [-Wimplicit-fallthrough=]
  908 |   if (dev->chiptype == M88DS3103_CHIPTYPE_3103B) {
      |      ^
/builddir/v4l/m88ds3103.c:913:2: note: here
  913 |  default:
      |  ^~~~~~~
cc1: some warnings being treated as errors
make[3]: *** [scripts/Makefile.build:266: /builddir/v4l/m88ds3103.o] Error 1
```

b-rad-NDi, 2020-02-10:
I haven't pushed up a v4 branch yet, but if you peep the media tree patchwork you'll find my latest series with this i2c_new_dummy -> i2c_new_dummy_device change in place.

The switch fall through doesn't do any harm, but i'll fix it.

CvH, 2020-02-10:
guess you mean this https://patchwork.kernel.org/project/linux-media/list/?series=235783 ?

so many places to follow :D

b-rad-NDi, 2020-02-26:
Indeed. I'll get everything sync'd back up when I have time.

closed 2020-02-26 by b-rad-NDi