b-rad-NDi/Ubuntu-media-tree-kernel-builder #95: pctv 461e tuner/demodulator issue

psyborg55, 2020-03-04:
trying to tune in certain transponders (mainly on hotbird but noticed on other satellites too) gives broken audio/video stream output.

tried all possible combinations of drivers/decoders both on windows and linux - the result is same
it happens even with official hauppauge software TVCenter 6

some suggest it may be caused when running the card in USB3.0 port, so i tried it on old laptop that has USB2.0 only ports. didn't help.

same cable/lnb/dish setup works fine on philips tv when trying to watch these same channels

there have been similar reports already:
https://forum.kodi.tv/showthread.php?tid=247076
https://forum.kodi.tv/showthread.php?tid=220129

this is one of the transponders that does not work properly with this device: http://www.lyngsat.com/muxes/Hotbird-13C_Wide_10758-V.html
continuity errors count increases as the streaming is tried to be made from this transponder.

here are two screenshots:

[image]
[image]

b-rad-NDi, 2020-03-12:
Unfortunately we have neither data sheet nor reference driver to assist in issues like this.

psyborg55, 2020-03-13:
how about this driver https://github.com/FilipDutina/PPuTViOS1/tree/master/18.12.2017/tdp_api/M88TS2022_adapt

b-rad-NDi, 2020-05-29:
That's probably the same oem code the driver was based off and others have reported it works even worse than what I've already done.

b-rad-NDi, 2020-05-29:
We are diagnosing this issue, would you be interested in trying out test kernels with modifications?

We suspect this is a usb bandwidth issue. DVB-T2 maximum bandwidth is ~53mbps whereas DVB-S2 maximum bandwidth is ~140mbps.

psyborg55, 2020-05-31:
sure, no problem. i think the problem is with RF parts of the device because it happens even on low-bandwidth transponders. the screenshots above are on channels that are clearly affected by this bug, but i've noticed at least one channel where the issue is on the edge, e.g. sometimes it gives few distortions (no green picture, only few pixelated blocks randomly on screen) and other times it is displayed fine. it is fashion tv on 16E (not sure of the transponder there are two)

b-rad-NDi, 2020-06-01:
Ok, if signals with low overall bandwidth are affected than it is not necessarily the same issue.

If you can find additional satellites which display issue it would be helpful. I will pass on "16E / fashion tv" and "13E / 4Fun.tv".

The problem is we have no source code, nor data sheet, nor kernel engineer in an area with dvb-s/s2.

psyborg55, 2020-06-03:
well the above linked repo seems to be source code, and there are datasheets available if you google for them.. fashion tv on 16E is at 11595 V, HD channel version

b-rad-NDi, 2020-06-04:
You've found a 3103b data sheet? Then send it to me. 3103b is not 3103.

And "source code" is irrelevant if it contains the exact same issues that were in the original "source code" i used as a reference.

psyborg55, 2020-06-04:
no. M88DS3103 and M88TS2022 datasheets. my device revision is B6H9

b-rad-NDi, 2020-06-05:
There's no compilation of any firmware. It's extracted from byte arrays inside of reference code, just like the source code you linked contains...

psyborg55, 2020-06-05:
i see.. did anyone ever tried contacting hauppauge regarding hw/sw problems? do they just tell people that product is outdated and to get newer supported model or what?

b-rad-NDi, 2020-06-05:
We are actively working on this with our suppliers. The thing is only a small subset of satellites do not work. Other satellites work 100% without issue. I am a hauppauge engineer, but I have no satellite access, which makes this harder to debug--especially considering there's no driver errors. The device is still actively sold and has had no issues until the beginning of this year, why issues just started I cannot say.

I'm working to get this issue prioritized.

b-rad-NDi, 2020-06-05:
@psyborg55 : Can you make TS captures of several "bad streams" for me? Preferably if you can get a couple from different satellites. I'm trying to get as much information as possible to relay this.

b-rad-NDi, 2020-06-05:
I also cannot find a ts2022 datasheet. If you have one could you send it to me?

b-rad-NDi, 2020-06-05:
Also @psyborg55 would you be interested in converting your 461e from isoc to bulk mode to see if this affects anything? There is an "official" hauppauge utility, and I have a work in progress Linux utility.

psyborg55, 2020-06-05:
>The device is still actively sold and has had no issues until the beginning of this year, why issues just started I cannot say.

i don't think this holds true. if you look at first comment where i linked threads on kodi forums they reported these same issues back in 2015.

i'll post streams once captured. when you have isoc to bulk util ready i can try it.

datasheet attached
[M88TS2022-Montage.pdf](https://github.com/b-rad-NDi/Ubuntu-media-tree-kernel-builder/files/4738601/M88TS2022-Montage.pdf)

b-rad-NDi, 2020-06-05:
The 461e is still sold, there is just a new revision, that apparently has similar/same issues. That's what I meant.

If you have windows available you can contact hauppauge support for the conversion utility. The Linux utility is hosted here as a git repo if you browse my profile and go to eeprom-tinker. There is just some assorted cwd/path related issues I need to fix.

Thanks for the datasheet, I wish I could find a 3103b demod data sheet now.

b-rad-NDi, 2020-06-05:
Also, those don't seem to be the same issues, although they might sound so. There is a lot of things that can trigger continuity/ts errors. That particular error linked to appears to be either usb subsystem or dvb-core subsystem related.

psyborg55, 2020-06-06:
ok, looks like your theory about USB throughput might be correct.
capured several streams, in 30 sec duration, all that were above 200MB have problems, the highest bandwidth one that still worked fine had 188MB which is about 50Mbps.
so streams above 50Mbps on transponder break

also, one thing the seems as pretty strong confirmation of the issue is fashion tv on 16E is no longer possible to watch it breaks completely just as other channels in the screenshots and what has changed in the meantime is a 4K version of the same channel was added to that transponder increasing overall bandwidth which the USB can no longer handle

here are the captures:

Eutelsat at 10E:
https://www56.zippyshare.com/v/wpw90Y6m/file.html

Hotbird at 13E:
https://www56.zippyshare.com/v/tDIYZrGM/file.html
https://www56.zippyshare.com/v/o2nccsrd/file.html
https://www56.zippyshare.com/v/0KxQipEA/file.html

Eutelsat at 16E:
https://www56.zippyshare.com/v/4lDyQw9R/file.html

b-rad-NDi, 2020-06-06:
Excellent find :-D and thank you for the captures.

I'm going to see about quickly fixing the issues with eeprom-tinker and convert the 461e I have to bulk mode. I'll ping you when it's ready to try.

This is the "official" tool, but I'm not sure if it allows conversion of 461e as the target devices were SoloHD/DualHD.
https://www.hauppauge.de/site/support/support_faq.php?n=FAQ.BulkOrIso

b-rad-NDi, 2020-06-06:
@psyborg55 The utility is read to use, I have confirmed that it successfully converted my 461e to bulk mode. It currently does not change the usb pid.

Full instructions are included in a README.md, but open it in a text editor github formatting makes it fubar.

psyborg55, 2020-06-06:
converted to bulk mode here too. everything works perfectly now!

[image]
[image]
[image]

b-rad-NDi, 2020-06-06:
SMFH :-D

I seriously wish I would have noticed this when all problems started cropping up as all the transponders began changing. *All* hauppauge devices are supposed to come from the factory in bulk mode now, but for some reason this device slipped through the cracks.

b-rad-NDi, 2020-06-06:
Thanks for being a guinea pig @psyborg55 I appreciate it and I think everyone else will as well.

Cheers.

closed 2020-06-06 by b-rad-NDi

b-rad-NDi, 2020-06-06:
Also @psyborg55 were my instructions clear enough for you to use eeprom-tinker or do you think I should expand them?

psyborg55, 2020-06-06:
absolutely simple and clear to do the conversion. i even did it on 16.04 with manually compiled kernel 5.6 used to test some other thing and manually compiled i2ctools 4.1 (only needed to soft link /usr/local/lib/libi2c-dev* to /usr/lib for app to work)

just to add one more detail here: sometime during beginning of this year i was after some feeds, one of them was posted online as 32APSK feed, the card was able to tune it and display content without issues. so it appears that 16APSK and 32APSK modes are also working fine with this card