b-rad-NDi/Ubuntu-media-tree-kernel-builder #97: linux-tools

WyseNynja, 2020-04-13:
I'm trying to develop some software on my server, but I'm getting this error:
```
WARNING: perf not found for kernel 4.15.0-72201912051620

  You may need to install the following packages for this specific kernel:
    linux-tools-4.15.0-72201912051620-generic
    linux-cloud-tools-4.15.0-72201912051620-generic

  You may also want to install one of the following packages to keep up to date:
    linux-tools-generic
    linux-cloud-tools-generic
```

Installing linux-tools-generic does not install the version I need (the version from mediatree).

Would it be possible to include packages for this? Thank you!

b-rad-NDi, 2020-05-13:
I'll look into it. I didn't like having those part of the ppa because they would get installed and then not auto remove themselves for some users.

The builds are totally reproducable, but at that point you'd have to build everything i think to compile the various tools.

closed 2020-05-29 by b-rad-NDi

b-rad-NDi, 2020-10-07:
I've added these kernel specific tools. I do not include the "common" package but that appears to be irrelevant.

800443959