b-rad-NDi/Ubuntu-media-tree-kernel-builder #98: Requesting Ubuntu 20.04 Focal Fossa Builds

spbuford, 2020-04-24:


awonglk, 2020-04-25:
Now that final Ubuntu 20.04 release is out, would appreciate a build is made available for it. Thanks!

okae, 2020-04-25:
Ubuntu 20.04 release is out!, we are waiting for a working build, there is lot of people waiting for it, thank you.

cjgreen, 2020-04-27:
Please let us know when the build is available. Waiting patiently.

LewisMagri, 2020-04-28:
Not meaning to be an eager beaver, but is there a totally rough ETA on this. If it's more than a month i may configure my server to use Bionic or Eoan (but i'd def rather hang around as Focal is LTS). Thanks in advance Brad!!!! ❤️

cytech, 2020-04-30:
same here. thank you

b-rad-NDi, 2020-05-13:
What devices do you all have? The only thing missing from kernel 5.4 (i think) is usb quadhd. Everything else is in mainline now. I intend on prioritizing this, but there are an insane amount of other things that have my attention due to what is explained below.

Note: hauppauge was greatly affected by covid and therefore i was not working for a long time. That has changed now.

LewisMagri, 2020-05-14:
Hey Brad, thanks for prioritising this. I have this one: https://www.amazon.co.uk/dp/B011Z7I0JY/ref=cm_sw_r_wa_apa_i_PTiVEbEH8HQKA

Swanseatom, 2020-05-14:
Also have this same device.

awonglk, 2020-05-14:
Hauppauge WinTV-dual HD

b-rad-NDi, 2020-05-14:
For both of your devices you only require linux-firmware-hauppauge from my ppa.

Your main benefit of using the ppa with those devices is you also get 100% of all future bug fixes and enhancements that wouldn't possibly be backported to older kernels.

nwskier, 2020-05-14:
Hey Brad, I have a WinTV Quad USB. Heard that you were gone due to Covid and been eagerly awaiting your return.Welcome back!

awonglk, 2020-05-14:
@b-read - Quite happy to use the linux-firmware-hauppauge from your PPA.
But I believe a release for Ubuntu 20.04 (focal fossa) is not available yet?

> $ sudo apt-get install linux-mediatree
> Reading package lists... Done
> Building dependency tree
> Reading state information... Done
> E: Unable to locate package linux-mediatree

b-rad-NDi, 2020-05-14:
@awonglk that's only the kernel that's missing, it will be worked on soon. Just the firmware package gets all but a couple devices up and running.

jksj461, 2020-05-14:
I have a pctv DVBS2 461e rev B8H9 which works fine with your recent fixes on ubuntu 19.10. I briefly plugged it into an ubuntu 20.04 system to check if the driver was included but the device was not recognised. I was waiting for your PPA before doing the full upgrade. I will give it another go if you think that driver is included.

cytech, 2020-05-14:
i have HAUPPAUGE WinTV-quadHD PCI Express TV Tuner Card

LewisMagri, 2020-05-14:
Hi Brad! Thanks for the heads up on the linux-firmware-hauppauge. This seems to get my Hauppauge WinTV-dual HD working, but I can't seem to get HD TV channels. Is this expected? Or do i just need to wait?

spbuford, 2020-05-14:
> What devices do you all have? The only thing missing from kernel 5.4 (i think) is usb quadhd. Everything else is in mainline now. I intend on prioritizing this, but there are an insane amount of [...]

I only needed the pci drivers which I picked up this week and seem to be working well.  Thanks!

b-rad-NDi, 2020-05-14:
@jksj461 : your device is in mainline now, but not in 5.4

@LewisMagri : You most likely need the DVB-T/T2 auto patch.

If I can find the time today I will be working on this, but I'll definitely get it done by tomorrow.

LewisMagri, 2020-05-14:
@b-rad-NDi Excuse my ignorance, but I'm guessing the "DVB-T/T2 auto patch" will be available in the linux-mediatree Focal PPA when you release right?

b-rad-NDi, 2020-05-14:
@LewisMagri : indeed it will be. It was rejected by kernel maintainers unfortunately, so I need to do some core extensions to get it mainlined.

LewisMagri, 2020-05-14:
Awesome, thanks @b-rad-NDi. Hope they don't reject it again 🤞

shspvr, 2020-05-23:
Brad any luck addressing the USB quad-HD ?.

b-rad-NDi, 2020-05-23:
Sorry it's taken a while, there is an insane backlog.

Here is my testing ppa:

https://launchpad.net/~b-rad/+archive/ubuntu/test-test-hw-staging

Everything is up there

b-rad-NDi, 2020-05-23:
@shspvr : still awaiting to hear from downstream. The current builds pushed do still have issue, but we're actively trying to sort things out.

shspvr, 2020-05-24:
> @shspvr : still awaiting to hear from downstream. The current builds pushed do still have issue, but we're actively trying to sort things out.

That good to hear I be look forward to up date when it ready

jksj461, 2020-05-24:
Thanks the PCTV DVB-S2 Stick (461e v2) is working fine on Ubuntu 20.04 using the testing ppa

salman79uk, 2020-05-25:
Hi Brad, trying to install this on an armhf device get the following dependency errors:

The following packages have unmet dependencies:
 linux-mediatree : Depends: linux-image-5.4.0-21-generic but it is not installable
                   Depends: linux-modules-5.4.0-21-generic but it is not installable
                   Depends: linux-modules-extra-5.4.0-21-generic but it is not installable
                   Depends: linux-headers-5.4.0-21-generic but it is not installable

This is running focal, how do I fix this?

b-rad-NDi, 2020-05-29:
@salman79uk : You're out of luck. Ubuntu ppa's are for x86/amd64 architectures only. You don't replace the kernel the same way on arm ubuntu. You need to build everything manually yourself with this:

https://git.linuxtv.org//media_build.git

If you need usb quad hd support then you'll have to stick my files in the build system yourself.

b-rad-NDi, 2020-05-29:
Another build has been pushed to the PPA, this one will very likely be migrated to the main PPA later this afternoon.

shspvr, 2020-05-29:
> Focal is live!

what command need for pop os with quad-hd

shspvr, 2020-05-29:
> Focal is live!

Thanks all well