b-rad-NDi/Ubuntu-media-tree-kernel-builder #100: How to uninstall?

thisisbenwoo, 2020-06-04:
Hi, I've followed the instructions and performed:
```
sudo add-apt-repository ppa:b-rad/kernel+mediatree+hauppauge
sudo apt-get update
sudo apt-get install linux-mediatree
sudo apt-get install linux-firmware-hauppauge

```

My problem is that there is some conflict with my NVIDIA GeForce GT730 card and Docker. I'm running Ubuntu 20.04. So my question is two-fold:

1/ Can I fully back out of the linux-mediatree kernel branch; and
2/ Is there something that I'm missing to make Docker + the WinTV-dualHD TV Tuner + NVIDIA GT730 on Ubuntu 20.04 play nice? 

Thanks in advance

shspvr, 2020-06-04:
You should not have needed to install anything with WinTV-dualHD it should work out the box
sudo add-apt-repository --remove ppa:b-rad/kernel+mediatree+hauppauge

There many option best read this
https://www.howtogeek.com/229699/how-to-uninstall-software-using-the-command-line-in-linux/
I would just start over it can easy go south when come unless it a reg app
What are you using docker for as in what app ?.

thisisbenwoo, 2020-06-04:
Hmm... well, that's not what the Hauppauge page suggests (https://hauppauge.com/pages/support/support_linux.html), or at least the way I read it. That's why I am confused. According to the page, it reads:

> Hauppauge's PPA has support for the following models of Hauppauge TV tuners
> ...
> WinTV-dualHD (all models, both tuners)

which is the version I have. 

WRT to what am I using docker for? I'm trying to run Plex, and use the WinTV for Live TV. I bought the GPU so that I can trasncode faster. I also run my home automation in docker containers.

Thanks for the link

thisisbenwoo, 2020-06-04:
... update ...
Just reversed the PPA kernel, and yes, by adding --device=/dev/dvb:/dev/dvb into my docker run/compose, everything worked out of the box.
Thanks again

closed 2020-06-04 by thisisbenwoo

shspvr, 2020-06-04:
Keep in mind that info I do believe is base on older ver of Ubuntu
If don't see one of the tuner status lights turn on after boot into Linux then you should only needed to run and should take care it.
sudo add-apt-repository ppa:b-rad/kernel+mediatree+hauppauge
sudo apt-get update
sudo apt-get install linux-firmware-hauppauge
reboot

With Ubuntu and it clone it should now just
sudo add-apt-repository ppa:b-rad/kernel+mediatree+hauppauge
sudo apt-get update
Ubuntu: sudo apt-get install linux-mediatree
Ubuntu LTS w/HWE: sudo apt-get install linux-hwe-mediatree

Mine work out of box under PoP! OS 20.04 and I didn't have do any thing and I have NextPVR being used as M3U Tuner that tied to Emby for LiveTV a long with MergerFS.
Plex should work natively but I have tested it under 20.04