b-rad-NDi/Ubuntu-media-tree-kernel-builder #102: No LTS/HWE Support for Ubuntu 20.04

okschlatter, 2020-06-27:
So I'm trying to get my WIN-HVR-1250 to work on my Ubuntu server (running Ubuntu 20.04 LTS/HWE). I ran `sudo add-apt-repository ppa:b-rad/kernel+mediatree+hauppauge` with no problem, but when I try `sudo apt-get install linux-hwe-mediatree`, I get this result:

`Reading package lists... Done`
`Building dependency tree`
`Reading state information... Done`
`E: Unable to locate package linux-hwe-mediatree`

Browsing https://launchpad.net/~b-rad/+archive/ubuntu/kernel+mediatree+hauppauge shows that there is no `linux-hwe-mediatree` release for Ubuntu 20.04 "Focal," which explains why I'm getting the error. 

So I guess this is really more of a request than an issue, but I would greatly appreciate Ubuntu 20.04 LTS/HWE compatibility.

b-rad-NDi, 2020-06-29:
I'm not tracking it until they they release 20.04.01, which would contain the actual hwe kernel. That'll happen sometime in August.

Right now standard focal kernel is 5.4 and hwe kernel is also 5.4...what's the point?

closed 2020-06-29 by b-rad-NDi

juniorRubyist, 2020-07-09:
20.04 does include the HWE kernel I believe. I tried installing the non-HWE drivers and it screwed up my system. Thankfully, I had ZFS snapshots to revert with, but still, this is concerning.

b-rad-NDi, 2020-07-15:
Well that's certainly strange. If you're running an HWE kernel right now what does this report:
```
uname -a
```
Because when I look at the 20.04 kernel repo and inspect the hwe branch it shows me 5.4.0

juniorRubyist, 2020-07-15:
Here's what `uname -a` outputs for me:
```
Linux josephgeis-lt 5.4.0-40-generic #44-Ubuntu SMP Tue Jun 23 00:01:04 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
```
Additionally, here's the output of `apt list --installed | grep hwe`, which shows that I have the HWE kernel installed.
```
linux-generic-hwe-20.04/focal-updates,focal-security,now 5.4.0.40.43 amd64 [installed]
linux-headers-generic-hwe-20.04/focal-updates,focal-security,now 5.4.0.40.43 amd64 [installed,automatic]
linux-image-generic-hwe-20.04/focal-updates,focal-security,now 5.4.0.40.43 amd64 [installed,automatic]
```
Interestingly, if I login through a `login` shell (like on a tty or over ssh), this is output in my `motd`:
```
Your Hardware Enablement Stack (HWE) is supported until April 2025.
```

b-rad-NDi, 2020-07-15:
and this is the non-hwe kernel (my build):
```
Linux ryzen 5.4.0-392007110711-generic #0+mediatree+hauppauge-Ubuntu SMP Sat Jul 11 19:13:56 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
```
They're the same kernel. HWE just gets a single version bump for the changelog.

shspvr, 2020-07-16:
That because you installed the experimental ZFS files,  Far I know of the HWE will not happen in tell 20.04.2 that what was last report.

thickhead, 2020-10-09:
Hi,
Ubuntu 20.04.1 LTS/HWE and following the directions to install PPA on Hauppauge website, I'm still getting same error as the OP and uname-a displays kernel 5.4.0-48-generic #52-ubuntu. Should it work now, or no?

b-rad-NDi, 2020-10-10:
There is essentially no different in "normal" and "HWE" kernels on focal 20.04 right now.

https://packages.ubuntu.com/focal-updates/linux-image-generic-hwe-20.04

Until you see the above package hit 5.8, there will be no linux-hwe-mediatree--just use linux-mediatree until then.