b-rad-NDi/Ubuntu-media-tree-kernel-builder #104: HD-PVR 2 / Colossus 2

shspvr, 2020-07-17:
I just wondering Bard do you have plan on converting Linux HD-PVR 2 / Colossus 2 drivers out of user mode application in reg driver package ?.

b-rad-NDi, 2020-07-17:
From what I understand there is one piece that cannot be open sourced. I've asked if I can work on all the other parts, but it would be fairly crippled as the encoder would be completely absent, so it hasn't been thought worth while.

b-rad-NDi, 2020-07-17:
I will inquire again though.

shspvr, 2020-07-17:
That very odd it as look like be all open sourced to me other then bin firmware file

b-rad-NDi, 2020-07-17:
The hw encoder had no open source driver when I last looked. The other two pieces do.

The user mode driver thats out there is definitely not open source, unless it entirely ignores the hw encoder.

b-rad-NDi, 2020-07-17:
Situation could have always changed over the years though I guess.

b-rad-NDi, 2020-07-17:
So, there is a lot of different source packages. Some of the parts are indeed open source. The magnum encoder parts look like reference code though, and they're heavily obfuscated in typical reference code format. The adv7842 does have a kernel driver, but the magnum encoder driver would have to be developed from scratch.

b-rad-NDi, 2020-07-17:
If I ever run out of stuff to do I'll see how much would be involved in the magnum porting, but it doesn't look trivial just inspecting the code.

shspvr, 2020-07-17:
Ok sound good when you get time