b-rad-NDi/Ubuntu-media-tree-kernel-builder #105: PCIe Hauppauge QuadHD Tuner -- Linux kernel hard reset

howardthomson, 2020-07-18:
Hi Brad,
I have emailed support@hauppauge.co.uk as follows:
============================
"I have a recently acquired Xeon 2U server box, to which I have just added a Hauppauge PCIe QuadHD DVB Tuner card.

Type: CX23885_BOARD_HAUPPAUGE_QUADHD_DVB in linux-stable/drivers/media/pci/cx23885/cx23885-cards.c [I think!].

w_scan recognized the card but detected no signals, at which point I twigged that firmware was needed.

Installing firmware from the OpenELEC download resulted in w_scan scanning up to the first frequency with a signal, and the
machine then crashing -- a hard reset.

Looking at the Hauppauge support page for Ubuntu [I run Linux Mint, currently Mate version 20], I re-installed LinuxMint
from scratch, and followed the instructions [ https://hauppauge.com/pages/support/support_linux.html ].

The machine now does not reach the graphical login, showing a static cursor in the centre of the screen, and crashing
to a hard reset ...

Do you have any suggestions ?

Should a try a HWE kernel ?

Should I retry an Ubuntu Desktop installation ?

I note that in the kernel file cx23885-core.c:

/*
 * Some platforms have been found to require periodic resetting of the DMA
 * engine. Ryzen and XEON platforms are known to be affected. The symptom
 * encountered is "mpeg risc op code error". Only Ryzen platforms employ
 * this workaround if the option equals 1. The workaround can be explicitly
 * disabled for all platforms by setting to 0, the workaround can be forced
 * on for any platform by setting to 2.
 */

I haven't found any evidence of the reported 'symptom encountered' so this may not be relevant.

Is there anything I can do to narrow down the cause of the problem ?

I have read that there are other satisfied Linux users of this card out there !

For trial purposes, I have a [much] older, but severely underpowered, box with a PCIe slot that I could test software with,
but I need to get it working in the new Xeon powered box to make any sensible use of it."
=======================
[dmesg.txt](https://github.com/b-rad-NDi/Ubuntu-media-tree-kernel-builder/files/4942831/dmesg.txt)
[lspci.txt](https://github.com/b-rad-NDi/Ubuntu-media-tree-kernel-builder/files/4942832/lspci.txt)

Applying the Haupauge support page installation guidance resulted in an essentially unusable system.

I note in one of the earlier closed issues that reverting to an earlier [unspecified] kernel resolved that issue,
but I would prefer to get the problem solved ...

What can I do ?

Howard

howardthomson, 2020-07-18:
I have now had a frozen machine booting from a live CD [Linux Mint 20] although I can still ping it ...

shspvr, 2020-07-19:
Have try taking the card out of system to see it will boot in Xeon 2U server box ?.
Far I know of the HWE will not happen in tell 20.04.2 that what was last report.
dmesg is saying
[  240.084752] si2168 2-0064: Direct firmware load for dvb-demod-si2168-b40-01.fw failed with error -2
[  240.084792] si2168 2-0064: Direct firmware load for dvb-demod-si2168-02.fw failed with error -2
[  240.084795] si2168 2-0064: firmware file 'dvb-demod-si2168-02.fw' not found
[  240.089818] si2157 3-0060: found a 'Silicon Labs Si2157-A30'
[  240.139113] si2157 3-0060: firmware version: 3.0.5
[  240.144575] si2168 2-0064: Direct firmware load for dvb-demod-si2168-02.fw failed with error -2
[  240.144600] si2168 2-0064: Direct firmware load for dvb-demod-si2168-02.fw failed with error -2
[  240.144603] si2168 2-0064: firmware file 'dvb-demod-si2168-02.fw' not found

Saentist, 2020-07-19:
Install firmware ;)

shspvr, 2020-07-19:
> Install firmware ;)

But you used OpenELEC which hasn't ben update in a very long time did actually check the /lib/firmware/ folder for that file

Have you tried
sudo add-apt-repository ppa:b-rad/kernel+mediatree+hauppauge
sudo apt-get update
sudo apt-get install linux-firmware-hauppauge

If nothing else you want check the hardware under windows just see if has the same problem who knows maybe a dumb device

b-rad-NDi, 2020-07-20:
Well, a missing firmware should not cause such catastrophic errors. If it does than I'll have to put a safeguard into the driver to gracefully fail. I don't have a dvb quadhd to test that out if that's the case though.

Installing my firmware package will at least ensure you have the files you need. I've had so many people say "i downloaded a firmware and it doesn't work!?!?!?!?!" only for me to tell them they downloaded an html page and not the firmware itself.

howardthomson, 2020-07-20:
When I installed the firmware and the media-tree update previously, the Xeon box did a hard reset before getting to a graphical login. I have since discovered that my older dual-AMD-Opteron box, that I thought had only PCI slots, actually has two PCIe slots as well, so I will work on that next. Unfortunately, I am also having some aerial cable problems.
If I can get it to work in the older box, I will get back to the Xeon box later.

b-rad-NDi, 2020-07-21:
I'd like to see the logs generated from a hard reset, if any are generated, if you get another. They should be in /var/log/

shspvr, 2020-07-21:
howardthomson just have ask did you install the quad-HD a 16x slots?.

howardthomson, 2020-07-21:
I have moved the QuadHD card to an old (!!) AMD Athlon X2 4000+ computer.
New Linux Mint 20 installation, install w_scan, follow hauppauge.com Linux setup process.
w_scan works OK. wc -l new-channels.conf = 85, as against wc -l channels.conf = 86 generated by dvb_usb_af9015 earlier
PCI dual tuner card.
Still work in progress ...

shspvr, 2020-07-22:
> I have moved the QuadHD card to an old (!!) AMD Athlon X2 4000+ computer.
> New Linux Mint 20 installation, install w_scan, follow hauppauge.com Linux setup process.
> w_scan works OK. wc -l [...]

I was ref to Xeon 2U server box

howardthomson, 2020-07-22:
This afternoon, I moved the card and the SATA disk, to the Xeon machine. On the AMD Athlon machine the card and disk resulted in a complete startup, and operational card, tuning OK. On the Xeon machine the result was, again, a hard reset before any useful interaction with the machine.
I have now booted the Xeon with a lived-DVD and looked at /var/log on that disk, with no useful file content from that failed startup, only the successful shutdown before moving the disk.
The Xeon machine has qty 2 x16 PCIe slots, and qty 2 x8 slots. The card was plugged into a x16 slot.
Next, I will move the card to my Dual Opteron box, Tyan S2927 Motherboard, which has a x8 PCIe slot available.
That looks like being the end resting location for my QuadHD PCIe Hauppauge Tuner card ...
Still work-in-progress, Howard

shspvr, 2020-07-22:
Look in the bios try changing that 16x slot to Gen 1 mode see that with compatibility I know on my Intel ASRock Z77 motherboard it dose like 16x slot at all but I never tested the diff gen mode as it now in my main media server

howardthomson, 2020-07-24:
I looked for settings for 'Gen 1' mode; found lots of settings but nothing that I understood! After my Dual-Opteron box froze, and the card no longer appearing in an lspci list, I have returned the card for a replacement, and will close this issue for the time being.
When my replacement card is delivered in late August (!), I will have another go ...

howardthomson, 2020-07-31:
A replacement card has been delivered! Initially installed in my ancient AMD Athlon X2 4000+ machine: tunes OK, works for DVB radio, but video fails -- CPU too slow. 

There is no 'Gen 1' mode for my Asus Xeon Z11PA-U12 motherboard. There are Gen 2 Settings ...
This is the system in which the previous card caused a hard reset during startup  ...

Does 'Above 4G Decoding' matter ? -- Disabled by default
Allows you to enable or disable 64-bit capable devices to be decoded in above 4G address
space. It only works if the system supports 64-bit PCI decoding.

Does 'Relaxed Ordering' matter ?
This option allows you to enable or disable PCI Express Device Relaxed Ordering.

Extended Tag [Disabled] ?
This option allows Device to use an 8-bit Tag field as a requester when set to Enabled.

Maximum Read Request [Auto] ?
This option allows you to set the Maximum Read Request of PCI Express Device or
allow System BIOS to select the value.
Configuration options: [Auto] [128 Bytes] [256 Bytes] [512 Bytes] [1024 Bytes]
[2048 Bytes] [4096 Bytes

There are other settings, all Disabled, that relate to additional PCIe options only relevant [I think] to Gen 2+

Howard

shspvr, 2020-07-31:
Try asking on Asus Forum I did look over the manuel it doesn't have the option to set each to own Gen mode

howardthomson, 2020-08-01:
I have made several BIOS settings changes, most of which I don't entirely understand, but which [in general] restrict the performance and negotiation options of the PCIe system.

Success !!!
vlc running on all four tuners, with Mate System Monitor indicating about 20% CPU on 16 cores. Only one frequency chosen for
all 4 adapters initially, but choosing different frequencies should hardly be an issue.
Thanks for all the advice.
I may do some more tweaking to establish what, precisely, made the difference between crashing and not crashing ...

closed 2020-08-01 by howardthomson