b-rad-NDi/Ubuntu-media-tree-kernel-builder #106: No linux-hwe-mediatree in ppa for Ubuntu 20.04 Focal Fossa?

michelalexander, 2020-07-19:
I just installed your ppa on Ubuntu 20.04 with HWE, but I failed to install linux-hwe-mediatree because it's not available for Focal Fossa. May I use linux-mediatree instead?

BTW: Thanks for your efforts!

shspvr, 2020-07-20:
HWE it not out in tell support schedule 20.04.2 you should be using reg install that

NumerousHats, 2020-09-28:
> HWE it not out in tell support schedule 20.04.2 you should be using reg install that

What does "you should be using reg install that" mean? Use the non-HWE package (i.e. `linux-mediatree`)? Something else?

shspvr, 2020-09-28:
> > HWE it not out in tell support schedule 20.04.2 you should be using reg install that
> 
> What does "you should be using reg install that" mean? Use the non-HWE package (i.e. `linux-mediatree`)? Something else?

That right the non hwe package

mtraxiot, 2020-12-27:
can I confirm we should just install linux-mediatree instead of HWE ?
Could someone update the support page for future ?
 https://www.hauppauge.com/pages/support/support_linux.html

mtraxiot, 2020-12-27:
I've also upgraded to LTS 20.04 and my Quad4 card doesn't appear to be working with MythTV now, however my other cards are..

b-rad-NDi, 2020-12-28:
There is no HWE kernel for Focal yet. They both use the *exact* same kernel. You install normal linux-mediatree for now.

As for quadHD not working, I have received no such reports. You can make your own issue and post full logs if yours does not.

closed 2020-12-28 by michelalexander

b-rad-NDi, 2021-01-21:
Focal HWE kernel 5.8 is incoming in my testing ppa now.