b-rad-NDi/Ubuntu-media-tree-kernel-builder #107: 2xWinTV-QuadHD-ATSC+AMD EPYC, second tuner failed to scan channels

d-averkiev, 2020-07-24:
Hi Brad,
I have 2 identical platforms: Supermicro H12SSW-NT + AMD EPYC EPYC 7302P + **two** WinTV-QuadHD-ATSC PCIe.
OS: Ubuntu 18.04, media tree hwe kernel installed - version 5.3.0-63202007061620-generic
Both platforms have 2 same issues:
1. At the first time no tuners was able to scan channels - w_scan returns usual (no data from PAT, no data from VCT) message. In the kernel log occurs  "**mpeg risc op code**" error message. Exactly this issue **was successfully fixed** by setting dma_reset_workaround=2 in module options. After the option set **first** tuner start working.

2. The issue #2 is that second tuner still failed to scan channels. Symptoms are exactly the same as in the first case, but no errors (or other suspicious messages ) in the kernel log. Starting from debug=5 as module option it drops a reg dump when failed to scan channels but I don't understand what is in this dump.
Do you have any idea what else I should try?
Thank you

b-rad-NDi, 2020-07-24:
Does the second tuner find signal on frequencies even if it does not find programs on those frequencies later?

d-averkiev, 2020-07-24:
Yes it found the signal on few frequencies end reports "signal ok".

b-rad-NDi, 2020-07-24:
I can't say I've heard of signal detected but nothing found, without a single error or warning printed. You'll have to provide dmesg during the scan. I probably don't want the debug at level 5 if it's crazy excessive.

d-averkiev, 2020-07-24:
Scan log is [here](https://pastebin.com/rxVurQZH)
The only messages in dmesg for the scan period is:
```
[860522.799128] si2157 12-0060: found a 'Silicon Labs Si2157-A30'
[860522.821562] si2157 12-0060: firmware version: 3.0.5
[860522.915135] si2157 12-0062: found a 'Silicon Labs Si2157-A30'
[860522.937611] si2157 12-0062: firmware version: 3.0.5
[860523.027131] si2157 15-0060: found a 'Silicon Labs Si2157-A30'
[860523.049857] si2157 15-0060: firmware version: 3.0.5
[860523.143122] si2157 15-0062: found a 'Silicon Labs Si2157-A30'
[860523.165630] si2157 15-0062: firmware version: 3.0.5
[860523.259141] si2157 12-0060: found a 'Silicon Labs Si2157-A30'
[860523.281520] si2157 12-0060: firmware version: 3.0.5
```
For now debug is disabled.

b-rad-NDi, 2020-07-24:
Well, I guess it's time to enable dynamic debug on si2157 and lgdt3306a. Do that and see if something is being hidden in debug logs that shouldn't be.

You shouldn't have to use the -A argument, but it shouldn't hurt. My scans are typically just 
```
w_scan -c US -f a -a X
```
where X is the particular adapter I'm testing.

d-averkiev, 2020-07-24:
This platform is under production load right now. I will wait until maintenance window (late night) will enable debug and re-initialize modules. Please can you tell me exact option name and values for both modules. Thank you.

b-rad-NDi, 2020-07-24:
dynamic debug doesn't require reboot or anything.

https://www.kernel.org/doc/html/v4.11/admin-guide/dynamic-debug-howto.html

si2157 and lgdt3306a are the modules. You just enable with options +pfl, which is print/function/line

b-rad-NDi, 2020-07-24:
Also, you shouldn't need the ppa for pcie quadhd and kernel 5.3 *unless* you needed analog tv support. So you should probably also test out the "vanilla" Ubuntu 5.3 kernel. Use of the ppa just ensures you get all fixes and additional hardware support that might not make it back to 5.3. I also have a "testing" ppa that has kernel releases with much more recent media tree. Those builds are still being validated, but the 5.3 releases seem perfectly fine, so that could be also another testing point.

I unfortunately only have a single quadhd, so I can't try and repro this myself. I've asked the main office to try and test out two quadhd's, but they don't have an epyc system.

d-averkiev, 2020-07-25:
After enabling dynamic debug I've got a new log. Here is the log from modules re-probe to scan complete: [kern.log.tgz.tar.gz](https://github.com/b-rad-NDi/Ubuntu-media-tree-kernel-builder/files/4975675/kern.log.tgz.tar.gz)
Also I tested latest, "vanilla", HWE kernel (it is 5.4 now) - same result, no channels from scan

b-rad-NDi, 2020-07-25:
Oddly I only see 4 dvb adapters registered when you should have 8...correct?

d-averkiev, 2020-07-25:
Good point. To perform the tests I have to switch to another, non-production platform. It has the same configuration (motherboard and cpu) but instead of first PCIe tuner there is USB-one (also WinTV-QuadHD). **The second tuner is PCIe WinTV-QuadHD**.
The symptoms on this platform is completely the same as for double PCIe cards.
I understand that this switch makes our test case not clean enough. But with non-prod platform I can easily stop, reboot or anything else.

b-rad-NDi, 2020-07-25:
Ok, so this is strange then. You're saying that in this pc there is:
- pcie quadhd (adapters 0-3)
- usb quadhd (adapters 4-7 ?)

and this causes the pcie quadhd to not work, while the usb quadhd works? But if the usb quadhd is not connected then the pcie quadhd works?

b-rad-NDi, 2020-07-25:
usb quadhd and pcie quadhd share not a single chip together, different bridges, different tuners, different demods. It would be bizarre for any cross effects.

d-averkiev, 2020-07-25:
Yes the setup just like that pcie quadhd (adapters 0-3), usb quadhd (adapters 4-7). Unfortunately I don't physical access to the platform to pull out the usb cable ). From the next Monday I'll try to free one of the "double pci" machines for the test.

> It would be bizarre for any cross effects.
Exactly! But it is happening ) For some reason.

b-rad-NDi, 2020-07-25:
I do have a usb quadhd and a pcie quadhd, alas only a simple ryzen system, but I can at least repro your current setup.

d-averkiev, 2020-07-25:
I also have "double pcie" setup on gigabyte x570 ud + AMD Ryzen 9 3900X. Everything running smoothly there.

EPYC is kind off system-on-chip, pcie controller built-in the cpu, so it is a difference from common amd-based setups.

b-rad-NDi, 2020-07-25:
Have you tried compiling mainline kernel and seeing if this issue exists there? This is starting to feel very epyc centric. Triply so if the same setup on ryzen works fine.

d-averkiev, 2020-08-10:
I finally got exact same piece of hardware that we are using on a production and perform some tests.
First, this is the server motherboard with pci-e raiser installed - [RSC-W-66G4](https://www.supermicro.com/en/support/resources/riser) (you can use model for search). Both tuner cards are installed in the riser.
All 8 devices are successfully initialized by Ubunutu 18.04 with 5.3.0-63202007061620-generic kernel.

So what I did, I've got the valid channels list (using working adapter):

```
root@changeme:~# cat c.conf
...
qubo:497000000:8VSB:65:68:4
...
```

Then I used this file to tune for the frequency the working adapter:

```
root@changeme:~# azap -a 0 -c c.conf qubo
using '/dev/dvb/adapter0/frontend0' and '/dev/dvb/adapter0/demux0'
Version: 5.11  	 FE_CAN { ATSC + DVB-C (B) }
tuning to 497000000 Hz
video pid 0x0041, audio pid 0x0044
status 1f | signal d709 | snr 00c2 | ber ffffffff | unc 000000ff | FE_HAS_LOCK
status 1f | signal d709 | snr 00c2 | ber ffffffff | unc 00000000 | FE_HAS_LOCK
status 1f | signal d709 | snr 00c2 | ber ffffffff | unc 00000000 | FE_HAS_LOCK
status 1f | signal d709 | snr 00c2 | ber ffffffff | unc 00000000 | FE_HAS_LOCK
status 1f | signal d709 | snr 00c2 | ber ffffffff | unc 00000000 | FE_HAS_LOCK
...
```

Then I've used dvbsnoop to analyze the TS for PAT:

```
root@changeme:~# dvbsnoop -adapter 0 -n 1 0x00
dvbsnoop V1.4.50 -- http://dvbsnoop.sourceforge.net/ 

------------------------------------------------------------
SECT-Packet: 00000001   PID: 0 (0x0000), Length: 36 (0x0024)
Time received: Mon 2020-08-10  12:25:57.554
------------------------------------------------------------
  0000:  00 b0 21 01 d9 cd 00 00  00 03 e0 30 00 04 e0 40   ..!........0...@
  0010:  00 05 e0 50 00 06 e0 60  00 07 e0 70 00 08 e0 80   ...P...`...p....
  0020:  68 a5 48 58                                        h.HX

PID:  0 (0x0000)  [= assigned for: ISO 13818-1 Program Association Table (PAT)]

Guess table from table id...
PAT-decoding....
Table_ID: 0 (0x00)  [= Program Association Table (PAT)]
section_syntax_indicator: 1 (0x01)
(fixed): 0 (0x00)
reserved_1: 3 (0x03)
Section_length: 33 (0x0021)
Transport_Stream_ID: 473 (0x01d9)
reserved_2: 3 (0x03)
Version_number: 6 (0x06)
current_next_indicator: 1 (0x01)  [= valid now]
Section_number: 0 (0x00)
Last_Section_number: 0 (0x00)

    Program_number: 3 (0x0003)
    reserved: 7 (0x07)
    Program_map_PID: 48 (0x0030)

    Program_number: 4 (0x0004)
    reserved: 7 (0x07)
    Program_map_PID: 64 (0x0040)

    Program_number: 5 (0x0005)
    reserved: 7 (0x07)
    Program_map_PID: 80 (0x0050)

    Program_number: 6 (0x0006)
    reserved: 7 (0x07)
    Program_map_PID: 96 (0x0060)

    Program_number: 7 (0x0007)
    reserved: 7 (0x07)
    Program_map_PID: 112 (0x0070)

    Program_number: 8 (0x0008)
    reserved: 7 (0x07)
    Program_map_PID: 128 (0x0080)

CRC: 1755662424 (0x68a54858)
==========================================================
```

Then I tuned the second (not working) adapter. As you can see the output is completely the same. So from the point of view of the frontend tuning is happened:

```
root@changeme:~# azap -a 4 -c c.conf qubo
using '/dev/dvb/adapter4/frontend0' and '/dev/dvb/adapter4/demux0'
Version: 5.11  	 FE_CAN { ATSC + DVB-C (B) }
tuning to 497000000 Hz
video pid 0x0041, audio pid 0x0044
status 1f | signal d1ea | snr 00b7 | ber ffffffff | unc 000000ff | FE_HAS_LOCK
status 1f | signal d47a | snr 00ba | ber ffffffff | unc 00000000 | FE_HAS_LOCK
status 1f | signal d47a | snr 00bb | ber ffffffff | unc 00000000 | FE_HAS_LOCK
status 1f | signal d47a | snr 00bb | ber ffffffff | unc 00000000 | FE_HAS_LOCK
status 1f | signal d47a | snr 00bc | ber ffffffff | unc 00000000 | FE_HAS_LOCK
...
```

Then dvbsnoop on that adapter is showing nothing until I hit ctrl+c:

```
root@changeme:~# dvbsnoop -adapter 4 -n 1 0x00
dvbsnoop V1.4.50 -- http://dvbsnoop.sourceforge.net/ 
^C
```

I have no idea why this is happening. But here is some thoughts. As I understand that the tuner and demodulator is virtually different devices. Can it be that OS is confusing a "routing" between them somehow?

b-rad-NDi, 2020-08-10:
This is kind of nonsensical imho. Have you switched slots between the QuadHD's and/or tested the "bad" one by itself to confirm there is not just some hardware issue? I've had users report to me using eight QuadHD's in a single system without issue.

There's no way to "confuse" and intermingle two individual devices. They are each instantiated individually and have their own memory spaces and discrete controls. There's no globally shared anything really. You would only see an issue like you are if the transport stream bus on the demod was shut off. A bug like that would affect every single board though. It's almost like the data lanes are just not connected.

While I do not have multiple QuadHD I do have additional boards using the same pcie bridge. I can use both boards perfectly fine. It would seem very bizarre for this to be a PCIE issue specifically with EPYC systems, but, the fact it works on other platforms how it should and not on EPYC is troublesome.

Do you happen to have any other motherboards with two empty pcie slots you can do a sanity test on?

Have you tried the riser with only a single card in each slot to see if only one works in each?

Maybe the riser is bad? Is the pcie in 16x mode? Can you force it to 4x or 8x mode?

I don't know enough how risers work and there's no documentation for that riser I can find.

Can you get me detailed lspci output from those two devices?

b-rad-NDi, 2020-08-11:
Also, have you tried forcing pcie from 4.0 to a lower standard for compatibility?