b-rad-NDi/Ubuntu-media-tree-kernel-builder #109: em28xx options em28xx_dvb adapter_nr

jksj461, 2020-08-19:
I have never had a lot of success using udev rules to distinguish between adaptors and create reliable persistent names. The problem has just got a lot worse when I changed the cpu to quad core. The linked adaptor names are created but for the Dual HD card they do not work.
Previously I have found setting the dvb adapter number in the driver to be very reliable. However I am trying to distinguish between a DualHD and a 461 E which to my level of understanding share the same driver.
A statement like "options em28xx_dvb adapter_nr=4,5" would set the numbers for a DualHD on its own.
Can this option be used when both cards are present to set unique, persistent numbers?

b-rad-NDi, 2020-08-19:
The parameter is a single integer, not an array. If you want to know what device is what look at my eeprom-tinker project in the file device-detection.c. It will identify individual devices using sysfs interface. You could then add some code around where it looks at ../idVendor and ../idProduct to instead get the devices at ./dvb/*

b-rad-NDi, 2020-08-19:
I might add that into eeprom-tinker just because.

b-rad-NDi, 2020-08-19:
Also, if there is an issue where the 461e identifies itself via some means, but the dualHD does not file a specific ticket with the information documented and i'll just fix that in the kernel.

jksj461, 2020-08-20:
Thanks will give eeprom tinker a go

closed 2026-02-03 by b-rad-NDi