b-rad-NDi/Ubuntu-media-tree-kernel-builder #110: No Raspberry Pi Ubuntu support from the ppa

poopingpenis, 2020-08-29:
Running

```
sudo apt-get install linux-mediatree
```

results in

```
The following packages have unmet dependencies:
 linux-mediatree : Depends: linux-image-5.4.0-392007061620-generic but it is not installable
                   Depends: linux-modules-5.4.0-392007061620-generic but it is not installable
                   Depends: linux-modules-extra-5.4.0-392007061620-generic but it is not installable
                   Depends: linux-headers-5.4.0-392007061620-generic but it is not installable
E: Unable to correct problems, you have held broken packages.
```

Going back to Raspbian isn't an option since I need to run 64-bit software which it doesn't allow. I'm running a [64-bit copy of Ubuntu 20.04.1 LTS](https://ubuntu.com/download/raspberry-pi).

Pls add support

b-rad-NDi, 2020-08-29:
I might look into it.

poopingpenis, 2020-09-02:
Thanks pls

b-rad-NDi, 2020-09-09:
So, it seems like it's probably not possible for me to add it to the ppa as the build scripts imply the entire process including packaging can only complete on an arm64 rpi itself. BUT, I'm seeing how far a cross compile gets me and if nothing else I can at least offer builds to download.

poopingpenis, 2020-09-10:
Builds at least would be greatly appreciated!

b-rad-NDi, 2020-11-12:
Take a peep in the releases section of this repo ;-)

closed 2020-11-12 by b-rad-NDi