b-rad-NDi/Ubuntu-media-tree-kernel-builder #112: WinTV SoloHD does not work after resume from suspend

puthre, 2020-09-06:
I'm using 
```
puthre@puthre-t7500:~$ uname -a
Linux puthre-t7500 **5.4.0-392007061620-generic #0+mediatree+hauppauge-Ubuntu** SMP Wed Jul 8 04:56:09 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
puthre@puthre-t7500:~$ cat /etc/issue
**Ubuntu 20.04.1 LTS** \n \l
```

KODI + tvheadend

Everything works fine until I put the system to sleep and them resume. After thet tvheadend reports that it has no singal on any channel.
I've tried to remove the em28xx_dvb module and reinsert it on resume but I can't, even if tvheadend service is stopped.

```
puthre@puthre-t7500:~$ modprobe -r em28xx_dvb
modprobe: FATAL: Module em28xx_dvb is in use.
puthre@puthre-t7500:~$ lsmod | grep em28xx_dvb
em28xx_dvb             36864  4
dvb_core              143360  1 em28xx_dvb
em28xx                 90112  2 em28xx_rc,em28xx_dvb
```

I've tried resetting the device as described at https://askubuntu.com/questions/645/how-do-you-reset-a-usb-device-from-the-command-line , but with no success.

The only thing that works is to physically remove the device from the port an insert it back, but that's not really convenient to do every time you resume the system.

th0ma7, 2020-09-08:
I believe this is similar to the issue hapening on Synology DSM ... 
Reference: https://github.com/b-rad-NDi/Embedded-MediaDrivers/issues/3

b-rad-NDi, 2020-09-08:
Completely unrelated.

There is a patch I disabled "Include b 4.0-11 when checking firmware version" which is apparently required for that firmware to resume from suspend. I'll re-enable it.

b-rad-NDi, 2020-09-12:
@puthre : The patch was re-enabled and new builds of everything have been pushed to the ppa. Please close the issue if you find the problem solved. If not do let me know.

closed 2020-10-22 by b-rad-NDi