b-rad-NDi/Ubuntu-media-tree-kernel-builder #113: Missing in repository

gunflint, 2020-09-07:
0-07-2020

sudo apt-get install linux-mediatree
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package linux-mediatree

b-rad-NDi, 2020-09-08:
You're going to have to provide much more information than that if you expect me to help.

gunflint, 2020-09-08:
I have the media tree installed an updated any heres the terminal output.

@Predator-Linux-Powered:~$ sudo apt-get install linux-mediatree
[sudo] password for dave:
Reading package lists... Done
Building dependency tree
Reading state information... Done
E: Unable to locate package linux-mediatree
@Predator-Linux-Powered:~$
Thank you.
Dave

b-rad-NDi, 2020-09-08:
I have no idea what OS you're running. I have no idea what version of that OS you're running. The package exists in my ppa for all currently supported versions of Ubuntu. Did you follow the instructions to add my ppa?

jamminjamesc, 2020-09-14:
I believe the problem has something to do with an authentication key? Getting an message about "no valid OpenPGP" when trying to add repository:

gpg: keybox '/tmp/tmplkw8ewrj/pubring.gpg' created
gpg: /tmp/tmplkw8ewrj/trustdb.gpg: trustdb created
gpg: key ED51750F9347DEC9: public key "Launchpad PPA for Brad Love" imported
gpg: Total number processed: 1
gpg:               imported: 1
gpg: no valid OpenPGP data found.

If you try to continue the install process; you end up getting the error OP posted about.

b-rad-NDi, 2020-09-15:
Still nothing on what OS and version of the OS, nor exactly what commands are being used. The ppa works on every version of Ubuntu currently supported when following the exact standard instructions to add a ppa.

gunflint, 2020-09-15:
Sorry I'm on Linux Mint 20 Ulyana. Which I consider a full independent
distributor now as many depencies on mint dont come from Ubuntu anymore

b-rad-NDi, 2020-09-15:
Ok, so then how did you enable the ppa?

gunflint, 2020-09-15:
I did. Updated the repositories then and trying to install nothing is found.

b-rad-NDi, 2020-09-15:
You did what? I see zero information on how you enabled the ppa.

gunflint, 2020-09-15:
I have the hauppauge repo installed and updated on Linux Mint 20 Uluana.

Terminal info:

@Predator-Linux-Powered:~$ sudo apt-get install linux-mediatree
Reading package lists... Done
Building dependency tree
Reading state information... Done
E: Unable to locate package linux-mediatree

b-rad-NDi, 2020-09-15:
Ok I get that loud and clear. What you have not said a single thing about is how you enabled the ppa. I want to know *exactly* how you enabled it. Did you even use add-apt-repository?

b-rad-NDi, 2020-09-15:
Do you even see the ppa getting checked when you do apt-get update?

You're providing almost no information.

gunflint, 2020-09-15:
sudo add-apt-repository ppa:b-rad/kernel+mediatree+hauppauge

You are about to add the following PPA:
This PPA provides the Ubuntu kernel, with the latest Linuxtv.org media tree
slipstreamed in and all new hardware enabled. There are also additional
patches applied to accommodate new Hauppauge hardware/features, while that
support is heading upstream.

Patches, build script, and bug tracker may be found here:
https://github.com/b-rad-NDi/Ubuntu-media-tree-kernel-builder
 More info:
https://launchpad.net/~b-rad/+archive/ubuntu/kernel+mediatree+hauppauge
Press Enter to continue or Ctrl+C to cancel
Executing: /tmp/apt-key-gpghome.ooLyZbFor1/gpg.1.sh --keyserver hkps://
keyserver.ubuntu.com:443 --recv-keys
9EC21B6EAABF32A03A92CA2EED51750F9347DEC9

gpg: key ED51750F9347DEC9: public key "Launchpad PPA for Brad Love" imported
gpg: Total number processed: 1
gpg:               imported: 1

sudo apt-get update *****

*@Predator-Linux-Powered*:*~*$ sudo apt-get install linux-mediatree

Reading package lists... Done
Building dependency tree
Reading state information... Done
E: Unable to locate package linux-mediatree

b-rad-NDi, 2020-09-15:
I want to see the output from apt-get update so I see what repositories you are hitting.

shspvr, 2020-09-15:
As far I know of PPAs is Ubuntu's thing and are not 100% compatible with some Debian out box's like Linux Min, I on other hand have like that OS in many year because of it compatible problem.
Any way try read this https://forums.linuxmint.com/viewtopic.php?t=323018 it may help
I hope this help you, Keep at it sooner or late you get fig out.

gunflint, 2020-09-15:
I used the staging ppa instead of stable.

"b-rad/test-test-hw-staging"

Dependencies where  installed  and kernel was built..

I plugged in the quad USB and the green led lit up.

Interesting.

b-rad-NDi, 2020-09-15:
My staging ppa isn't necessarily meant to be used by the general public. It's my testing grounds and often times does not have virtual packages included. All builds go there first, are tested, then are copied over to the main ppa for rebuilding.

Right now it includes every package but that can change at any time and when it does there will be no virtual packages (linux-mediatree & linux-hwe-mediatree) until the release is ready.

gunflint, 2020-09-15:
Apparently the  ppa:b-rad/kernel+mediatree+hauppauge ppa does not work on
Linux Mint 20 and the staging does.

b-rad-NDi, 2020-09-15:
https://launchpad.net/~b-rad/+archive/ubuntu/kernel+mediatree+hauppauge/+packages

https://launchpad.net/~b-rad/+archive/ubuntu/test-test-hw-staging/+packages

Every single package in staging is in the main ppa. No clue what your issue could be. Still haven't seen your apt-get update repo hit list to see if there's something wrong there somehow.

gunflint, 2020-09-15:
Well I'm no geek I don't code just letting you know I used the staging and
it worked. Maybe this info helps with Bugzilla for others experiences the
same issues

Anyway thank you for supporting Linux an FOSS.

b-rad-NDi, 2020-09-15:
It unfortunately doesn't help to address issues when I can't get all the information about what could be wrong.

I would still like to see your apt-get update repo hit list.

gunflint, 2020-09-15:
Terminal output from the stable ppa, I hope this helps

redator-Linux-Powered:~$
dave@Predator-Linux-Powered:~$ sudo add-apt-repository
ppa:b-rad/kernel+mediatree+hauppauge

You are about to add the following PPA:
 This PPA provides the Ubuntu kernel, with the latest Linuxtv.org media
tree slipstreamed in and all new hardware enabled. There are also
additional patches applied to accommodate new Hauppauge hardware/features,
while that support is heading upstream.

Patches, build script, and bug tracker may be found here:
https://github.com/b-rad-NDi/Ubuntu-media-tree-kernel-builder
 More info:
https://launchpad.net/~b-rad/+archive/ubuntu/kernel+mediatree+hauppauge
Press Enter to continue or Ctrl+C to cancel
Executing: /tmp/apt-key-gpghome.ooLyZbFor1/gpg.1.sh --keyserver hkps://
keyserver.ubuntu.com:443 --recv-keys
9EC21B6EAABF32A03A92CA2EED51750F9347DEC9

gpg: key ED51750F9347DEC9: public key "Launchpad PPA for Brad Love" imported
gpg: Total number processed: 1
gpg:               imported: 1
gpg: invalid key resource URL
'/etc/apt/trusted.gpg.d/home:hawkeye116477:waterfox.gpg'
gpg: keyblock resource '(null)': General error
gpg: key 76F1A20FF987672F: 1 signature not checked due to a missing key
gpg: key 7721F63BD38B4796: 2 signatures not checked due to missing keys
gpg: key 5E62D791625A271E: 1 signature not checked due to a missing key
gpg: key 0FAD31CA8719FCE4: 1 signature not checked due to a missing key
gpg: key 85A3D26506C4AE2A: 2 signatures not checked due to missing keys
gpg: key A6616109451BBBF2: 12 signatures not checked due to missing keys
gpg: key 3B4FE6ACC0B21F32: 3 signatures not checked due to missing keys
gpg: key Predator-Linux-Powered:~$ sudo apt-get update
Hit:1 http://repository.spotify.com stable InRelease

Hit:2 https://dl.winehq.org/wine-builds/ubuntu focal InRelease

Get:3 https://deb.opera.com/opera-beta stable InRelease [2,591 B]

Hit:4 http://dl.google.com/linux/chrome/deb stable InRelease

Get:5 https://deb.opera.com/opera-developer stable InRelease [2,591 B]

Get:6 https://deb.opera.com/opera-stable stable InRelease [2,591 B]

Ign:7 http://dl.google.com/linux/earth/deb stable InRelease

Hit:8 https://brave-browser-apt-beta.s3.brave.com stable InRelease

Hit:9 https://brave-browser-apt-release.s3.brave.com stable InRelease

Err:3 https://deb.opera.com/opera-beta stable InRelease

  The following signatures couldn't be verified because the public key is
not available: NO_PUBKEY 4B8EC3BAABDC4346
Hit:10 http://dl.google.com/linux/earth/deb stable Release

Hit:11 https://brave-browser-apt-nightly.s3.brave.com stable InRelease

Err:5 https://deb.opera.com/opera-developer stable InRelease

  The following signatures couldn't be verified because the public key is
not available: NO_PUBKEY 4B8EC3BAABDC4346
Ign:12 http://repo.vivaldi.com/snapshot/deb stable InRelease

Hit:13 https://repo.skype.com/deb stable InRelease

Err:6 https://deb.opera.com/opera-stable stable InRelease
[... 992 lines cut ...]
E: The repository 'http://ppa.launchpad.net/thomas-schiex/blender/ubuntu
focal InRelease' is not signed.
N: Updating from such a repository can't be done securely, and is therefore
disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration
details.
W: GPG error: http://ppa.launchpad.net/transmissionbt/ppa/ubuntu focal
InRelease: The following signatures couldn't be verified because the public
key is not available: NO_PUBKEY 976B5901365C5CA1
E: The repository 'http://ppa.launchpad.net/transmissionbt/ppa/ubuntu focal
InRelease' is not signed.
N: Updating from such a repository can't be done securely, and is therefore
disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration
details.
dave@Predator-Linux-Powered:~$ sudo apt-get install linux-mediatree
Reading package lists... Done
Building dependency tree
Reading state information... Done
E: Unable to locate package linux-mediatree
D94AA3F0EFE21092: 3 signatures not checked due to missing keys
gpg: key C8CAB6595FDFF622: 2 signatures not checked due to missing keys
gpg: key 871920D1991BC93C: 1 signature not checked due to a missing key
gpg: Total number processed: 36
gpg:       skipped new keys: 36

shspvr, 2020-09-15:
> I used the staging ppa instead of stable. "b-rad/test-test-hw-staging" Dependencies where installed and kernel was built.. I plugged in the quad USB and the green led lit up. Interesting.
> […](#)

Yes you get should get two green light there number software package you used try testing Kaffeine TV, MeTV, SageTV, NextPVR and VDR and royal pain butt MythTV and many other but there DVB only so no ATSC support.

b-rad-NDi, 2020-09-15:
@gunflint : You have inherent issues with your gpg key management that happened somehow. You are missing tons of keys causing the issues you are experiencing. Maybe they're all corrupt, I don't know, but you'll have to figure out how to manually wipe all the bad ones and re-import them. You should be able to remove my ppa, then wipe whatever key mint thinks it has, then re-add the key. Whatever the case your system requires manual intervention to fix the issues.

Consult mint forums for information on how to clean up your apt gpg cache.

shspvr, 2020-09-15:
> @gunflint : You have inherent issues with your gpg key management that happened somehow. You are missing tons of keys causing the issues you are experiencing. Maybe they're all corrupt, I don't [...]

I agree with @b-rad-NDi that a huge mess above

gunflint, 2020-09-15:
Thank you ill use mint software center to clean up the mess. I use
y-ppa-manager on occasion as well.

jamminjamesc, 2020-09-15:
> Still nothing on what OS and version of the OS, nor exactly what commands are being used. The ppa works on every version of Ubuntu currently supported when following the exact standard instructions to add a ppa.

Oh, OK. I’m running debian and am very fresh to linux. Please forgive.  I just figured it might not be me since I was getting the same error as other poster.

Maybe he’s new too...  Hey, other dude, run lsb_release -d and check your OS version.

b-rad-NDi, 2020-09-15:
@jamminjamesc What instructions did you follow to enable the ppa? Did you use add-apt-repository?

jamminjamesc, 2020-09-17:
I followed the instructions here:
https://hauppauge.com/pages/support/support_linux.html

Yes, I used add-apt-repository.

I was not successful.

gunflint, 2020-09-17:
Oh I see. I would respond on getub  but don't know how. Thanks.

gunflint, 2020-09-17:
I was with the experimental.

b-rad-NDi, 2020-09-17:
@jamminjamesc : Those instructions are for ubuntu specifically. On debian it probably would not set up the ppa correctly. You would probably have to edit a file in /etc/apt.d/ to point it to somewhere suitable.

If you show the output from sudo apt-get update it should be obvious the error.

gunflint, 2020-09-17:
Experimental works on Linux mint 20. Mint is a disro in its own right now a
day's.

closed 2020-10-07 by b-rad-NDi