b-rad-NDi/Ubuntu-media-tree-kernel-builder #115: WinTV-dual HD Linux continuity counter issues

gurabli, 2020-10-21:
Hi,

I'm having some problem with the Dual HD stick on Ubuntu 18.04 server and Tvheadend.
For some reason I receive quite a lot of Continuity Counter Errors. Using DVB-C source.
Strange, that on a NanopiNeo2 running Armbian it works perfectly fine.

Linux szimat.net 5.4.0-52-generic #57~18.04.1-Ubuntu SMP Thu Oct 15 14:04:49 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
HTS Tvheadend 4.3-1906~g7e1dac822

PPA added, installed, including firmwares.
Let me know what should I provide further.

b-rad-NDi, 2020-10-22:
Well, for one, that's not a ppa kernel. Do you see these errors on ppa kernel? You can attach lsusb and dmesg logs to this issue as well.

gurabli, 2020-10-22:
Strange, I added the ppa and also did a diát-upgrade. Any specific way to install ppa kernel? 
Will check and return with logs.

b-rad-NDi, 2020-10-22:
You installed linux-mediatree or linux-hwe-mediatree? That will install the kernels.

gurabli, 2020-10-22:
I installed hwe. I have a hwe kernel. Strange why is it not installed.

b-rad-NDi, 2020-10-22:
Well do you see the files in /boot/ ? Perhaps another kernel is pre-empting it, you might have to select it manually in grub or set the default number to boot from 1 to something else.

b-rad-NDi, 2020-10-22:
Wait a second. According to Ubuntu's git repos the 18.04 HWE kernel is 5.3.0-64, yet what they're publishing is 5.4.0-52...that's certainly strange and is definitely what's causing the issue. It's booting the 5.4 kernel since it's "newer"

gurabli, 2020-10-29:
> Wait a second. According to Ubuntu's git repos the 18.04 HWE kernel is 5.3.0-64, yet what they're publishing is 5.4.0-52...that's certainly strange and is definitely what's causing the issue. It's booting the 5.4 kernel since it's "newer"

Sorry for the late reply. What do you suggest, ehat should I try?

b-rad-NDi, 2020-12-28:
I figured out what Ubuntu was doing and updated git repo locations and did some branch overriding. The Bionic HWE kernel is now 5.4.

gurabli, 2021-01-05:
> I figured out what Ubuntu was doing and updated git repo locations and did some branch overriding. The Bionic HWE kernel is now 5.4.

Thanks! So I reinstalled Ubuntu Server few weeks ago, now running non HWE kernel. I added your PPA and the kernel and firmwares all installed fine.

```
uname -a
Linux server 4.15.0-126202012221313-generic #0+mediatree+hauppauge-Ubuntu SMP Wed Dec 23 21:52:47 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux

```
In Tvheadend I still get lot of continuity errors on DVB-C. Running on a Nanopi device with the same tuner it is much better. Still I have some cont errors there, but not so much. Signal is fine when watching DVB-C connected directly to TV.

```
2021-01-05 09:25:48.686 TS: XX: H264 @ #1501 Continuity counter error (total 4)
2021-01-05 09:26:47.510 TS: XX: H264 @ #1501 Continuity counter error (total 8)
2021-01-05 09:27:30.492 TS: XX: H264 @ #1501 Continuity counter error (total 11)
2021-01-05 09:29:01.790 TS: XX: H264 @ #1501 Continuity counter error (total 12)
2021-01-05 09:31:07.072 TS: XX: H264 @ #1501 Continuity counter error (total 18)
```
What do you think, what can be the problem? And how could we start debuging/fixing this?

b-rad-NDi, 2021-01-05:
Do you have an ISOC or bulk DualHD? This will be indicated by the usb vid:pid and by statements in dmesg.

gurabli, 2021-01-05:
> Do you have an ISOC or bulk DualHD? This will be indicated by the usb vid:pid and by statements in dmesg.

I think it will be ISOC
dmesg:
```
[   20.057717] em28xx 1-1.2:1.0: New device HCW dualHD @ 480 Mbps (2040:0265, interface 0, class 0)
[   20.057721] em28xx 1-1.2:1.0: DVB interface 0 found: isoc
[   20.116294] em28xx 1-1.2:1.0: chip ID is em28174
[   20.290617] intel_rapl: Found RAPL domain package
[   20.290620] intel_rapl: Found RAPL domain core
[   21.320800] em28xx 1-1.2:1.0: EEPROM ID = 26 00 01 00, EEPROM hash = 0x03e7c3c8
[   21.320805] em28xx 1-1.2:1.0: EEPROM info:
[   21.320808] em28xx 1-1.2:1.0:        microcode start address = 0x0004, boot configuration = 0x01
[   21.328621] em28xx 1-1.2:1.0:        AC97 audio (5 sample rates)
[   21.328626] em28xx 1-1.2:1.0:        500mA max power
[   21.328631] em28xx 1-1.2:1.0:        Table at offset 0x27, strings=0x0e6a, 0x1888, 0x087e
[   21.388101] em28xx 1-1.2:1.0: Identified as Hauppauge WinTV-dualHD DVB (card=99)
[   21.391614] tveeprom: Hauppauge model 204109, rev B3I6, serial# 13970025
[   21.391618] tveeprom: tuner model is SiLabs Si2157 (idx 186, type 4)
[   21.391621] tveeprom: TV standards PAL(B/G) NTSC(M) PAL(I) SECAM(L/L') PAL(D/D1/K) ATSC/DVB Digital (eeprom 0xfc)
[   21.391622] tveeprom: audio processor is None (idx 0)
```
In TVheadend I have signal strength -29 dBm and SNR 33.8 dB for example for a HD channel. Although I don't know if these values matter or mean anything or how reliable they are.

b-rad-NDi, 2021-01-05:
Use my tool eeprom-tinker and change it to bulk mode.

gurabli, 2021-01-05:
> Use my tool eeprom-tinker and change it to bulk mode.

I converted to bulk mode with your utility. It looks fine now! Running the same HD stream and only have 1 continuity error reported, and that doesn't even show up in the log, just in the counter. Many thanks for your excellent work and support! 
This is why it is worth buying quality products with great support!

Out of curiosity, why not shipping the USB tuners in bulk mode by default? Or are there any downsides?

b-rad-NDi, 2021-01-05:
Glad to hear you're fixed up. The manufacturing change to bulk mode was done after my own research into such matters indicated the necessity. They've been shipped in bulk mode for a couple years now. You'll notice the pid (should have) changed from 0x0265 to 0x8265. Some places had old stock around for a while though.

closed 2021-01-05 by b-rad-NDi

parkerlreed, 2021-01-05:
Thanks for the tip. Mine has been working fine but it is in isoc mode. I'll be sure to change out the mode.

```
[ 5327.268699] usb 3-1.4.4: New USB device found, idVendor=2040, idProduct=026d, bcdDevice= 1.00
[ 5327.268706] usb 3-1.4.4: New USB device strings: Mfr=3, Product=1, SerialNumber=2
[ 5327.268717] usb 3-1.4.4: Product: 955D
[ 5327.268720] usb 3-1.4.4: Manufacturer: HCW
[ 5327.268724] usb 3-1.4.4: SerialNumber: 0013811222
[ 5327.270940] em28xx 3-1.4.4:1.0: New device HCW 955D @ 480 Mbps (2040:026d, interface 0, class 0)
[ 5327.270946] em28xx 3-1.4.4:1.0: DVB interface 0 found: isoc
[ 5327.331483] em28xx 3-1.4.4:1.0: chip ID is em28174
[ 5328.527892] em28xx 3-1.4.4:1.0: EEPROM ID = 26 00 01 00, EEPROM hash = 0xfa790dca
[ 5328.527897] em28xx 3-1.4.4:1.0: EEPROM info:
[ 5328.527901] em28xx 3-1.4.4:1.0:      microcode start address = 0x0004, boot configuration = 0x01
[ 5328.534301] em28xx 3-1.4.4:1.0:      AC97 audio (5 sample rates)
[ 5328.534305] em28xx 3-1.4.4:1.0:      500mA max power
[ 5328.534310] em28xx 3-1.4.4:1.0:      Table at offset 0x27, strings=0x0a72, 0x187c, 0x086a
[ 5328.594761] em28xx 3-1.4.4:1.0: Identified as Hauppauge WinTV-dualHD 01595 ATSC/QAM (card=100)
```