b-rad-NDi/Ubuntu-media-tree-kernel-builder #116: PPA not working on L4T (Linux for Tegra)

is-lonely, 2020-11-10:
I have L4T installed on my modded Nintendo Switch which has the Tegra TX1 chip, and I'm getting this error when using the PPA. I haven't tried building from source yet, but I just thought I'd make y'all aware of this.
[image]

shspvr, 2020-11-10:
Tegra TX1 is ARM base which not supported here

is-lonely, 2020-11-11:
That makes since. Totally forgot about that

closed 2020-11-11 by is-lonely

b-rad-NDi, 2020-11-11:
Unless Ubuntu has a branch for Tegra X1 I couldn't support it, externally at least, by building packages.

parkerlreed, 2020-11-12:
That wont work anyways as L4T uses a custom 4.9 kernel. It has to be manually recompiled with the modules you need.

Hop over to the Discord https://discord.gg/hw8e6ZyD

parkerlreed, 2020-11-12:
Depending on the card you have support is already in the kernel. 

em28xx is included. I just poked a day ago because it's missing the lgt tuner module for the Dual HD. It has worked previously so it just needs enabled in the config and recompiled.