b-rad-NDi/Ubuntu-media-tree-kernel-builder #117: [Question] IR setup for Win TV Dual HD

parkerlreed, 2020-11-11:
Does this patchset handle the rc0 handling for this card?

I'm currently debugging upstream where the /dev/lirc0 interface gets created but I am unable to use it in lircd.

Trying to cat it manually which should give me some gibberish, leads to invalid argument

I double checked and lirc0 is coming from this card. Not sure what could be going on.

```
[root@stealth parker]# cat /dev/lirc0
cat: /dev/lirc0: Invalid argument
```

```
[ 5327.268699] usb 3-1.4.4: New USB device found, idVendor=2040, idProduct=026d, bcdDevice= 1.00
[ 5327.268706] usb 3-1.4.4: New USB device strings: Mfr=3, Product=1, SerialNumber=2
[ 5327.268717] usb 3-1.4.4: Product: 955D
[ 5327.268720] usb 3-1.4.4: Manufacturer: HCW
[ 5327.268724] usb 3-1.4.4: SerialNumber: 0013811222
[ 5327.270940] em28xx 3-1.4.4:1.0: New device HCW 955D @ 480 Mbps (2040:026d, interface 0, class 0)
[ 5327.270946] em28xx 3-1.4.4:1.0: DVB interface 0 found: isoc
[ 5327.331483] em28xx 3-1.4.4:1.0: chip ID is em28174
[ 5328.527892] em28xx 3-1.4.4:1.0: EEPROM ID = 26 00 01 00, EEPROM hash = 0xfa790dca
[ 5328.527897] em28xx 3-1.4.4:1.0: EEPROM info:
[ 5328.527901] em28xx 3-1.4.4:1.0:      microcode start address = 0x0004, boot configuration = 0x01
[ 5328.534301] em28xx 3-1.4.4:1.0:      AC97 audio (5 sample rates)
[ 5328.534305] em28xx 3-1.4.4:1.0:      500mA max power
[ 5328.534310] em28xx 3-1.4.4:1.0:      Table at offset 0x27, strings=0x0a72, 0x187c, 0x086a
[ 5328.594761] em28xx 3-1.4.4:1.0: Identified as Hauppauge WinTV-dualHD 01595 ATSC/QAM (card=100)
[ 5328.599565] tveeprom: Hauppauge model 204101, rev B2I6, serial# 13811222
[ 5328.599569] tveeprom: tuner model is SiLabs Si2157 (idx 186, type 4)
[ 5328.599572] tveeprom: TV standards PAL(B/G) NTSC(M) PAL(I) SECAM(L/L') PAL(D/D1/K) ATSC/DVB Digital (eeprom 0xfc)
[ 5328.599574] tveeprom: audio processor is None (idx 0)
[ 5328.599576] tveeprom: has no radio, has IR receiver, has no IR transmitter
[ 5328.599583] em28xx 3-1.4.4:1.0: dvb set to isoc mode.
[ 5328.654819] em28xx 3-1.4.4:1.0: chip ID is em28174
[ 5329.837667] em28xx 3-1.4.4:1.0: EEPROM ID = 26 00 01 00, EEPROM hash = 0xfa790dca
[ 5329.837672] em28xx 3-1.4.4:1.0: EEPROM info:
[ 5329.837676] em28xx 3-1.4.4:1.0:      microcode start address = 0x0004, boot configuration = 0x01
[ 5329.843967] em28xx 3-1.4.4:1.0:      AC97 audio (5 sample rates)
[ 5329.843971] em28xx 3-1.4.4:1.0:      500mA max power
[ 5329.843975] em28xx 3-1.4.4:1.0:      Table at offset 0x27, strings=0x0a72, 0x187c, 0x086a
[ 5329.898096] em28xx 3-1.4.4:1.0: Identified as Hauppauge WinTV-dualHD 01595 ATSC/QAM (card=100)
[ 5329.902728] tveeprom: Hauppauge model 204101, rev B2I6, serial# 13811222
[ 5329.902733] tveeprom: tuner model is SiLabs Si2157 (idx 186, type 4)
[ 5329.902739] tveeprom: TV standards PAL(B/G) NTSC(M) PAL(I) SECAM(L/L') PAL(D/D1/K) ATSC/DVB Digital (eeprom 0xfc)
[ 5329.902742] tveeprom: audio processor is None (idx 0)
[ 5329.902745] tveeprom: has no radio, has IR receiver, has no IR transmitter
[ 5329.902754] em28xx 3-1.4.4:1.0: dvb ts2 set to isoc mode.
[ 5330.102882] em28xx 3-1.4.4:1.0: Binding DVB extension
[ 5330.119306] i2c i2c-14: Added multiplexed i2c bus 17
[ 5330.119309] lgdt3306a 14-0059: LG Electronics LGDT3306A successfully identified
[ 5330.122941] si2157 17-0060: Silicon Labs Si2147/2148/2157/2158 successfully attached
[ 5330.122969] dvbdev: DVB: registering new adapter (3-1.4.4:1.0)
[ 5330.122972] em28xx 3-1.4.4:1.0: DVB: registering adapter 0 frontend 0 (LG Electronics LGDT3306A VSB/QAM Frontend)...
[ 5330.122974] dvbdev: dvb_create_media_entity: media entity 'LG Electronics LGDT3306A VSB/QAM Frontend' registered.
[ 5330.123370] dvbdev: dvb_create_media_entity: media entity 'dvb-demux' registered.
[ 5330.125413] em28xx 3-1.4.4:1.0: DVB extension successfully initialized
[ 5330.125417] em28xx 3-1.4.4:1.0: Binding DVB extension
[ 5330.136416] i2c i2c-16: Added multiplexed i2c bus 18
[ 5330.136419] lgdt3306a 16-000e: LG Electronics LGDT3306A successfully identified
[ 5330.141104] si2157 18-0062: Silicon Labs Si2147/2148/2157/2158 successfully attached
[ 5330.141131] dvbdev: DVB: registering new adapter (3-1.4.4:1.0)
[ 5330.141134] em28xx 3-1.4.4:1.0: DVB: registering adapter 1 frontend 0 (LG Electronics LGDT3306A VSB/QAM Frontend)...
[ 5330.141137] dvbdev: dvb_create_media_entity: media entity 'LG Electronics LGDT3306A VSB/QAM Frontend' registered.
[ 5330.141504] dvbdev: dvb_create_media_entity: media entity 'dvb-demux' registered.
[ 5330.142812] em28xx 3-1.4.4:1.0: DVB extension successfully initialized
[ 5330.142815] em28xx 3-1.4.4:1.0: Registering input extension
[ 5330.142930] Registered IR keymap rc-hauppauge
[ 5330.143077] rc rc0: Hauppauge WinTV-dualHD 01595 ATSC/QAM as /devices/pci0000:00/0000:00:1c.4/0000:02:00.0/0000:03:01.0/0000:05:00.0/0000:06:04.0/0000:08:00.0/usb3/3-1/3-1.4/3-1.4.4/3-1.4.4:1.0/rc/rc0
[ 5330.143120] rc rc0: lirc_dev: driver em28xx registered at minor = 0, scancode receiver, no transmitter
[ 5330.143159] input: Hauppauge WinTV-dualHD 01595 ATSC/QAM as /devices/pci0000:00/0000:00:1c.4/0000:02:00.0/0000:03:01.0/0000:05:00.0/0000:06:04.0/0000:08:00.0/usb3/3-1/3-1.4/3-1.4.4/3-1.4.4:1.0/rc/rc0/input74
[ 5330.143197] em28xx 3-1.4.4:1.0: Input extension successfully initialized
[ 5330.143198] em28xx 3-1.4.4:1.0: Remote control support is not available for this card.
```

parkerlreed, 2020-11-11:
Ok I realized it shows up as devinput 

```
/dev/input/event257:    Hauppauge WinTV-dualHD 01595 ATSC/QAM
```

I get no data checking that device but no errors as well. I may need to check if the remote needs new batteries. It shows some light through a camera but not very bright.

b-rad-NDi, 2020-11-11:
The IR worked the last time I tried it, but admittedly that was probably ages ago. I'll queue up testing it out.

parkerlreed, 2020-11-12:
I think the kernel side is fine. I just tried the remote with my Flirc USB receiver and it picks up nothing as well. I'll have to get out sometime and get a new battery.

parkerlreed, 2020-11-12:
Doh. It was just the battery

```
Select the device event number [0-262]: 29
Input driver version is 1.0.1
Input device ID: bus 0x3 vendor 0x2040 product 0x26d version 0x100
Input device name: "Hauppauge WinTV-dualHD 01595 ATSC/QAM"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 28 (KEY_ENTER)
    Event code 102 (KEY_HOME)
    Event code 103 (KEY_UP)
    Event code 105 (KEY_LEFT)
    Event code 106 (KEY_RIGHT)
    Event code 108 (KEY_DOWN)
    Event code 113 (KEY_MUTE)
    Event code 114 (KEY_VOLUMEDOWN)
    Event code 115 (KEY_VOLUMEUP)
    Event code 116 (KEY_POWER)
    Event code 119 (KEY_PAUSE)
    Event code 128 (KEY_STOP)
    Event code 139 (KEY_MENU)
    Event code 142 (KEY_SLEEP)
    Event code 163 (KEY_NEXTSONG)
    Event code 165 (KEY_PREVIOUSSONG)
    Event code 167 (KEY_RECORD)
    Event code 168 (KEY_REWIND)
    Event code 174 (KEY_EXIT)
    Event code 207 (KEY_PLAY)
    Event code 208 (KEY_FASTFORWARD)
    Event code 212 (KEY_CAMERA)
    Event code 352 (KEY_OK)
    Event code 353 (KEY_SELECT)
    Event code 354 (KEY_GOTO)
    Event code 356 (KEY_POWER2)
    Event code 365 (KEY_EPG)
    Event code 370 (KEY_SUBTITLE)
    Event code 372 (KEY_ZOOM)
    Event code 377 (KEY_TV)
    Event code 385 (KEY_RADIO)
    Event code 388 (KEY_TEXT)
    Event code 392 (KEY_AUDIO)
    Event code 393 (KEY_VIDEO)
    Event code 398 (KEY_RED)
    Event code 399 (KEY_GREEN)
    Event code 400 (KEY_YELLOW)
    Event code 401 (KEY_BLUE)
    Event code 402 (KEY_CHANNELUP)
    Event code 403 (KEY_CHANNELDOWN)
    Event code 405 (KEY_LAST)
    Event code 407 (KEY_NEXT)
    Event code 412 (KEY_PREVIOUS)
    Event code 512 (KEY_NUMERIC_0)
    Event code 513 (KEY_NUMERIC_1)
    Event code 514 (KEY_NUMERIC_2)
    Event code 515 (KEY_NUMERIC_3)
    Event code 516 (KEY_NUMERIC_4)
    Event code 517 (KEY_NUMERIC_5)
    Event code 518 (KEY_NUMERIC_6)
    Event code 519 (KEY_NUMERIC_7)
    Event code 520 (KEY_NUMERIC_8)
    Event code 521 (KEY_NUMERIC_9)
  Event type 2 (EV_REL)
    Event code 0 (REL_X)
    Event code 1 (REL_Y)
  Event type 4 (EV_MSC)
    Event code 4 (MSC_SCAN)
Key repeat handling:
  Repeat type 20 (EV_REP)
    Repeat code 0 (REP_DELAY)
      Value    500
    Repeat code 1 (REP_PERIOD)
      Value    125
Properties:
  Property type 5 (INPUT_PROP_POINTING_STICK)
Testing ... (interrupt to exit)
Event: time 1605206595.016593, type 4 (EV_MSC), code 4 (MSC_SCAN), value 1c01
Event: time 1605206595.016593, type 1 (EV_KEY), code 513 (KEY_NUMERIC_1), value 1
Event: time 1605206595.016593, -------------- SYN_REPORT ------------
Event: time 1605206595.119926, type 4 (EV_MSC), code 4 (MSC_SCAN), value 1c01
Event: time 1605206595.119926, -------------- SYN_REPORT ------------
Event: time 1605206595.366505, type 1 (EV_KEY), code 513 (KEY_NUMERIC_1), value 0
Event: time 1605206595.366505, -------------- SYN_REPORT ------------
Event: time 1605206595.636563, type 4 (EV_MSC), code 4 (MSC_SCAN), value 1c02
Event: time 1605206595.636563, type 1 (EV_KEY), code 514 (KEY_NUMERIC_2), value 1
Event: time 1605206595.636563, -------------- SYN_REPORT ------------
```

closed 2020-11-12 by parkerlreed