b-rad-NDi/Ubuntu-media-tree-kernel-builder #118: pctv 461e tuner/continuity problems on linux

psyborg55, 2020-12-01:
there are frequent problems using this card on linux, i've read previously some couldn't get it work at all...

1. it appears the driver sometimes doesn't properly initialize the tuner, in this case i can see signal level at about -82dBm - no stream can be watched in this case, regardless of the band/polarization used
disabling/re-enabling the card from tvheadend GUI or re-plugging it (once or several times) gets the tuner to initialize and the signal level of the same transponder previously measured gets to e.g. -25dBm - some streams can be watched in this case
one observation here: it seems the failure rate is much higher if the first transponder being tuned is on horizontal polarization

2. after tuner init as described above, some streams still cannot be watched, status page in tvheadend GUI displays stronger signal levels than expected (e.g. -3dBm, -10dBm) and continuity errors keeps increasing.
there is a nasty workaround to this problem: disable card, and very quickly after enabling it again hit the play button to tune stream immediately, then streaming works and no continuity errors appear. however switching to a stream on a different transponder the problem is back. all the transponders with this problem can be tuned fine with windows dvb apps.

i do have ubuntu 16.04 with kernel 5.6 and one of the latest tvheadend revisions. is there anyone that doesn't get such an issues and what distro are you using?
i tried experimenting with some tvheadend settings, like accept 0 value, or setting higher LNB voltage but these didn't help

b-rad-NDi, 2020-12-09:
Have you converted the device from ISOC transport to bulk using my eeprom-tinker tool?

psyborg55, 2020-12-09:
of course, that solved transport problems. these appear to be linux driver related

b-rad-NDi, 2020-12-10:
what is your vid:pid? If you're running the device on kernel 5.6 and not using my kernel I'm guessing you have one of the earlier 461e because the 0x461/0x259 pids are supported in 5.7+. I'll have to ask support if they've received any complaints, but they haven't mentioned any to me.

psyborg55, 2020-12-10:
2013:0258. probably people got tired of filling complaints or support got tired of dealing with them... you can usually find these on 3rd party software message boards, such as kodi, rpi etc.
even on wiki page is listed as "Knows issues" https://www.linuxtv.org/wiki/index.php/Pinnacle_PCTV_DVB-S2_Stick_(461e)

psyborg55, 2021-04-27:
After cold start, it is in fact impossible to use 461e if tuning channel on horizontal polarization first, plugged in small power adapter into wattmeter, it reads out 0.1W for a second when attempting to load channel and remains at 0W no matter what else I try. Need to unplug the card to be able to use it.
When loading channel on vertical polarization first, consumption goes to about 2W, and if switching to horizontal it is about 2.8W

psyborg55, 2021-12-23:
after installing recent driver tree and tuning utility from https://github.com/deeptho/blindscan i can tune both polarisations right after plugging in card.
so the problem was fixed meanwhile, in case it existed at all. if not then it was tvheadend issue and don't belong here.
closing this either way now

closed 2021-12-23 by psyborg55