b-rad-NDi/Ubuntu-media-tree-kernel-builder #119: QuadHD no signal after weeks of uptime Ubuntu 18.04

VoiletDragon, 2020-12-01:
Hi,

I have a issue with my QuadHD Tuner on Ubuntu 18.04 after 3 weeks of the Server being on i have a issue with Kodi and tvheadend saying No Signal on pretty much all of the channels some works and most dont and radio channels do not work neither, rebooting the server fixes this issue. This problem didn't start until a few months i originally thought the problem was motherboard related because after a reboot i just kept getting a scrambled screen when signal goes, changed the motherboard to a Intel S1200BTL been fine until now but problem has started again with no Signal after being on for 3 weeks or so. scrambled screen doesn't happen with this motherboard. Any ideas on this? I followed the guide on the hauppauge website. Cheers.

Saentist, 2020-12-01:
Is this happens on all 4 yuners or just two of them?

b-rad-NDi, 2020-12-09:
Have you tried forcing that pcie slot to gen1?

VoiletDragon, 2020-12-14:
> Is this happens on all 4 tuners or just two of them?

On all of them. I believe, dmesg | grep -i dvb indicates no issues.

 

> Have you tried forcing that pcie slot to gen1?

The Intel S1200BTL motherboard has no option in the BIOS for PCI-E Revisions neither does IPMI. When i had the Asus N3050M-E with the Tuner it was worst, but with this S1200 Server board it only happens every now and then.

shspvr, 2020-12-14:
> S1200BTL

Your motherboard is to old for the option as for your problem have try tested the tuner under Windows to be sure that it working I just install WinTV 8.5/10 with the key that come with and put in Four tuner picture-in-picture by right click in video windows do all 4 tuner display a working video ?.
[image]

VoiletDragon, 2020-12-14:
> > S1200BTL
> 
> Your motherboard is to old for the option as for your problem have try tested the tuner under Windows to be sure that it working you should just install WinTV 8.5/10 with the key that it come with and put in Four tuner picture-in-picture by right click in video windows do all 4 tuner display a working video ?.
> [image]

I will take a look i don't use windows, I only use Linux and FreeBSD but i will test on a machine running Windows 7 and see but i can record on each of the tuners, record 3 shows and watch live tv with Tvheadend which works great its just this issue i am facing with no Signal after so long. i have noticed that its got worst with Ubuntu 18.04 i was running 16.04 with no issues.

shspvr, 2020-12-14:
> I will take a look i don't use windows, I only use Linux and FreeBSD but i will test on a machine running Windows 7 and see but i can record on each of the tuners, record 3 shows and watch live tv [...]

That why it good idea test it to rule out a possible hardware issues

VoiletDragon, 2020-12-14:
> That why it good idea test it to rule out a possible hardware issues

Yeah leave it with me and i shall test when the Server is not being used. Cheers.

b-rad-NDi, 2020-12-15:
I'm going to add a module option to disable the analog capability of QuadHD that have analog. The reports I've reviewed lately seem to indicate it's the analog ports that are dying and taking out everything else.

Can you attach dmesg with the logs including the time of the failures? It will probably include opcode error messages.

VoiletDragon, 2020-12-16:
> I'm going to add a module option to disable the analog capability of QuadHD that have analog. The reports I've reviewed lately seem to indicate it's the analog ports that are dying and taking out [...]

Hi Brad,

Here is the output of dmesg with the Tuners working correctly. I cant seem to find what is causing this problem with the Tuners but when it happens. The Montage Technology M88RS6000 is a DVBSky S952 S/S2 Tuner i have removed this Tuner along with its Firmware but the problem still happens without. i will report with dmesg output,

`[    7.163399] tveeprom: Hauppauge model 166101, rev B4I6, serial# 4036075844
[    7.163401] tveeprom: MAC address is 00:0d:fe:91:a1:44
[    7.163402] tveeprom: tuner model is SiLabs Si2157 (idx 186, type 4)
[    7.163403] tveeprom: TV standards PAL(B/G) PAL(I) SECAM(L/L') PAL(D/D1/K) ATSC/DVB Digital (eeprom 0xf4)
[    7.163404] tveeprom: audio processor is CX23888 (idx 40)
[    7.163404] tveeprom: decoder processor is CX23888 (idx 34)
[    7.163404] tveeprom: has no radio
[    7.163405] cx23885: cx23885[1]: hauppauge eeprom: model=166101
[    7.165323] cx25840 12-0044: cx23888 A/V decoder found @ 0x88 (cx23885[1])
[    7.786305] cx25840 12-0044: loaded v4l-cx23885-avcore-01.fw firmware (16382 bytes)
[    7.824189] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[    7.864104] cx23885: cx23885[1]: registered device video1 [v4l2]
[    7.864132] cx23885: cx23885[1]: registered device vbi1
[    7.864204] cx23885: cx23885[1]: alsa: registered ALSA audio device
[    7.864205] cx23885: cx23885_dvb_register() allocating 1 frontend(s)
[    7.864206] cx23885: cx23885[1]: cx23885 based dvb card
[    7.864207] cx23885: dvb_register(): board=56 port=1
[    7.873365] i2c i2c-10: Added multiplexed i2c bus 13
[    7.873367] si2168 10-0064: Silicon Labs Si2168-B40 successfully identified
[    7.873368] si2168 10-0064: firmware version: B 4.0.2
[    7.874427] si2157 11-0060: Silicon Labs Si2147/2148/2157/2158 successfully attached
[    7.874440] cx23885: dvb_register(): QUADHD_DVB analog setup
[    7.874443] dvbdev: DVB: registering new adapter (cx23885[1])
[    7.874444] cx23885 0000:05:00.0: DVB: registering adapter 4 frontend 0 (Silicon Labs Si2168)...
[    7.874631] cx23885: cx23885_dvb_register() allocating 1 frontend(s)
[    7.874632] cx23885: cx23885[1]: cx23885 based dvb card
[    7.874633] cx23885: dvb_register(): board=56 port=2
[    7.883599] i2c i2c-10: Added multiplexed i2c bus 14
[    7.883600] si2168 10-0066: Silicon Labs Si2168-B40 successfully identified
[    7.883601] si2168 10-0066: firmware version: B 4.0.2
[    7.884611] si2157 11-0062: Silicon Labs Si2147/2148/2157/2158 successfully attached
[    7.884627] dvbdev: DVB: registering new adapter (cx23885[1])
[    7.884630] cx23885 0000:05:00.0: DVB: registering adapter 5 frontend 0 (Silicon Labs Si2168)...
[    7.885060] cx23885: cx23885_dev_checkrevision() Hardware revision = 0xd0
[    7.885065] cx23885: cx23885[1]/0: found at 0000:05:00.0, rev: 4, irq: 19, latency: 0, mmio: 0xc1a00000
[   11.874031] e1000e: eth0 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: None
[   11.874069] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   17.199293] FS-Cache: Loaded
[   17.210336] FS-Cache: Netfs 'nfs' registered for caching
[   17.221252] NFS: Registering the id_resolver key type
[   17.221259] Key type id_resolver registered
[   17.221260] Key type id_legacy registered
[   17.903054] new mount options do not match the existing superblock, will be ignored
[   18.592445] si2168 10-0066: downloading firmware from file 'dvb-demod-si2168-b40-01.fw'
[   19.198860] si2168 10-0066: firmware version: B 4.0.11
[   19.205102] si2157 11-0062: found a 'Silicon Labs Si2157-A30'
[   19.253680] si2157 11-0062: firmware version: 3.0.5
[   19.256787] cx23885 0000:05:00.0: DVB: adapter 5 frontend 0 frequency 0 out of range (48000000..870000000)
[   19.268274] si2168 10-0064: downloading firmware from file 'dvb-demod-si2168-b40-01.fw'
[   19.275317] si2157 11-0062: found a 'Silicon Labs Si2157-A30'
[   19.297644] si2157 11-0062: firmware version: 3.0.5
[   19.885677] si2168 10-0064: firmware version: B 4.0.11
[   19.891926] si2157 11-0060: found a 'Silicon Labs Si2157-A30'
[   19.940519] si2157 11-0060: firmware version: 3.0.5
[   19.943552] cx23885 0000:05:00.0: DVB: adapter 4 frontend 0 frequency 0 out of range (48000000..870000000)
[   19.952326] si2168 5-0066: downloading firmware from file 'dvb-demod-si2168-b40-01.fw'
[   19.958229] si2157 11-0060: found a 'Silicon Labs Si2157-A30'
[   19.980587] si2157 11-0060: firmware version: 3.0.5
[   20.560519] si2168 5-0066: firmware version: B 4.0.11
[   20.566765] si2157 6-0062: found a 'Silicon Labs Si2157-A30'
[   20.615362] si2157 6-0062: firmware version: 3.0.5
[   20.618514] cx23885 0000:04:00.0: DVB: adapter 3 frontend 0 frequency 0 out of range (48000000..870000000)
[   20.626174] si2168 5-0064: downloading firmware from file 'dvb-demod-si2168-b40-01.fw'
[   21.235764] si2168 5-0064: firmware version: B 4.0.11
[   21.242002] si2157 6-0060: found a 'Silicon Labs Si2157-A30'
[   21.290102] si2157 6-0060: firmware version: 3.0.5
[   21.293176] cx23885 0000:04:00.0: DVB: adapter 2 frontend 0 frequency 0 out of range (48000000..870000000)
[   21.305201] m88ds3103 2-0069: found a 'Montage Technology M88RS6000' in cold state
[   21.305694] m88ds3103 2-0069: downloading firmware from file 'dvb-demod-m88rs6000.fw'
[... cut ...]
[   23.918509] m88ds3103 1-0069: found a 'Montage Technology M88RS6000' in cold state
[   23.918569] m88ds3103 1-0069: downloading firmware from file 'dvb-demod-m88rs6000.fw'
[   26.371828] m88ds3103 1-0069: found a 'Montage Technology M88RS6000' in warm state
[   26.371831] m88ds3103 1-0069: firmware version: 4.1
[   26.477952] SMI PCIe driver 0000:01:00.0: DVB: adapter 0 frontend 0 frequency 0 out of range (950000..2150000)
[   43.726179] SGI XFS with ACLs, security attributes, realtime, no debug enabled
`
Cheers.

b-rad-NDi, 2020-12-16:
yikes. can you do something about the formatting of that?

VoiletDragon, 2020-12-16:
> yikes. can you do something about the formatting of that?

I used the code option then copied it.

shspvr, 2020-12-16:
LoL got Love Linux any way here you got Brad

[ 7.163399] tveeprom: Hauppauge model 166101, rev B4I6, serial# 4036075844
[ 7.163401] tveeprom: MAC address is 00:0d:fe:91:a1:44
[ 7.163402] tveeprom: tuner model is SiLabs Si2157 (idx 186, type 4)
[ 7.163403] tveeprom: TV standards PAL(B/G) PAL(I) SECAM(L/L') PAL(D/D1/K) ATSC/DVB Digital (eeprom 0xf4)
[ 7.163404] tveeprom: audio processor is CX23888 (idx 40)
[ 7.163404] tveeprom: decoder processor is CX23888 (idx 34)
[ 7.163404] tveeprom: has no radio
[ 7.163405] cx23885: cx23885[1]: hauppauge eeprom: model=166101
[ 7.165323] cx25840 12-0044: cx23888 A/V decoder found @ 0x88 (cx23885[1])
[ 7.786305] cx25840 12-0044: loaded v4l-cx23885-avcore-01.fw firmware (16382 bytes)
[ 7.824189] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[ 7.864104] cx23885: cx23885[1]: registered device video1 [v4l2]
[ 7.864132] cx23885: cx23885[1]: registered device vbi1
[ 7.864204] cx23885: cx23885[1]: alsa: registered ALSA audio device
[ 7.864205] cx23885: cx23885_dvb_register() allocating 1 frontend(s)
[ 7.864206] cx23885: cx23885[1]: cx23885 based dvb card
[ 7.864207] cx23885: dvb_register(): board=56 port=1
[ 7.873365] i2c i2c-10: Added multiplexed i2c bus 13
[ 7.873367] si2168 10-0064: Silicon Labs Si2168-B40 successfully identified
[ 7.873368] si2168 10-0064: firmware version: B 4.0.2
[ 7.874427] si2157 11-0060: Silicon Labs Si2147/2148/2157/2158 successfully attached
[ 7.874440] cx23885: dvb_register(): QUADHD_DVB analog setup
[ 7.874443] dvbdev: DVB: registering new adapter (cx23885[1])
[ 7.874444] cx23885 0000:05:00.0: DVB: registering adapter 4 frontend 0 (Silicon Labs Si2168)...
[ 7.874631] cx23885: cx23885_dvb_register() allocating 1 frontend(s)
[ 7.874632] cx23885: cx23885[1]: cx23885 based dvb card
[ 7.874633] cx23885: dvb_register(): board=56 port=2
[ 7.883599] i2c i2c-10: Added multiplexed i2c bus 14
[ 7.883600] si2168 10-0066: Silicon Labs Si2168-B40 successfully identified
[ 7.883601] si2168 10-0066: firmware version: B 4.0.2
[ 7.884611] si2157 11-0062: Silicon Labs Si2147/2148/2157/2158 successfully attached
[ 7.884627] dvbdev: DVB: registering new adapter (cx23885[1])
[ 7.884630] cx23885 0000:05:00.0: DVB: registering adapter 5 frontend 0 (Silicon Labs Si2168)...
[ 7.885060] cx23885: cx23885_dev_checkrevision() Hardware revision = 0xd0
[ 7.885065] cx23885: cx23885[1]/0: found at 0000:05:00.0, rev: 4, irq: 19, latency: 0, mmio: 0xc1a00000
[ 11.874031] e1000e: eth0 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: None
[ 11.874069] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[ 17.199293] FS-Cache: Loaded
[ 17.210336] FS-Cache: Netfs 'nfs' registered for caching
[ 17.221252] NFS: Registering the id_resolver key type
[ 17.221259] Key type id_resolver registered
[ 17.221260] Key type id_legacy registered
[ 17.903054] new mount options do not match the existing superblock, will be ignored
[ 18.592445] si2168 10-0066: downloading firmware from file 'dvb-demod-si2168-b40-01.fw'
[ 19.198860] si2168 10-0066: firmware version: B 4.0.11
[ 19.205102] si2157 11-0062: found a 'Silicon Labs Si2157-A30'
[ 19.253680] si2157 11-0062: firmware version: 3.0.5
[ 19.256787] cx23885 0000:05:00.0: DVB: adapter 5 frontend 0 frequency 0 out of range (48000000..870000000)
[ 19.268274] si2168 10-0064: downloading firmware from file 'dvb-demod-si2168-b40-01.fw'
[ 19.275317] si2157 11-0062: found a 'Silicon Labs Si2157-A30'
[ 19.297644] si2157 11-0062: firmware version: 3.0.5
[ 19.885677] si2168 10-0064: firmware version: B 4.0.11
[ 19.891926] si2157 11-0060: found a 'Silicon Labs Si2157-A30'
[ 19.940519] si2157 11-0060: firmware version: 3.0.5
[ 19.943552] cx23885 0000:05:00.0: DVB: adapter 4 frontend 0 frequency 0 out of range (48000000..870000000)
[ 19.952326] si2168 5-0066: downloading firmware from file 'dvb-demod-si2168-b40-01.fw'
[ 19.958229] si2157 11-0060: found a 'Silicon Labs Si2157-A30'
[ 19.980587] si2157 11-0060: firmware version: 3.0.5
[ 20.560519] si2168 5-0066: firmware version: B 4.0.11
[ 20.566765] si2157 6-0062: found a 'Silicon Labs Si2157-A30'
[ 20.615362] si2157 6-0062: firmware version: 3.0.5
[ 20.618514] cx23885 0000:04:00.0: DVB: adapter 3 frontend 0 frequency 0 out of range (48000000..870000000)
[ 20.626174] si2168 5-0064: downloading firmware from file 'dvb-demod-si2168-b40-01.fw'
[ 21.235764] si2168 5-0064: firmware version: B 4.0.11
[ 21.242002] si2157 6-0060: found a 'Silicon Labs Si2157-A30'
[ 21.290102] si2157 6-0060: firmware version: 3.0.5
[ 21.293176] cx23885 0000:04:00.0: DVB: adapter 2 frontend 0 frequency 0 out of range (48000000..870000000)
[ 21.305201] m88ds3103 2-0069: found a 'Montage Technology M88RS6000' in cold state
[ 21.305694] m88ds3103 2-0069: downloading firmware from file 'dvb-demod-m88rs6000.fw'
[ 21.307905] si2157 6-0060: found a 'Silicon Labs Si2157-A30'
[ 21.330321] si2157 6-0060: firmware version: 3.0.5
[ 23.781086] m88ds3103 2-0069: found a 'Montage Technology M88RS6000' in warm state
[ 23.781088] m88ds3103 2-0069: firmware version: 4.1
[ 23.886131] SMI PCIe driver 0000:01:00.0: DVB: adapter 1 frontend 0 frequency 0 out of range (950000..2150000)
[ 23.918509] m88ds3103 1-0069: found a 'Montage Technology M88RS6000' in cold state
[ 23.918569] m88ds3103 1-0069: downloading firmware from file 'dvb-demod-m88rs6000.fw'
[ 26.371828] m88ds3103 1-0069: found a 'Montage Technology M88RS6000' in warm state
[ 26.371831] m88ds3103 1-0069: firmware version: 4.1
[ 26.477952] SMI PCIe driver 0000:01:00.0: DVB: adapter 0 frontend 0 frequency 0 out of range (950000..2150000)
[ 43.726179] SGI XFS with ACLs, security attributes, realtime, no debug enabled

b-rad-NDi, 2020-12-16:
@VoiletDragon : That is just the startup logs. If you can let the quadhd run until it fails then get me any dmesg logs around that time it would be ideal. I want to see what errors are displayed in regards to the failure.

VoiletDragon, 2020-12-16:
> @VoiletDragon : That is just the startup logs. If you can let the quadhd run until it fails then get me any dmesg logs around that time it would be ideal. I want to see what errors are displayed in regards to the failure.

Yeah sure i can do that. Not sure if this is related but i installed MiniDLNA Server alongside tvheadend for a DAB Radio but at the time of configuring that months after thats when this problem occurred.

b-rad-NDi, 2020-12-18:
There's a module option that exists currently to disable analog audio. It seems most of the issues I've seen are due to the audio, since desktop audio mixers just automatically keep them open to mix in with the system when they should only be opened during captures.

So, you can try and set this module option in /etc/modprobe.d/linuxtv.conf:
```
options cx23885 disable_analog_audio=1
```
Then reboot.

To disable analog video would require a lot of ugly patching in small spots

VoiletDragon, 2020-12-18:
> There's a module option that exists currently to disable analog audio. It seems most of the issues I've seen are due to the audio, since desktop audio mixers just automatically keep them open to mix [...]

Hi thanks for your reply. I will give that ago my Intel Server Board doesn't have onboard Audio and im using Ubuntu Server headless and control it over IPMI i guess i need to disable that. I shall give that ago 

Cheers.