b-rad-NDi/Ubuntu-media-tree-kernel-builder #120: Quad HD no signal after OS upgrade

tommycw1, 2020-12-22:
I've been using a QuadHD tuner on Ubuntu 16.04 LTS for about 1.5 years as a tuner in a MythTV backed.

I recently had a mobo die so I put in another old mobo and installed a fresh copy of Ubuntu 20.04 LTS onto the box

Here is my kernel:

```
tom@mythfe:~$ uname -a
Linux mythfe 5.4.0-58-generic #64-Ubuntu SMP Wed Dec 9 08:16:25 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
```

The OS sees the hardware on the PCI bus:
```
05:00.0 Multimedia video controller: Conexant Systems, Inc. CX23887/8 PCIe Broadcast Audio and Video Decoder with 3D Comb (rev 04)
06:00.0 Multimedia video controller: Conexant Systems, Inc. CX23887/8 PCIe Broadcast Audio and Video Decoder with 3D Comb (rev 04)
```

The OS creates all 4 DVB devs:
```
tom@mythfe:~$ ls -al /dev/dvb/
total 0
drwxr-xr-x  6 root root  120 Dec 22 17:37 .
drwxr-xr-x 20 root root 4480 Dec 22 17:38 ..
drwxr-xr-x  2 root root  120 Dec 22 17:37 adapter0
drwxr-xr-x  2 root root  120 Dec 22 17:37 adapter1
drwxr-xr-x  2 root root  120 Dec 22 17:37 adapter2
drwxr-xr-x  2 root root  120 Dec 22 17:37 adapter3
```
and the OS loads the appropriate module:

```
tom@mythfe:~$ dmesg | grep cx238
[   21.717375] cx23885: cx23885 driver version 0.0.4 loaded
[   21.717583] cx23885: CORE cx23885[0]: subsystem: 0070:6a18, board: Hauppauge WinTV-QuadHD-ATSC [card=57,autodetected]
[   22.073454] cx23885: cx23885[0]: hauppauge eeprom: model=165100
[   22.222952] cx25840 7-0044: cx23888 A/V decoder found @ 0x88 (cx23885[0])
[   22.911591] cx25840 7-0044: loaded v4l-cx23885-avcore-01.fw firmware (16382 bytes)
[   22.947144] cx23885: cx23885_dvb_register() allocating 1 frontend(s)
[   22.947146] cx23885: cx23885[0]: cx23885 based dvb card
[   22.947147] cx23885: dvb_register(): board=57 port=1
[   23.128451] dvbdev: DVB: registering new adapter (cx23885[0])
[   23.128455] cx23885 0000:05:00.0: DVB: registering adapter 0 frontend 0 (LG Electronics LGDT3306A VSB/QAM Frontend)...
[   23.129827] cx23885: cx23885_dvb_register() allocating 1 frontend(s)
[   23.129829] cx23885: cx23885[0]: cx23885 based dvb card
[   23.129831] cx23885: dvb_register(): board=57 port=2
[   23.137531] dvbdev: DVB: registering new adapter (cx23885[0])
[   23.137534] cx23885 0000:05:00.0: DVB: registering adapter 1 frontend 0 (LG Electronics LGDT3306A VSB/QAM Frontend)...
[   23.138081] cx23885: cx23885_dev_checkrevision() Hardware revision = 0xd0
[   23.138085] cx23885: cx23885[0]/0: found at 0000:05:00.0, rev: 4, irq: 38, latency: 0, mmio: 0xf7400000
[   23.138314] cx23885: CORE cx23885[1]: subsystem: 0070:6b18, board: Hauppauge WinTV-QuadHD-ATSC [card=57,autodetected]
[   23.493163] cx23885: cx23885[1]: hauppauge eeprom: model=165101
[   23.515523] cx25840 14-0044: cx23888 A/V decoder found @ 0x88 (cx23885[1])
[   24.143101] cx25840 14-0044: loaded v4l-cx23885-avcore-01.fw firmware (16382 bytes)
[   24.175360] cx23885: cx23885_dvb_register() allocating 1 frontend(s)
[   24.175362] cx23885: cx23885[1]: cx23885 based dvb card
[   24.175364] cx23885: dvb_register(): board=57 port=1
[   24.183056] dvbdev: DVB: registering new adapter (cx23885[1])
[   24.183059] cx23885 0000:06:00.0: DVB: registering adapter 2 frontend 0 (LG Electronics LGDT3306A VSB/QAM Frontend)...
[   24.183563] cx23885: cx23885_dvb_register() allocating 1 frontend(s)
[   24.183564] cx23885: cx23885[1]: cx23885 based dvb card
[   24.183566] cx23885: dvb_register(): board=57 port=2
[   24.191197] dvbdev: DVB: registering new adapter (cx23885[1])
[   24.191200] cx23885 0000:06:00.0: DVB: registering adapter 3 frontend 0 (LG Electronics LGDT3306A VSB/QAM Frontend)...
[   24.191674] cx23885: cx23885_dev_checkrevision() Hardware revision = 0xd0
[   24.191679] cx23885: cx23885[1]/0: found at 0000:06:00.0, rev: 4, irq: 34, latency: 0, mmio: 0xf7200000
```

I had started seeing issues that appeared that the OS was intermittentally seeing the device, losing contact and findign again over and over.  I thought that maybe when my mobo died this card was also damaged so I contacted Hauppague tech support and they told me to send in my card.  I sent my card into Hauppauge under warranty they ran it, say it is fine and returned it to me.

I received from tech support this:
[... 68 lines cut ...]
[  351.205203] cx23885: cx23885[1]:   iq 9: 0x00000000 [ arg #2 ]
[  351.205207] cx23885: cx23885[1]:   (0x00010658) iq a: 
[  351.205207] 0x1c0002f0 [ write sol eol count=752 ]
[  351.205212] cx23885: cx23885[1]:   iq b: 0xffeda630 [ arg #1 ]
[  351.205216] cx23885: cx23885[1]:   iq c: 0x00000000 [ arg #2 ]
[  351.205219] cx23885: cx23885[1]:   (0x00010664) iq d: 
[  351.205220] 0x1c0002f0 [ write sol eol count=752 ]
[  351.205225] cx23885: cx23885[1]:   iq e: 0xffeda920 [ arg #1 ]
[  351.205228] cx23885: cx23885[1]:   iq f: 0x00000000 [ arg #2 ]
[  351.205230] cx23885: cx23885[1]: fifo: 0x00005000 -> 0x6000
[  351.205232] cx23885: cx23885[1]: ctrl: 0x00010630 -> 0x10690
[  351.205235] cx23885: cx23885[1]:   ptr1_reg: 0x00005700
[  351.205238] cx23885: cx23885[1]:   ptr2_reg: 0x00010898
[  351.205241] cx23885: cx23885[1]:   cnt1_reg: 0x00000012
[  351.205244] cx23885: cx23885[1]:   cnt2_reg: 0x00000005

```
Next I added this repo and installed the kernel:
```
tom@mythfe:~$ uname -a
Linux mythfe 5.4.0-512010201020-generic #0+mediatree+hauppauge-Ubuntu SMP Wed Oct 21 18:57:14 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
```
I have verified I still have the same issues, I cannot tune to any channel on any of the 4 QuadHD tuners and get the same dmesg errors.

Is there any other recommendations for troubleshooting?

shspvr, 2020-12-23:
What Motherboard do you have tommycw1  ?.

tommycw1, 2020-12-23:
Hi shspvr

```
tom@mythfe:~$ sudo dmidecode | grep -A3 "type 2"
Handle 0x0002, DMI type 2, 15 bytes
Base Board Information
	Manufacturer: Gigabyte Technology Co., Ltd.
	Product Name: 970A-UD3P
```

shspvr, 2020-12-23:
You don't need to in if in the small 1x slot

shspvr, 2020-12-23:
Now as for motherboard it is only 2.0 standard you need more up to date motherboard with 3.0 standard in order to have that option gen mode option and it only apply to long 16x Slot or 8x Half Slot

tommycw1, 2020-12-23:
@shspvr  - Thanks - makes sense

I've been googling around about this and found several other similar issues like this:

#17 - this seems like the first discovery of this issue and it was suggsted that using km option of 
`options cx23885 debug=7`

or

`options cx23885 debug=6`

May be workarounds for this issue.

Later, #51 & #69 reported similar issues - in these thread it seems people are instructing to use a km option of 
`options cx23885 dma_reset_workaround=2`

These are a couple years old though so I'm not sure if this is still valid on a 5.4 kernel

@b-rad-NDi  - can you confirm if any of these options are still valid for the 5.4 Ubuntu 20.04 kernel?

tommycw1, 2020-12-23:
@b-rad-NDi  

I was able to confirm that the dma_reset_workaround option was still a valid option in the kernel module for 5.4 by listing all of the available parameters with the following:

```
tom@mythfe:~$ modinfo cx23885 | grep parm
parm:           disable_analog_audio:disable analog audio ALSA driver (int)
parm:           audio_debug:enable debug messages [analog audio] (int)
parm:           ci_dbg:Enable CI debugging (int)
parm:           ci_irq_enable:Enable IRQ from CAM (int)
parm:           ir_888_debug:enable debug messages [CX23888 IR controller] (int)
parm:           mpegbufs:number of mpeg buffers, range 2-32 (int)
parm:           mpeglines:number of lines in an MPEG buffer, range 2-32 (int)
parm:           mpeglinesize:number of bytes in each line of an MPEG buffer, range 512-1024 (int)
parm:           v4l_debug:enable V4L debug messages (int)
parm:           alt_tuner:Enable alternate tuner configuration (int)
parm:           adapter_nr:DVB adapter numbers (array of short)
parm:           i2c_debug:enable debug messages [i2c] (int)
parm:           i2c_scan:scan i2c bus at insmod time (int)
parm:           dma_reset_workaround:periodic RiSC dma engine reset; 0-force disable, 1-driver detect (default), 2-force enable (int)
parm:           debug:enable debug messages (int)
parm:           card:card type (array of int)
parm:           vbibufs:number of vbi buffers, range 2-32 (int)
parm:           vbi_debug:enable debug messages [vbi] (int)
parm:           video_nr:video device numbers (array of int)
parm:           vbi_nr:vbi device numbers (array of int)
parm:           video_debug:enable debug messages [video] (int)
parm:           irq_debug:enable debug messages [IRQ handler] (int)
parm:           vid_limit:capture memory limit in megabytes (int)
parm:           netup_card_rev:NetUP Dual DVB-T/C CI card revision (int)
parm:           enable_885_ir:Enable integrated IR controller for supported
```

So I added:

`options cx23885 dma_reset_workaround=2`

to /etc/modprobe.d/cx23885.conf.  I rebooted the computer and I now have a working QuadHD card!

I do however get a slew of issues where it looks like the card is lost and rediscovered over and over:

```
[  154.524682] si2157 6-0060: found a 'Silicon Labs Si2157-A30'
[  154.573205] si2157 6-0060: firmware version: 3.0.5
[  154.665225] si2157 6-0060: found a 'Silicon Labs Si2157-A30'
[  154.687644] si2157 6-0060: firmware version: 3.0.5
[  154.785252] si2157 6-0060: found a 'Silicon Labs Si2157-A30'
[  154.807912] si2157 6-0060: firmware version: 3.0.5
[  155.768693] si2157 6-0062: found a 'Silicon Labs Si2157-A30'
[  155.817212] si2157 6-0062: firmware version: 3.0.5
[  155.909265] si2157 6-0062: found a 'Silicon Labs Si2157-A30'
[  155.931890] si2157 6-0062: firmware version: 3.0.5
[  156.029259] si2157 6-0062: found a 'Silicon Labs Si2157-A30'
[  156.051920] si2157 6-0062: firmware version: 3.0.5
[  156.845298] si2157 6-0060: found a 'Silicon Labs Si2157-A30'
[  156.867819] si2157 6-0060: firmware version: 3.0.5
[  156.965254] si2157 6-0060: found a 'Silicon Labs Si2157-A30'
[  156.987844] si2157 6-0060: firmware version: 3.0.5
[  158.817355] si2157 6-0060: found a 'Silicon Labs Si2157-A30'
[  158.839767] si2157 6-0060: firmware version: 3.0.5
[  158.937235] si2157 6-0060: found a 'Silicon Labs Si2157-A30'
[  158.959796] si2157 6-0060: firmware version: 3.0.5
[  160.748690] si2157 13-0060: found a 'Silicon Labs Si2157-A30'
[  160.798111] si2157 13-0060: firmware version: 3.0.5
[  160.889260] si2157 13-0060: found a 'Silicon Labs Si2157-A30'
[  160.911574] si2157 13-0060: firmware version: 3.0.5
[  161.004750] si2157 13-0062: found a 'Silicon Labs Si2157-A30'
[  161.053969] si2157 13-0062: firmware version: 3.0.5
[  161.145387] si2157 13-0062: found a 'Silicon Labs Si2157-A30'
[  161.167908] si2157 13-0062: firmware version: 3.0.5
[... 33 lines cut ...]
[ 3130.702826] si2157 13-0060: firmware version: 3.0.5
[ 3130.796301] si2157 13-0060: found a 'Silicon Labs Si2157-A30'
[ 3130.818960] si2157 13-0060: firmware version: 3.0.5
[ 3130.912336] si2157 13-0060: found a 'Silicon Labs Si2157-A30'
[ 3130.934863] si2157 13-0060: firmware version: 3.0.5
[ 3131.028304] si2157 13-0060: found a 'Silicon Labs Si2157-A30'
[ 3131.050962] si2157 13-0060: firmware version: 3.0.5
[ 4758.586589] si2157 13-0062: found a 'Silicon Labs Si2157-A30'
[ 4758.608939] si2157 13-0062: firmware version: 3.0.5
[ 4758.702631] si2157 13-0062: found a 'Silicon Labs Si2157-A30'
[ 4758.725073] si2157 13-0062: firmware version: 3.0.5
[ 4763.610702] si2157 13-0062: found a 'Silicon Labs Si2157-A30'
[ 4763.633082] si2157 13-0062: firmware version: 3.0.5
[ 4763.726808] si2157 13-0062: found a 'Silicon Labs Si2157-A30'
[ 4763.749226] si2157 13-0062: firmware version: 3.0.5
[ 4763.842738] si2157 13-0062: found a 'Silicon Labs Si2157-A30'
[ 4763.865124] si2157 13-0062: firmware version: 3.0.5
[ 4763.958705] si2157 13-0062: found a 'Silicon Labs Si2157-A30'
[ 4763.981021] si2157 13-0062: firmware version: 3.0.5
```

Wondering two things:  

1. Since this is called a "workaround" is there a plan to do some type of other fix in the future or does this solve the original issue?
2. Should this issue and its solution be added to the documentation somewhere?  I found it only after a bunch of effort to understand the issue and then found similar solutions.

b-rad-NDi, 2020-12-24:
These messages:
```
Dec 23 16:13:39 mythfe kernel: [ 6241.798482] si2157 6-0060: found a 'Silicon Labs Si2157-A30'
Dec 23 16:13:39 mythfe kernel: [ 6241.820899] si2157 6-0060: firmware version: 3.0.5
```
Are red herrings, they are output every single time a card is opened. It really signifies nothing. It's a bad design decision to have done it that way, but it wasn't mine.

I have added a couple new iommu pci id's to my build. If yours is not one of them you'll have to figure out what it is and then I can add it to the list of cpu's the fix is applied to by default.

tommycw1, 2020-12-24:
Thanks for this @b-rad-NDi - glad to know this is a non-event.  Looks like the si2157 module doesn't have any options to turn this off unfortunately...

```
tom@mythfe:~$ modinfo si2157 | grep parm
parm:           tuner_lock_debug:if set, signal lock is briefly waited on after setting params (int)
```
Oh well - nbd I guess.

I poked around your github commits quickly and didn't see anything that jumped out at me about IOMMUs, so not sure where to look, but here is mine:
```
tom@mythfe:~$ sudo dmidecode | grep -A3 "type 2"
Handle 0x0002, DMI type 2, 15 bytes
Base Board Information
	Manufacturer: Gigabyte Technology Co., Ltd.
	Product Name: 970A-UD3P

tom@mythfe:~$ lspci | grep IOMMU
00:00.2 IOMMU: Advanced Micro Devices, Inc. [AMD/ATI] RD890S/RD990 I/O Memory Management Unit (IOMMU)
```

So to make sure I'm clear on this, I changed motherboards and OSs at the same time.  It sounds like my issue wasn't related to the OS change from Ubuntu 16.04->20.04 but related to my mobo change and it's IOMMU.  Is that correct?

b-rad-NDi, 2020-12-24:
This is almost exclusively an AMD issue. If you look in cx23885-core.c you'll see the list of current id's. I can't recall offhand how to find them, but I don't think you get it from lspci.

Newer Intel mobos can have issues, but they are fixed by forcing the pcie slot to gen1 mode.

tommycw1, 2020-12-24:
@b-rad-NDi - OK, yes, I see in the code now here:

https://github.com/torvalds/linux/blob/58cf05f597b03a8212d9ecf2c79ee046d3ee8ad9/drivers/media/pci/cx23885/cx23885-core.c#L2068

Interestingly it talks about this being an issue with Ryzen, but I'm running an older mobo/processor.  My media server PC that I have this card installed in is using this processor:

```
tom@mythfe:~$ cat /proc/cpuinfo | grep "model name"
model name	: AMD FX(tm)-6300 Six-Core Processor
```
on this mobo:
```
tom@mythfe:~$ sudo dmidecode |grep -i product
	Product Name: 970A-UD3P
```

Anyway, my desktop (not the box with this card installed) happens to be Ryzen 7 and when I look at that IOMMU, I see the same vendor ID number as one of the two refered to in cx23885-core.c **0x1451**

```
tom@orbital:~$ lspci -nn | grep IOMMU
00:00.2 IOMMU [0806]: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) I/O Memory Management Unit [1022:1451]
```

Doing the same in the box this card IS installed in shows a very different ID

```
tom@mythfe:~$ sudo lspci -nn | grep IOMMU
00:00.2 IOMMU [0806]: Advanced Micro Devices, Inc. [AMD/ATI] RD890S/RD990 I/O Memory Management Unit (IOMMU) [1002:5a23]
```

So, I believe the ID that needs to be added is 0x5a23

shspvr, 2020-12-25:
Is the SVM and IOMMU Controller disable or enable it not need unless your running a virtual machines it be the source of your problem.
Besure Cool & Quiet disable

b-rad-NDi, 2021-01-04:
@tommycw1 : I'll add this to my builds and submit it for inclusion upstream.

sergiud, 2021-06-24:
I found this issue while trying to run a Hauppauge QuadHD on Debian Buster with Linux kernel 5.10 on an AMD Ryzen system. I can confirm that using `options cx23885 dma_reset_workaround=2` allowed me to resolve the no signal issue.

sergiud, 2021-07-02:
Unfortunately, `options cx23885 dma_reset_workaround=2` still causes `mpeg risc op code error` messages. Is this expected?
<details>

```
[    7.484802] si2168 14-0066: firmware version: B 4.0.11
[    7.491967] si2157 15-0062: found a 'Silicon Labs Si2157-A30'
[    7.514281] si2157 15-0062: firmware version: 3.0.5
[    7.517424] cx23885 0000:29:00.0: DVB: adapter 3 frontend 0 frequency 0 out of range (48000000..870000000)
[    7.526765] si2168 14-0064: firmware: direct-loading firmware dvb-demod-si2168-b40-01.fw
[    7.526769] si2168 14-0064: downloading firmware from file 'dvb-demod-si2168-b40-01.fw'
[    7.534035] si2157 15-0062: found a 'Silicon Labs Si2157-A30'
[    7.556396] si2157 15-0062: firmware version: 3.0.5
[    8.019490] r8169 0000:25:00.0 enp37s0: Link is Up - 1Gbps/Full - flow control rx/tx
[    8.019500] IPv6: ADDRCONF(NETDEV_CHANGE): enp37s0: link becomes ready
[    8.158381] si2168 14-0064: firmware version: B 4.0.11
[    8.165536] si2157 15-0060: found a 'Silicon Labs Si2157-A30'
[    8.187817] si2157 15-0060: firmware version: 3.0.5
[    8.190934] cx23885 0000:29:00.0: DVB: adapter 2 frontend 0 frequency 0 out of range (48000000..870000000)
[    8.197559] si2168 3-0066: firmware: direct-loading firmware dvb-demod-si2168-b40-01.fw
[    8.197562] si2168 3-0066: downloading firmware from file 'dvb-demod-si2168-b40-01.fw'
[    8.203288] si2157 15-0060: found a 'Silicon Labs Si2157-A30'
[    8.225551] si2157 15-0060: firmware version: 3.0.5
[    8.830289] si2168 3-0066: firmware version: B 4.0.11
[    8.837442] si2157 4-0062: found a 'Silicon Labs Si2157-A30'
[    8.859718] si2157 4-0062: firmware version: 3.0.5
[    8.862868] cx23885 0000:28:00.0: DVB: adapter 1 frontend 0 frequency 0 out of range (48000000..870000000)
[    8.872377] si2168 3-0064: firmware: direct-loading firmware dvb-demod-si2168-b40-01.fw
[    8.872383] si2168 3-0064: downloading firmware from file 'dvb-demod-si2168-b40-01.fw'
[    8.879749] si2157 4-0062: found a 'Silicon Labs Si2157-A30'
[    8.902034] si2157 4-0062: firmware version: 3.0.5
[    9.503910] si2168 3-0064: firmware version: B 4.0.11
[    9.511072] si2157 4-0060: found a 'Silicon Labs Si2157-A30'
[    9.533362] si2157 4-0060: firmware version: 3.0.5
[    9.536508] cx23885 0000:28:00.0: DVB: adapter 0 frontend 0 frequency 0 out of range (48000000..870000000)
[    9.548750] si2157 4-0060: found a 'Silicon Labs Si2157-A30'
[    9.571049] si2157 4-0060: firmware version: 3.0.5
[ 2128.566454] cx23885 0000:29:00.0: dma in progress detected 0x00000001 0x00000001, clearing
[ 2627.963214] cx23885 0000:29:00.0: dma in progress detected 0x00000001 0x00000001, clearing
[ 3075.105414] cx23885 0000:29:00.0: dma in progress detected 0x00000001 0x00000001, clearing
[ 3505.380351] cx23885 0000:29:00.0: dma in progress detected 0x00000001 0x00000001, clearing
[ 3845.362959] cx23885 0000:29:00.0: dma in progress detected 0x00000001 0x00000001, clearing
[ 5102.916345] cx23885 0000:29:00.0: dma in progress detected 0x00000001 0x00000001, clearing
[ 5581.059497] cx23885 0000:29:00.0: dma in progress detected 0x00000001 0x00000001, clearing
[ 6869.616838] cx23885 0000:28:00.0: dma in progress detected 0x00000001 0x00000001, clearing
[ 7862.831238] cx23885 0000:28:00.0: dma in progress detected 0x00000001 0x00000001, clearing
[ 8469.112658] cx23885 0000:28:00.0: dma in progress detected 0x00000001 0x00000001, clearing
[ 8534.455189] cx23885 0000:28:00.0: dma in progress detected 0x00000001 0x00000001, clearing
[ 8945.308730] cx23885 0000:28:00.0: dma in progress detected 0x00000001 0x00000001, clearing
[ 9895.729454] cx23885 0000:28:00.0: dma in progress detected 0x00000001 0x00000001, clearing
[10158.245325] cx23885 0000:28:00.0: dma in progress detected 0x00000001 0x00000001, clearing
[10158.245383] cx23885: cx23885[0]: mpeg risc op code error
[10158.245393] cx23885: cx23885[0]: TS1 B - dma channel status dump
[10158.245398] cx23885: cx23885[0]:   cmds: init risc lo   : 0x358f8000
[10158.245402] cx23885: cx23885[0]:   cmds: init risc hi   : 0x00000000
[10158.245406] cx23885: cx23885[0]:   cmds: cdt base       : 0x00010870
[10158.245410] cx23885: cx23885[0]:   cmds: cdt size       : 0x0000000a
[10158.245413] cx23885: cx23885[0]:   cmds: iq base        : 0x00010630
[10158.245417] cx23885: cx23885[0]:   cmds: iq size        : 0x00000010
[10158.245423] cx23885: cx23885[0]:   cmds: risc pc lo     : 0x358f8018
[10158.245427] cx23885: cx23885[0]:   cmds: risc pc hi     : 0x00000000
[10158.245431] cx23885: cx23885[0]:   cmds: iq wr ptr      : 0x00004192
[10158.245434] cx23885: cx23885[0]:   cmds: iq rd ptr      : 0x0000418c
[10158.245438] cx23885: cx23885[0]:   cmds: cdt current    : 0x00010878
[10158.245442] cx23885: cx23885[0]:   cmds: pci target lo  : 0x0a226050
[10158.245445] cx23885: cx23885[0]:   cmds: pci target hi  : 0x00000000
[10158.245449] cx23885: cx23885[0]:   cmds: line / byte    : 0x000b0000
[10158.245453] cx23885: cx23885[0]:   risc0: 
[10158.245455] 0x1c0002f0 [ write sol eol count=752 ]
[10158.245461] cx23885: cx23885[0]:   risc1: 
[10158.245462] 0x0a225d60 [ INVALID sol irq2 21 cnt1 14 12 count=3424 ]
[... 17 lines cut ...]
[10158.245527] 0x1c0002f0 [ write sol eol count=752 ]
[10158.245533] cx23885: cx23885[0]:   iq a: 0x0a225780 [ arg #1 ]
[10158.245536] cx23885: cx23885[0]:   iq b: 0x00000000 [ arg #2 ]
[10158.245541] cx23885: cx23885[0]:   (0x00010660) iq c: 
[10158.245542] 0x1c0002f0 [ write sol eol count=752 ]
[... cut ...]
[10158.245565] cx23885: cx23885[0]:   iq 11: 0x1c0002f0 [ arg #2 ]
[10158.245567] cx23885: cx23885[0]: fifo: 0x00005000 -> 0x6000
[10158.245568] cx23885: cx23885[0]: ctrl: 0x00010630 -> 0x10690
[10158.245572] cx23885: cx23885[0]:   ptr1_reg: 0x00005480
[10158.245576] cx23885: cx23885[0]:   ptr2_reg: 0x00010888
[10158.245579] cx23885: cx23885[0]:   cnt1_reg: 0x00000019
[10158.245583] cx23885: cx23885[0]:   cnt2_reg: 0x00000007
```

</details>

```bash
$ uname -a
Linux pony 5.10.0-0.bpo.7-amd64 #1 SMP Debian 5.10.40-1~bpo10+1 (2021-06-04) x86_64 GNU/Linux
```

sotiris-bos, 2021-07-14:
I am having issues with this card on two systems:

1. AMD Ryzen Threadripper 2920x, Arch Linux 5.12.13-arch1-2
2. Dell R720xd (Intel E5-2650 v2), Arch Linux 5.10.47-1-lts

The AMD system posts:

```
Audio risc op code error
```

while the Intel system posts:

```
mpeg risc op code error
```

IOMMU is enabled on both systems as it is required for PCIe passthrough of devices to VMs.

On the other hand, I am getting no problems with Ubuntu server 20.04.2 LTS 5.4.0-77-generic on an HP T620 Plus thin client running an AMD GX-420CA SOC, no IOMMU enabled by me. I believe my issues are related to IOMMU. I will try on the Ryzen system with amd_iommu=off and report back, unfortunately the Intel system is a production machine so I cannot afford much downtime.

NatVac, 2021-08-15:
I'm posting here because my issue is stated in the title. I'm dual-booting Ubuntu 16.04/kernel 4.15 (Linux Mint 18.2) and 20.04/kernel 5.4 (Linux Mint 20.2) on the same Intel-based hardware. The Hauppauge WinTV-QuadHD-ATSC (card 57) is recognized on both systems.

The QuadHD card works _just fine_ on the 16.04 system with Tvheadend and Kodi, but Tvheadend says "Scan - no data" with the registered cx23885 devices on the 20.04 OS with kernel 5.4.

My processor is an Intel Haswell model i5-4690S quad-core CPU @3.20 GHz so the AMD Ryzen workarounds don't apply. I confess to trying them anyway. There's no way to set the gen mode on the PCI-e bus in the BIOS. There's no IOMMU found with `lspci -nn | grep IOMMU`.

I was perfectly happy with the very stable and fully-functional older OS and hardware, but the LTS support for 16.04 recently expired. (The new 20.04/5.4 setup also had frequent link up/down trouble with the Intel onboard NIC, which works flawlessly with 16.04/4.15. The boot option to limit the sleep state didn't help. I'm now using a USB-to-Ethernet adapter that works fine for both OSes.)

I tried a spare WinTV-HVR-955Q USB tuner with the newer OS. It has the same issue, reporting as a single version of the same LG Electronics LGDT3306A VSB/QAM devices in the QuadHD card. It has the same Tvheadend "Scan - no data" in 20.04/5.4.

The same "cx23885 driver version 0.0.4 loaded" appears in the system logs of 16.04/4.15 and 20.04/5.4. There are no error messages other than a single set of "frequency 0 out of range (54000000..858000000)" for all four adapters at startup, something that's been there even with the 16.04 system for as long as I can recall. (The 1609 QuadHD card was purchased in October 2017.) They don't cause any problems.

```
$ uname -a
Linux Onyx 5.4.0-81-generic #91-Ubuntu SMP Thu Jul 15 19:09:17 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux
```

Here's some relevant lines from `dmesg | grep cx238`:

```
[    4.375105] cx23885: cx23885 driver version 0.0.4 loaded
[    4.375226] cx23885: CORE cx23885[0]: subsystem: 0070:6a18, board: Hauppauge WinTV-QuadHD-ATSC [card=57,autodetected]
[    4.734365] cx23885: cx23885[0]: hauppauge eeprom: model=165100
[    4.783947] cx25840 11-0044: cx23888 A/V decoder found @ 0x88 (cx23885[0])
[    5.459381] cx25840 11-0044: loaded v4l-cx23885-avcore-01.fw firmware (16382 bytes)
[    5.491184] cx23885: cx23885_dvb_register() allocating 1 frontend(s)
[    5.491186] cx23885: cx23885[0]: cx23885 based dvb card
[    5.491187] cx23885: dvb_register(): board=57 port=1
[    5.504339] dvbdev: DVB: registering new adapter (cx23885[0])
[    5.504351] cx23885 0000:03:00.0: DVB: registering adapter 0 frontend 0 (LG Electronics LGDT3306A VSB/QAM Frontend)...
... same for the other three
[    6.527240] cx23885: cx23885_dev_checkrevision() Hardware revision = 0xd0
[    6.527244] cx23885: cx23885[1]/0: found at 0000:04:00.0, rev: 4, irq: 18, latency: 0, mmio: 0xc0400000
[    6.856017] cx23885 0000:04:00.0: DVB: adapter 3 frontend 0 frequency 0 out of range (54000000..858000000)
[    6.992413] cx23885 0000:04:00.0: DVB: adapter 2 frontend 0 frequency 0 out of range (54000000..858000000)
[    7.132029] cx23885 0000:03:00.0: DVB: adapter 1 frontend 0 frequency 0 out of range (54000000..858000000)
[    7.279214] cx23885 0000:03:00.0: DVB: adapter 0 frontend 0 frequency 0 out of range (54000000..858000000)
```

What's different between kernel 4.15 and kernel 5.4? The Hauppauge drivers are reporting the same version. The hardware is exactly the same. I'm thinking it is something OS related. For now, I'm okay with dual-booting to record shows, but I'd hate to need another computer just to record my shows and have a supported OS, too.