b-rad-NDi/Ubuntu-media-tree-kernel-builder #121: Hauppauge WinTV-HVR2200 (saa7164) - REV2 use different firmware

rb0135, 2020-12-31:
Hi,

I have the above board and for a while, I have had to patch the drivers .c file so that Rev2 uses a different firmware (from Stephen Toth). 

The wrong firmware error in dmesg is shown as this:

` dmesg | grep saa
[    3.084293] CORE saa7164[0]: subsystem: 0070:8901, board: Hauppauge WinTV-HVR2200 [card=6,autodetected]
[    3.084298] saa7164[0]/0: found at 0000:06:00.0, rev: 129, irq: 16, latency: 0, mmio: 0xf7800000
[    3.376423] saa7164_downloadfirmware() no first image
[    3.376431] saa7164_downloadfirmware() Waiting for firmware upload (NXP7164-2010-03-10.1.fw)
[    3.429355] saa7164_downloadfirmware() firmware read 4019072 bytes.
[    3.429357] saa7164_downloadfirmware() firmware loaded.
[    3.429360] saa7164_downloadfirmware() SecBootLoader.FileSize = 4019072
[    3.429364] saa7164_downloadfirmware() FirmwareSize = 0x1fd6
[    3.429365] saa7164_downloadfirmware() BSLSize = 0x0
[    3.429365] saa7164_downloadfirmware() Reserved = 0x0
[    3.429366] saa7164_downloadfirmware() Version = 0x1661c00
[    9.392484] saa7164_downloadimage() image corrupt
[    9.392858] saa7164 driver loaded`

I have been able to update the driver file and recompile for a few years, with 100% success rate of the correct firmware loading and working. With Ubuntu 20.04 LTS, it was working, but now I am on 20.04.1 LTS, and when try to compile the media tree, I am getting these errors (yes, I refetched the mediatree, etc):

`******************
* Start building *
******************
make -C /home/mediapc/media_build/v4l allyesconfig
make[1]: Entering directory '/home/mediapc/media_build/v4l'
No version yet, using 5.4.0-562012221313-generic
make[2]: Entering directory '/home/mediapc/media_build/linux'
Applying patches for kernel 5.4.0-562012221313-generic
patch -s -f -N -p1 -i ../backports/api_version.patch
patch -s -f -N -p1 -i ../backports/pr_fmt.patch
patch -s -f -N -p1 -i ../backports/debug.patch
patch -s -f -N -p1 -i ../backports/drx39xxj.patch
patch -s -f -N -p1 -i ../backports/ccs.patch
patch -s -f -N -p1 -i ../backports/v5.9_tasklet.patch
patch -s -f -N -p1 -i ../backports/v5.7_mmap_read_lock.patch
patch -s -f -N -p1 -i ../backports/v5.7_vm_map_ram.patch
1 out of 1 hunk FAILED
make[2]: *** [Makefile:132: apply_patches] Error 1
make[2]: Leaving directory '/home/mediapc/media_build/linux'
make[1]: *** [Makefile:378: allyesconfig] Error 2
make[1]: Leaving directory '/home/mediapc/media_build/v4l'
make: *** [Makefile:26: allyesconfig] Error 2
can't select all drivers at ./build line 531`

The saa7164 driver file (in this location of the media tree: "drivers/media/pci/saa7164/saa7164-fw.c" has two define lines for Ver2 and Ver3 but has the same firmware version for both which does not work.

In that saa7164 driver file, these two lines need to have the values changed to those shown here: 	
	#define SAA7164_REV2_FIRMWARE 	"v4l-saa7164-1.0.2-3.fw"
	#define SAA7164_REV2_FIRMWARE	4038864

So, I am wondering, since this seems the most active for Ubuntu and Media Tree building, is it possible for you to make the permanent change in the Driver File so that the correct firmware is loaded for VER2 saa7164 boards or maybe I'm just missing something in the compile so I can continue to compile myself.

Im a bit of a loss as to why Im getting compile errors.

Dont mind testing/supplying more information.

Thanks,
Rob

b-rad-NDi, 2021-01-04:
I'll add this to my builds here and submit it upstream.

rb0135, 2021-01-05:
Hi Brad,

Thanks for replying... and thanks for adding the change, much appreciated

I've been trying to figure out why it wont compile with the latest tree/version.

Not familiar with PATCH files, but to me the backports/v5.7_vm_map_ram.patch file seems to end prematurely (as seen below, the file ends on the IF line). Could this be the reason I cant build the tree? I have tried to download several times and each time I get that HUNK error.

'  GNU nano 4.8                                        backports/v5.7_vm_map_ram.patch                                        Modified  
diff --git a/drivers/media/common/videobuf2/videobuf2-dma-sg.c b/drivers/media/common/videobuf2/videobuf2-dma-sg.c
index 92072a08af25..6db60e9d5183 100644
--- a/drivers/media/common/videobuf2/videobuf2-dma-sg.c
+++ b/drivers/media/common/videobuf2/videobuf2-dma-sg.c
@@ -309,7 +309,8 @@ static void *vb2_dma_sg_vaddr(void *buf_priv)
                if (buf->db_attach)
                        buf->vaddr = dma_buf_vmap(buf->db_attach->dmabuf);
                else
-                       buf->vaddr = vm_map_ram(buf->pages, buf->num_pages, -1);
+                       buf->vaddr = vm_map_ram(buf->pages,
+                                       buf->num_pages, -1, PAGE_KERNEL);
        }

        /* add offset in case userptr is not page-aligned */
diff --git a/drivers/media/common/videobuf2/videobuf2-vmalloc.c b/drivers/media/common/videobuf2/videobuf2-vmalloc.c
index c66fda4a65e4..1a4f0ca87c7c 100644
--- a/drivers/media/common/videobuf2/videobuf2-vmalloc.c
+++ b/drivers/media/common/videobuf2/videobuf2-vmalloc.c
@@ -107,7 +107,8 @@ static void *vb2_vmalloc_get_userptr(struct device *dev, unsigned long vaddr,
backports/v5.7_vm_map_ram.patch                 buf->vaddr = (__force void *)
                        ioremap(__pfn_to_phys(nums[0]), size + offset);
        } else {
-               buf->vaddr = vm_map_ram(frame_vector_pages(vec), n_pages, -1);
+               buf->vaddr = vm_map_ram(frame_vector_pages(vec), n_pages, -1,
+                                       PAGE_KERNEL);
        }

        if (!buf->vaddr)
'

Thanks Rob

b-rad-NDi, 2021-01-26:
Hi @rb0135 : I'm unsure of you build failure. It might have been fixed by now since it's been a few weeks. Could you verify my patch to address your firmware issue is correct so I can upstream it?

6c88e84

rb0135, 2021-01-27:
Hi Brad...

Thanks for that... I've just tried to rebuild again as I have been doing in the past (by retrieving the lastest direct from LinuxTV) and this time it compiled without issues.. Maybe there was something astray in the previous code.

So, I then cleaned that up and tried yours and your patch worked perfectly....

Thanks, Rob

b-rad-NDi, 2021-01-27:
Excellent, if you'd like me to cc you on the patch when I submit it so you can add a tested-by tag just send me your email address.

rb0135, 2021-07-13:
Hi Brad.

Is there any update coming soon? The reason I ask is that my system updated to GNU/Linux 5.8.0-59-generic x86_64 in June and the drivers arent loadin properly again. I tried to do an update from your repo, but it says it is up-to-date.

Reason I ask is the repo seems to be back to April, but from your notes, they state "The PPA will be updated quarterly, or immediately upon new Ubuntu kernel release"

Hope Im not jumping the gun too soon in asking.

Thanks,
Rob

b-rad-NDi, 2021-09-02:
New builds have just been pushed for every release. This patch should be in all of them.

rb0135, 2021-09-07:
Hi Brad.. Thanks, but I still have the problem that the  proper firmware is not being loaded.

So, I run the commands:
sudo add-apt-repository ppa:b-rad/kernel+mediatree+hauppauge
sudo apt-get update

and although it goes through the process, nothing gets updated.

I then try:
sudo apt-get install linux-mediatree

and it tells me I am up-to-date:
Reading package lists... Done
Building dependency tree
Reading state information... Done
linux-mediatree is already the newest version (0.1.7+focal).
Starting pkgProblemResolver with broken count: 0
Starting 2 pkgProblemResolver with broken count: 0
Done
0 to upgrade, 0 to newly install, 0 to remove and 0 not to upgrade.

This is the info of my system version:
Linux <pcname> 5.11.0-27-generic #29~20.04.1-Ubuntu SMP Wed Aug 11 15:58:17 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux

Any ideas?

Have I missed something?

Thanks,
Rob

b-rad-NDi, 2021-09-07:
The latest packages all exist in the ppa now. There was one minor issue blocking them. Everything is good now. If you have an issue feel free to re-open.

closed 2021-09-07 by b-rad-NDi