b-rad-NDi/Ubuntu-media-tree-kernel-builder #122: Unable to add ppa for Linux Mint 20.1 ulyssa

Shanghai-Tom, 2021-01-19:
$sudo add-apt-repository ppa:b-rad/kernel+mediatree+hauppauge
[sudo] password:           
LSB codename: 'sh: 0: getcwd() failed: No such file or directory
ulyssa'.
Version of base-files: 'sh: 0: getcwd() failed: No such file or directory
20.1.0'.
Your LSB codename isn't a valid Linux Mint codename.
Please check your LSB information with "lsb_release -a".

$ lsb_release -a
No LSB modules are available.
Distributor ID:	Linuxmint
Description:	Linux Mint 20.1
Release:	20.1
Codename:	ulyssa

Shanghai-Tom, 2021-01-19:
I using hwe Kernel

linux-image-generic-hwe-20.04-edge         5.8.0.38.43~20.04.23                  amd64        Generic Linux kernel image

b-rad-NDi, 2021-01-20:
Well it works on Ubuntu which is what is designed for. If you to want use my ppa you might have to manually add it into /etc/apt/sources.list.d/ yourself.

Shanghai-Tom, 2021-01-20:
Thanks Brad, I'll add it manually, Mint 20.1 is based on Ubuntu 20.1 focal

I'm closing the ticket

closed 2021-01-20 by Shanghai-Tom