b-rad-NDi/Ubuntu-media-tree-kernel-builder #123: Hauppauge WinTV-QuadHD-DVB-T2 Fails to be found after a certain amount of time

thonatron, 2021-02-07:
Ive been struggling to to get this card working for about a week now, from a cold boot it works fine for a while then eventually it silently disappears from the computer. A warm reboot of the computer leaves the card not found on the next boot, but the pcie splitter/bridge is still seen, just nothing that is attached to it
i have
```
cat /etc/modprobe.d/quadhd.conf 
options cx23885 dma_reset_workaround=2
```
ive tried setting this to 0 or 2 and letting the driver handle it automatically but at some point it always seams to crash out and have no tuners found. 
Dmesg gives nothing useful, nothing about why the card just disappears, i must need debugging enabled somewhere.
The card is in a gen2 pciex4 slot, (i cant set it gen1), i could try the pcie3 slots but i have a nvme drive and a hba card in them so its not realistic option to move it to one of those slots
I moved over to this kernel from the default one in ubuntu 20.04LTS due to the errors about mpeg risc op code that kept crashing the card.
The main use of the card is with mythtv, The problem usually starts with a failure to get a signal lock, but interestingly if i switch to a different tuner pair from (1,2) -> (3,4) or the other way, that tuner is usually working, until it doesnt work, then i cant get a signal at all. Then to server need at cold start like mentioned above. I think i have the intel Vt-d extensions enabled as this computer is going to be running some vms once i get this card and mythtv playing nice, so id rather not have to disable those

```
$ uname -a
Linux Thor 5.4.0-642101212021-generic #0+mediatree+hauppauge-Ubuntu SMP Mon Jan 25 22:07:20 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux
```
cpu motherboard
```
Model name:                      Intel(R) Xeon(R) CPU E3-1231 v3 @ 3.40GHz

      description: Motherboard
       product: X10SLM-F
       vendor: Supermicro
```

dmesg when card is working
```
[   11.158072] videodev: Linux video capture interface: v2.00
[   11.158073] WARNING: You are using an experimental version of the media stack.
               	As the driver is backported to an older kernel, it doesn't offer
               	enough quality for its usage in production.
               	Use it with care.
               Latest git patches (needed if you report a bug to linux-media@vger.kernel.org):
               	c386e0797d26a32e354daf4480c5d40165db66a1 media: dvb-frontends/drxk_hard.c: fix uninitialized variable warning
               	2311072d9905b1f810a3c0665362ee552f076c26 media: tvp7002: fix uninitialized variable warning
               	7124ae48f2a45399789bcc14389614c899f6aa7c media: s5k5baf: drop 'data' field in struct s5k5baf_fw
[   11.195476] WARNING: You are using an experimental version of the media stack.
               	As the driver is backported to an older kernel, it doesn't offer
               	enough quality for its usage in production.
               	Use it with care.
               Latest git patches (needed if you report a bug to linux-media@vger.kernel.org):
               	c386e0797d26a32e354daf4480c5d40165db66a1 media: dvb-frontends/drxk_hard.c: fix uninitialized variable warning
               	2311072d9905b1f810a3c0665362ee552f076c26 media: tvp7002: fix uninitialized variable warning
               	7124ae48f2a45399789bcc14389614c899f6aa7c media: s5k5baf: drop 'data' field in struct s5k5baf_fw
[   11.198906] WARNING: You are using an experimental version of the media stack.
               	As the driver is backported to an older kernel, it doesn't offer
               	enough quality for its usage in production.
               	Use it with care.
               Latest git patches (needed if you report a bug to linux-media@vger.kernel.org):
               	c386e0797d26a32e354daf4480c5d40165db66a1 media: dvb-frontends/drxk_hard.c: fix uninitialized variable warning
               	2311072d9905b1f810a3c0665362ee552f076c26 media: tvp7002: fix uninitialized variable warning
               	7124ae48f2a45399789bcc14389614c899f6aa7c media: s5k5baf: drop 'data' field in struct s5k5baf_fw
[   11.219418] cx23885: cx23885 driver version 0.0.4 loaded
[   11.219593] cx23885: CORE cx23885[0]: subsystem: 0070:6a28, board: Hauppauge WinTV-QuadHD-DVB [card=56,autodetected]
[   11.232719] ipmi_si IPI0001:00: IPMI message handler: Found new BMC (man_id: 0x002a7c, prod_id: 0x0811, dev_id: 0x20)
[   11.235033] intel_rapl_common: Found RAPL domain package
[   11.235034] intel_rapl_common: Found RAPL domain core
[   11.235035] intel_rapl_common: Found RAPL domain dram
[   11.278232] ipmi_si IPI0001:00: IPMI kcs interface initialized
[   11.279773] ipmi_ssif: IPMI SSIF Interface driver
[   11.325334] XFS (sdg2): Mounting V4 Filesystem
[   11.372114] zlua: loading out-of-tree module taints kernel.
[   11.372117] zlua: module license 'MIT' taints kernel.
[   11.372117] Disabling lock debugging due to kernel taint
[   11.528035] XFS (sdg2): Ending clean mount
[   11.528055] xfs filesystem being mounted at /disks/b supports timestamps until 2038 (0x7fffffff)
[   11.577333] tveeprom: Hauppauge model 166100, rev B4I6, serial# 4036162655
[   11.577334] tveeprom: MAC address is 00:0d:fe:92:f4:5f
[   11.577334] tveeprom: tuner model is SiLabs Si2157 (idx 186, type 4)
[   11.577335] tveeprom: TV standards PAL(B/G) PAL(I) SECAM(L/L') PAL(D/D1/K) ATSC/DVB Digital (eeprom 0xf4)
[... 86 lines cut ...]
00:00.0 Host bridge: Intel Corporation Xeon E3-1200 v3 Processor DRAM Controller (rev 06)
00:01.0 PCI bridge: Intel Corporation Xeon E3-1200 v3/4th Gen Core Processor PCI Express x16 Controller (rev 06)
00:01.1 PCI bridge: Intel Corporation Xeon E3-1200 v3/4th Gen Core Processor PCI Express x8 Controller (rev 06)
00:14.0 USB controller: Intel Corporation 8 Series/C220 Series Chipset Family USB xHCI (rev 05)
00:16.0 Communication controller: Intel Corporation 8 Series/C220 Series Chipset Family MEI Controller #1 (rev 04)
00:16.1 Communication controller: Intel Corporation 8 Series/C220 Series Chipset Family MEI Controller #2 (rev 04)
00:19.0 Ethernet controller: Intel Corporation Ethernet Connection I217-LM (rev 05)
00:1a.0 USB controller: Intel Corporation 8 Series/C220 Series Chipset Family USB EHCI #2 (rev 05)
00:1c.0 PCI bridge: Intel Corporation 8 Series/C220 Series Chipset Family PCI Express Root Port #1 (rev d5)
00:1c.1 PCI bridge: Intel Corporation 8 Series/C220 Series Chipset Family PCI Express Root Port #2 (rev d5)
00:1c.4 PCI bridge: Intel Corporation 8 Series/C220 Series Chipset Family PCI Express Root Port #5 (rev d5)
00:1d.0 USB controller: Intel Corporation 8 Series/C220 Series Chipset Family USB EHCI #1 (rev 05)
00:1f.0 ISA bridge: Intel Corporation C224 Series Chipset Family Server Standard SKU LPC Controller (rev 05)
00:1f.2 SATA controller: Intel Corporation 8 Series/C220 Series Chipset Family 6-port SATA Controller 1 [AHCI mode] (rev 05)
00:1f.3 SMBus: Intel Corporation 8 Series/C220 Series Chipset Family SMBus Controller (rev 05)
00:1f.6 Signal processing controller: Intel Corporation 8 Series Chipset Family Thermal Management Controller (rev 05)
01:00.0 Serial Attached SCSI controller: Broadcom / LSI SAS2008 PCI-Express Fusion-MPT SAS-2 [Falcon] (rev 03)
02:00.0 Non-Volatile memory controller: Phison Electronics Corporation Device 5013 (rev 01)
03:00.0 PCI bridge: ASPEED Technology, Inc. AST1150 PCI-to-PCI Bridge (rev 03)
04:00.0 VGA compatible controller: ASPEED Technology, Inc. ASPEED Graphics Family (rev 30)
05:00.0 Ethernet controller: Intel Corporation I210 Gigabit Network Connection (rev 03)
06:00.0 PCI bridge: Pericom Semiconductor PI7C9X2G304 EL/SL PCIe2 3-Port/4-Lane Packet Switch (rev 05)
07:01.0 PCI bridge: Pericom Semiconductor PI7C9X2G304 EL/SL PCIe2 3-Port/4-Lane Packet Switch (rev 05)
07:02.0 PCI bridge: Pericom Semiconductor PI7C9X2G304 EL/SL PCIe2 3-Port/4-Lane Packet Switch (rev 05)
```

thonatron, 2021-02-08:
Ive just had the a tuner die and me again with the other one still working, the only thing in dmesg between last known working and tuner been gone is this
```
[17558.772023] si2168 7-0064: downloading firmware from file 'dvb-demod-si2168-b40-01.fw'
[17559.459250] si2168 7-0064: firmware version: B 4.0.11
[17559.467044] si2157 8-0060: found a 'Silicon Labs Si2157-A30'
[17559.489318] si2157 8-0060: firmware version: 3.0.5
[17837.335797] perf: interrupt took too long (2503 > 2500), lowering kernel.perf_event_max_sample_rate to 79750
[23727.218983] kauditd_printk_skb: 19 callbacks suppressed
[23727.218985] audit: type=1400 audit(1612742401.084:31): apparmor="DENIED" operation="capable" profile="/usr/sbin/cups-browsed" pid=22052 comm="cups-browsed" capability=23  capname="sys_nice"
[28106.267154] si2168 2-0064: downloading firmware from file 'dvb-demod-si2168-b40-01.fw'
[28106.961138] si2168 2-0064: firmware version: B 4.0.11
[28106.968912] si2157 3-0060: found a 'Silicon Labs Si2157-A30'
[28106.991198] si2157 3-0060: firmware version: 3.0.5
[34135.110378] perf: interrupt took too long (3137 > 3128), lowering kernel.perf_event_max_sample_rate to 63750
[59399.157784] si2168 2-0064: downloading firmware from file 'dvb-demod-si2168-b40-01.fw'
[59399.851997] si2168 2-0064: firmware version: B 4.0.11
[59399.859771] si2157 3-0060: found a 'Silicon Labs Si2157-A30'
[59399.882056] si2157 3-0060: firmware version: 3.0.5
```
lspci shows the card still there but the rev is now FF instead of 04
```
05:00.0 Ethernet controller: Intel Corporation I210 Gigabit Network Connection (rev 03)
06:00.0 PCI bridge: Pericom Semiconductor PI7C9X2G304 EL/SL PCIe2 3-Port/4-Lane Packet Switch (rev 05)
07:01.0 PCI bridge: Pericom Semiconductor PI7C9X2G304 EL/SL PCIe2 3-Port/4-Lane Packet Switch (rev 05)
07:02.0 PCI bridge: Pericom Semiconductor PI7C9X2G304 EL/SL PCIe2 3-Port/4-Lane Packet Switch (rev 05)
08:00.0 Multimedia video controller: Conexant Systems, Inc. CX23887/8 PCIe Broadcast Audio and Video Decoder with 3D Comb (rev 04)
09:00.0 Multimedia video controller: Conexant Systems, Inc. CX23887/8 PCIe Broadcast Audio and Video Decoder with 3D Comb (rev ff)
```
```
lscpi -vvv
08:00.0 Multimedia video controller: Conexant Systems, Inc. CX23887/8 PCIe Broadcast Audio and Video Decoder with 3D Comb (rev 04)
	Subsystem: Hauppauge computer works Inc. CX23887/8 PCIe Broadcast Audio and Video Decoder with 3D Comb
	Control: I/O- Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Latency: 0, Cache Line Size: 64 bytes
	Interrupt: pin A routed to IRQ 17
	Region 0: Memory at f7400000 (64-bit, non-prefetchable) [size=2M]
	Capabilities: <access denied>
	Kernel driver in use: cx23885
	Kernel modules: cx23885

09:00.0 Multimedia video controller: Conexant Systems, Inc. CX23887/8 PCIe Broadcast Audio and Video Decoder with 3D Comb (rev ff) (prog-if ff)
	!!! Unknown header type 7f
	Kernel driver in use: cx23885
	Kernel modules: cx23885
```
rebooting gives
```
lspci 
05:00.0 Ethernet controller: Intel Corporation I210 Gigabit Network Connection (rev 03)
06:00.0 PCI bridge: Pericom Semiconductor PI7C9X2G304 EL/SL PCIe2 3-Port/4-Lane Packet Switch (rev 05)
07:01.0 PCI bridge: Pericom Semiconductor PI7C9X2G304 EL/SL PCIe2 3-Port/4-Lane Packet Switch (rev 05)
07:02.0 PCI bridge: Pericom Semiconductor PI7C9X2G304 EL/SL PCIe2 3-Port/4-Lane Packet Switch (rev 05)
08:00.0 Multimedia video controller: Conexant Systems, Inc. CX23887/8 PCIe Broadcast Audio and Video Decoder with 3D Comb (rev 04)
```

Now something different happens, i thought id try to see what bios options are available, rebooted spammed F2 to enter Bios over ipmi, and the board just hangs at entering setup. Rebooted and this time let it fully boot while i figure out whats going on with entering setup, and both cards are back working, didnt require a cold boot. So thinking that the card is somehow stopping me entering the bios, i reboot again and im now in the bios.
So crashed card -> reboot -> no bios
not crashed card -> reboot -> bios
crashed card -> reboot -> reboot whilst rebooting -> card fine

Ive found a setting in bios about logging pcie errors to either the bios or ipmi so ill see how that works out

shspvr, 2021-02-09:
Sound like there something with the tuner you may want test it under Windows to see if exhibit the same problem by using PiP mode or or SageTV where you can test each of the tuner.

thonatron, 2021-02-09:
Unfortunately i don't have any windows machines, but mythtv does allow me to switch tuners, havent tried PiP yet.
My bios pcie debugging option did bring some good(?) news

```
Event ID | Time Stamp | Sensor Name | Sensor Type | Description
-- | -- | -- | -- | --
1 | 2021/02/07 17:26:10 | OEM | OOB | Activate Node Product Key - Assertion
2 | 2021/02/09 03:58:06 | #0x00 | Critical Interrupt | PCI SERR - Assertion@Bus08(DevFn00)
3 | 2021/02/09 08:27:22 | #0x00 | Critical Interrupt | PCI SERR - Assertion@Bus07(DevFn10)
```
At least i now have a timestamp of what time each one error-ed out, not that i know what it means yet. At 03:58 and 08:27 mythtv wasnt doing anything to do with tuning, the cards are set to open on demand
```
Feb  9 04:32:29 Thor mythbackend: mythbackend[2776]: N Expire autoexpire.cpp:241 (CalcParams) AutoExpire: CalcParams(): Max required Free Space: 100.0 GB w/freq: 15 min
Feb  9 04:47:29 Thor mythbackend: mythbackend[2776]: N Expire autoexpire.cpp:241 (CalcParams) AutoExpire: CalcParams(): Max required Free Space: 100.0 GB w/freq: 15 min
Feb  9 05:02:29 Thor mythbackend: mythbackend[2776]: N Expire autoexpire.cpp:241 (CalcParams) AutoExpire: CalcParams(): Max required Free Space: 100.0 GB w/freq: 15 min
...
Feb  9 08:06:55 Thor mythbackend: mythbackend[2776]: I HouseKeeping housekeeper.cpp:159 (Run) HouseKeeperTask 'RecordedArtworkUpdate' Finished Successfully.
Feb  9 08:17:30 Thor mythbackend: mythbackend[2776]: N Expire autoexpire.cpp:241 (CalcParams) AutoExpire: CalcParams(): Max required Free Space: 100.0 GB w/freq: 15 min
Feb  9 08:32:30 Thor mythbackend: mythbackend[2776]: N Expire autoexpire.cpp:241 (CalcParams) AutoExpire: CalcParams(): Max required Free Space: 100.0 GB w/freq: 15 min
Feb  9 08:47:30 Thor mythbackend: mythbackend[2776]: N Expire autoexpire.cpp:241 (CalcParams) AutoExpire: CalcParams(): Max required Free Space: 100.0 GB w/freq: 15 min
Feb  9 09:02:30 Thor mythbackend: mythbackend[2776]: N Expire autoexpire.cpp:241 (CalcParams) AutoExpire: CalcParams(): Max required Free Space: 100.0 GB w/freq: 15 min
```
dmesg still doesnt give anything which is not surprising since since it still thinks the cards are there
```
 ls -al /dev/dvb/
total 0
drwxr-xr-x  6 root root  120 Feb  8 11:46 .
drwxr-xr-x 20 root root 5000 Feb  8 11:46 ..
drwxr-xr-x  2 root root  120 Feb  8 11:46 adapter0
drwxr-xr-x  2 root root  120 Feb  8 11:46 adapter1
drwxr-xr-x  2 root root  120 Feb  8 11:46 adapter2
drwxr-xr-x  2 root root  120 Feb  8 11:46 adapter3
```
but both are dead by the (rev ff) in lspci
```
 lspci
06:00.0 PCI bridge: Pericom Semiconductor PI7C9X2G304 EL/SL PCIe2 3-Port/4-Lane Packet Switch (rev 05)
07:01.0 PCI bridge: Pericom Semiconductor PI7C9X2G304 EL/SL PCIe2 3-Port/4-Lane Packet Switch (rev 05)
07:02.0 PCI bridge: Pericom Semiconductor PI7C9X2G304 EL/SL PCIe2 3-Port/4-Lane Packet Switch (rev 05)
08:00.0 Multimedia video controller: Conexant Systems, Inc. CX23887/8 PCIe Broadcast Audio and Video Decoder with 3D Comb (rev ff)
09:00.0 Multimedia video controller: Conexant Systems, Inc. CX23887/8 PCIe Broadcast Audio and Video Decoder with 3D Comb (rev ff)
```
Ive put in a request with supermicro about the PCI SERR to try and find out what that means, hopefully ill get some good news but the board is now EOL. Im going to see if i can put the card in another computer, but this might take a bit of time

shspvr, 2021-02-09:
Keep in mind that consumer TV tuner where never meant for server system
I don't know if this well help but in the bios try disable PCI SERR
Maybe when brad come back in he will have some better suggestion for you.

thonatron, 2021-02-09:
ive got in in one of the PCIE3x8 slots now, it wasnt visible, in fact i couldnt get in to the bios, until i removed it and set the slot to gen1. Now i can get in the bios see the card and linux can see the card, ill just have to wait to see if it falls again, but its not ideal as my hba is in a slower PCIE2x4 slot, luckily ive only got 4 HDD attached to it at the moment.

b-rad-NDi, 2021-02-09:
If it's pcie2 it will be perfectly fine in a pcie3 slot. On intel systems, for whatever strange reason, their implementation of pcie requires to be set to gen1. The card should continue to work without issue for you that way.

thonatron, 2021-02-09:
is there anyway to make it work in the pcie2 slot if i cant set the gen on the slot, ive only got the option to set gen on the 2 pcie3 slots

b-rad-NDi, 2021-02-09:
This is specifically an Intel issue in this regard. I don't know what they are doing electrically or systematically to make the boards behave as such. As you've seen when the pcie chipset becomes unhappy the board just disappears. This is not a driver issue, but some inherent issue in the Intel chipset.

thonatron, 2021-02-11:
little update, it eventually managed to error out again, The log in the ipmi did come up with about 15 of the serr# errors before it did drop off so i guess thats an improvement of sorts. Supermicro got back to me, but as the board is EOL not a lot could be done. In the original manual it mentions Serr and perr disable, but the latest bios doesn't have those options, supermicro provided me with some old bios files but the i got to a point where the m.2 drive in a pcie caddy wasn't recognized in so it wouldn't boot for me.
For the moment ive pulled the card, looking to test it in this desktop, but the pciex1 slot is blocked by a graphics card.
so for now it looks like supermicro x10slm-f and a quadhd is a no go.