b-rad-NDi/Ubuntu-media-tree-kernel-builder #127: QuadHD mpeg risc op code error and stalls

sergiud, 2021-07-06:
I'm using a QuadHD in a Debian Buster system running on an AMD Ryzen 3200G and MSI B450M PRO-M2 MAX mainboard.

```
$ uname -a
Linux pony 5.10.0-0.bpo.7-amd64 #1 SMP Debian 5.10.40-1~bpo10+1 (2021-06-04) x86_64 GNU/Linux
```

I use tvheadend to manage the DVB card.

Unfortunately, the card stops randomly receiving data while recording and the only way to stop the stall is to change the channel. However, this is not a viable solution since the recording remains incomplete.

I run the cx23885 module with the `dma_reset_workaround=2` parameter because it is otherwise not possible to scan the channels. When a stall occurs I can see in the dmesg output that an `mpeg risc op code error` is triggered:

<details>

```
[    5.561739] cx23885 0000:29:00.0: enabling device (0000 -> 0002)
[    5.561824] cx23885: CORE cx23885[1]: subsystem: 0070:6b28, board: Hauppauge WinTV-QuadHD-DVB [card=56,autodetected]
[    5.913982] tveeprom: Hauppauge model 166101, rev B4I6, serial# redacted
[    5.913984] tveeprom: MAC address is redacted
[    5.913985] tveeprom: tuner model is SiLabs Si2157 (idx 186, type 4)
[    5.913986] tveeprom: TV standards PAL(B/G) PAL(I) SECAM(L/L') PAL(D/D1/K) ATSC/DVB Digital (eeprom 0xf4)
[    5.913987] tveeprom: audio processor is CX23888 (idx 40)
[    5.913988] tveeprom: decoder processor is CX23888 (idx 34)
[    5.913989] tveeprom: has no radio
[    5.913990] cx23885: cx23885[1]: hauppauge eeprom: model=166101
[    5.914759] kauditd_printk_skb: 2 callbacks suppressed
[    5.914760] audit: type=1400 audit(1625509597.705:13): apparmor="STATUS" operation="profile_load" profile="unconfined" name="docker-default" pid=1005 comm="apparmor_parser"
[    5.917248] cx25840 16-0044: cx23888 A/V decoder found @ 0x88 (cx23885[1])
[    5.924496] cx25840 16-0044: firmware: direct-loading firmware v4l-cx23885-avcore-01.fw
[    6.223483] NFSD: Using UMH upcall client tracking operations.
[    6.223487] NFSD: starting 90-second grace period (net f0000098)
[    6.578088] cx25840 16-0044: loaded v4l-cx23885-avcore-01.fw firmware (16382 bytes)
[    6.615021] bridge: filtering via arp/ip/ip6tables is no longer available by default. Update your scripts to load br_netfilter if you need this.
[    6.616102] Bridge firewalling registered
[    6.661656] cx23885: cx23885[1]: registered device video1 [v4l2]
[    6.661707] cx23885: cx23885[1]: registered device vbi1
[    6.661821] cx23885: cx23885[1]: alsa: registered ALSA audio device
[    6.661823] cx23885: cx23885_dvb_register() allocating 1 frontend(s)
[    6.661824] cx23885: cx23885[1]: cx23885 based dvb card
[    6.661826] cx23885: dvb_register(): board=56 port=1
[    6.664562] Initializing XFRM netlink socket
[    6.669000] i2c i2c-14: Added multiplexed i2c bus 17
[    6.669003] si2168 14-0064: Silicon Labs Si2168-B40 successfully identified
[    6.669005] si2168 14-0064: firmware version: B 4.0.2
[    6.670968] si2157 15-0060: Silicon Labs Si2147/2148/2157/2158 successfully attached
[    6.670983] cx23885: dvb_register(): QUADHD_DVB analog setup
[    6.670985] dvbdev: DVB: registering new adapter (cx23885[1])
[    6.670988] cx23885 0000:29:00.0: DVB: registering adapter 2 frontend 0 (Silicon Labs Si2168)...
[    6.671258] cx23885: cx23885_dvb_register() allocating 1 frontend(s)
[    6.671259] cx23885: cx23885[1]: cx23885 based dvb card
[    6.671262] cx23885: dvb_register(): board=56 port=2
[    6.677829] i2c i2c-14: Added multiplexed i2c bus 18
[    6.677832] si2168 14-0066: Silicon Labs Si2168-B40 successfully identified
[    6.677833] si2168 14-0066: firmware version: B 4.0.2
[    6.681260] si2157 15-0062: Silicon Labs Si2147/2148/2157/2158 successfully attached
[    6.681814] dvbdev: DVB: registering new adapter (cx23885[1])
[    6.681817] cx23885 0000:29:00.0: DVB: registering adapter 3 frontend 0 (Silicon Labs Si2168)...
[    6.682057] cx23885: cx23885_dev_checkrevision() Hardware revision = 0xd0
[    6.682062] cx23885: cx23885[1]/0: found at 0000:29:00.0, rev: 4, irq: 38, latency: 0, mmio: 0xfc400000
[    6.763869] si2168 14-0066: firmware: direct-loading firmware dvb-demod-si2168-b40-01.fw
[    6.763873] si2168 14-0066: downloading firmware from file 'dvb-demod-si2168-b40-01.fw'
[    7.398142] si2168 14-0066: firmware version: B 4.0.11
[    7.405360] si2157 15-0062: found a 'Silicon Labs Si2157-A30'
[    7.427734] si2157 15-0062: firmware version: 3.0.5
[    7.430859] cx23885 0000:29:00.0: DVB: adapter 3 frontend 0 frequency 0 out of range (48000000..870000000)
[    7.439417] si2168 14-0064: firmware: direct-loading firmware dvb-demod-si2168-b40-01.fw
[    7.439420] si2168 14-0064: downloading firmware from file 'dvb-demod-si2168-b40-01.fw'
[    7.446727] si2157 15-0062: found a 'Silicon Labs Si2157-A30'
[    7.469013] si2157 15-0062: firmware version: 3.0.5
[... 1287 lines cut ...]
[71330.083168] cx23885: cx23885[0]: ctrl: 0x00010670 -> 0x106d0
[71330.083171] cx23885: cx23885[0]:   ptr1_reg: 0x00006460
[71330.083174] cx23885: cx23885[0]:   ptr2_reg: 0x000108e8
[71330.083177] cx23885: cx23885[0]:   cnt1_reg: 0x00000017
[71330.083180] cx23885: cx23885[0]:   cnt2_reg: 0x00000007
[72216.848449] cx23885 0000:29:00.0: dma in progress detected 0x00000001 0x00000001, clearing
[72490.204475] cx23885 0000:28:00.0: dma in progress detected 0x00000001 0x00000001, clearing
[72835.467092] cx23885 0000:29:00.0: dma in progress detected 0x00000001 0x00000001, clearing
[72891.009963] cx23885 0000:29:00.0: dma in progress detected 0x00000001 0x00000001, clearing
[72891.052125] cx23885 0000:28:00.0: dma in progress detected 0x00000001 0x00000001, clearing
[73761.113334] cx23885 0000:29:00.0: dma in progress detected 0x00000001 0x00000001, clearing
[73816.073935] cx23885 0000:29:00.0: dma in progress detected 0x00000001 0x00000001, clearing
[74023.573960] cx23885 0000:29:00.0: dma in progress detected 0x00000001 0x00000001, clearing
[74424.252363] cx23885 0000:28:00.0: dma in progress detected 0x00000001 0x00000001, clearing
[74493.063205] cx23885 0000:28:00.0: dma in progress detected 0x00000001 0x00000001, clearing
[74949.075463] cx23885 0000:29:00.0: dma in progress detected 0x00000001 0x00000001, clearing
[75529.064557] cx23885 0000:28:00.0: dma in progress detected 0x00000001 0x00000001, clearing
[75611.936455] cx23885 0000:29:00.0: dma in progress detected 0x00000001 0x00000001, clearing
```

</details>

I already updated the mainboard BIOS to the latest version, tried running the kernel with `iommu=soft`, `pci=noats`, both parameters combined, enabling and disabling virtualization (SVM) to no avail.

Any suggestions?

b-rad-NDi, 2022-01-28:
Have you tried forcing the pcie slot to gen1 mode in bios?

b-rad-NDi, 2022-01-28:
and jeez, wow, I don't think I've ever seen a board so affected after noticing your dmesg log just now.

sergiud, 2022-01-28:
Thanks for looking into the issue. While this is already some time ago, as tried to find a workaround, I believe the BIOS does not provide an option to force the mode.

In the meantime, I've updated to Debian Bullseye currently running
```console
Linux pony 5.14.0-0.bpo.2-amd64 #1 SMP Debian 5.14.9-2~bpo11+1 (2021-10-10) x86_64 GNU/Linux
```
and still see the issue.

shspvr, 2022-01-28:
Have install the latest bios to 7B84vAE ?

sergiud, 2022-01-28:
Not yet. I just saw that the version was released only last week. Does it provide this option now?

sotiris-bos, 2022-01-30:
Hi, I have the same problem. It seems to happen with a particular DVB-T Mux for some reason..

<details>
[  430.219418] si2168 8-0066: downloading firmware from file 'dvb-demod-si2168-b40-01.fw'
[  430.226132] audit: type=1130 audit(1643551436.900:416): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=systemd-networkd-wait-online comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[  430.832245] si2168 8-0066: firmware version: B 4.0.11
[  430.839182] si2157 9-0062: found a 'Silicon Labs Si2157-A30'
[  430.861453] si2157 9-0062: firmware version: 3.0.5
[  430.864500] cx23885 0000:46:00.0: DVB: adapter 3 frontend 0 frequency 0 out of range (48000000..870000000)
[  430.872786] si2168 8-0064: downloading firmware from file 'dvb-demod-si2168-b40-01.fw'
[  430.879760] si2157 9-0062: found a 'Silicon Labs Si2157-A30'
[  430.902011] si2157 9-0062: firmware version: 3.0.5
[  431.483195] si2168 8-0064: firmware version: B 4.0.11
[  431.490140] si2157 9-0060: found a 'Silicon Labs Si2157-A30'
[  431.512226] si2157 9-0060: firmware version: 3.0.5
[  431.515323] cx23885 0000:46:00.0: DVB: adapter 2 frontend 0 frequency 0 out of range (48000000..870000000)
[  431.521259] si2168 3-0066: downloading firmware from file 'dvb-demod-si2168-b40-01.fw'
[  431.526755] si2157 9-0060: found a 'Silicon Labs Si2157-A30'
[  431.549119] si2157 9-0060: firmware version: 3.0.5
[  432.131927] si2168 3-0066: firmware version: B 4.0.11
[  432.138854] si2157 4-0062: found a 'Silicon Labs Si2157-A30'
[  432.160909] si2157 4-0062: firmware version: 3.0.5
[  432.163940] cx23885 0000:45:00.0: DVB: adapter 1 frontend 0 frequency 0 out of range (48000000..870000000)
[  432.172200] si2168 3-0064: downloading firmware from file 'dvb-demod-si2168-b40-01.fw'
[  432.179257] si2157 4-0062: found a 'Silicon Labs Si2157-A30'
[  432.201460] si2157 4-0062: firmware version: 3.0.5
[  432.782428] si2168 3-0064: firmware version: B 4.0.11
[  432.789379] si2157 4-0060: found a 'Silicon Labs Si2157-A30'
[  432.811913] si2157 4-0060: firmware version: 3.0.5
[  432.814970] cx23885 0000:45:00.0: DVB: adapter 0 frontend 0 frequency 0 out of range (48000000..870000000)
[  432.826249] si2157 4-0060: found a 'Silicon Labs Si2157-A30'
[  432.848677] si2157 4-0060: firmware version: 3.0.5
[  433.003962] audit: type=1334 audit(1643551439.680:417): prog-id=35 op=LOAD
[  435.093240] new mount options do not match the existing superblock, will be ignored
[  450.618067] cx23885: cx23885[0]: mpeg risc op code error
[  450.618071] cx23885: cx23885[0]: TS2 C - dma channel status dump
[  450.618074] cx23885: cx23885[0]:   cmds: init risc lo   : 0x76b14000
[  450.618077] cx23885: cx23885[0]:   cmds: init risc hi   : 0x00000000
[  450.618079] cx23885: cx23885[0]:   cmds: cdt base       : 0x000108d0
[  450.618082] cx23885: cx23885[0]:   cmds: cdt size       : 0x0000000a
[  450.618084] cx23885: cx23885[0]:   cmds: iq base        : 0x00010670
[  450.618086] cx23885: cx23885[0]:   cmds: iq size        : 0x00000010
[  450.618089] cx23885: cx23885[0]:   cmds: risc pc lo     : 0x76b14018
[  450.618091] cx23885: cx23885[0]:   cmds: risc pc hi     : 0x00000000
[  450.618093] cx23885: cx23885[0]:   cmds: iq wr ptr      : 0x000041a2
[  450.618095] cx23885: cx23885[0]:   cmds: iq rd ptr      : 0x0000419c
[  450.618098] cx23885: cx23885[0]:   cmds: cdt current    : 0x000108e8
[  450.618100] cx23885: cx23885[0]:   cmds: pci target lo  : 0x76db5d60
[  450.618102] cx23885: cx23885[0]:   cmds: pci target hi  : 0x00000000
[  450.618104] cx23885: cx23885[0]:   cmds: line / byte    : 0x000a0000
[  450.618107] cx23885: cx23885[0]:   risc0: 
[  450.618108] 0x1c0002f0 [ write sol eol count=752 ]
[  450.618113] cx23885: cx23885[0]:   risc1: 
[  450.618113] 0x76db5a70 [ jump eol irq2 23 22 20 19 cnt1 cnt0 14 12 count=2672 ]
[  450.618120] cx23885: cx23885[0]:   risc2: 
[  450.618121] 0x00000000 [ INVALID count=0 ]
[  450.618124] cx23885: cx23885[0]:   risc3: 
[  450.618124] 0x1c0002f0 [ write sol eol count=752 ]
[  450.618128] cx23885: cx23885[0]:   (0x00010670) iq 0: 
[  450.618129] 0x1c0002f0 [ write sol eol count=752 ]
[  450.618132] cx23885: cx23885[0]:   iq 1: 0x76db6050 [ arg #1 ]
[  450.618135] cx23885: cx23885[0]:   iq 2: 0x00000000 [ arg #2 ]
[  450.618137] cx23885: cx23885[0]:   (0x0001067c) iq 3: 
[  450.618138] 0x1c0002f0 [ write sol eol count=752 ]
[  450.618141] cx23885: cx23885[0]:   iq 4: 0x76db6340 [ arg #1 ]
[  450.618143] cx23885: cx23885[0]:   iq 5: 0x00000000 [ arg #2 ]
[  450.618146] cx23885: cx23885[0]:   (0x00010688) iq 6: 
[  450.618146] 0x1c0002f0 [ write sol eol count=752 ]
[  450.618150] cx23885: cx23885[0]:   iq 7: 0x76db5490 [ arg #1 ]
[  450.618152] cx23885: cx23885[0]:   iq 8: 0x00000000 [ arg #2 ]
[... 53 lines cut ...]
[  450.623351] cx23885: cx23885[1]:   iq 8: 0x00000000 [ arg #2 ]
[  450.623353] cx23885: cx23885[1]:   (0x00010694) iq 9: 
[  450.623354] 0x1c0002f0 [ write sol eol count=752 ]
[  450.623358] cx23885: cx23885[1]:   iq a: 0x76b55780 [ arg #1 ]
[  450.623360] cx23885: cx23885[1]:   iq b: 0x00000000 [ arg #2 ]
[  450.623363] cx23885: cx23885[1]:   (0x000106a0) iq c: 
[  450.623363] 0x1c0002f0 [ write sol eol count=752 ]
[  450.623367] cx23885: cx23885[1]:   iq d: 0x76b55a70 [ arg #1 ]
[  450.623370] cx23885: cx23885[1]:   iq e: 0x00000000 [ arg #2 ]
[  450.623372] cx23885: cx23885[1]:   (0x000106ac) iq f: 
[  450.623373] 0x00000000 [ INVALID count=0 ]
[... cut ...]
[  450.623380] cx23885: cx23885[1]:   ptr2_reg: 0x000108f8
[  450.623382] cx23885: cx23885[1]:   cnt1_reg: 0x00000009
[  450.623385] cx23885: cx23885[1]:   cnt2_reg: 0x00000005
</details>

I am on Arch Linux: `Linux nas 5.10.74-1-lts #1 SMP Sun, 17 Oct 2021 12:15:59 +0000 x86_64 GNU/Linux
`

The hardware is a Dell R720xd server with 2x Intel Xeon E5-2650 v2 CPUs.

There is no option in the BIOS to force the port to gen1 but I have done so with a Linux script. I am also using the dma_reset_workaround=2 driver option but the problem persists.

sergiud, 2022-01-30:
Meanwhile, I've updated to the latest BIOS. There's an option for forcing the chipset gen but not for a particular PCIe slot. The option was presumably present prior to the update but I've avoided it since it would have affected the performance of the whole system.

After some of the BIOS updates, I could observe another `dmesg` entry close to `dma in progress` logs appearing once or twice after the boot:
```
[108860.487289] cx23885 0000:28:00.0: dma in progress detected 0x00000001 0x00000001, clearing
[109175.071109] perf: interrupt took too long (2502 > 2500), lowering kernel.perf_event_max_sample_rate to 79750
```
Could this be an important cue?

Lovett1991, 2022-02-01:
Figured I'd add to this... running Ubuntu 20.04 lts
I was previously running this on an install of manjaro just fine but now this is running in a kvm VM with pci passthrough. w_scan finds muxes but then can't get the service

```
538000: (time: 01:13.944)         signal ok:	QAM_AUTO f = 538000 kHz I999B8C999D999T999G999Y999 (0:0:0)
        Info: no data from PAT after 2 seconds
        deleting (QAM_AUTO f = 538000 kHz I999B8C999D999T999G999Y999 (0:0:0))

```

I bring it up under this issue because when tvheadend is scanning I get the same error in dmesg: `[   23.136825] cx23885: cx23885[0]: mpeg risc op code error`

```
[   12.788472] si2168 6-0066: downloading firmware from file 'dvb-demod-si2168-b40-01.fw'
[   14.018514] si2168 6-0066: firmware version: B 4.0.11
[   14.027433] si2157 7-0062: found a 'Silicon Labs Si2157-A30'
[   14.050373] si2157 7-0062: firmware version: 3.0.5
[   14.058509] cx23885 0000:07:00.0: DVB: adapter 3 frontend 0 frequency 0 out of range (48000000..870000000)
[   14.074896] si2168 6-0064: downloading firmware from file 'dvb-demod-si2168-b40-01.fw'
[   15.249573] si2168 6-0064: firmware version: B 4.0.11
[   15.262399] si2157 7-0060: found a 'Silicon Labs Si2157-A30'
[   15.286062] si2157 7-0060: firmware version: 3.0.5
[   15.292617] cx23885 0000:07:00.0: DVB: adapter 2 frontend 0 frequency 0 out of range (48000000..870000000)
[   15.304130] si2168 1-0066: downloading firmware from file 'dvb-demod-si2168-b40-01.fw'
[   16.437817] si2168 1-0066: firmware version: B 4.0.11
[   16.448364] si2157 2-0062: found a 'Silicon Labs Si2157-A30'
[   16.478064] si2157 2-0062: firmware version: 3.0.5
[   16.484623] cx23885 0000:06:00.0: DVB: adapter 1 frontend 0 frequency 0 out of range (48000000..870000000)
[   16.498891] si2168 1-0064: downloading firmware from file 'dvb-demod-si2168-b40-01.fw'
[   17.325960] si2168 1-0064: firmware version: B 4.0.11
[   17.333334] si2157 2-0060: found a 'Silicon Labs Si2157-A30'
[   17.355674] si2157 2-0060: firmware version: 3.0.5
[   17.358940] cx23885 0000:06:00.0: DVB: adapter 0 frontend 0 frequency 0 out of range (48000000..870000000)
[   17.368633] si2168 1-0064: downloading firmware from file 'dvb-demod-si2168-b40-01.fw'
[   18.336750] si2168 1-0064: firmware version: B 4.0.11
[   18.346393] si2157 2-0060: found a 'Silicon Labs Si2157-A30'
[   18.368945] si2157 2-0060: firmware version: 3.0.5
[   23.136825] cx23885: cx23885[0]: mpeg risc op code error
[   23.138265] cx23885: cx23885[0]: TS1 B - dma channel status dump
[   23.138270] cx23885: cx23885[0]:   cmds: init risc lo   : 0x71cea000
[   23.138273] cx23885: cx23885[0]:   cmds: init risc hi   : 0x00000000
[   23.138277] cx23885: cx23885[0]:   cmds: cdt base       : 0x00010870
[   23.138280] cx23885: cx23885[0]:   cmds: cdt size       : 0x0000000a
[   23.138283] cx23885: cx23885[0]:   cmds: iq base        : 0x00010630
[   23.138287] cx23885: cx23885[0]:   cmds: iq size        : 0x00000010
[   23.138290] cx23885: cx23885[0]:   cmds: risc pc lo     : 0x71cea018
[   23.138293] cx23885: cx23885[0]:   cmds: risc pc hi     : 0x00000000
[   23.138297] cx23885: cx23885[0]:   cmds: iq wr ptr      : 0x00004192
[   23.138300] cx23885: cx23885[0]:   cmds: iq rd ptr      : 0x0000418c
[   23.138303] cx23885: cx23885[0]:   cmds: cdt current    : 0x00010898
[   23.138307] cx23885: cx23885[0]:   cmds: pci target lo  : 0x71d171f0
[   23.138310] cx23885: cx23885[0]:   cmds: pci target hi  : 0x00000000
[   23.138313] cx23885: cx23885[0]:   cmds: line / byte    : 0x00110000
[   23.138317] cx23885: cx23885[0]:   risc0: 
[   23.138318] 0x1c0002f0 [ write sol eol count=752 ]
[   23.138323] cx23885: cx23885[0]:   risc1: 
[   23.138324] 0x71d171f0 [ jump irq1 23 22 20 cnt0 14 13 12 count=496 ]
[   23.138330] cx23885: cx23885[0]:   risc2: 
[   23.138330] 0x00000000 [ INVALID count=0 ]
[   23.138334] cx23885: cx23885[0]:   risc3: 
[   23.138335] 0x1c0002f0 [ write sol eol count=752 ]
[   23.138340] cx23885: cx23885[0]:   (0x00010630) iq 0: 
[   23.138340] 0x18000250 [ write sol count=592 ]
[   23.138345] cx23885: cx23885[0]:   iq 1: 0x71d17db0 [ arg #1 ]
[   23.138348] cx23885: cx23885[0]:   iq 2: 0x00000000 [ arg #2 ]
[   23.138351] cx23885: cx23885[0]:   (0x0001063c) iq 3: 
[   23.138352] 0x140000a0 [ write eol count=160 ]
[   23.138356] cx23885: cx23885[0]:   iq 4: 0x71d12000 [ arg #1 ]
[   23.138360] cx23885: cx23885[0]:   iq 5: 0x00000000 [ arg #2 ]
[   23.138363] cx23885: cx23885[0]:   (0x00010648) iq 6: 
[   23.138363] 0x00000000 [ INVALID count=0 ]
[   23.138367] cx23885: cx23885[0]:   (0x0001064c) iq 7: 
[   23.138368] 0x1c0002f0 [ write sol eol count=752 ]
[   23.138372] cx23885: cx23885[0]:   iq 8: 0x71d174e0 [ arg #1 ]
[   23.138376] cx23885: cx23885[0]:   iq 9: 0x00000000 [ arg #2 ]
[   23.138379] cx23885: cx23885[0]:   (0x00010658) iq a: 
[   23.138379] 0x1c0002f0 [ write sol eol count=752 ]
[   23.138384] cx23885: cx23885[0]:   iq b: 0x71d177d0 [ arg #1 ]
[   23.138387] cx23885: cx23885[0]:   iq c: 0x00000000 [ arg #2 ]
[   23.138390] cx23885: cx23885[0]:   (0x00010664) iq d: 
[   23.138391] 0x1c0002f0 [ write sol eol count=752 ]
[   23.138395] cx23885: cx23885[0]:   iq e: 0x71d17ac0 [ arg #1 ]
[   23.138399] cx23885: cx23885[0]:   iq f: 0x00000000 [ arg #2 ]
[   23.138400] cx23885: cx23885[0]: fifo: 0x00005000 -> 0x6000
[   23.138401] cx23885: cx23885[0]: ctrl: 0x00010630 -> 0x10690
[   23.138404] cx23885: cx23885[0]:   ptr1_reg: 0x00005750
[   23.138408] cx23885: cx23885[0]:   ptr2_reg: 0x00010898
[   23.138411] cx23885: cx23885[0]:   cnt1_reg: 0x00000017
[   23.138414] cx23885: cx23885[0]:   cnt2_reg: 0x00000005

```

b-rad-NDi, 2022-02-01:
@sotiris-bos Can you help @sergiud by providing the script you're using to force the pcie slot to gen1 mode?

sotiris-bos, 2022-02-07:
> @sotiris-bos Can you help @sergiud by providing the script you're using to force the pcie slot to gen1 mode?

Yep, sorry for the late reply, here is the link to the script: https://www.alexforencich.com/wiki/en/pcie/set-speed

Usage example: ./pcie_set_speed.sh 45:00.0 1

45:00.0 = device
1 = pcie link speed