b-rad-NDi/Ubuntu-media-tree-kernel-builder #128: No Obvious Way to Getting a WinTV-QuadHD Working in Debian

prime-minister-of-fun, 2021-07-08:
I'm currently running Debian Bullseye (Testing) and a WinTV-QuadHD USB doesn't create anything in /dev like /dev/video0.  Kernel version is 5.10.40.  I just use Debian's repos.  Nothing fancy.

There's no support for Debian in your project. (Ironic) My attempt to build the media_tree repo at the v4l usinghttps://www.linuxtv.org/wiki/index.php/How_to_Obtain,_Build_and_Install_V4L-DVB_Device_Drivers as instructions fails.

ERROR: modpost: "__devm_regmap_init_sccb" [/home/jester/media-build/media_build/v4l/ov9650.ko] undefined!
ERROR: modpost: "__devm_regmap_init_sccb" [/home/jseter/media-build/media_build/v4l/ov772x.ko] undefined!

Deleting the files in the media_tree directory don't stop it from building, so, I'm assuming it's not even using the "git clone" files to build the modules???  It also doesn't seem to be using the .config file either as commenting out the modules doesn't stop them from being built.

Can you describe some general steps to getting the device to be recognized in Debian?  The patches are many and varied and I'm WAY out of my element discerning useful patches.  I'm fine doing some trial-and-error to figure it out, but it seems like the scripts to get the kernel patched require VERY specific knowledge.

prime-minister-of-fun, 2021-07-08:
Here's the post describing the same error.  But, the options in the post don't exist in make menu-config..  
https://github.com/tbsdtv/linux_media/issues/227

b-rad-NDi, 2021-09-02:
You can try and just install the ppa kernel packages. Ubuntu is a debian derivative, the kernel packages might "just work"

MrTopCat, 2021-09-23:
Going off what @b-rad-NDi mentioned, I used the ppa and after some trial and error I got this to work. 

I got this working on debian 10 using the bionic release. After adding the ppa, having no luck with installing the packages, I looked at the `.list` file I noticed it was pointing me to the `impish` release of Ubuntu (I didn't even know this existed). After some trial and error I changed the entry in `/etc/apt/sources.list.d/b-rad-ubuntu-kernel_mediatree_hauppauge-impish.list` from: 

```
deb http://ppa.launchpad.net/b-rad/kernel+mediatree+hauppauge/ubuntu impish main
```

To:

```
deb http://ppa.launchpad.net/b-rad/kernel+mediatree+hauppauge/ubuntu bionic main
```

After this run `sudo apt update` so you're pointing at the right release.

And then followed the instructions on [the hauppauge website](https://hauppauge.com/pages/support/support_linux.html). For my model I needed two packages;

```
sudo apt install linux-mediatree
sudo apt-get install linux-firmware-hauppauge
```