b-rad-NDi/Ubuntu-media-tree-kernel-builder #129: Ubuntu 20.04 issues

shspvr, 2021-08-05:
Hi bard  when get sometime USB quad-HD stop working I think right after kennel update 5.11.0-25 was install thank god ness the reg USB dual-HD still work

b-rad-NDi, 2021-09-02:
I have pushed up 20.04 HWE 5.11 kernel support today to my testing ppa. It will be in the main ppa by tomorrow.

shspvr, 2021-09-02:
> I have pushed up 20.04 HWE 5.11 kernel support today to my testing ppa. It will be in the main ppa by tomorrow.

Thanks I be looking forward to it

shspvr, 2021-09-05:
> I have pushed up 20.04 HWE 5.11 kernel support today to my testing ppa. It will be in the main ppa by tomorrow.

any news ?

b-rad-NDi, 2021-09-07:
A couple errors crept into one build. They're all up and ready now.

shspvr, 2021-09-07:
No go here the tuner light are not coming on at all so I guesting the firmware maybe not loading

b-rad-NDi, 2021-09-07:
Paste dmesg here. Which device is not working? You did install the firmware package?

shspvr, 2021-09-08:
The quad-HD the dual-HD works as NextPVR can see it 

[    0.000000] Linux version 5.11.0-342109010901-generic (buildd@lgw01-amd64-030) (gcc (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #1+mediatree+hauppauge~hwe-Ubuntu SMP Tue Sep 7 19:22:15 UTC 202 (Ubuntu 5.11.0-342109010901.1+mediatree+hauppauge~hwe-generic 5.11.22)
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-5.11.0-342109010901-generic root=UUID=37c807a1-ff7e-46b0-90af-4e2d3eec1617 ro quiet splash vt.handoff=7
[    0.000000] KERNEL supported cpus:
[    0.000000]   Intel GenuineIntel
[    0.000000]   AMD AuthenticAMD
[    0.000000]   Hygon HygonGenuine
[    0.000000]   Centaur CentaurHauls
[    0.000000]   zhaoxin   Shanghai  
[    0.000000] random: get_random_u32 called from bsp_init_amd+0x22a/0x250 with crng_init=0
[    0.000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x004: 'AVX registers'
[    0.000000] x86/fpu: xstate_offset[2]:  576, xstate_sizes[2]:  256
[    0.000000] x86/fpu: Enabled xstate features 0x7, context size is 832 bytes, using 'standard' format.
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009ffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000008d6f1fff] usable
[    0.000000] BIOS-e820: [mem 0x000000008d6f2000-0x000000008d9d2fff] reserved
[    0.000000] BIOS-e820: [mem 0x000000008d9d3000-0x000000008d9dcfff] ACPI data
[    0.000000] BIOS-e820: [mem 0x000000008d9dd000-0x000000008dd30fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x000000008dd31000-0x000000008e9dcfff] reserved
[    0.000000] BIOS-e820: [mem 0x000000008e9dd000-0x000000008e9ddfff] usable
[    0.000000] BIOS-e820: [mem 0x000000008e9de000-0x000000008ebe3fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x000000008ebe4000-0x000000008f04efff] usable
[    0.000000] BIOS-e820: [mem 0x000000008f04f000-0x000000008f7dffff] reserved
[    0.000000] BIOS-e820: [mem 0x000000008f7e0000-0x000000008f7fffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000fec00000-0x00000000fec00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec10000-0x00000000fec10fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed00000-0x00000000fed00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed40000-0x00000000fed44fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed80000-0x00000000fed8ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ff000000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100001000-0x000000023effffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] e820: update [mem 0x5e4c5018-0x5e4d4a57] usable ==> usable
[    0.000000] e820: update [mem 0x5e4c5018-0x5e4d4a57] usable ==> usable
[    0.000000] e820: update [mem 0x5e4b4018-0x5e4c4057] usable ==> usable
[    0.000000] e820: update [mem 0x5e4b4018-0x5e4c4057] usable ==> usable
[    0.000000] extended physical RAM map:
[    0.000000] reserve setup_data: [mem 0x0000000000000000-0x000000000009ffff] usable
[    0.000000] reserve setup_data: [mem 0x0000000000100000-0x000000005e4b4017] usable
[    0.000000] reserve setup_data: [mem 0x000000005e4b4018-0x000000005e4c4057] usable
[    0.000000] reserve setup_data: [mem 0x000000005e4c4058-0x000000005e4c5017] usable
[    0.000000] reserve setup_data: [mem 0x000000005e4c5018-0x000000005e4d4a57] usable
[    0.000000] reserve setup_data: [mem 0x000000005e4d4a58-0x000000008d6f1fff] usable
[    0.000000] reserve setup_data: [mem 0x000000008d6f2000-0x000000008d9d2fff] reserved
[    0.000000] reserve setup_data: [mem 0x000000008d9d3000-0x000000008d9dcfff] ACPI data
[    0.000000] reserve setup_data: [mem 0x000000008d9dd000-0x000000008dd30fff] ACPI NVS
[    0.000000] reserve setup_data: [mem 0x000000008dd31000-0x000000008e9dcfff] reserved
[    0.000000] reserve setup_data: [mem 0x000000008e9dd000-0x000000008e9ddfff] usable
[    0.000000] reserve setup_data: [mem 0x000000008e9de000-0x000000008ebe3fff] ACPI NVS
[    0.000000] reserve setup_data: [mem 0x000000008ebe4000-0x000000008f04efff] usable
[    0.000000] reserve setup_data: [mem 0x000000008f04f000-0x000000008f7dffff] reserved
[    0.000000] reserve setup_data: [mem 0x000000008f7e0000-0x000000008f7fffff] usable
[    0.000000] reserve setup_data: [mem 0x00000000fec00000-0x00000000fec00fff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000fec10000-0x00000000fec10fff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000fed00000-0x00000000fed00fff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000fed40000-0x00000000fed44fff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000fed80000-0x00000000fed8ffff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000ff000000-0x00000000ffffffff] reserved
[    0.000000] reserve setup_data: [mem 0x0000000100001000-0x000000023effffff] usable
[    0.000000] efi: EFI v2.31 by American Megatrends
[    0.000000] efi: ACPI=0x8d9d5000 ACPI 2.0=0x8d9d5000 SMBIOS=0xf04c0 MOKvar=0x82729000 
[    0.000000] secureboot: Secure boot disabled
[    0.000000] SMBIOS 2.7 present.
[    0.000000] DMI: Gigabyte Technology Co., Ltd. To be filled by O.E.M./F2A85X-D3H, BIOS F4b 06/14/2013
[    0.000000] tsc: Fast TSC calibration using PIT
[    0.000000] tsc: Detected 3793.386 MHz processor
[... 1090 lines cut ...]
[... cut ...]
[   53.010726] usb 4-1: device not accepting address 5, error -62
[   53.166012] usb 4-1: new full-speed USB device number 6 using ohci-pci
[   53.578699] usb 4-1: device not accepting address 6, error -62
[   53.578774] usb usb4-port1: unable to enumerate USB device
[  140.670712] usb 3-1: USB disconnect, device number 2
[  142.373168] usb 8-1: USB disconnect, device number 2
[  142.373180] usb 8-1.1: USB disconnect, device number 3
[  142.374110] usb 8-1.3: USB disconnect, device number 4
shspvr@Linux-TV:~$

b-rad-NDi, 2021-09-08:
Investigating. Must be a missing kernel config for mxl692.

shspvr, 2021-09-08:
Just so you know mine is ATSC tuner

b-rad-NDi, 2021-09-08:
usb quadHD right? I think there is only one model

shspvr, 2021-09-08:
> usb quadHD right? I think there is only one model

yes oh I guest no DVB model sorry I thought for sure there be one by now

b-rad-NDi, 2021-09-08:
snafu on my part, patched the wrong kernel config for hwe.

b-rad-NDi, 2021-09-08:
Hmmm....actually my usb quadHD driver just isn't in 5.11. I guess I thought it was. My bad, I've gotta regroup.

shspvr, 2021-09-08:
> Hmmm....actually my usb quadHD driver just isn't in 5.11. I guess I thought it was. My bad, I've gotta regroup.

Ok

poopingpenis, 2021-09-10:
When I try installing linux-hwe-mediatree on 20.04 raspi I get

`E: Unable to locate package linux-hwe-mediatree`

When I try installing linux-mediatree I get

```
linux-mediatree : Depends: linux-image-5.4.0-802109010901-generic but it is n``ot installable
                   Depends: linux-modules-5.4.0-802109010901-generic but it is not installable
                   Depends: linux-modules-extra-5.4.0-802109010901-generic but it is not installable
                   Depends: linux-headers-5.4.0-802109010901-generic but it is not installable
E: Unable to correct problems, you have held broken packages.
```

I'm also using a WinTV-quadHD :(

b-rad-NDi, 2021-09-11:
@poopingpenis My ppa is for amd64 and i386 only. If you want raspi aarch64 builds you'll have to download them from the releases page of this repo and install them manually.

b-rad-NDi, 2021-09-11:
@shspvr All new builds with fresh media tree sync are in my testing ppa now. I won't have a chance to fully test them until monday, but you're free to grab and test drive.

shspvr, 2021-09-12:
> @shspvr All new builds with fresh media tree sync are in my testing ppa now. I won't have a chance to fully test them until monday, but you're free to grab and test drive.

Fresh Install Ubuntu 20.04.3
sudo add-apt-repository ppa:b-rad/kernel+mediatree+hauppauge
sudo apt-get update

the got to
shspvr@Linux-TV:~$ sudo apt-get install linux-hwe-mediatree
[sudo] password for shspvr: 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package linux-hwe-mediatree
shspvr@Linux-TV:~$

shspvr, 2021-09-16:
> Testing ppa is located here:
> 
> https://launchpad.net/~b-rad/+archive/ubuntu/test-test-hw-staging
> 
> I'll be validating all the new builds today and migrating them to the main ppa.

Ok I don't when I be able to get to it as are internet was most of the week

b-rad-NDi, 2021-09-16:
All releases are up in the main ppa now

shspvr, 2021-09-17:
> All releases are up in the main ppa now

Was able to install but same bad new for you when I do scan it goes real fast and it not pick up any channel

b-rad-NDi, 2021-09-17:
Odd. I'll install it on an actual machine and give it a test.

shspvr, 2021-09-17:
> Odd. I'll install it on an actual machine and give it a test.

No idea but it new fresh install of Ubuntu 20.04.3 LTS even Kaffeine don't work the dual-HD work fine

b-rad-NDi, 2021-09-17:
So, the usb QuadHD does not work, but dualHD still does? I'll get to the bottom of it.

shspvr, 2021-09-17:
Here output of dmesg if it is any help
shspvr@Linux-TV: $ dmesg
[    0.000000] Linux version 5.11.0-342109090909-generic (buildd@lcy01-amd64-021) (gcc (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #1+mediatree+hauppauge~hwe-Ubuntu SMP Mon Sep 13 19:03:05 UTC 20 (Ubuntu 5.11.0-342109090909.1+mediatree+hauppauge~hwe-generic 5.11.22)
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-5.11.0-342109090909-generic root=UUID=e38b2bf3-6548-4e0e-94ef-25f472c20138 ro quiet splash vt.handoff=7
[    0.000000] KERNEL supported cpus:
[    0.000000]   Intel GenuineIntel
[    0.000000]   AMD AuthenticAMD
[    0.000000]   Hygon HygonGenuine
[    0.000000]   Centaur CentaurHauls
[    0.000000]   zhaoxin   Shanghai  
[    0.000000] random: get_random_u32 called from bsp_init_amd+0x22a/0x250 with crng_init=0
[    0.000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x004: 'AVX registers'
[    0.000000] x86/fpu: xstate_offset[2]:  576, xstate_sizes[2]:  256
[    0.000000] x86/fpu: Enabled xstate features 0x7, context size is 832 bytes, using 'standard' format.
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009ffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000008d6f1fff] usable
[    0.000000] BIOS-e820: [mem 0x000000008d6f2000-0x000000008d9d2fff] reserved
[    0.000000] BIOS-e820: [mem 0x000000008d9d3000-0x000000008d9dcfff] ACPI data
[    0.000000] BIOS-e820: [mem 0x000000008d9dd000-0x000000008dd30fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x000000008dd31000-0x000000008e9dcfff] reserved
[    0.000000] BIOS-e820: [mem 0x000000008e9dd000-0x000000008e9ddfff] usable
[    0.000000] BIOS-e820: [mem 0x000000008e9de000-0x000000008ebe3fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x000000008ebe4000-0x000000008f04efff] usable
[    0.000000] BIOS-e820: [mem 0x000000008f04f000-0x000000008f7dffff] reserved
[    0.000000] BIOS-e820: [mem 0x000000008f7e0000-0x000000008f7fffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000fec00000-0x00000000fec00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec10000-0x00000000fec10fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed00000-0x00000000fed00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed40000-0x00000000fed44fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed80000-0x00000000fed8ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ff000000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100001000-0x000000023effffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] e820: update [mem 0x5e4e1018-0x5e4f0a57] usable ==> usable
[    0.000000] e820: update [mem 0x5e4e1018-0x5e4f0a57] usable ==> usable
[    0.000000] e820: update [mem 0x5e4d0018-0x5e4e0057] usable ==> usable
[    0.000000] e820: update [mem 0x5e4d0018-0x5e4e0057] usable ==> usable
[    0.000000] extended physical RAM map:
[    0.000000] reserve setup_data: [mem 0x0000000000000000-0x000000000009ffff] usable
[    0.000000] reserve setup_data: [mem 0x0000000000100000-0x000000005e4d0017] usable
[    0.000000] reserve setup_data: [mem 0x000000005e4d0018-0x000000005e4e0057] usable
[    0.000000] reserve setup_data: [mem 0x000000005e4e0058-0x000000005e4e1017] usable
[    0.000000] reserve setup_data: [mem 0x000000005e4e1018-0x000000005e4f0a57] usable
[    0.000000] reserve setup_data: [mem 0x000000005e4f0a58-0x000000008d6f1fff] usable
[    0.000000] reserve setup_data: [mem 0x000000008d6f2000-0x000000008d9d2fff] reserved
[    0.000000] reserve setup_data: [mem 0x000000008d9d3000-0x000000008d9dcfff] ACPI data
[    0.000000] reserve setup_data: [mem 0x000000008d9dd000-0x000000008dd30fff] ACPI NVS
[    0.000000] reserve setup_data: [mem 0x000000008dd31000-0x000000008e9dcfff] reserved
[    0.000000] reserve setup_data: [mem 0x000000008e9dd000-0x000000008e9ddfff] usable
[    0.000000] reserve setup_data: [mem 0x000000008e9de000-0x000000008ebe3fff] ACPI NVS
[    0.000000] reserve setup_data: [mem 0x000000008ebe4000-0x000000008f04efff] usable
[    0.000000] reserve setup_data: [mem 0x000000008f04f000-0x000000008f7dffff] reserved
[    0.000000] reserve setup_data: [mem 0x000000008f7e0000-0x000000008f7fffff] usable
[    0.000000] reserve setup_data: [mem 0x00000000fec00000-0x00000000fec00fff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000fec10000-0x00000000fec10fff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000fed00000-0x00000000fed00fff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000fed40000-0x00000000fed44fff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000fed80000-0x00000000fed8ffff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000ff000000-0x00000000ffffffff] reserved
[    0.000000] reserve setup_data: [mem 0x0000000100001000-0x000000023effffff] usable
[    0.000000] efi: EFI v2.31 by American Megatrends
[    0.000000] efi: ACPI=0x8d9d5000 ACPI 2.0=0x8d9d5000 SMBIOS=0xf04c0 MOKvar=0x82723000 
[    0.000000] secureboot: Secure boot disabled
[    0.000000] SMBIOS 2.7 present.
[    0.000000] DMI: Gigabyte Technology Co., Ltd. To be filled by O.E.M./F2A85X-D3H, BIOS F4b 06/14/2013
[    0.000000] tsc: Fast TSC calibration using PIT
[    0.000000] tsc: Detected 3793.438 MHz processor
[... 1146 lines cut ...]
[... cut ...]
[  101.560539] scsi 8:0:0:0: Direct-Access     PNY      USB 2.0 FD       1100 PQ: 0 ANSI: 4
[  101.561102] sd 8:0:0:0: Attached scsi generic sg4 type 0
[  101.563238] sd 8:0:0:0: [sdd] 15950592 512-byte logical blocks: (8.17 GB/7.61 GiB)
[  101.563961] sd 8:0:0:0: [sdd] Write Protect is off
[  101.563968] sd 8:0:0:0: [sdd] Mode Sense: 43 00 00 00
[  101.564718] sd 8:0:0:0: [sdd] No Caching mode page found
[  101.564730] sd 8:0:0:0: [sdd] Assuming drive cache: write through
[  101.589979]  sdd: sdd1
[  101.628330] sd 8:0:0:0: [sdd] Attached SCSI removable disk

b-rad-NDi, 2021-09-17:
This is definitely the issue:
```
[ 17.969963] mxl692 19-0063: Direct firmware load for dvb-demod-mxl692.fw failed with error -2
[ 18.126300] mxl692 17-0060: Direct firmware load for dvb-demod-mxl692.fw failed with error -2
[ 18.285433] mxl692 15-0063: Direct firmware load for dvb-demod-mxl692.fw failed with error -2
[ 18.445430] mxl692 6-0060: Direct firmware load for dvb-demod-mxl692.fw failed with error -2
```

b-rad-NDi, 2021-09-17:
-2 is ENOENT, which indicates no such file or directory. Do you have linux-firmware-hauppauge installed?

What is md5sum of /lib/firmware/dvb-demod-mxl692.fw ?

shspvr, 2021-09-17:
I forgot that you haven't included it as part of the process package so it fix now

closed 2021-09-17 by shspvr

b-rad-NDi, 2021-09-17:
Excellent.