b-rad-NDi/Ubuntu-media-tree-kernel-builder #132: Running inside Ubuntu VM on Proxmox (Hauppauge WinTV-QuadHD)

samip5, 2021-09-21:
I'm seeing wierd issues with the driver or my card, as it was working fine previously and I don't know what changed.
My vm host is running Promox 7 with intel_ionmu enabled and the DVB card passed though to the VM.

dmesg log snippet:
```
[  545.185986] cx23885: cx23885[0]: mpeg risc op code error
[  545.186035] cx23885: cx23885[0]: TS1 B - dma channel status dump
[  545.186039] cx23885: cx23885[0]:   cmds: init risc lo   : 0x32596000
[  545.186042] cx23885: cx23885[0]:   cmds: init risc hi   : 0x00000000
[  545.186045] cx23885: cx23885[0]:   cmds: cdt base       : 0x00010870
[  545.186048] cx23885: cx23885[0]:   cmds: cdt size       : 0x0000000a
[  545.186051] cx23885: cx23885[0]:   cmds: iq base        : 0x00010630
[  545.186054] cx23885: cx23885[0]:   cmds: iq size        : 0x00000010
[  545.186057] cx23885: cx23885[0]:   cmds: risc pc lo     : 0x32596018
[  545.186060] cx23885: cx23885[0]:   cmds: risc pc hi     : 0x00000000
[  545.186062] cx23885: cx23885[0]:   cmds: iq wr ptr      : 0x00004192
[  545.186065] cx23885: cx23885[0]:   cmds: iq rd ptr      : 0x0000418c
[  545.186068] cx23885: cx23885[0]:   cmds: cdt current    : 0x00010888
[  545.186080] cx23885: cx23885[0]:   cmds: pci target lo  : 0x32afe630
[  545.186083] cx23885: cx23885[0]:   cmds: pci target hi  : 0x00000000
[  545.186086] cx23885: cx23885[0]:   cmds: line / byte    : 0x000d0000
[  545.186098] cx23885: cx23885[0]:   risc0: 
[  545.186106] 0x1c0002f0 [ write sol eol count=752 ]
[  545.186110] cx23885: cx23885[0]:   risc1: 
[  545.186111] 0x32afe340 [ INVALID irq2 23 21 19 18 cnt1 cnt0 resync 14 13 count=832 ]
```

dmesg|grep -i dvb:
```
[    2.949888] cx23885: CORE cx23885[0]: subsystem: 0070:6a28, board: Hauppauge WinTV-QuadHD-DVB [card=56,autodetected]
[    3.299366] tveeprom: TV standards PAL(B/G) PAL(I) SECAM(L/L') PAL(D/D1/K) ATSC/DVB Digital (eeprom 0xf4)
[    3.299369] cx23885: cx23885_dvb_register() allocating 1 frontend(s)
[    3.299370] cx23885: cx23885[0]: cx23885 based dvb card
[    3.299373] cx23885: dvb_register(): board=56 port=1
[    3.315367] dvbdev: DVB: registering new adapter (cx23885[0])
[    3.315369] cx23885 0000:00:10.0: DVB: registering adapter 0 frontend 0 (Silicon Labs Si2168)...
[    3.315912] cx23885: cx23885_dvb_register() allocating 1 frontend(s)
[    3.315912] cx23885: cx23885[0]: cx23885 based dvb card
[    3.315915] cx23885: dvb_register(): board=56 port=2
[    3.327099] dvbdev: DVB: registering new adapter (cx23885[0])
[    3.327101] cx23885 0000:00:10.0: DVB: registering adapter 1 frontend 0 (Silicon Labs Si2168)...
[    9.122615] si2168 1-0066: downloading firmware from file 'dvb-demod-si2168-b40-01.fw'
[    9.794382] cx23885 0000:00:10.0: DVB: adapter 1 frontend 0 frequency 0 out of range (48000000..870000000)
[    9.803662] si2168 1-0066: downloading firmware from file 'dvb-demod-si2168-b40-01.fw'
[    9.806426] si2168 1-0064: downloading firmware from file 'dvb-demod-si2168-b40-01.fw'
[   11.163959] cx23885 0000:00:10.0: DVB: adapter 0 frontend 0 frequency 0 out of range (48000000..870000000)
[   11.173058] si2168 1-0064: downloading firmware from file 'dvb-demod-si2168-b40-01.fw'
```

b-rad-NDi, 2022-01-27:
This is an issue that crops up for some people on occasion, depending on their system. Have you tried any of the reported working solutions? Like the dma workaround kernel parameter or forcing your pcie slot to gen 1 in the bios?

b-rad-NDi, 2022-01-27:
Also, we don't really support use of the hardware in vm's. It works for a lot of people, but there is a lot involved and things can break even in so called "pass thru" modes

samip5, 2022-01-27:
Yeah, I don't know what it actually was, but after a fresh Ubuntu 20.04 and the PPA driver aka mediatree kernel, it works again. None of the reported "working" solutions worked for me before fresh install, as I did try the dma workaround too.
I tried before fresh installing pretty much all of the workarounds (except forcing pcie slot), in a VM and non-vm environment but no dice.

For record working with `uname -a`: `Linux home-srv 5.4.0-882110311031-generic #0+mediatree+hauppauge-Ubuntu SMP Sat Nov 6 00:51:47 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux`

closed 2022-01-27 by samip5