b-rad-NDi/Ubuntu-media-tree-kernel-builder #133: quadHD can't find any sources

poopingpenis, 2021-09-25:
I have WinTV-quadHD USB on Ubuntu 20.04 for the Raspberry Pi scanning for sources with both w-scan and tvheadend with the pre-defined "United States: us-ATSC-center-frequencies-8VSB" muxes and it receives nothing.

I am using the card in Puerto Rico and connecting the same antenna directly to my TV picks up around 32(!) channels. :( What can I do?

shspvr, 2021-09-25:
Try this sudo apt-get install linux-firmware-hauppauge

poopingpenis, 2021-09-25:
> If 2 of the tuner light are not on then try this sudo apt-get install linux-firmware-hauppauge

There are indeed only two lights on, I'll try this; it works for the RasPi on Ubuntu 20.04 arm64, right?

shspvr, 2021-09-26:
> > If 2 of the tuner light are not on then try this sudo apt-get install linux-firmware-hauppauge
> 
> There are indeed only two lights on, I'll try this; it works for the RasPi on Ubuntu 20.04 arm64, right?
> 
> EDIT: Unable to locate package linux-firmware-hauppauge

That should be right you only see 2 main light on boot up beside the power light unless one of other tuner is being use by the what ever app your using then the other will come on too it could be RasPi thing I only run req PC ver of Ubuntu not arm

poopingpenis, 2021-09-26:
I did install the deb drivers, so I think it shouldn't be necessary anyway? And the device is recognised by tvheadend.

shspvr, 2021-09-26:
> I did install the deb drivers, so I think it shouldn't be necessary anyway? And the device is recognised by tvheadend and w_scan.

You need 20.04.1 or 20.04.2 LTS and DON'T UPDATE THE KERNEL now if it is 20.04.3 LTS package you need is newer arm64 package for it which Bard hasn't update the Raspberry Pi Ubuntu builds for that latest Linux kernel that could be why it no working

poopingpenis, 2021-09-26:
> You need 20.04.1 or 20.04.2 LTS and DON'T UPDATE THE KERNEL now if it is 20.04.3 LTS package you need is newer arm64 package for it which Bard hasn't update the Raspberry Pi Ubuntu builds for that [...]

I do indeed have Ubuntu 20.04.3 LTS with GNU/Linux 5.4.0-592101082101-raspi aarch64 kernel :(

b-rad-NDi, 2021-09-29:
I'll prioritize building new RPi kernels soon.

poopingpenis, 2021-10-02:
> I'll prioritize building new RPi kernels soon.

awesome :)

poopingpenis, 2021-10-14:
polite reminder pls

poopingpenis, 2021-11-03:
My kernel got updated recently to Ubuntu 20.04.3 LTS (GNU/Linux 5.4.0-592101082101-raspi aarch64).

Don't mean to rush I know it must be hard work but I haven't been able to use this since I got it ;w;

b-rad-NDi, 2022-01-27:
I will get this building today or tomorrow. Apologies.

poopingpenis, 2022-02-05:
> I will get this building today or tomorrow. Apologies.

Appreciate it. :)

emveepee, 2022-02-06:
> Appreciate it. :)

@poopingpenis could you please change your username.  This is a respectable adult support and and I don't like getting the email notifications.  Hopefully b-rad will stop responding to your offensive name which children might think funny.

poopingpenis, 2022-02-16:
> > Appreciate it. :)
> 
> @poopingpenis could you please change your username. This is a respectable adult support and and I don't like getting the email notifications. Hopefully b-rad will stop responding to your offensive name which children might think funny.

I've used this name for too long and tied it with too many projects to change it right now. Maybe you can mute this issue...

b-rad-NDi, 2022-02-16:
A name is a name as far as I'm concerned. I'm trying to prioritize this for you. I'll try and set up a build system today.

b-rad-NDi, 2022-03-03:
This has been derailed a bit. I have the kernels patches up enough to compile, but now Ubuntu raspi requires more than a 32GB sdcard to full build everything. I have several 64GB cards on order so I can build everything.

b-rad-NDi, 2022-03-09:
Okie. Done. Please let me know if there are any issues.

Note: you must install linux-firmware-hauppauge from my ppa unless you obtain your firmwares elsewhere.

b-rad-NDi, 2022-03-09:
Just so you know. Building each kernel package natively, on RPi4, because that is required takes 4-6 hours. The process is a slow ordeal.

shspvr, 2022-03-10:
> Just so you know. Building each kernel package natively, on RPi4, because that is required takes 4-6 hours. The process is a slow ordeal.

WoW that long to bad there not away do with desktop PC