b-rad-NDi/Ubuntu-media-tree-kernel-builder #134: HVR 5525 DVB-S tuner not working

jlaba, 2021-12-05:
The DVB-T/DVB-C tuner is showing up as an adapter in /dev, but the DVB-S not. In the dmesg log (see below), it says, that it tries to load the m88ds3103 instead of the m88rs6000t mentioned in linuxtv.org wiki. Loading of the m88ds3103 fails also.

I'm running Gentoo with kernel 5.10.76, but also tested libreelec and libreelec test image with kernel 5.15. The problem is the same.

If you need any further input, please let me know! 

[ 11.840919] cx25840 6-0044: loaded v4l-cx23885-avcore-01.fw firmware (16382 bytes)
[ 11.920093] cx23885: cx23885[0]: registered device video0 [v4l2]
[ 11.920182] cx23885: cx23885[0]: registered device vbi0
[ 11.920349] cx23885: cx23885[0]: alsa: registered ALSA audio device
[ 11.920351] cx23885: cx23885_dvb_register() allocating 1 frontend(s)
[ 11.920353] cx23885: cx23885[0]: cx23885 based dvb card
[ 11.920706] m88ds3103 4-0069: Unknown device. Chip_id=00
[ 11.920728] cx23885: cx23885[0]: frontend initialization failed
[ 11.920730] cx23885: cx23885_dvb_register() dvb_register failed err = -22
[ 11.920731] cx23885: cx23885_dev_setup() Failed to register dvb adapters on VID_B
[ 11.920732] cx23885: cx23885_dvb_register() allocating 1 frontend(s)
[ 11.920733] cx23885: cx23885[0]: cx23885 based dvb card
[ 11.932121] i2c i2c-4: Added multiplexed i2c bus 7
[ 11.932126] si2168 4-0064: Silicon Labs Si2168-B40 successfully identified
[ 11.932127] si2168 4-0064: firmware version: B 4.0.2
[ 11.955354] si2157 5-0060: Silicon Labs Si2147/2148/2157/2158 successfully attached
[ 11.955378] dvbdev: DVB: registering new adapter (cx23885[0])
[ 11.955382] cx23885 0000:01:00.0: DVB: registering adapter 0 frontend 0 (Silicon Labs Si2168)...
[ 11.955752] cx23885: cx23885_dev_checkrevision() Hardware revision = 0xd0
[ 11.955758] cx23885: cx23885[0]/0: found at 0000:01:00.0, rev: 4, irq: 22, latency: 0, mmio: 0xa1000000
[ 14.818436] r8169 0000:03:00.0 eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[ 14.818462] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[ 14.860679] 8021q: 802.1Q VLAN Support v1.8
[ 22.337493] logitech-hidpp-device 0003:046D:4024.0004: HID++ 2.0 device connected.
[ 156.803608] si2168 4-0064: downloading firmware from file 'dvb-demod-si2168-b40-01.fw'
[ 158.267824] si2168 4-0064: firmware version: B 4.0.25
[ 158.274242] si2157 5-0060: found a 'Silicon Labs Si2157-A30'
[ 158.323799] si2157 5-0060: firmware version: 3.0.5
[ 158.326878] cx23885 0000:01:00.0: DVB: adapter 0 frontend 0 frequency 0 out of range (48000000..870000000)
[ 401.499754] si2168 4-0064: downloading firmware from file 'dvb-demod-si2168-b40-01.fw'
[ 402.962388] si2168 4-0064: firmware version: B 4.0.25
[ 402.969376] si2157 5-0060: found a 'Silicon Labs Si2157-A30'
[ 402.991805] si2157 5-0060: firmware version: 3.0.5

Fabri-73, 2022-01-03:
Hi, 
I'm having the same issue with Ubuntu 20.04  :

5.11.0-43-generic #47~20.04.2-Ubuntu SMP Mon Dec 13 11:06:56 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux

[    5.843188] cx23885: cx23885 driver version 0.0.4 loaded
[    5.843225] cx23885 0000:03:00.0: enabling device (0100 -> 0102)
[    5.843333] cx23885: cx23885_dev_setup: cx23885_dev_setup() Memory configured for PCIe bridge type 888
[    5.843335] cx23885: cx23885_init_tsport: cx23885_init_tsport(portno=1)
[    5.843336] cx23885: cx23885_init_tsport: cx23885_init_tsport(portno=2)
[    5.843350] cx23885: CORE cx23885[0]: subsystem: 0070:f038, board: Hauppauge WinTV-HVR5525 [card=52,autodetected]
[    5.843353] cx23885: cx23885_pci_quirks: cx23885_pci_quirks()
[    5.843354] cx23885: cx23885_dev_setup: cx23885_dev_setup() tuner_type = 0x4 tuner_addr = 0x0 tuner_bus = 0
[    5.843355] cx23885: cx23885_dev_setup: cx23885_dev_setup() radio_type = 0x0 radio_addr = 0x0
[    5.843357] cx23885: cx23885_reset: cx23885_reset()
....
[    6.897471] cx23885: cx23885[0]: registered device video0 [v4l2]
[    6.897625] cx23885: cx23885[0]: registered device vbi0
[    6.897627] cx23885: cx23885_dvb_register() allocating 1 frontend(s)
[    6.897629] cx23885: cx23885[0]: cx23885 based dvb card
[    6.897944] m88ds3103 1-0069: Unknown device. Chip_id=00
[    6.897957] cx23885: cx23885[0]: frontend initialization failed
[    6.897958] cx23885: cx23885_dvb_register() dvb_register failed err = -22
[    6.897959] cx23885: cx23885_dev_setup() Failed to register dvb adapters on VID_B
[    6.897960] cx23885: cx23885_dvb_register() allocating 1 frontend(s)

Any Idea?

BR.!

jlaba, 2022-01-05:
I looked around in the code a bit. In file cx23885-dvb.c from cx23885 kernel driver it fails.
```
case 1:
	/* attach frontend */
	fe0->dvb.frontend = dvb_attach(m88ds3103_attach,
			&hauppauge_hvr5525_m88ds3103_config,
			&i2c_bus->i2c_adap, &adapter);
	if (fe0->dvb.frontend == NULL)
		break;
```

The problem is in file m88ds3103.c in the switch block, where it tries to figure out, what frontend chip is there (M88RS6000 or M88DS3103). It has no valid chip id.
```
switch (dev->chip_id) {
	case M88RS6000_CHIP_ID:
	case M88DS3103_CHIP_ID:
		break;
	default:
		ret = -ENODEV;
		dev_err(&client->dev, "Unknown device. Chip_id=%02x\n", dev->chip_id);
		goto err_kfree;
	}
```

So the problem is one of:

- Different chip on the card then M88DS3103 or M88RS6000
- Failed to read the correct chip id and it is an M88DS3103 or M88RS6000 with the known IDs
- Either M88DS3103 or M88RS6000 with an unknown ID

I appreciate any hints, how to figure out, which of the possible problems it is.

Fabri-73, 2022-01-06:
Hi, 

Well i can only describe my own case.
According to the code in 
https://elixir.bootlin.com/linux/v5.11/source/drivers/media/dvb-frontends/m88ds3103.c#L1816

The value of the dev->chip_id is read via i2c just 2 lines above the switch case.
```
	/* 0x00: chip id[6:0], 0x01: chip ver[7:0], 0x02: chip ver[15:8] */
	ret = regmap_read(dev->regmap, 0x00, &utmp);
	if (ret)
		goto err_kfree;

	dev->chip_id = utmp >> 1;
	dev->chiptype = (u8)id->driver_data;
```

So I enabled i2c_debug for the module cx23885 (using options cx23885 i2c_debug=1) and i can see the returned value of the card over i2c is not valid, at least not the expected one:
```
[    6.980613] kernel: cx23885: cx23885[0]: registered device video0 [v4l2]
[    6.980687] kernel: cx23885: cx23885[0]: registered device vbi0
[    6.980857] kernel: cx23885: cx23885[0]: alsa: registered ALSA audio device
[    6.980859] kernel: cx23885: cx23885_dvb_register() allocating 1 frontend(s)
[    6.980860] kernel: cx23885: cx23885[0]: cx23885 based dvb card
[    6.980917] kernel: cx23885: i2c_xfer: i2c:i2c_xfer(num = 2)
[    6.980919] kernel: cx23885: i2c_xfer: i2c:i2c_xfer(num = 2) addr = 0x69  len = 0x1
[    6.980921] kernel: cx23885: i2c_sendbytes: i2c:i2c_sendbytes(msg->wlen=1, nextmsg->rlen=1)
[    6.981057] kernel:  <W d2 00
[    6.981058] kernel: cx23885: i2c_readbytes: i2c: R
[    6.981195] kernel: cx23885: i2c_readbytes: i2c: 00
[    6.981195] kernel: cx23885: i2c_readbytes: i2c: >
[    6.981197] kernel: m88ds3103 1-0069: Unknown device. Chip_id=00
[    6.981211] kernel: cx23885: cx23885[0]: frontend initialization failed
[    6.981212] kernel: cx23885: cx23885_dvb_register() dvb_register failed err = -22
[    6.981213] kernel: cx23885: cx23885_dev_setup() Failed to register dvb adapters on VID_B
```

As a comparison the value returned for the DVB-C/T card which is correctly detected:
```
[    6.981214] kernel: cx23885: cx23885_dvb_register() allocating 1 frontend(s)
[    6.981215] kernel: cx23885: cx23885[0]: cx23885 based dvb card
[    6.983606] kernel: cx23885: i2c_xfer: i2c:i2c_xfer(num = 1)
[    6.983608] kernel: cx23885: i2c_xfer: i2c:i2c_xfer(num = 1) addr = 0x64  len = 0xd
[    6.983612] kernel: cx23885: i2c_sendbytes: i2c:i2c_sendbytes(msg->len=13)
[    6.983815] kernel:  <W c8 c0 12 00 0c 00 0d 16 00 00 00 00 00 00 >
[    6.985076] kernel: cx23885: i2c_xfer: i2c:i2c_xfer(num = 1)
[    6.985076] kernel: cx23885: i2c_xfer: i2c:i2c_xfer(num = 1) addr = 0x64  len = 0x8
[    6.985077] kernel: cx23885: i2c_sendbytes: i2c:i2c_sendbytes(msg->len=8)
[    6.985280] kernel:  <W c8 c0 06 01 0f 00 20 20 01 >
[    6.986029] kernel: cx23885: i2c_xfer: i2c:i2c_xfer(num = 1)
[    6.986029] kernel: cx23885: i2c_xfer: i2c:i2c_xfer(num = 1) addr = 0x64  len = 0x1
[    6.986030] kernel: cx23885: i2c_readbytes: i2c:i2c_readbytes(msg->len=1)
[    6.986031] kernel: cx23885: i2c_readbytes: i2c: <R c9
[    6.986267] kernel: cx23885: i2c_readbytes: i2c: 00
[    6.986268] kernel: cx23885: i2c_readbytes: i2c: >
[    6.986269] kernel: cx23885: i2c_xfer: i2c:i2c_xfer(num = 1)
[    6.986269] kernel: cx23885: i2c_xfer: i2c:i2c_xfer(num = 1) addr = 0x64  len = 0x1
[    6.986270] kernel: cx23885: i2c_readbytes: i2c:i2c_readbytes(msg->len=1)
[    6.986271] kernel: cx23885: i2c_readbytes: i2c: <R c9
[    6.986507] kernel: cx23885: i2c_readbytes: i2c: 00
[    6.986508] kernel: cx23885: i2c_readbytes: i2c: >
[    6.986508] kernel: cx23885: i2c_xfer: i2c:i2c_xfer(num = 1)
[    6.986509] kernel: cx23885: i2c_xfer: i2c:i2c_xfer(num = 1) addr = 0x64  len = 0x1
[    6.986510] kernel: cx23885: i2c_readbytes: i2c:i2c_readbytes(msg->len=1)
[    6.986510] kernel: cx23885: i2c_readbytes: i2c: <R c9
[    6.986747] kernel: cx23885: i2c_readbytes: i2c: 80
[    6.986747] kernel: cx23885: i2c_readbytes: i2c: >
```

I tried then using i2cdetect/i2cdump to show the bytes available in the i2c table 
```
i2c-3	i2c       	cx23885[0]                      	I2C adapter
i2c-10	i2c       	NVIDIA i2c adapter 11 at 1:00.0 	I2C adapter
i2c-1	i2c       	cx23885[0]                      	I2C adapter

```
Checking with i2cdump the DVB-S device card (in my case is i2c-3 with address 0x69) i see also that all returned values are set to 
60... 
Is this an indication of an hardware issue? Any idea/advice?

Best Regards!

jlaba, 2022-01-06:
My devices are
```
i2c-3   i2c             i915 gmbus misc                         I2C adapter
i2c-1   i2c             i915 gmbus dpb                          I2C adapter
i2c-6   i2c             cx23885[0]                              I2C adapter
i2c-4   i2c             cx23885[0]                              I2C adapter
i2c-2   i2c             i915 gmbus dpc                          I2C adapter
i2c-0   smbus           SMBus I801 adapter at f040              SMBus adapter
i2c-7   i2c             i2c-4-mux (chan_id 0)                   I2C adapter
i2c-5   i2c             cx23885[0]                              I2C adapter
```

And the output of i2cdetect for bus 6 is:
```
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: 30 31 32 33 34 35 36 37 -- -- -- -- -- -- -- -- 
40: -- -- -- -- UU -- -- -- -- -- -- -- 4c -- -- -- 
50: 50 51 52 53 54 55 56 57 58 59 5a 5b 5c 5d 5e 5f 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- -- 
```

for bus 4:
```
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- 0b -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: 30 31 32 33 34 35 36 37 -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: 50 51 52 53 54 55 56 57 58 59 5a 5b 5c 5d 5e 5f 
60: -- -- -- -- UU -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- -- 
```

And for bus 7:
```
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: 30 31 32 33 34 35 36 37 -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: 50 51 52 53 54 55 56 57 58 59 5a 5b 5c 5d 5e 5f 
60: UU -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- -- 
```

wakkorottix, 2022-01-23:
Same Problem with Starburst 2 (DVB-S/DVB-S2 only) with Ubuntu 20.04.
I've installed Haupauge PPA with kernel + firmware but still no /dev/dvb device

dmesg:
[So Jan 23 11:43:05 2022] cx23885: cx23885 driver version 0.0.4 loaded
[So Jan 23 11:43:05 2022] cx23885 0000:01:00.0: enabling device (0000 -> 0002)
[So Jan 23 11:43:05 2022] cx23885: CORE cx23885[0]: subsystem: 0070:f02a, board: Hauppauge WinTV-Starburst2 [card=59,autodetected]
[So Jan 23 11:43:06 2022] tveeprom: Hauppauge model 150300, rev C2I9, serial# 4036172330
[So Jan 23 11:43:06 2022] tveeprom: MAC address is 00:0d:fe:93:1a:2a
[So Jan 23 11:43:06 2022] tveeprom: tuner model is unknown (idx 189, type 4)
[So Jan 23 11:43:06 2022] tveeprom: TV standards UNKNOWN (eeprom 0x01)
[So Jan 23 11:43:06 2022] tveeprom: audio processor is CX23888 (idx 40)
[So Jan 23 11:43:06 2022] tveeprom: decoder processor is CX23888 (idx 34)
[So Jan 23 11:43:06 2022] tveeprom: has no radio, has IR receiver, has no IR transmitter
[So Jan 23 11:43:06 2022] cx23885: cx23885[0]: warning: unknown hauppauge model #150300
[So Jan 23 11:43:06 2022] cx23885: cx23885[0]: hauppauge eeprom: model=150300
[So Jan 23 11:43:06 2022] cx23885: cx23885_dvb_register() allocating 1 frontend(s)
[So Jan 23 11:43:06 2022] cx23885: cx23885[0]: cx23885 based dvb card
[So Jan 23 11:43:06 2022] m88ds3103 4-0069: Unknown device. Chip_id=00
[So Jan 23 11:43:06 2022] cx23885: cx23885[0]: frontend initialization failed
[So Jan 23 11:43:06 2022] cx23885: cx23885_dvb_register() dvb_register failed err = -22
[So Jan 23 11:43:06 2022] cx23885: cx23885_dev_setup() Failed to register dvb adapters on VID_B
[So Jan 23 11:43:06 2022] cx23885: cx23885_dev_checkrevision() Hardware revision = 0xd0
[So Jan 23 11:43:06 2022] cx23885: cx23885[0]/0: found at 0000:01:00.0, rev: 4, irq: 22, latency: 0, mmio: 0xa1000000

shspvr, 2022-01-23:
Check sees if the dvb-demod-m88rs6000.fw in /lib/firmware ?

wakkorottix, 2022-01-23:
downloaded from openelec repo

ls *dvb*
dvb-demod-m88ds3103b.fw  dvb-demod-m88rs6000.fw  dvb-demod-si2168-b40-01.fw  dvb-fe-xc4000-1.4.1.fw    dvb-fe-xc5000c-4.1.30.7.fw  dvb-usb-dib0700-1.20.fw  dvb-usb-it9135-02.fw
dvb-demod-m88ds3103.fw   dvb-demod-mxl692.fw     dvb-demod-si2168-d60-01.fw  dvb-fe-xc5000-1.6.114.fw  dvb-tuner-si2157-a30-01.fw  dvb-usb-it9135-01.fw     dvb-usb-terratec-h5-drxk.fw

shspvr, 2022-01-23:
Reboot and dose "dmesg" output show it
You want try reading https://www.linuxtv.org/wiki/index.php/Hauppauge_WinTV-Starburst2
Sorry I don't have this card as I'm in USA

wakkorottix, 2022-01-23:
already did, still not working

wakkorottix, 2022-01-27:
Yes, i've bought it on 21.01.2022 - but sent back

b-rad-NDi, 2022-04-04:
I've reproduced this and we're trying to figure out what's up.

nicsimi, 2023-04-18:
I have an identical issue like jlaba has (on HVR-5525 card).
Are there any solution to initialise  the DVB-S tuner? The DVB-C & DVB-T tuners are working ok.

plinkerz, 2023-05-03:
Same issue here

- HVR 5525 card
- Ubuntu 20.04 fresh installed
- DVB-T and DVB-C tuners showing up properly but not DVB-S

let me know if you need some extra inputs from the machine that can help you out with the issue

falo2k, 2023-07-01:
I also have this issue with a HVR-5525 running on Unraid or Ubuntu 22.04 (same behaviour either way).  In my case it's reporting a chip_id of 71 for the DVB-S tuner.  This card was bought this week direct from Hauppauge.  I also followed it back to that chip_id switch in the m88ds3103 driver as the issue (given the returned ID and the limited IDs is looks for).

Looking at the LinuxTV wiki for the HVR-5525, their card was a revision C316.  Mine is a C616 (and I've found other references out there to the same problem with revision C516).  It also didn't detect the card properly because of the different subsystem (0070:f039 vs 0070:f038) so I had to force the module with a card=52 argument.

`Jun 30 15:24:38 servermonkey3 kernel: cx23885: cx23885 driver version 0.0.4 loaded
Jun 30 15:24:38 servermonkey3 kernel: cx23885: CORE cx23885[0]: subsystem: 0070:f039, board: Hauppauge WinTV-HVR5525 [card=52,insmod option]
Jun 30 15:24:38 servermonkey3 kernel: tveeprom: Hauppauge model 150339, rev C6I6, serial# 4036192849
Jun 30 15:24:38 servermonkey3 kernel: tveeprom: MAC address is 00:0d:fe:93:6a:51
Jun 30 15:24:38 servermonkey3 kernel: tveeprom: tuner model is SiLabs Si2157 (idx 186, type 4)
Jun 30 15:24:38 servermonkey3 kernel: tveeprom: TV standards PAL(B/G) PAL(I) SECAM(L/L') PAL(D/D1/K) ATSC/DVB Digital (eeprom 0xf4)
Jun 30 15:24:38 servermonkey3 kernel: tveeprom: audio processor is CX23888 (idx 40)
Jun 30 15:24:38 servermonkey3 kernel: tveeprom: decoder processor is CX23888 (idx 34)
Jun 30 15:24:38 servermonkey3 kernel: tveeprom: has radio, has IR receiver, has no IR transmitter
Jun 30 15:24:38 servermonkey3 kernel: cx23885: cx23885[0]: warning: unknown hauppauge model #150339
Jun 30 15:24:38 servermonkey3 kernel: cx23885: cx23885[0]: hauppauge eeprom: model=150339
Jun 30 15:24:38 servermonkey3 kernel: cx25840 2-0044: cx23888 A/V decoder found @ 0x88 (cx23885[0])
Jun 30 15:24:39 servermonkey3 kernel: cx25840 2-0044: loaded v4l-cx23885-avcore-01.fw firmware (16382 bytes)
Jun 30 15:24:39 servermonkey3 kernel: cx23885: cx23885[0]: registered device video0 [v4l2]
Jun 30 15:24:39 servermonkey3 kernel: cx23885: cx23885[0]: registered device vbi0
Jun 30 15:24:39 servermonkey3 kernel: cx23885: cx23885[0]: alsa: registered ALSA audio device
Jun 30 15:24:39 servermonkey3 kernel: cx23885: cx23885_dvb_register() allocating 1 frontend(s)
Jun 30 15:24:39 servermonkey3 kernel: cx23885: cx23885[0]: cx23885 based dvb card
Jun 30 15:24:39 servermonkey3 kernel: m88ds3103 0-0069: Unknown device. Chip_id=71
Jun 30 15:24:39 servermonkey3 kernel: cx23885: cx23885[0]: frontend initialization failed
Jun 30 15:24:39 servermonkey3 kernel: cx23885: cx23885_dvb_register() dvb_register failed err = -22
Jun 30 15:24:39 servermonkey3 kernel: cx23885: cx23885_dev_setup() Failed to register dvb adapters on VID_B
Jun 30 15:24:39 servermonkey3 kernel: cx23885: cx23885_dvb_register() allocating 1 frontend(s)
Jun 30 15:24:39 servermonkey3 kernel: cx23885: cx23885[0]: cx23885 based dvb card
Jun 30 15:24:39 servermonkey3 kernel: i2c i2c-0: Added multiplexed i2c bus 3
Jun 30 15:24:39 servermonkey3 kernel: si2168 0-0064: Silicon Labs Si2168-D60 successfully identified
Jun 30 15:24:39 servermonkey3 kernel: si2168 0-0064: firmware version: D 6.0.1
Jun 30 15:24:39 servermonkey3 kernel: si2157 1-0060: Silicon Labs Si2157 successfully attached
Jun 30 15:24:39 servermonkey3 kernel: dvbdev: DVB: registering new adapter (cx23885[0])
Jun 30 15:24:39 servermonkey3 kernel: cx23885 0000:04:00.0: DVB: registering adapter 0 frontend 0 (Silicon Labs Si2168)...
Jun 30 15:24:39 servermonkey3 kernel: cx23885: cx23885_dev_checkrevision() Hardware revision = 0xd0
Jun 30 15:24:39 servermonkey3 kernel: cx23885: cx23885[0]/0: found at 0000:04:00.0, rev: 4, irq: 33, latency: 0, mmio: 0xfb200000`

stulluk, 2023-11-07:
I should have read this before buying a brand new HVR-5525 :(

I connected the board to my pci-e slot , and tried everything possible... DVB-T/C tuner seems to be detected & working, but DVB-S/S2 does not !

I tested following kernels:
```bash
stulluk ~ $  dpkg -l | grep linux-image*
ii  linux-image-5.15.0-522304060810-generic           5.15.0-522304060810.0+mediatree+hauppauge   amd64        Linux kernel image with slipstreamed mediatree drivers for version 5.15.0 on 64 bit x86 SMP
ii  linux-image-5.19.0-50-generic                     5.19.0-50.50                                amd64        Signed kernel image generic
ii  linux-image-6.2.0-33-generic                      6.2.0-33.33~22.04.1                         amd64        Signed kernel image generic
ii  linux-image-unsigned-6.4.0-060400-generic         6.4.0-060400.202306271339                   amd64        Linux kernel image for version 6.4.0 on 64 bit x86 SMP
ii  linux-image-unsigned-6.5.7-060507-generic         6.5.7-060507.202310102154                   amd64        Linux kernel image for version 6.5.7 on 64 bit x86 SMP
stulluk ~ $ 
```

Even with following override:
```bash
stulluk ~ $  cat /etc/modprobe.d/hauppauge.conf 
options cx23885 card=52
stulluk ~ $  
```

All of the kernels above fails with following:

```bash
...
[    8.745307] m88ds3103 3-0069: Unknown device. Chip_id=71
[    8.745334] cx23885: cx23885[0]: frontend initialization failed
[    8.745345] cx23885: cx23885_dvb_register() dvb_register failed err = -22
[    8.745356] cx23885: cx23885_dev_setup() Failed to register dvb adapters on VID_B
...
```

I am attaching full dmesg output .
[hvr-5525-dvbs-fails-in-all-kernels.txt](https://github.com/b-rad-NDi/Ubuntu-media-tree-kernel-builder/files/13288851/hvr-5525-dvbs-fails-in-all-kernels.txt)

@b-rad-NDi @falo2k @jlaba @wakkorottix any suggestions ?

Pimmetje, 2023-11-25:
I have the same problem. Did anyone get it working?

stulluk, 2023-11-25:
I haven't get it working, and contacted hauppauge : [support@hauppauge.co.uk](mailto:support@hauppauge.co.uk)

They told me that @b-rad-NDi  will check this week but this also didn't happen. I so returned my card back to amazon.

I am still wondering, does anybody knows, is there any DVB-S2 ( USB or PCI ) card that works out-of-box without requiring recompilation of media_tree ? I am is there any DVB-S2 card that works as plug-n-play with ubuntu kernels ?

Pimmetje, 2023-11-25:
@stulluk thanks for your reply. I have a 15 days return policy. So ill return it at the end of next week if i can't get it fixed.

Pimmetje, 2023-11-26:
@stulluk I used one of https://digitaldevices.de/en/manuals/duoflex-s2/ before. But one kernel it works the next it does not. Yet another kernel update it works. I got crazy about it and decided i wanted another card. I took a hauppage because it looked like it had decent linux support. And now i have a hardware revision that does not work ...

stulluk, 2023-11-26:
@Pimmetje  I am searching for a "compatible" DVB-S tuner since for months.

I have a TBS5530 -> This requires "rebuilding + installing" media-tree everytime you update your kernel. And this has many consequences, such as your webcam doesn't work, your soundcard doesn'twork..etc ( Because some drivers rely on others and it affects other devices)

I bought Hauppauge HVR5525 -> This also requires "reinstallation of media-tree" . I checked what does this repository is doing: Basically it patches & rebuilds media-tree for Ubuntu kernels, and when you install the PPA, it loads additional *.ko files to your system. And Again, this is not very sustainable, the drivers will still depend on the kernel version that you are using.. 

And for HVR5525, I had another issue: When I use this device to my PCIe slot on my system, during POST ( Power On Self-Test - UEFI/BIOS) , it sporadically causing my system to hang as well. This was yet another red flag for me.

I am currently checking this: https://shop.sundtek.de/en/DVB-S-S2-S2X/Sundtek-SkyTV-Dual-2x-DVB-S-S2-S2X.html

I send an email asking if they have tested this with Ubuntu 22.04 -> But it seems they did not and their architecture is a little interesting: They have a userspace driver which interacts with kernel .. I don't know the consequences of doing this when running with TSDuck / Kaffeine, so I requested them to try it.

If they try and give green light, then I will buy it.

Sadly, the only "dream" device that seems to be working out of box with recent ubuntu kernels is http://dvbsky.net/Products_S960.html  However, I couldn't find this device anywhere on the internet. If I can find somebody selling it, I will immediately buy it.